networkupstools/nut #962: Voltage value from upsc is 0 on AVR750U (0x3024).

tjmc, 2021-01-23:
From upsc
```                                                                                                                                                                                                                                          battery.charge: 100                                                                                                                                                                                                                          battery.charge.warning: 30                                                                                                                                                                                                                   battery.runtime: 1480
battery.type: PbAc
battery.voltage: 0.0
battery.voltage.nominal: 12.0                                                                                                                                                                                                                device.mfr: Tripp Lite                                                                                                                                                                                                                       device.model: AVR750U                                                                                                                                                                                                                        device.serial: 3028BV4OM882401230                                                                                                                                                                                                            device.type: ups                                                                                                                                                                                                                             driver.name: usbhid-ups                                                                                                                                                                                                                      driver.parameter.pollfreq: 30                                                                                                                                                                                                                driver.parameter.pollinterval: 2                                                                                                                                                                                                             driver.parameter.port: /dev/bus/usb/002/009                                                                                                                                                                                                  driver.parameter.productid: 3024                                                                                                                                                                                                             driver.parameter.synchronous: no
driver.version: 2.7.4-2112-gb263e476
driver.version.data: TrippLite HID 0.82
driver.version.internal: 0.43                                                                                                                                                                                                                input.frequency: 6020.0                                                                                                                                                                                                                      input.transfer.high: 145.0
input.transfer.low: 89.0
input.voltage: 0.0
input.voltage.nominal: 120                                                                                                                                                                                                                   output.current: 120.0                                                                                                                                                                                                                        output.frequency: 6020.0
output.frequency.nominal: 60
output.voltage: 0.0
output.voltage.nominal: 120                                                                                                                                                                                                                  ups.beeper.status: muted                                                                                                                                                                                                                     ups.delay.shutdown: 20                                                                                                                                                                                                                       ups.delay.start: 30                                                                                                                                                                                                                          ups.load: 28                                                                                                                                                                                                                                 ups.mfr: Tripp Lite                                                                                                                                                                                                                          ups.model: AVR750U                                                                                                                                                                                                                           ups.power: 131.0                                                                                                                                                                                                                             ups.power.nominal: 750                                                                                                                                                                                                                       ups.productid: 3024                                                                                                                                                                                                                          ups.serial: 3028BV4OM882401230                                                                                                                                                                                                               ups.status: OL                                                                                                                                                                                                                               ups.timer.reboot: 65535                                                                                                                                                                                                                      ups.timer.shutdown: 65535
ups.timer.start: 65535
ups.vendorid: 09ae
ups.watchdog.status: 0
```

/lib/nut/usbhid-ups appears off by a factor of 10000

```/lib/nut/usbhid-ups -D -a office -u root```
```
   0.044669     [D1] Path: UPS.PowerSummary.Input.Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 0.001192
   0.065565     [D1] Path: UPS.BatterySystem.Battery.Voltage, Type: Feature, ReportID: 0x20, Offset: 0, Size: 16, Value: 0.000137
   0.079124     [D1] Path: UPS.PowerConverter.Input.Voltage, Type: Feature, ReportID: 0x18, Offset: 0, Size: 16, Value: 0.001192

```
also frequency and current
```
   0.080625     [D1] Path: UPS.PowerConverter.Input.Frequency, Type: Feature, ReportID: 0x19, Offset: 0, Size: 16, Value: 6020
   0.085140     [D1] Path: UPS.PowerConverter.Output.Current, Type: Feature, ReportID: 0x46, Offset: 0, Size: 16, Value: 90
```

```
lsusb
Bus 001 Device 006: ID 09ae:3024 Tripp Lite
```

Full dump:
```
/lib/nut/usbhid-ups -D -a office -u root
   0.000000     [D1] debug level is '1'
[... cut ...]
   0.128865     [D1] Path: UPS.ffff00ff.ffff00ef, Type: Input, ReportID: 0xe3, Offset: 0, Size: 8, Value: 0
   0.128884     [D1] Path: UPS.ffff00ff.ffff00ef, Type: Feature, ReportID: 0xe3, Offset: 0, Size: 8, Value: 0
   0.128910     [D1] Detected a UPS: Tripp Lite/AVR750U
   0.132043     [D1] upsdrv_initinfo...
   0.132064     [D1] upsdrv_updateinfo...
   0.135251     [D1] Got 1 HID objects...
   0.135278     [D1] Quick update...
   0.135366     [D1] upsdrv_updateinfo...
   0.138385     [D1] Got 1 HID objects...
```

jimklimov, 2021-04-20:
PR merged, testing by users of the hardware (with builds of NUT master branch) would be welcome.

jimklimov, 2023-08-29:
Cheers, is there an update about how well this device got supported by NUT 2.8.0 (or current master)?