networkupstools/nut #964: [HCL] Emerson Network Power, Liebert GXT4 somewhat supported by usbhid-ups

xztraz, 2021-01-23:
It responds but not much data.  online status seem ok but not many voltages show up and such.

`upsc ups@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
battery.charge.warning: 0
battery.type: PbAc
battery.voltage: 10000000.
battery.voltage.nominal: 0.0
device.mfr: Emerson Network Power
device.model: Liebert GXT4
device.serial: 1601600192AFE13
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Belkin/Liebert HID 0.17
driver.version.internal: 0.41
ups.mfr: Emerson Network Power
ups.model: Liebert GXT4
ups.productid: 0000
ups.serial: 1601600192AFE17
ups.status: OL CHRG
ups.vendorid: 10af
`

conf
`[ups]

        driver = usbhid-ups
        port = auto
        desc = "Emerson Network Power Liebert GXT4"
        productid = 0000
`

`upscmd -l ups
Instant commands supported on UPS [ups]:

`
(none)

`lsusb -d 10af:0000 -v

Bus 001 Device 004: ID 10af:0000 Liebert Corp. UPS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x10af Liebert Corp.
  idProduct          0x0000 UPS
  bcdDevice            0.00
  iManufacturer           1 Emerson Network Power
  iProduct                2 Liebert GXT4
  iSerial                 3 1601600192AFE17
  bNumConfigurations      1
OTG Descriptor:
  bLength                 3
  bDescriptorType         9
  bmAttributes         0x00
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0025
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 EMERSON NETWORK-GXT4G
    bmAttributes         0x40
      (Missing must-be-set bit!)
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              5 HID-LEDs
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     505
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              32
`

product manufacturer web:
https://www.vertiv.com/en-cn/products-catalog/critical-power/uninterruptible-power-supplies-ups/liebert-gxt4-ups-700-10000-va/

mattrivett, 2021-02-21:
I have the same UPS and am seeing similar results.  I'd be happy to help gather data.

pi@vpn02alp:~ $ /lib/nut/usbhid-ups -DD -a Liebert
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '2'
   0.002143     upsdrv_initups...
   0.035059     Checking device (1D6B/0003) (002/001)
   0.064237     - VendorID: 1d6b
   0.065204     - ProductID: 0003
   0.066101     - Manufacturer: unknown
   0.067003     - Product: unknown
   0.067896     - Serial Number: unknown
   0.068791     - Bus: 002
   0.069708     - Device release number: 0510
   0.070601     Trying to match device
   0.071509     Device does not match - skipping
   0.072816     Checking device (10AF/0000) (001/004)
   0.082366     - VendorID: 10af
   0.083299     - ProductID: 0000
   0.084229     - Manufacturer: Emerson Network Power
   0.085124     - Product: Liebert GXT4
   0.085842     - Serial Number: 1718700042AFB93
   0.085994     - Bus: 001
   0.086373     - Device release number: 0000
   0.086519     Trying to match device
   0.086730     Device matches
   0.089401     HID descriptor length 505
   0.107887     Report Descriptor size = 505
   0.108703     Using subdriver: Belkin/Liebert HID 0.17
   0.109475     59 HID objects found
   0.110660     Path: UPS.PowerSummary.Undefined, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 0
   0.113400     Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 4, Value: 3
   0.113575     Path: UPS.PowerSummary.iManufacturer, Type: Feature, ReportID: 0x02, Offset: 4, Size: 4, Value: 1
   0.113736     Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x02, Offset: 8, Size: 4, Value: 1
   0.113939     Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x02, Offset: 12, Size: 4, Value: 6
   0.114103     Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x02, Offset: 16, Size: 8, Value: 100
   0.116157     Path: UPS.PowerSummary.RemainingTimeLimit, Type: Feature, ReportID: 0x03, Offset: 0, Size: 24, Value: 250
   0.116328     Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x03, Offset: 24, Size: 2, Value: 2
   0.116487     Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x03, Offset: 26, Size: 1, Value: 1
   0.116646     Path: UPS.PowerSummary.Undefined, Type: Feature, ReportID: 0x03, Offset: 27, Size: 5, Value: 1
   0.118897     Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x13, Offset: 0, Size: 8, Value: 20
   0.121645     Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 100
   0.121812     Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x06, Offset: 0, Size: 8, Value: 100
   0.124404     Path: UPS.PowerSummary.Charging, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 1, Value: 1
   0.124570     Path: UPS.PowerSummary.Discharging, Type: Feature, ReportID: 0x0c, Offset: 1, Size: 1, Value: 0
   0.124726     Path: UPS.PowerSummary.ACPresent, Type: Feature, ReportID: 0x0c, Offset: 2, Size: 1, Value: 1
   0.124882     Path: UPS.PowerSummary.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x0c, Offset: 3, Size: 1, Value: 0
   0.125036     Path: UPS.PowerSummary.BatteryPresent, Type: Feature, ReportID: 0x0c, Offset: 4, Size: 1, Value: 1
   0.125192     Path: UPS.PowerSummary.Undefined, Type: Feature, ReportID: 0x0c, Offset: 5, Size: 3, Value: 0
   0.125346     Path: UPS.PowerSummary.Charging, Type: Input, ReportID: 0x0c, Offset: 0, Size: 1, Value: 1
   0.125502     Path: UPS.PowerSummary.Discharging, Type: Input, ReportID: 0x0c, Offset: 1, Size: 1, Value: 0
   0.125656     Path: UPS.PowerSummary.ACPresent, Type: Input, ReportID: 0x0c, Offset: 2, Size: 1, Value: 1
   0.125811     Path: UPS.PowerSummary.BelowRemainingCapacityLimit, Type: Input, ReportID: 0x0c, Offset: 3, Size: 1, Value: 0
   0.125966     Path: UPS.PowerSummary.BatteryPresent, Type: Input, ReportID: 0x0c, Offset: 4, Size: 1, Value: 1
   0.126120     Path: UPS.PowerSummary.Undefined, Type: Input, ReportID: 0x0c, Offset: 5, Size: 3, Value: 0
   0.127149     Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x04, Offset: 0, Size: 8, Value: 0
   0.127317     Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x04, Offset: 8, Size: 8, Value: 0
   0.127474     Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x04, Offset: 16, Size: 8, Value: 1
   0.127632     Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x04, Offset: 24, Size: 8, Value: 1
   0.129914     Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x05, Offset: 0, Size: 16, Value: 1
   0.130085     Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x05, Offset: 16, Size: 16, Value: 0
   0.132649     Path: UPS.PowerSummary.Test, Type: Feature, ReportID: 0x08, Offset: 0, Size: 8, Value: 1
   0.135407     Path: UPS.PowerSummary.Undefined, Type: Feature, ReportID: 0x09, Offset: 0, Size: 8, Value: 1
   0.138147     Path: UPS.PowerSummary.Undefined, Type: Feature, ReportID: 0x32, Offset: 0, Size: 8, Value: 1
   0.140878     refresh_report_buffer: expected 5 bytes, but got 2 instead
   0.141044     Path: UPS.PowerSummary.Undefined, Type: Feature, ReportID: 0x33, Offset: 0, Size: 8, Value: 1
   0.143646     Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 0
[... cut ...]
   2.256839     Path: UPS.PowerSummary.Undefined, Type: Input, ReportID: 0x0c, Offset: 5, Size: 3, Value: 0
   2.257013     Quick update...
   2.259715     Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 100
   2.259808     Path: UPS.PowerSummary.Discharging, Type: Feature, ReportID: 0x0c, Offset: 1, Size: 1, Value: 0
   2.259874     Path: UPS.PowerSummary.Charging, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 1, Value: 1
   2.259936     Path: UPS.PowerSummary.ACPresent, Type: Feature, ReportID: 0x0c, Offset: 2, Size: 1, Value: 1

clepple, 2021-03-04:
@mattrivett what is interesting is that the debug log you posted on nut-upsuser showed a different path through the scaling code, though I guess that was from code newer than v2.7.4. From above:

```
   0.211353     Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x05, Offset: 16, Size: 16, Value: 0
   0.211484     ConfigVoltage = 0 -> assuming correction factor = 1e+08
```

From the list:
```
   0.208056     [D3] Report[buf]: (5 bytes) => 05 36 00 30 00
   0.208187     [D2] Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x05, Offset: 0, Size: 16, Value: 0
   0.208311     LineVoltage exponent looks wrong, but not correcting.
   0.208463     [D3] Report[buf]: (5 bytes) => 05 36 00 30 00
   0.208593     [D2] Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x05, Offset: 16, Size: 16, Value: 0
   0.208718     ConfigVoltage exponent looks wrong, but not correcting.
```

Clearly needs some more debug info, especially since this version seems to still be truncating the scaled "Value" printouts.

@xztraz I posted an explanation on the list about where to look for some of the other values. It might be hard to implement with the current code structure, though, because NUT uses the "Path:" IDs for looking things up, and the "Undefined" items are indistinguishable from one another (except via the ReportID, which is really more of an internal ID, and is not exposed to the HID-to-NUT matching tables like in [drivers/belkin-hid.c](https://github.com/networkupstools/nut/blob/v2.7.4/drivers/belkin-hid.c#L429)).

https://alioth-lists.debian.net/pipermail/nut-upsuser/2021-March/012321.html

mattrivett, 2021-03-04:
In my earlier posts I installed NUT using apt-get nut on an RPI3.  The RPI
I also run as to get VPN access into my home network.  I wanted to upgrade
to get faster ethernet and processing however it really isn't required.
The latest output is from compiling nut master on the RPI4.

Thanks for all the information.  My goal is to get this data out of the UPS
into MQTT that I then historize into a PI server.  I am doing that using a
modified version of this script for a CyberPower UPS and it works well.
https://github.com/dniklewicz/ups-mqtt

[image: image.png]

I guess I can just leave the driver running and pipe it into a file for
now.  I've recorded video of this type of thing to provide data to decode
sensors for rtl_433.  Then you can see in real-time the output and the LCD
and the data being received.  That might help.  I also have the software
that came with the UPS that I haven't even installed yet.  I could see if
that provides any useful information.

xztraz, 2021-03-04:
> @xztraz I posted an explanation on the list about where to look for some of the other values..

Great! Anything i can do to help. photo of screen and some different loads with some log?

Loizzus, 2021-04-07:
I'm in the same boat, I also have one of these things and need to get it to work properly. Anything I can do to help let me know.

MGlants, 2021-09-05:
I have the same thing with LIEBERT PSI 1000VA

jimklimov, 2022-04-25:
Note that since 2020 several PRs were merged to update liebert-hid.c, so recent builds from NUT master branch (and eventual 2.8.0) might support this better. Any tests and reports back if it actually does would be welcome.

jimklimov, 2022-04-25:
HCL/DDL part of the issue was handled with https://github.com/networkupstools/nut-ddl/pull/21 and #1396

sebastian-tennant, 2023-12-04:
I have a similar issue where input and output voltage seem to be missing. If the UPS is plugged into my Synology, it will provide me the values - but when I plug it into a system with full NUT, I don't get them. I should see what they are doing..

```
# upsc -V
Network UPS Tools upscmd 2.8.0.1
```

```
# upsc ups@localhost   
battery.charge: 100
battery.charge.low: 38
battery.charge.warning: 38
battery.type: PbAc
battery.voltage: 0.0
battery.voltage.nominal: 0.0
device.mfr: Emerson Network Power
device.model: LiebertPSI
device.serial:           
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.0.1
driver.version.data: Belkin/Liebert HID 0.18
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.0 (API: 0x1000102)
input.frequency: 49.9
input.voltage: 0.0
output.voltage: 0.0
ups.load: 12
ups.mfr: Emerson Network Power
ups.model: LiebertPSI
ups.productid: 0001
ups.serial:           
ups.status: OL CHRG
ups.vendorid: 10af
```

jimklimov, 2023-12-04:
@sebastian-tennant : what does Synology say (assuming it has another NUT version) about the `driver.version.data` (and the version itself)? Is it by chance MGE, maybe 2.7.4 based?

There is a overlap and difference of the two mapping tables, with some devices with same VID:PID marketed as one or other brand with different capabilities.

NUT 2.8.1 release (and later master branch code) added a `subdriver` option for `usbhid-ups` so users can pick their preference (and use USB IDs not built into binary drivers).

sebastian-tennant, 2023-12-15:
@jimklimov They might be doing some proprietary magic. This is the output from the same UPS, just plugged into the Synology:

`# upsc ups@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 38
battery.charge.warning: 38
battery.type: PbAc
battery.voltage: 43.3
battery.voltage.nominal: 36.0
device.mfr: Emerson Network Power
device.model: LiebertPSI
device.serial:           
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 5
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: DSM7-1-1-42930-workplus-version2-repack-42930-220712
driver.version.data: Belkin/Liebert HID 0.17
driver.version.internal: 0.41
input.frequency: 50.0
input.voltage: 240.5
output.voltage: 242.5
ups.load: 13
ups.mfr: Emerson Network Power
ups.model: LiebertPSI
ups.productid: 0001
ups.serial:           
ups.status: OL CHRG
ups.vendorid: 10af`

`# upsc -V
Network UPS Tools upscmd DSM7-1-1-42930-workplus-version2-repack-42930-220712`

jimklimov, 2023-12-15:
Thanks! So, diff'ing the two data samples - I see they have a different `synchronous` setting (impacts how drivers send into the pipe to `upsd`) but with so few data points it should not matter regarding buffer overflows. Still, worth trying to flip yours on the "system with full NUT".

Another is the apparent difference of NUT releases involved, with Synology version looking like NUT v2.7.4; at least its sources have the identifying strings:
````
drivers/belkin-hid.c:#define BELKIN_HID_VERSION      "Belkin/Liebert HID 0.17"
drivers/usbhid-ups.c:#define DRIVER_VERSION             "0.41"
````

I'm not sure which exact commit your "2.8.0.1" build is made from, but versions 0.52 and 0.18 are around for some time, and are newer than 2.8.0 release.

From this, two possibilities pop out:
* 2.7.x and before were libusb-0.1 based (some packages I think did use libusb-1.x, maybe with the compat API) and 2.8.x has the option of libusb-1.0 -- which is a largely different implementation, so something might differ here. This can be checked by a custom build `--with-libusb=0.1` at least.
* NUT code differences in the driver (`usbhid-ups.c`) are significant, too many to overview while commuting :) But those in the subdriver not so much (mostly about changed internal structures - adding NULL fields). Between 2.7.4 and 2.8.0 a feature appeared to "fix" USB report descriptors that seem to not match the spec about value representations (see #1024 and a breadcrumb from there if interested). Later on we added a config toggle `disable_fix_report_desc` to disable this fix easily (if it is suspected to introduce problems) - so that's another thing you can try quickly.

Also at least on the "full" system you can bump debug of the driver to see what it reports anything interesting about the data it sees (if it applies the descriptor fixes, something else...) - see https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity