networkupstools/nut #973: NUT losing connection when UPS switches to battery

Neji3971, 2021-02-04:
I have the following scenario:
- UPS with megatec USB protocol (using the blaze_usb driver) connected to Raspberry Pi via USB
- Works perfectly fine as long as the UPS is on power, I can execute `upsc ups`
- As soon as the UPS switches to Battery, NUT reports lost communications
- The USB-device is still connected, as checked with `lsusb`
- If I restart the driver, the UPS is shortly connected and NUT reports being on battery
- A few seconds later, it reports lost communications again
- This can be repeated as long as the UPS is on battery
- When the UPS switches back to line power, it reestablishes connections again

I'm sorry, this is my first time I use GitHub, so please tell me if you need anything.

Neji3971, 2021-03-29:
If it would help, I'd be down to provide a user on my Raspberry Pi as well as demonstrate in a Discord-Call.

jimklimov, 2021-03-29:
There was another issue raised a while ago but reminded of recently, #703 - also with a USB driver on Raspberry. In that case, one glaring problem was a "value too big for defined type" possibly related to a 256-byte long message (if there is something limiting it to a 1-byte length field anywhere, e.g. 255 max - no idea yet, but sounded like a possibility). Did you see anything like this on your system, possibly with a driver that you would start with higher debug verbosity by adding `-DDDDDD` argument?

Also, just in case:
* do you use a packaged NUT (so probably 4+ years old) or one built from recent sources?
* are you in position to experiment with building from source, e.g. from master branch or one of the branches with candidate code for libusb-1.0 upgrade of NUT?
* which version of Raspberry, which OS distro;
* is it 32-bit or 64-bit?

Neji3971, 2021-03-29:
> Did you see anything like this on your system, possibly with a driver that you would start with higher debug verbosity by adding `-DDDDDD` argument?

I didn't see anything like that, but I am sadly not a pro. I have no idea what you mean with starting a driver and adding `-DDDDDD`, sorry.

> do you use a packaged NUT (so probably 4+ years old) or one built from recent sources?
I used a packaged NUT

> are you in position to experiment with building from source, e.g. from master branch or one of the branches with candidate code for libusb-1.0 upgrade of NUT?
I guess I could pull it from GitHub, but I need to find out how to build it, because I don't know.

> which version of Raspberry, which OS distro;
I run a Raspberry Pi 2 Model B Rev 1.1. I could also use a Raspberry Pi 3 because my dad has one that he doesn't use.

> is it 32-bit or 64-bit?
It is 32-bit.

I am pretty good with Linux, but I don't know what to do as I am not that good with Linux. Sorry.