networkupstools/nut #977: Problem with all numbers either being rounded up or truncated to integer

denisyurin, 2021-02-20:
Hi there, 

I have a problem with NUT that it rounds up all UPS digital data to integer values. I have an Eaton 9SX 6000iR with a Network M2 card with Environmental Sensor connected to it. Using Web-interface I can see that Temperature and Humidity data goes at least with one decimal place precision. However when I receive this data from the card with NUT’s snmp-ups driver and display it with upsc and other nut’s tools, all numbers either rounded up or truncated to integer. Here is my setup:

Ubuntu Server 20.04.2
nut 2.7.4 (standard ubuntu package)
Eaton 9SX 6000iR + Network M2 card + Environmental Probe 

```
/etc/nut/ups.conf

[snmpv1]
        driver = snmp-ups
        port = 192.168.0.8
        community = public
        snmp_version = v1
        pollfreq = 15
```

```
> uspc snmpv1
ambient.humidity: 25.00
ambient.humidity.high: 90.00
ambient.humidity.low: 10.00
ambient.temperature: 22.0
ambient.temperature.high: 80.00
ambient.temperature.low: 0.00
…
```

Any help is extremely welcome!

closed 2021-03-22 by denisyurin

masterwishx, 2024-10-03:
@denisyurin Hi, do you still have Eaton 9sx?

denisyurin, 2024-10-15:
Yes, I do.

masterwishx, 2024-10-15:
> Yes, I do.

Maybe you using eco mode or only online ?
I'm almost finished pr for enabling bypass and eco mode for Eaton.
Tested with my 9E 2000i and entered to eco, but some commands sometime not work for me do some issues on my side but would be cool if someone can test it also

masterwishx, 2024-10-15:
Oh you are not using usb...

denisyurin, 2024-10-15:
Regarding my problem I found a solution (thanks to Eaton engineers): new Eaton MIBs have a branch .1.3.6.1.4.1.534.6.8 where the humidity and temperature are given with 0.1 precision

jimklimov, 2024-10-15:
...and OIDs from those MIB branches are present in many current NUT SNMP mapping tables.