networkupstools/nut #982: Cyber Power LX1500GU incorrect output voltage reading

JasonNaughton, 2021-02-28:
Working on FreeBSD 11.4, managed to configure ups.conf, so that my Cyberpower 1500 UPS was reporting but unfortunately some of the values are a bit off:

# upsc cyberups

battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 1650
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: LX1500GU
device.serial: QARGU2000275
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 116.0
input.voltage.nominal: 120
**output.voltage: 140.0**
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 30
ups.mfr: CPS
ups.model: LX1500GU
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: QARGU2000275
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

My, I'm sure the output voltage is 120 not 140.

JasonNaughton, 2021-02-28:
Also noticed that ups_upslog values are not reporting properly:

# cat /var/log/nut/ups.log 
20210228 174727 NA NA NA [NA] NA NA
20210228 175019 NA NA NA [NA] NA NA
20210228 182639 NA NA NA [NA] NA NA
20210228 183139 NA NA NA [NA] NA NA
20210228 183639 NA NA NA [NA] NA NA
20210228 184139 NA NA NA [NA] NA NA
20210228 184639 NA NA NA [NA] NA NA
20210228 185139 NA NA NA [NA] NA NA

JasonNaughton, 2021-02-28:
Configuration files:

# grep -v "^#" ups.conf
[cyberups]
        driver = usbhid-ups
        port = auto
        desc = "UPS CP1500VA"

# grep -v "^#" upsd.conf |grep -v '^$'
LISTEN 127.0.0.1 3493

# grep -v "^#" nut.conf | grep -v '^$'
MODE=standalone

# grep -v "^#" upsmon.conf | grep -v '^$'
RUN_AS_USER root
MONITOR cyberups@localhost 1 monmaster password_removed master
MINSUPPLIES 1
SHUTDOWNCMD "echo Hello >> /tmp/foo"
POLLFREQ 5
POLLFREQALERT 5
HOSTSYNC 15
DEADTIME 15
POWERDOWNFLAG /etc/killpower
RBWARNTIME 43200
NOCOMMWARNTIME 300
FINALDELAY 5

JasonNaughton, 2021-03-01:
Did a 

/usr/local/libexec/nut/usbhid-ups -DDDDD -a cyberups 2> /tmp/ouput

Found the lines which the UPS was reporting:

   1.455619     Entering libusb_get_report
   1.462009     Report[get]: (3 bytes) => 12 **74** 00
   1.462042     PhyMax = 0, PhyMin = 0, LogMax = 142, LogMin = 136
   1.462069     Unit = 00f0d121, UnitExp = 7
   1.462086     Exponent = 0
   1.462104     hid_lookup_path: 00840004 -> UPS
   1.462122     hid_lookup_path: 0084001c -> Output
   1.462140     hid_lookup_path: 00840030 -> Voltage
   1.462159     Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: **140**

Yet Hex 0x74 is 116 not 140

   8.350308     Report[buf]: (3 bytes) => 12 **74** 00
   8.350325     PhyMax = 0, PhyMin = 0, LogMax = 142, LogMin = 136
   8.350342     Unit = 00f0d121, UnitExp = 7
   8.350359     Exponent = 0
   8.350377     Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 140
   8.350396     send_to_all: SETINFO output.voltage "**140.0**"

clepple, 2021-03-04:
see #581 and other issues tagged [CyberPower](https://github.com/networkupstools/nut/issues?q=is%3Aopen+is%3Aissue+label%3A%22CyberPower+%28CPS%29%22). The HID report descriptors are broken.

flashydave, 2022-01-17:
What Vendorid and Poductid does your UPS report?

JasonNaughton, 2022-01-17:
Unfortunately I disconnected the UPS from the server as I wasn't getting any response a while back.    Sorry I can't give you much information on it now.

arcenik, 2022-09-24:
I have simmilar issue

The input voltage is 226v not 21.6v and the battery is 40.1v not 400v

```
battery.charge: 100
battery.charge.low: 20
battery.charge.warning: 20
battery.mfr.date: CyberPower Systems
battery.runtime: 6180
battery.runtime.low: 300
battery.type: PbAcid
**battery.voltage: 401.0**
battery.voltage.nominal: 360
device.mfr: CyberPower Systems
device.model: OL1000EXL
device.serial: WD9KN2000022
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0601
driver.parameter.synchronous: no
driver.parameter.vendorid: 0764
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 300
input.transfer.low: 190
**input.voltage: 21.2**
input.voltage.nominal: 230
output.voltage: 230.5
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 5
ups.mfr: CyberPower Systems
ups.model: OL1000EXL
ups.productid: 0601
ups.realpower.nominal: 900
ups.serial: WD9KN2000022
ups.status: OL
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

jimklimov, 2022-09-25:
Can you please run the driver with higher debug verbosity (3 or more) so it dumps content of the USB Report Descriptors (lots of hex) and interpretation (Path lines) annd copy that log here?

Are you in position to build and run current NUT from github master? @flashydave proposed some fixes for broken CPS descriptors, maybe your issue is already resolved.

arcenik, 2022-09-25:
Hi,

Here is the debug log from the Debian package driver.

```
/lib/nut/usbhid-ups -DDDDD -a cyberpower 2>&1 | tee /tmp/nut-driver-debug.log
```

[nut-driver-debug.log](https://github.com/networkupstools/nut/files/9640962/nut-driver-debug.log)

I've compiled nut from the source but it does not work, do I need some lib or parameter in configure call ?

```
   0.005749     Can't connect to UPS [cyberpower] (usbhid-ups-cyberpower): Connection refused
   0.005821     Found 1 UPS defined in ups.conf
```

jimklimov, 2022-09-25:
Thanks, the latter looks like a message from upsd while the driver is not started?

darsys, 2023-03-30:
I have same issue.  I wanted to share my information just so this issue is well documented.  Everything works properly except the output voltage, as far as I can tell.

Version:  Generic HID driver 0.41 (2.7.4)

Excerpt from usbhid-ups:

   0.653344     Entering libusb_get_report
   0.654006     Report[get]: (3 bytes) => 12 **_7c_** 00
   0.654049     PhyMax = 0, PhyMin = 0, LogMax = 142, LogMin = 136
   0.654091     Unit = 00f0d121, UnitExp = 7
   0.654125     Exponent = 0
   0.654159     hid_lookup_path: 00840004 -> UPS
   0.654194     hid_lookup_path: 0084001c -> Output
   0.654230     hid_lookup_path: 00840030 -> Voltage
   0.654295     Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: **_140_**

Hex 7c = 124 but the value reported is 140.

Attached is the complete log my `sudo /lib/nut/usbhid-ups -DDDDD -a cyberpower` run

jimklimov, 2024-12-18:
PR #2718 is presumed to fix the broken CPS input/output voltage reports discussed in this issue.

closed 2024-12-18 by jimklimov