networkupstools/nut #985: Support of generic HID-compliant UPS with Interface greater than 0

abratchik, 2021-03-04:
Current implementation of HID-compliant UPS support (usbhid_ups) assumes only Interface 0. This may or may not be valid for composite devices, which may combine the UPS function with some other interface. 
Besides, [the USB HID Power Device specification](https://www.usb.org/sites/default/files/pdcv11.pdf) does not limit the end point to Interface 0 only.  
Is it possible to make the interface number configurable or auto-detecting the correct interface? That would help to make networkupstools more universal.

jimklimov, 2021-11-14:
With the PR merged, I think this issue may be closed. Thanks for the info and the effort!

closed 2021-11-14 by jimklimov

DavidLin1577, 2024-08-26:
mark