networkupstools/nut #999: OEM UPS support query

CindyZhan19891207, 2021-03-26:
Hello NUT developers:

         I'm seeking to list our  UPSs  to the compatibility list.  My question is:  
        There's 5 support levels, and what should I do if I want to start level 5 compatibility test (send samples to someone? post our own test result?)?  MGE driver and hidusb-ups driver can support, but not level 5. 
        Thanks.

BR
Cindy

aquette, 2021-03-30:
@CindyZhan19891207 
as discussed, to add your units to the list of supported ones, you will want to provide a pull request similar to this:
https://github.com/networkupstools/nut/commit/d56ac7712fa6c3920460a08c649ad03d2c2e82e7#diff-5e045e2ffcd1c6dd261fbc08a0315f6ed2103626e1e7a9efe657654c33536867

(obviously adapting the information).
Support level would be 4, since the protocol is public, and completed with manufacturer provided data (through MGE and then Eaton). Going for level 5 would not make a big difference, and would require that you provide units.
if you need assistance in doing so, just tell us back.

Note that such submission should be accompanied with a device dump, to ensure that the right data/values are actually provided, and so that we can publish on the separate [NUT Devices Dump Library](https://networkupstools.org/ddl/)

side note: @jimklimov  as discussed, a level 6 for vendor developed drivers.

jimklimov, 2022-01-19:
Related driver and driver list changes were merged. Original question also answered, so closing this issue for now.

closed 2022-01-19 by jimklimov