networkupstools/nut #1004: usbhid-ups driver: Not working on VMware Flings' ESXi ARM edition

NeverEatYellowSwissSnow, 2021-04-05:
I was running nut server all the time without any issues inside an Ubuntu Server 20.04 VM running on a VMware ESXi 7.0 host (x86 architecture) with the "Eaton 5P650IR" UPS' USB port passed through the VM.
Since VMware has released its free [ESXi ARM edition for the Raspberry Pi 4](https://flings.vmware.com/esxi-arm-edition) it was obvious to me to set up a new VM to run nut server as a VM on the Raspberry Pi 4 (8GB RAM version) with the same VM settings.
I have tried all the way upside down to get this to work, unfortunately without success.
Here is what I did:

Ubuntu Server (minimal installation with just OpenSSH server installed):
Logged in with default user and did the following steps, step-by-step:
`$ sudo apt-get update && sudo apt-get dist-upgrade`
`$ sudo reboot now`
`$ uname -a` (Which gives the following oputput:
_Linux ubuntuserver02 5.4.0-70-generic #78-Ubuntu SMP Fri Mar 19 13:29:32 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux_ )
`$ sudo apt-get install nut`
`$ sudo cp /lib/udev/rules.d/62-nut-usbups.rules /etc/udev/rules.d/`

Now I have configured the nut config files like this (left all settings on default but only added content):

**/etc/nut/nut.conf**:
`MODE=standalone`

**/etc/nut/ups.conf**:
```
pollinterval = 10
[ups]
        driver = usbhid-ups
        port = auto
        desc = "Eaton 5P650IR"
        serial = G114L21147
        vendorid = 0463
        productid = ffff
```

**/etc/nut/upsd.conf**:
`LISTEN localhost 3493`

**/etc/nut/upsd.users**:
```
[upsmon]
        password  = 1234
        upsmon master
```

**/etc/nut/upsmon.conf**:
`MONITOR ups@localhost 1 monuser 1234 master`

Finally I executed the following commands:

`$ sudo systemctl enable nut-monitor.service`
`$ sudo systemctl enable nut-server.service`
`$ sudo systemctl start nut-monitor.service`
`$ sudo systemctl start nut-server.service`
`$ reboot now`

Now when executing the following command:
`$ lsusb`
It gives back the following output:

_Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 0463:ffff MGE UPS Systems UPS
Bus 002 Device 004: ID 0e0f:0006 VMware, Inc. Virtual Keyboard
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub_

When executing the following command:
`$ nut-scanner`
It gives back the following output:

_SNMP library not found. SNMP search disabled.
Neon library not found. XML search disabled.
IPMI library not found. IPMI search disabled.
Scanning USB bus.
No start IP, skipping NUT bus (old connect method)
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0463"
        productid = "FFFF"
        bus = "002"_

So this indicates the UPS is definitely being recognized.

Now whenever the following command is being executed:
`$ sudo systemctl status nut-server`
It gives back the following error(s):

_● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2021-01-12 15:55:01 UTC; 2min 17s ago
    Process: 930 ExecStartPre=/bin/sleep 15 (code=exited, status=0/SUCCESS)
    Process: 943 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
   Main PID: 946 (upsd)
      Tasks: 1 (limit: 2233)
     Memory: 2.0M
     CGroup: /system.slice/nut-server.service
             └─946 /lib/nut/upsd_

_Jan 12 15:54:47 ubuntuserver systemd[1]: Starting Network UPS Tools - power devices information server...
Jan 12 15:55:01 ubuntuserver upsd[943]: fopen /run/nut/upsd.pid: No such file or directory
Jan 12 15:55:01 ubuntuserver upsd[943]: listening on localhost port 3493
Jan 12 15:55:01 ubuntuserver upsd[943]: listening on localhost port 3493
Jan 12 15:55:01 ubuntuserver upsd[943]: Can't connect to UPS [ups] (usbhid-ups-ups): No such file or directory
Jan 12 15:55:01 ubuntuserver upsd[943]: Can't connect to UPS [ups] (usbhid-ups-ups): No such file or directory
Jan 12 15:55:01 ubuntuserver upsd[946]: Startup successful
Jan 12 15:55:01 ubuntuserver systemd[1]: Started Network UPS Tools - power devices information server._

So the nut service is stuck somewhere and somehow but I was not able to figure out why.

When executing the following command as root user:
`$ /lib/nut/usbhid-ups -DDD -a ups`
It gives back the following output:
 
_Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.002398     upsdrv_initups...
   0.049019     Checking device (0463/FFFF) (002/005)
   0.236934     - VendorID: 0463
   0.236994     - ProductID: ffff
   0.237005     - Manufacturer: unknown
   0.237035     - Product: Eaton 5P
   0.237045     - Serial Number: G114L21147
   0.237054     - Bus: 002
   0.237063     - Device release number: 0202
   0.237072     Trying to match device
   0.237351     Device matches
   0.237500     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.286503     HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 55 08
   0.286557     HID descriptor length (method 1) 2133
   0.286574     HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 25 04
   0.286602     HID descriptor length (method 2) 1061
   0.286612     Eaton device v2.02. Using full report descriptor
   0.286622     Warning: two different HID descriptors retrieved (Reportlen = 2133 vs. 1061)
   0.286632     HID descriptor length 2133
   2.335596     Unable to get Report descriptor: Invalid or incomplete multibyte or wide character
   2.335701     Checking device (0E0F/0006) (002/004)
   2.335783     - VendorID: 0e0f
   2.335827     - ProductID: 0006
   2.335837     - Manufacturer: unknown
   2.335846     - Product: unknown
   2.335855     - Serial Number: unknown
   2.335864     - Bus: 002
   2.335873     - Device release number: 0100
   2.335882     Trying to match device
   2.335905     Device does not match - skipping
   2.335929     Checking device (0E0F/0002) (002/003)
   2.335995     - VendorID: 0e0f
   2.336011     - ProductID: 0002
   2.336020     - Manufacturer: unknown
   2.336030     - Product: unknown
   2.336039     - Serial Number: unknown
   2.336048     - Bus: 002
   2.336057     - Device release number: 0100
   2.336066     Trying to match device
   2.336082     Device does not match - skipping
   2.336110     Checking device (0E0F/0003) (002/002)
   2.336171     - VendorID: 0e0f
   2.336190     - ProductID: 0003
   2.336199     - Manufacturer: unknown
   2.336209     - Product: unknown
   2.336218     - Serial Number: unknown
   2.336226     - Bus: 002
   2.336235     - Device release number: 0102
   2.336244     Trying to match device
   2.336258     Device does not match - skipping
   2.336283     Checking device (1D6B/0001) (002/001)
   2.336461     - VendorID: 1d6b
   2.336500     - ProductID: 0001
   2.336510     - Manufacturer: unknown
   2.336519     - Product: unknown
   2.336528     - Serial Number: unknown
   2.336537     - Bus: 002
   2.336546     - Device release number: 0504
   2.336555     Trying to match device
   2.336570     Device does not match - skipping
   2.336599     Checking device (1D6B/0002) (001/001)
   2.337068     - VendorID: 1d6b
   2.337108     - ProductID: 0002
   2.337117     - Manufacturer: unknown
   2.337126     - Product: unknown
   2.337135     - Serial Number: unknown
   2.337143     - Bus: 001
   2.337152     - Device release number: 0504
   2.337161     Trying to match device
   2.337176     Device does not match - skipping
   2.337198     No appropriate HID device found
   2.337211     No matching HID UPS found_

For virtual machine settings in VMware ESXi I have also tried between virtual USB2.0 and virtual USB3.1 controller and for both controllers I have tried to switch the UPS USB-connection from the Raspberry Pi's physical USB2.0 and USB3.1 ports but all combinations ended with the same behavior.

What I have also tried is to add the following line:
`ExecStartPre=/bin/sleep 15`
on top of the the _[Service]_ section inside the following file:
**/etc/systemd/system/multi-user.target.wants/nut-server.service**
and did a system reboot but that also did not help.

I hope you can understand everything in this topic.

Of yourse -if you need more informations- I can send you all the log files you will ask for.

I have also tried the same on a CentOS ARM system but the behaviour was exactly the same.

Thank you many times for your appreciated help in advance. It would be so nice to run nut-server successfully on a ESXi ARM VM.

NeverEatYellowSwissSnow, 2021-04-09:
Update:
Here is the output of the following command:

`$ systemctl status nut-driver.service`

● _nut-driver.service - Network UPS Tools - power device driver controller
     Loaded: loaded (/lib/systemd/system/nut-driver.service; static; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2021-04-09 10:49:37 UTC; 3min 47s ago
    Process: 789 ExecStart=/sbin/upsdrvctl start (code=exited, status=1/FAILURE)_

_Apr 09 10:49:29 ubuntuserver02 upsdrvctl[990]: No matching HID UPS found
Apr 09 10:49:29 ubuntuserver02 upsdrvctl[789]: Driver failed to start (exit status=1)
Apr 09 10:49:37 ubuntuserver02 upsdrvctl[1004]: Network UPS Tools - Generic HID driver 0.41 (2.7.4)
Apr 09 10:49:37 ubuntuserver02 upsdrvctl[1004]: USB communication driver 0.33
Apr 09 10:49:37 ubuntuserver02 upsdrvctl[1004]: No matching HID UPS found
Apr 09 10:49:37 ubuntuserver02 upsdrvctl[789]: Driver failed to start (exit status=1)
Apr 09 10:49:37 ubuntuserver02 upsdrvctl[789]: Network UPS Tools - UPS driver controller 2.7.4
Apr 09 10:49:37 ubuntuserver02 systemd[1]: nut-driver.service: Control process exited, code=exited, status=1/FAILURE
Apr 09 10:49:37 ubuntuserver02 systemd[1]: nut-driver.service: Failed with result 'exit-code'.
Apr 09 10:49:37 ubuntuserver02 systemd[1]: Failed to start Network UPS Tools - power device driver controller._

jimklimov, 2021-04-10:
Thank you for this report, unfortunately at the moment I am a bit at a loss: during the past few weeks we had several reports of issues with different generations of Raspberry Pi and various OSes and USB UPSes. So far my two main guesses were that there can be something flaky with the hardware (USB chips involved), after all low price was a hard goal, or bitness especially in older Pi generations which I think were 32-bit (and/or also OS bitness). In one report specifically there was a logged attempt to fit IIRC 256 bytes into a 255-byte buffer.

In your report, the varied HID descriptors and length mismatches look fishy; given that this is a pass-through of USB into a VM on hardware family that is already implicated in other issues even bare-metal, I am currently not sure what to state so would refrain. The rest looks logical, with the NUT driver failing to start, the corresponding service fails and upsd server has no driver to talk to and represent to upsmon client.

@aquette: do you have a chance to check about Eaton hardware involved? And/or with a Raspberry and varied OSes on that?

NeverEatYellowSwissSnow, 2021-04-10:
Thank you for your answer. In my case I can report that everything works smoothly on the same Raspberry Pi 4 when installing it bare metal on a Raspberry Pi OS.

It's only not working when using the Raspberry Pi 4 in an ESXi ARM virtual environment. In the meantime I have tried a couple of more virtual ARM based OS'es on the Raspberry Pi 4 ESXi ARM host, unfortunateliy without success:

- CentOS 8 aarch64
- Ubuntu server 20.04 arm64
- Debian 10 arm64
- Raspberry Pi OS arm64
- Alpine3 aarch64
- OpenSUSE aarch64

Like already mentioned on a generic x86 server PC with ESXi x86 virtual environment everything works as it should in my case.

NeverEatYellowSwissSnow, 2021-06-03:
Any news on this?
@aquette Were you able to check things according to @jimklimov recommendations?
Than you in advance.

aquette, 2021-06-03:
I'll try to check from work on a real PC tomorrow (not a phone). In the meantime, could you send the debug output level 5 (-DDDDD)
I gotta a few things in mind but need more traces

NeverEatYellowSwissSnow, 2021-06-04:
`$ /lib/nut/usbhid-ups -DDDDD -a ups`
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	send_to_all: SETINFO driver.parameter.port "auto"
   0.000036	send_to_all: SETINFO driver.parameter.serial "G114L21147"
   0.000143	send_to_all: SETINFO driver.parameter.vendorid "0463"
   0.000160	send_to_all: SETINFO driver.parameter.productid "ffff"
   0.000188	debug level is '5'
   0.002700	upsdrv_initups...
   0.048303	Checking device (0463/FFFF) (002/005)
   0.263941	- VendorID: 0463
   0.264000	- ProductID: ffff
   0.264019	- Manufacturer: unknown
   0.264064	- Product: Eaton 5P
   0.264078	- Serial Number: G114L21147
   0.264087	- Bus: 002
   0.264096	- Device release number: 0202
   0.264105	Trying to match device
   0.264289	Device matches
   0.264437	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.314468	HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 55 08
   0.314526	HID descriptor length (method 1) 2133
   0.314551	i=0, extra[i]=09, extra[i+1]=21
   0.314571	HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 25 04
   0.314581	HID descriptor length (method 2) 1061
   0.314592	Eaton device v2.02. Using full report descriptor
   0.314602	Warning: two different HID descriptors retrieved (Reportlen = 2133 vs. 1061)
   0.314611	HID descriptor length 2133
   2.363137	Unable to get Report descriptor: Invalid or incomplete multibyte or wide character
   2.363247	Checking device (0E0F/0006) (002/004)
   2.363323	- VendorID: 0e0f
   2.363337	- ProductID: 0006
   2.363347	- Manufacturer: unknown
   2.363356	- Product: unknown
   2.363365	- Serial Number: unknown
   2.363374	- Bus: 002
   2.363383	- Device release number: 0100
   2.363392	Trying to match device
   2.363413	Device does not match - skipping
   2.363455	Checking device (0E0F/0002) (002/003)
   2.363508	- VendorID: 0e0f
   2.363528	- ProductID: 0002
   2.363537	- Manufacturer: unknown
   2.363546	- Product: unknown
   2.363554	- Serial Number: unknown
   2.363563	- Bus: 002
   2.363572	- Device release number: 0100
   2.363588	Trying to match device
   2.363611	Device does not match - skipping
   2.363638	Checking device (0E0F/0003) (002/002)
   2.363690	- VendorID: 0e0f
   2.363702	- ProductID: 0003
   2.363711	- Manufacturer: unknown
   2.363720	- Product: unknown
   2.363728	- Serial Number: unknown
   2.363778	- Bus: 002
   2.363795	- Device release number: 0102
   2.363804	Trying to match device
   2.363906	Device does not match - skipping
   2.363935	Checking device (1D6B/0001) (002/001)
   2.363996	- VendorID: 1d6b
   2.364020	- ProductID: 0001
   2.364030	- Manufacturer: unknown
   2.364943	- Product: unknown
   2.364993	- Serial Number: unknown
   2.365010	- Bus: 002
   2.365029	- Device release number: 0504
   2.365053	Trying to match device
   2.365079	Device does not match - skipping
   2.365123	Checking device (1D6B/0002) (001/001)
   2.365593	- VendorID: 1d6b
   2.365639	- ProductID: 0002
   2.365655	- Manufacturer: unknown
   2.365674	- Product: unknown
   2.365694	- Serial Number: unknown
   2.365716	- Bus: 001
   2.365733	- Device release number: 0504
   2.365757	Trying to match device
   2.365784	Device does not match - skipping
   2.365817	No appropriate HID device found
   2.365839	No matching HID UPS found

NeverEatYellowSwissSnow, 2021-06-04:
@aquette Let me know if you need anything else.

NeverEatYellowSwissSnow, 2021-06-10:
@aquette Did you find out something about my issue?
Than you in advance.

NeverEatYellowSwissSnow, 2021-06-16:
@aquette Tried VMware Flings' ESXi ARM Version 1.4 which was released yesterday but unfortunately still no luck. The behavior is still the same.

aquette, 2021-06-17:
sorry, had no time to check. crazy life.
one additional point is to set the debug of libusb. export USB_DEBUG=3
that may give us more clues.

NeverEatYellowSwissSnow, 2021-06-17:
`$ export USB_DEBUG=3 && sudo /lib/nut/usbhid-ups -DDDDD -a ups -u root`
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     send_to_all: SETINFO driver.parameter.port "auto"
   0.000091     send_to_all: SETINFO driver.parameter.serial "G114L21147"
   0.000119     send_to_all: SETINFO driver.parameter.vendorid "0463"
   0.000145     send_to_all: SETINFO driver.parameter.productid "ffff"
   0.000297     debug level is '5'
   0.001829     upsdrv_initups...
   0.049636     Checking device (0463/FFFF) (002/005)
   0.242370     - VendorID: 0463
   0.242431     - ProductID: ffff
   0.242462     - Manufacturer: unknown
   0.242474     - Product: Eaton 5P
   0.242482     - Serial Number: G114L21147
   0.242491     - Bus: 002
   0.242500     - Device release number: 0202
   0.242509     Trying to match device
   0.242703     Device matches
   0.242795     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.292137     HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 55 08
   0.292198     HID descriptor length (method 1) 2133
   0.292210     i=0, extra[i]=09, extra[i+1]=21
   0.292267     HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 25 04
   0.292310     HID descriptor length (method 2) 1061
   0.292352     Eaton device v2.02. Using full report descriptor
   0.292406     Warning: two different HID descriptors retrieved (Reportlen = 2133 vs. 1061)
   0.292453     HID descriptor length 2133
   2.342449     Unable to get Report descriptor: Invalid or incomplete multibyte or wide character
   2.343240     Checking device (0E0F/0006) (002/004)
   2.361371     - VendorID: 0e0f
   2.361490     - ProductID: 0006
   2.361509     - Manufacturer: VMware
   2.361526     - Product: VMware Virtual USB Keyboard
   2.361536     - Serial Number: unknown
   2.361545     - Bus: 002
   2.361554     - Device release number: 0100
   2.361563     Trying to match device
   2.361585     Device does not match - skipping
   2.361628     Checking device (0E0F/0002) (002/003)
   2.381554     - VendorID: 0e0f
   2.381624     - ProductID: 0002
   2.381634     - Manufacturer: VMware, Inc.
   2.381651     - Product: VMware Virtual USB Hub
   2.381667     - Serial Number: unknown
   2.381676     - Bus: 002
   2.381686     - Device release number: 0100
   2.381694     Trying to match device
   2.381713     Device does not match - skipping
   2.381749     Checking device (0E0F/0003) (002/002)
   2.403380     - VendorID: 0e0f
   2.403448     - ProductID: 0003
   2.403465     - Manufacturer: VMware
   2.403482     - Product: VMware Virtual USB Mouse
   2.403492     - Serial Number: unknown
   2.403500     - Bus: 002
   2.403509     - Device release number: 0102
   2.403518     Trying to match device
   2.403538     Device does not match - skipping
   2.403571     Checking device (1D6B/0001) (002/001)
   2.403959     - VendorID: 1d6b
   2.404010     - ProductID: 0001
   2.404021     - Manufacturer: Linux 5.4.0-74-generic uhci_hcd
   2.404030     - Product: UHCI Host Controller
   2.404039     - Serial Number: 0000:01:00.0
   2.404048     - Bus: 002
   2.404056     - Device release number: 0504
   2.404065     Trying to match device
   2.404080     Device does not match - skipping
   2.404112     Checking device (1D6B/0002) (001/001)
   2.404807     - VendorID: 1d6b
   2.404863     - ProductID: 0002
   2.404872     - Manufacturer: Linux 5.4.0-74-generic ehci_hcd
   2.404890     - Product: EHCI Host Controller
   2.404900     - Serial Number: 0000:01:01.0
   2.404909     - Bus: 001
   2.404917     - Device release number: 0504
   2.404926     Trying to match device
   2.404943     Device does not match - skipping
   2.417758     No appropriate HID device found
   2.417848     No matching HID UPS found

aquette, 2021-06-18:
You got fooled by exporting before sudoing 😉
Try again

NeverEatYellowSwissSnow, 2021-06-18:
@aquette Oops, got that now:

`$ USB_DEBUG=3 /lib/nut/usbhid-ups -DDDDD -a ups -u root`

_Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	send_to_all: SETINFO driver.parameter.port "auto"
   0.000078	send_to_all: SETINFO driver.parameter.serial "G114L21147"
   0.000325	send_to_all: SETINFO driver.parameter.vendorid "0463"
   0.000359	send_to_all: SETINFO driver.parameter.productid "ffff"
   0.000406	debug level is '5'
   0.001706	upsdrv_initups...
usb_set_debug: Setting debugging level to 3 (on)
usb_os_init: Found USB VFS at /dev/bus/usb
usb_os_find_busses: Found 002
usb_os_find_busses: Found 001
usb_os_find_devices: Found 001 on 002
skipping descriptor 0x30
skipped 1 class/vendor specific endpoint descriptors
usb_os_find_devices: Found 006 on 001
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 005 on 001
usb_os_find_devices: Found 004 on 001
usb_os_find_devices: Found 003 on 001
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 002 on 001
skipped 1 class/vendor specific interface descriptors
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 001 on 001
error obtaining child information: Inappropriate ioctl for device
error obtaining child information: Inappropriate ioctl for device
error obtaining child information: Inappropriate ioctl for device
   0.327884	Checking device (1D6B/0003) (002/001)
   0.328268	- VendorID: 1d6b
   0.328314	- ProductID: 0003
   0.328325	- Manufacturer: Linux 5.4.0-74-generic xhci-hcd
   0.328347	- Product: xHCI Host Controller
   0.328366	- Serial Number: 0000:0a:00.0
   0.328386	- Bus: 002
   0.328403	- Device release number: 0504
   0.328420	Trying to match device
   0.328445	Device does not match - skipping
   0.328708	Checking device (0463/FFFF) (001/006)
   0.609781	- VendorID: 0463
   0.609836	- ProductID: ffff
   0.609854	- Manufacturer: EATON
   0.609874	- Product: Eaton 5P
   0.609892	- Serial Number: G114L21147
   0.609910	- Bus: 001
   0.609926	- Device release number: 0202
   0.609945	Trying to match device
   0.610121	Device matches
   0.610244	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.644848	HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 55 08
   0.644905	HID descriptor length (method 1) 2133
   0.644917	i=0, extra[i]=09, extra[i+1]=21
   0.644931	HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 25 04
   0.644952	HID descriptor length (method 2) 1061
   0.644963	Eaton device v2.02. Using full report descriptor
   0.644972	Warning: two different HID descriptors retrieved (Reportlen = 2133 vs. 1061)
   0.644982	HID descriptor length 2133
USB error: error sending control message: Protocol error
   2.690173	Unable to get Report descriptor: Protocol error
   2.690278	Checking device (0E0F/0002) (001/005)
   2.804984	- VendorID: 0e0f
   2.805040	- ProductID: 0002
   2.805058	- Manufacturer: VMware, Inc.
   2.805077	- Product: VMware Virtual USB Hub
   2.805086	- Serial Number: unknown
   2.805095	- Bus: 001
   2.805104	- Device release number: 0100
   2.805114	Trying to match device
   2.805135	Device does not match - skipping
   2.843674	Checking device (0E0F/0002) (001/004)
   2.844360	- VendorID: 0e0f
   2.844412	- ProductID: 0002
   2.844431	- Manufacturer: VMware, Inc.
   2.844448	- Product: VMware Virtual USB Hub
   2.844457	- Serial Number: unknown
   2.844466	- Bus: 001
   2.844475	- Device release number: 0100
   2.844484	Trying to match device
   2.844508	Device does not match - skipping
   2.844546	Checking device (0E0F/0006) (001/003)
   2.845169	- VendorID: 0e0f
   2.845216	- ProductID: 0006
   2.845235	- Manufacturer: VMware
   2.845250	- Product: VMware Virtual USB Keyboard
   2.845260	- Serial Number: unknown
   2.845269	- Bus: 001
   2.845278	- Device release number: 0100
   2.845292	Trying to match device
   2.845315	Device does not match - skipping
   2.845342	Checking device (0E0F/0003) (001/002)
   2.845883	- VendorID: 0e0f
   2.845929	- ProductID: 0003
   2.845947	- Manufacturer: VMware
   2.845962	- Product: VMware Virtual USB Mouse
   2.845972	- Serial Number: unknown
   2.845981	- Bus: 001
   2.845990	- Device release number: 0102
   2.846002	Trying to match device
   2.846026	Device does not match - skipping
   2.846055	Checking device (1D6B/0002) (001/001)
   2.846215	- VendorID: 1d6b
   2.846239	- ProductID: 0002
   2.846248	- Manufacturer: Linux 5.4.0-74-generic xhci-hcd
   2.846257	- Product: xHCI Host Controller
   2.846271	- Serial Number: 0000:0a:00.0
   2.846285	- Bus: 001
   2.846294	- Device release number: 0504
   2.846302	Trying to match device
   2.846323	Device does not match - skipping
   2.846344	No appropriate HID device found
   2.846361	No matching HID UPS found_

NeverEatYellowSwissSnow, 2021-06-23:
@aquette Did you find anything within the logs? It doesn't seem to provide much more than without `USB_DEBUG=3`, does it? Thank you.

NeverEatYellowSwissSnow, 2021-07-09:
No news on this?
Trying so hard to make nut work on ESXi ARM.

aquette, 2021-07-12:
sadly, had no time to check / think more about this. and vacations tomorrow.
I'd be interested in the same tests with the same unit, but on a working setup, just to see these 2 HID descriptors len (i.e. wrongly reported size Vs protocol issue)

NeverEatYellowSwissSnow, 2021-07-14:
@aquette That would be awesome. Just let me know in case I can provide something to help.

aquette, 2021-07-15:
Yes, plz provide the same command output, but on a working system (x86 Linux) 😉 for me to check the desc len thing

NeverEatYellowSwissSnow, 2021-07-22:
@aquette I'm in holidays right now until the end of July without possibility for accessing the infrastructure. I will provide it when I'm back ASAP.

NeverEatYellowSwissSnow, 2021-08-01:
@aquette Came back from holidays and my x64 ESXi host crashed somehow. Not sure when I will be able to set all up and running again. Very frustrating situation. Will you be able to do your tests anyway?
Thanks.

NeverEatYellowSwissSnow, 2022-01-19:
@aquette Any news on this? Were you able to try this out? Any hope for the next nut release this is going to work? Or is it really a USB only issue of Raspberriy 4 in general?

jimklimov, 2022-01-20:
Looking at issue labels, Raspberries of different models were implicated in
USB issues rather often.

Recent changes to NUT master branch now include support for libusb-1.0
which might be more stable by itself than 0.1 before it - and it is a large
milestone, getting us nearer to a new release.

IIRC some reports indicated issues with interrupts on some platforms, so
`pollonly` option might help too.

NeverEatYellowSwissSnow, 2022-01-27:
> IIRC some reports indicated issues with interrupts on some platforms, so
`pollonly` option might help too.>

Thanks for the hint. Tried the `pollonly` option you mentioned but unfortunately it does not change anything.

Rimers, 2022-04-27:
It seems to me that it is more of a general issue affecting devices using  passthru to on ESXI.
I have 29 servers running vmware esxi 6.5 Update 3 that all have this issue.
Our setup is HP ProLiant DL20 Gen9 server with Eaton 5SC UPS connected to the server using Eaton supplied USB cable.
The UPS is passthru to an Ubuntu 18.04 or 20.04 VM.
Both Ubuntu versions have same issue and are reporting Driver not connected.

Below info is pulled from Ubuntu 20.04 server 
# nut-scanner --usb_scan
SNMP library not found. SNMP search disabled.
Neon library not found. XML search disabled.
IPMI library not found. IPMI search disabled.
Scanning USB bus.
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0463"
        productid = "FFFF"
        product = "Eaton 5SC"
        serial = "G145H51037"
        vendor = "EATON"
        bus = "001"

/etc/nut/ups.conf content:
[eaton]
        driver = "usbhid-ups"
        port = "auto"
        ignorelb
        override.battery.charge.low = 75
        override.battery.charge.restart = 75

Above setup has been working before on all of our servers.

# USB_DEBUG=3 /lib/nut/usbhid-ups -DDDDD -a eaton -u root
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     send_to_all: SETINFO driver.parameter.port "auto"
   0.000241     send_to_all: SETINFO driver.flag.ignorelb "enabled"
   0.000342     send_to_all: SETINFO battery.charge.low "75"
   0.000427     send_to_all: SETFLAGS battery.charge.low
   0.000503     send_to_all: SETINFO battery.charge.restart "75"
   0.000586     send_to_all: SETFLAGS battery.charge.restart
   0.000648     debug level is '5'
   0.001888     upsdrv_initups...
usb_set_debug: Setting debugging level to 3 (on)
usb_os_init: Found USB VFS at /dev/bus/usb
usb_os_find_busses: Found 002
usb_os_find_busses: Found 001
usb_os_find_devices: Found 001 on 002
skipping descriptor 0x30
skipped 1 class/vendor specific endpoint descriptors
usb_os_find_devices: Found 013 on 001
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 005 on 001
usb_os_find_devices: Found 004 on 001
usb_os_find_devices: Found 002 on 001
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 001 on 001
error obtaining child information: Inappropriate ioctl for device
error obtaining child information: Inappropriate ioctl for device
   0.697191     Checking device (1D6B/0003) (002/001)
   0.725100     - VendorID: 1d6b
   0.725335     - ProductID: 0003
   0.725523     - Manufacturer: Linux 5.4.0-109-generic xhci-hcd
   0.725647     - Product: xHCI Host Controller
   0.725752     - Serial Number: 0000:0b:00.0
   0.725842     - Bus: 002
   0.725952     - Device release number: 0504
   0.726065     Trying to match device
   0.726193     Device does not match - skipping
   0.726476     Checking device (0463/FFFF) (001/013)
   1.012462     - VendorID: 0463
   1.012711     - ProductID: ffff
   1.012903     - Manufacturer: EATON
   1.013111     - Product: Eaton 5SC
   1.013279     - Serial Number: G145H51037
   1.013362     - Bus: 001
   1.013440     - Device release number: 0202
   1.013517     Trying to match device
   1.013598     Device matches
   1.013689     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   1.064580     HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 6a 06
   1.064728     HID descriptor length (method 1) 1642
   1.064847     i=0, extra[i]=09, extra[i+1]=21
   1.064977     HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 d3 03
   1.065096     HID descriptor length (method 2) 979
   1.065210     Eaton device v2.02. Using full report descriptor
   1.065323     Warning: two different HID descriptors retrieved (Reportlen = 1642 vs. 979)
   1.065483     HID descriptor length 1642
USB error: error sending control message: Protocol error
   3.109400     Unable to get Report descriptor: Protocol error
   3.109615     Checking device (0E0F/0002) (001/005)
   3.225426     - VendorID: 0e0f
   3.225670     - ProductID: 0002
   3.225862     - Manufacturer: unknown
   3.226048     - Product: VMware Virtual USB Hub
   3.226242     - Serial Number: unknown
   3.226362     - Bus: 001
   3.226441     - Device release number: 0100
   3.226560     Trying to match device
   3.226676     Device does not match - skipping
   3.264956     Checking device (0E0F/0002) (001/004)
   3.381408     - VendorID: 0e0f
   3.381646     - ProductID: 0002
   3.381837     - Manufacturer: unknown
   3.382026     - Product: VMware Virtual USB Hub
   3.382177     - Serial Number: unknown
   3.382257     - Bus: 001
   3.382370     - Device release number: 0100
   3.382494     Trying to match device
   3.382621     Device does not match - skipping
   3.421069     Checking device (0E0F/0003) (001/002)
   3.421505     - VendorID: 0e0f
   3.421715     - ProductID: 0003
   3.421902     - Manufacturer: VMware
   3.422099     - Product: VMware Virtual USB Mouse
   3.422248     - Serial Number: unknown
   3.422352     - Bus: 001
   3.422440     - Device release number: 0103
   3.422562     Trying to match device
   3.422690     Device does not match - skipping
   3.422807     Checking device (1D6B/0002) (001/001)
   3.423010     - VendorID: 1d6b
   3.423207     - ProductID: 0002
   3.423395     - Manufacturer: Linux 5.4.0-109-generic xhci-hcd
   3.423586     - Product: xHCI Host Controller
   3.423751     - Serial Number: 0000:0b:00.0
   3.423850     - Bus: 001
   3.423971     - Device release number: 0504
   3.424088     Trying to match device
   3.424209     Device does not match - skipping
   3.424339     No appropriate HID device found
   3.424467     No matching HID UPS found

Grandma-Betty, 2022-04-27:
@Rimers Yes, I mentioned this too on both of my x86 systems when I tested it again recently. Running vSphere 7, so it's not only vSphere 6 related I guess.

marcelmah, 2022-05-03:
I just noticed the same

Running Home Assistant OS, with the NUT addon on a VMware ESXi 7 U3d, used to work... now the driver wont start:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] nut: applying... 
[fix-attrs.d] nut: exited 0.
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.10.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 7.6  (amd64 / qemux86-64)
 Home Assistant Core: 2022.4.7
 Home Assistant Supervisor: 2022.04.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] nut.sh: executing... 
[12:32:59] INFO: Setting mode to netserver...
[12:32:59] INFO: Connected USB devices:
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 002 Device 004: ID 0463:ffff MGE UPS Systems UPS
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[12:32:59] INFO: Generating /etc/nut/upsd.users...
[12:32:59] INFO: Configuring user: ups
[12:32:59] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[12:33:00] INFO: Configuring Device named myups...
[12:33:00] INFO: Starting the UPS drivers...
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
No matching HID UPS found
Driver failed to start (exit status=1)
Network UPS Tools - UPS driver controller 2.7.4
[cont-init.d] nut.sh: exited 1.
[cont-finish.d] executing container finish scripts...
[cont-finish.d] 99-message.sh: executing... 
-----------------------------------------------------------
                Oops! Something went wrong.

 We are so sorry, but something went terribly wrong when
 starting or running this add-on.
 
 Be sure to check the log above, line by line, for hints.
-----------------------------------------------------------
[cont-finish.d] 99-message.sh: exited 0.
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

jimklimov, 2022-05-04:
For vendorid 0x0463 in particular seems there were "usb quirks" in some
linux kernel versions which can be related to the issue and only were
useful for some firmware versions.

Otherwise general suggestions about devfs permission (udev rules etc.)

Jim

Grandma-Betty, 2022-06-13:
> Otherwise general suggestions about devfs permission (udev rules etc.) 

Does this also count when `RUN_AS_USER root` is configured within `upsmon.conf`?

jimklimov, 2022-06-13:
udev rules apply to NUT driver daemons that talk to HW. roots used for
other daemons are not related to that and serve other purposes (e.g.
shutdown from upsmon), note it usually forks a non-root part and that
suffices.

Grandma-Betty, 2022-06-13:
> udev rules apply to NUT driver daemons that talk to HW. roots used for other daemons are not related to that and serve other purposes (e.g. shutdown from upsmon), note it usually forks a non-root part and that suffices.

Thanks for clearing things up.
Unfortunately NUT 2.8.0 has not been released yet within the standard ubuntu and/or debian packages repository. Hopefully the latest NUT version will change those issues we are having.

Andr3asB, 2022-11-06:
I see the same issue on an Intel NUC running VMware ESXi 7 and Debian 11 with the nut-packages upgraded to testing (bookworm) as this contains nut 2.8.0 (2.8.0-5). So I've tried to setup nut with my Eaton 5P1550iR:

```
# /lib/nut/usbhid-ups -V
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
```
and 
```
# /lib/nut/usbhid-ups -a myeaton -DDDDD
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '5'
   0.000887     [D5] send_to_all: SETINFO device.type "ups"
   0.000973     [D2] Initializing an USB-connected UPS with library libusb-1.0.24 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.001037     [D1] upsdrv_initups (non-SHUT)...
   0.003686     [D2] Checking device 1 of 5 (1D6B/0002)
   0.003846     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.003904     [D2] Checking device 2 of 5 (0463/FFFF)
   0.102691     [D1] nut_libusb_open get iManufacturer failed, retrying...
   1.104995     [D1] nut_libusb_open get iManufacturer failed, retrying...
   3.106244     [D1] nut_libusb_open get iManufacturer failed, retrying...
   4.107564     [D1] nut_libusb_open get iProduct failed, retrying...
   5.110150     [D1] nut_libusb_open get iProduct failed, retrying...
   6.110588     [D1] nut_libusb_open get iProduct failed, retrying...
   7.111117     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   8.114161     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   9.114659     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   9.114684     [D2] - VendorID: 0463
   9.114689     [D2] - ProductID: ffff
   9.114693     [D2] - Manufacturer: unknown
   9.114698     [D2] - Product: unknown
   9.114702     [D2] - Serial Number: unknown
   9.114706     [D2] - Bus: 002
   9.114710     [D2] - Device: unknown
   9.114715     [D2] - Device release number: 0202
   9.114720     [D2] Trying to match device
   9.114725     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   9.114761     [D3] match_function_regex: matching a device...
   9.114769     [D2] Device matches
   9.114773     [D2] Reading first configuration descriptor
   9.114790     [D3] libusb_kernel_driver_active() returned 0
   9.114845     [D2] Claimed interface 0 successfully
   9.114851     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
  11.167386     [D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 55 08
  11.167409     [D3] HID descriptor length (method 1) 2133
  11.167414     [D4] i=0, extra[i]=09, extra[i+1]=21
  11.167421     [D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 25 04
  11.167427     [D3] HID descriptor length (method 2) 1061
  11.167431     [D1] Eaton device v2.02. Using full report descriptor
  11.167436     [D2] Warning: two different HID descriptors retrieved (Reportlen = 2133 vs. 1061)
  11.167441     [D2] HID descriptor length 2133
  13.212939     [D2] Unable to get Report descriptor: Resource temporarily unavailable
  13.212986     [D2] Checking device 3 of 5 (0E0F/0002)
  13.213006     [D1] Failed to open device (0E0F/0002), skipping: Access denied (insufficient permissions)
  13.213009     [D2] Checking device 4 of 5 (0E0F/0003)
  13.213019     [D1] Failed to open device (0E0F/0003), skipping: Access denied (insufficient permissions)
  13.213022     [D2] Checking device 5 of 5 (1D6B/0001)
  13.213028     [D1] Failed to open device (1D6B/0001), skipping: Access denied (insufficient permissions)
  13.213031     [D2] libusb1: No appropriate HID device found
  13.213034     libusb1: Could not open any HID devices: insufficient permissions on everything
  13.213037     No matching HID UPS found
```
but the UPS seems to be available and recognized properly:
```
# usb-devices
T:  Bus=02 Lev=02 Prnt=03 Port=00 Cnt=01 Dev#=  4 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=0463 ProdID=ffff Rev=02.02
S:  Manufacturer=EATON
S:  Product=Eaton 5P
S:  SerialNumber=G117M14077
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=20mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=(none)
```
Can you see where this is going wrong and why I get the "[D1] nut_libusb_open get iManufacturer failed, retrying..." etc. errors?

I was also thinking about setting up a new VM with debian testing so any interdependencies between stable/testing packages could be ruled out.

jimklimov, 2022-11-07:
Just wanted to confirm if it is not about a conflict with another running driver instance (e.g. the new `nut-driver-enumerator` units, if integrated and enabled by packaging, could automatically generate you a `nut-driver@myeaton.service` based on `ups.conf` content changes). But at least in my system, running two copies of same driver config looks different:
````
root@pve:~# /usr/local/ups/bin/usbhid-ups -a eco650 -DDDDDD
Network UPS Tools - Generic HID driver 0.49 (2.8.0-Windows-58-g326f90866)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '6'
   0.000621     [D1] Succeeded to become_user(nut): now UID=77 GID=77
   0.000637     [D5] send_to_all: SETINFO device.type "ups"
   0.000646     [D2] Initializing an USB-connected UPS with library libusb-1.0.24 (API: 0x1000108) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000651     [D1] upsdrv_initups (non-SHUT)...
   0.003886     [D2] Checking device 1 of 9 (1D6B/0003)
   0.003908     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.003914     [D2] Checking device 2 of 9 (03F0/0D4A)
   0.003923     [D1] Failed to open device (03F0/0D4A), skipping: Access denied (insufficient permissions)
   0.003927     [D2] Checking device 3 of 9 (1D6B/0002)
   0.003934     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.003938     [D2] Checking device 4 of 9 (1D6B/0003)
   0.003945     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.003949     [D2] Checking device 5 of 9 (0463/FFFF)
   0.425586     [D2] - VendorID: 0463
   0.425596     [D2] - ProductID: ffff
   0.425601     [D2] - Manufacturer: EATON
   0.425605     [D2] - Product: Ellipse ECO
   0.425610     [D2] - Serial Number: 000000000
   0.425614     [D2] - Bus: 003
   0.425618     [D2] - Device: unknown
   0.425622     [D2] - Device release number: 0100
   0.425626     [D2] Trying to match device
   0.425630     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.425637     [D3] match_function_regex: matching a device...
   0.425705     [D2] Device matches
   0.425710     [D2] Reading first configuration descriptor
   0.425720     [D3] libusb_kernel_driver_active() returned 0
   0.425726     [D2] failed to claim USB device: Resource busy
   0.425731     [D2] Kernel driver already detached
   0.425735     [D2] failed to claim USB device: Resource busy
   0.425740     [D2] Kernel driver already detached
   0.425744     [D2] failed to claim USB device: Resource busy
   0.425750     [D2] Kernel driver already detached
   0.425753     [D2] failed to claim USB device: Resource busy
   0.425758     [D2] Kernel driver already detached
   0.425765     Can't claim USB device [0463:ffff]@0/0: Entity not found
````

I do remember seeing the pattern with three retries and lack of basic entries recently, but can't well put my memory index on it... possibly during NUT for Windows experiments and some lack of low-level libUSB driver on that system.

Virtualization pass-through has its quirks, but one thing to rule out is that the host system is not grabbing the device for itself (more probable for HID) and honestly is passing that through. Also check if the driver run-time user (built-in or in settings) matches what is in udev.rules (or newly udev.hwdb) file. Though that would probably manifest in explicit `Access denied` verdicts...

cuppeanuts, 2023-06-03:
I see the same issue on EXSI 7.0 update 2.0. I create a VM with OMV 5 and passthru the USB device (Santak TG BOX 850) to the VM. I can see the device with `lsusb` in OMV. However, running with the following command `/sbin/usbhid-ups -a ups -u root -DDD` in OMV will give an error that "Unable to get Report descriptor: Invalid or incomplete multibyte or wide character".  

Then I passthru the whole USB controller ( which is "Interl 200 Series/Z370 Chipset Family USB 3.0 xHCI Controller" in my case）to OMV. The NUT works fine. I believe @Rimers was right. It seems a general issue of EXSI and passthru the whole USB controller might solve the problem.

Andr3asB, 2023-08-02:
I solved my issue by updating to ESXi 8.0 (U1) and installing a new Debian 12 server. For the setup I used this quick guide: https://justit.eu/quick-setup-network-ups-tools-nut-server-on-debian-10/

closed 2023-08-03 by jimklimov

marcelmah, 2023-09-30:
I can confirm it working again after updating ESXi U3g to U3o