networkupstools/nut #1014: Upsilon 2000 values never updated (Crown CMU-SP1200IEC USB)

JulyIghor, 2021-04-13:
I used this config

blazer_usb -a Crown -DDDD -u root

JulyIghor, 2021-05-08:
Also the same numbers persistent even after PC restart.
And they start changing once I connect it to a Windows with Upsilon 2000 driver installed

Yellow-Horror, 2021-06-20:
Due to proprietary Crown software debug log, it sends `BL` command to the UPS between `Q1`. I think, that `BL` is required to renew `Q1` output.

JulyIghor, 2021-06-20:
> Due to proprietary Crown software debug log, it sends `BL` command to the UPS between `Q1`. I think, that `BL` is required to renew `Q1` output.

Thanks for the reply.
How can I test that?

Yellow-Horror, 2021-06-20:
After some research, I revisited the previous conclusion. I looked at the wrong log in a proprietary program (UART log instead of USB). Now I think we are dealing with a case where order of commands is important to UPS. Try reading the discussion on [this request](https://github.com/networkupstools/nut/pull/638) and using a similar config.

JulyIghor, 2021-06-20:
> After some research, I revisited the previous conclusion. I looked at the wrong log in a proprietary program (UART log instead of USB). Now I think we are dealing with a case where order of commands [...]

I tried that, I'm getting this output
[image]

Yellow-Horror, 2021-06-20:
I think, we have to build NUT from sources to use the new 'hunnox' protocol subdriver. Or wait for a new release, which may take unknown amount of time (the current version 2.7.4 was released in 2016)
:(

JulyIghor, 2021-06-20:
> I think, we have to build NUT from sources to use the new 'hunnox' protocol subdriver. Or wait for a new release, which may take unknown amount of time (the current version 2.7.4 was released in 2016)
> :(

Compiled from source. Here is output with the same config.

````   0.188576	[D2] Checking device (0001/0000) (002/006)
   0.216200	[D2] - VendorID: 0001
   0.216251	[D2] - ProductID: 0000
   0.216297	[D2] - Manufacturer: unknown
   0.216347	[D2] - Product: MEC0003
   0.216392	[D2] - Serial Number: unknown
   0.216452	[D2] - Bus: 002
   0.216498	[D2] - Device: 006
   0.216526	[D2] - Device release number: 0100
   0.216554	[D2] Trying to match device
   0.216598	[D3] match_function_regex: matching a device...
   0.216863	[D2] Device matches
   0.216954	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.225819	[D1] First supported language ID: 0x409 (please report to the NUT maintainer!)
   0.225870	[D2] Skipping protocol Voltronic 0.06
   0.225921	[D2] Skipping protocol Voltronic-QS 0.07
   0.225967	[D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.225996	[D2] Skipping protocol Mustek 0.07
   0.226024	[D2] Skipping protocol Megatec/old 0.07
   0.226052	[D2] Skipping protocol BestUPS 0.06
   0.226098	[D2] Skipping protocol Mecer 0.07
   0.226127	[D2] Skipping protocol Megatec 0.06
   0.226155	[D2] Skipping protocol Zinto 0.06
   0.226201	[D2] Skipping protocol Masterguard 0.01
   0.226232	[D3] send: Q1
   0.226262	[D4] command index: 0x03
   0.275849	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.275945	[D3] send: FW?
   0.276021	[D3] read: FW?
   0.276080	[D2] qx_process_answer: short reply (ups.firmware)
   0.276119	[D3] send: Q1
   0.276155	[D4] command index: 0x03
   0.322911	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.323010	[D3] send: FW?
   0.323094	[D3] read: FW?
   0.323161	[D2] qx_process_answer: short reply (ups.firmware)
   0.323198	[D3] send: Q1
   0.323234	[D4] command index: 0x03
   0.372287	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.372440	[D3] send: FW?
   0.372510	[D3] read: FW?
   0.372570	[D2] qx_process_answer: short reply (ups.firmware)
   0.372608	[D2] Skipping protocol Q1 0.07
   0.372665	Device not supported!
   0.372701	Device not supported!````

JulyIghor, 2021-06-20:
> I think, we have to build NUT from sources to use the new 'hunnox' protocol subdriver. Or wait for a new release, which may take unknown amount of time (the current version 2.7.4 was released in 2016)
> :(

I compiled it from source.
With this config:
```
[Crown]
driver = "nutdrv_qx"
port = "auto"
vendorid = "0001"
productid = "0000"
subdriver = fabula
protocol = hunnox
langid_fix = 0x0409
novendor
noscanlangid
```
I'm getting that output:
```
   0.198767	[D3] match_function_regex: matching a device...
   0.199152	[D2] Device matches
   0.199307	[D2] failed to claim USB device: could not claim interface 0: Device or resource busy
   0.199452	[D2] detached kernel driver from USB device...
   0.199623	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.213877	[D1] First supported language ID: 0x409 (please report to the NUT maintainer!)
   0.213944	[D2] Skipping protocol Voltronic 0.06
   0.213982	[D2] Skipping protocol Voltronic-QS 0.07
   0.214014	[D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.214041	[D2] Skipping protocol Mustek 0.07
   0.214067	[D2] Skipping protocol Megatec/old 0.07
   0.214093	[D2] Skipping protocol BestUPS 0.06
   0.214189	[D2] Skipping protocol Mecer 0.07
   0.214216	[D2] Skipping protocol Megatec 0.06
   0.214265	[D2] Skipping protocol Zinto 0.06
   0.214297	[D2] Skipping protocol Masterguard 0.01
   0.214327	[D3] send: Q1
   0.214354	[D4] command index: 0x03
   0.253629	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.253736	Using protocol: Hunnox 0.01
   0.253775	[D2] blazer_initups: skipping device.mfr
   0.253802	[D2] blazer_initups: skipping device.model
   0.253828	[D2] blazer_initups: skipping ups.firmware
   0.253871	[D1] upsdrv_initinfo...
   0.253903	[D3] send: Q1
   0.253929	[D4] command index: 0x03
   0.301755	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.301873	[D3] send: F
   0.301907	[D4] command index: 0x0d
   0.326072	[D3] read: #000.0 0.0 00.00 00.0
   0.326185	No values for battery high/low voltages
   0.326223	Using 'guesstimation' (low: 0.000000, high: 0.000000)!
   0.326250	Battery runtime will not be calculated (runtimecal not set)
   0.326286	[D1] upsdrv_updateinfo...
   0.326314	[D1] Quick update...
   0.326341	[D3] send: Q1
   0.326367	[D4] command index: 0x03
   0.364177	[D3] read: (000.0 000.0 000.0 000 00.0 0.0 No  No  No  No
   0.364255	[D2] ups_infoval_set: non numerical value [ups.temperature: o  N]
   0.364287	[D1] Communications with the UPS lost: status read failed!
   0.367045	[D2] dstate_init: sock /var/state/ups/nutdrv_qx-Crown open on fd 5
   0.367117	[D1] upsdrv_updateinfo...
   0.367171	[D1] Quick update...
   0.367205	[D3] send: Q1
   0.367243	[D4] command index: 0x03
   0.421513	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000 No
   0.421623	[D3] blazer_process_status_bits: unexpected value  @43->
   0.421664	[D4] ups_infoval_set: failed to preprocess value [ups.status:  ]
   0.421697	[D1] Communications with the UPS lost: status read failed!
   2.369219	[D1] upsdrv_updateinfo...
   2.369292	[D1] Quick update...
   2.369323	[D3] send: Q1
   2.369351	[D4] command index: 0x03
   2.408624	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   4.371315	[D1] upsdrv_updateinfo...
   4.371399	[D1] Quick update...
   4.371444	[D3] send: Q1
   4.371485	[D4] command index: 0x03
   4.418288	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   6.373493	[D1] upsdrv_updateinfo...
   6.373594	[D1] Quick update...
   6.373632	[D3] send: Q1
   6.373686	[D4] command index: 0x03
   6.419247	[D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   ```

JulyIghor, 2021-06-20:
When used this config
```
[Crown]
driver = "nutdrv_qx"
port = "auto"
vendorid = "0001"
productid = "0000"
protocol = hunnox
langid_fix = 0x0409
novendor
noscanlangid
```
Looks like it kick out USB device and I'm getting Input/output error (-5) until replugged

JulyIghor, 2021-06-20:
> I think, we have to build NUT from sources to use the new 'hunnox' protocol subdriver. Or wait for a new release, which may take unknown amount of time (the current version 2.7.4 was released in 2016)
> :(

Thanks, it is working with hunnox

closed 2021-06-20 by JulyIghor

Yellow-Horror, 2021-06-20:
> Thanks, it is working with hunnox
Can you post here the working config?

JulyIghor, 2021-06-20:
This one works for me
```
[ups]
driver = nutdrv_qx
port = auto
vendorid = "0001"
productid = "0000"
protocol = hunnox
langid_fix = 0x0409
novendor
noscanlangid
```

JulyIghor, 2021-06-20:
Also it fails sometimes with 'read: Input/output error (-5)'
I have to re plug usb cable to get it work again.

commit 3929bdf6 (2022-04-25): driver.list.in: Add HCL info for Crown CMU-SP1200IEC (USB)

Closes: #1014