networkupstools/nut #1017: feature request : Add support for Huawei Modbus UPSs

sumsethan, 2021-04-16:
here comes the Huawei UPS2000 series UPS Modbus  Protocol  Manual
https://support.huawei.com/view/PdfRead/EDOC1000110696/SUPE_DOC/k001/document.pdf

It has a USB port and a RS232 port
USB port seems like behind a translation chip " ID 04e2:1410 Exar Corp. XR21V1410 USB-UART IC"(on lsusb command) though RS232

Any ideas guys?

sumsethan, 2021-04-16:
connecting usb port recognized as a  /dev/XRUSB0 device

jimklimov, 2021-04-19:
I believe currently there is one modbus-aware driver, `phoenixcontact_modbus`, with some efforts discussed to introduce a second. While the media connection is serial (or emulated to that), though I think RS-485 is more frequent than RS-232 for that (not that they would be easy to tell apart by looking), NUT may be lacking a driver that talks an actual protocol for that.

Actually, one of the discussed efforts is indeed a Huawei UPS2000 driver in #954 which was not merged yet due to a burst of activity and then silence from the contributor. But I suppose it would help if you can build that branch from biergaizi:ups2000 and confirm whether it works for your device.

whc2001, 2021-07-21:
Any progress on that?

biergaizi, 2021-09-23:
> @sumsethan: connecting usb port recognized as a /dev/ttyXRUSB0 device

This is new information for me. What system are you using? 

The RX21V1410 chip in UPS2000 doesn't work on the official upstream Linux kernel, it can be recognized as a CDC device, but not functional (presumably because the chip is in a special configuration). On your system, if you see that the device shows as a nonstandard name like `ttyXRUSB0`, it means you're using an unofficial, out-of-tree driver unsupported by the upstream Linux kernel, so it must be a local patch from your Linux distro.

I didn't even realize there is actually a distro that included out-of-tree driver for the chip. In all my documentation I always claim that USB is entirely unsupported, but apparently with the unofficial driver installed, USB should actually work.

Again, what is your distro?

biergaizi, 2021-09-23:
Following the same line of investigation, apparently the driver for MaxLinear RX21V1410 has just been mainlined to the upstream in January and April, and an official driver (with the standard serial port names, not the nonstandard `ttyXRUSB0`) is already included in Linux 5.13, 5.14, and 5.15-RC. This is an important development.

Perhaps it was a good decision not to merge my patch back in January. Time to retest the driver and update the documentation.

biergaizi, 2021-09-24:
Never mind the question.

Linux 5.12 and newer kernels already have a mainline driver `xr_serial` and it's confirmed to work.

biergaizi, 2021-09-24:
@whc2001: now it's 100% completed. Waiting for a code review and a merge.

whc2001, 2021-09-24:
Glad to hear, thanks for the effort!

jimklimov, 2021-11-14:
Now that the driver became part of NUT master branch, may I ask people watching this issue to try a build and report if it works for them? :)

ultinous-dancsa, 2022-01-24:
We've just bought a UPS2000-A-1KTTS. I'm not very familiar with building C code, i had some issues. (fresh master 88e8a83)
Linux mgmt01 4.4.0-210-generic #242-Ubuntu SMP Fri Apr 16 09:57:56 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux                                                                                                                                                                     
Ubuntu 16.04 (yeah, old, currently under replacement)
config: ./configure --with-user=ups --with-group=nut --with-usb --with-drivers=huawei-ups2000
i'm not sure if it's a bug, but configure won't complain about missing modbus lib, but the make will fail. apt-get install libmodbus-dev is needed. 

root@mgmt01:/usr/local/ups/etc# cat ups.conf
[huvi]
        driver = huawei-ups2000
        port = /dev/ttyXRUSB0
        desc = "Workstation"
root@mgmt01:/usr/local/ups/sbin# /usr/local/ups/bin/upsc huvi@localhost 
battery.capacity: 9
battery.charge: 100.0
battery.charger.status: charging
battery.packs: 2
battery.runtime: 8692
battery.voltage: 27.4
device.mfr: Huawei
device.model: UPS2000A
...
ups.firmware: V2R1C1SPC50
ups.firmware.aux: P1.0-D1.0

seems working

jimklimov, 2022-01-26:
Thanks, I suppose that can be treated as a bug. It seems that `configure`
script does not really know or care about the ever-changing population of
drivers, and is geared to enable/disable whole categories (e.g.
with/without modbus) so did not require to have libmodbus as a dependency
for particular enabled driver - just opted to use one if available.

jimklimov, 2022-02-08:
PR #1277 should handle the drive-by `--with-drivers=...` deficiency found above (thanks!)

I suppose the original issue for Huawei Modbus can be marked as achieved.

closed 2022-02-08 by jimklimov

davidmarquesneves, 2022-09-29:
Here is the page of NUT for this product:

https://networkupstools.org/docs/man/huawei-ups2000.html

biergaizi, 2022-10-21:
> root@mgmt01:/usr/local/ups/etc# cat ups.conf
> [huvi]
> driver = huawei-ups2000
> port = /dev/ttyXRUSB0
> desc = "Workstation"
> root@mgmt01:/usr/local/ups/sbin# /usr/local/ups/bin/upsc huvi@localhost [...]
...
> ups.firmware: V2R1C1SPC50
> ups.firmware.aux: P1.0-D1.0
>
> seems working
> (I'm still working myself through the documentation as i'm new to nut)

Thanks for reporting. I'll add your model to the list of tested units in the documentation.