networkupstools/nut #1022: Poll UPS failed - Data stale

HyP3r-, 2021-04-21:
Hello I have a Bluewalker UPS Powerwalker VI 2200 LCD UPS. Unfortunately, always after some time comes the following error messages:

> Apr 02 22:30:27 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: could not claim interface 0: Device or resource busy
> Apr 02 22:30:27 ffwzie-vm01 usbhid-ups[2387]: libusb_get_report: error sending control message: No such device
> Apr 04 04:52:24 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: could not claim interface 0: Device or resource busy
> Apr 04 04:52:24 ffwzie-vm01 usbhid-ups[2387]: libusb_get_report: error sending control message: No such device
> Apr 05 10:37:36 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: error submitting URB: No such device
> Apr 06 14:49:44 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: could not claim interface 0: Device or resource busy
> Apr 06 14:49:59 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: error submitting URB: No such device
> Apr 07 20:52:55 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: could not claim interface 0: Device or resource busy
> Apr 07 20:52:55 ffwzie-vm01 usbhid-ups[2387]: libusb_get_report: error sending control message: No such device
> Apr 11 00:53:15 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: error submitting URB: No such device
> Apr 16 06:50:12 ffwzie-vm01 systemd[1]: /lib/systemd/system/nut-monitor.service:6: PIDFile= references path below legacy directory /var/run/, updating /var/run/nut/upsmon.pid → /run/nut/upsmon.pid; pleas
> Apr 16 06:50:13 ffwzie-vm01 systemd[1]: nut-monitor.service: Supervising process 2399 which is not our child. We'll most likely not notice when it exits.
> Apr 16 06:50:14 ffwzie-vm01 systemd[1]: /lib/systemd/system/nut-monitor.service:6: PIDFile= references path below legacy directory /var/run/, updating /var/run/nut/upsmon.pid → /run/nut/upsmon.pid; pleas
> Apr 16 06:50:14 ffwzie-vm01 systemd[1]: nut-monitor.service: Supervising process 2399 which is not our child. We'll most likely not notice when it exits.
> Apr 16 13:18:24 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: error sending control message: Value too large for defined data type
> Apr 16 13:18:24 ffwzie-vm01 usbhid-ups[2387]: libusb_get_report: error sending control message: No such device
> Apr 17 19:41:05 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: error submitting URB: No such device
> Apr 17 23:27:38 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: error submitting URB: No such device
> Apr 18 19:46:40 ffwzie-vm01 usbhid-ups[2387]: libusb_get_interrupt: error submitting URB: No such device
> Apr 21 18:01:01 ffwzie-vm01 upsd[2391]: Data for UPS [ffwzie-vm01] is stale - check driver
> Apr 21 18:01:04 ffwzie-vm01 upsmon[2399]: Poll UPS [ffwzie-vm01@localhost] failed - Data stale
> Apr 21 18:01:04 ffwzie-vm01 upsmon[2399]: Communications with UPS ffwzie-vm01@localhost lost
> Apr 21 18:01:09 ffwzie-vm01 upsmon[2399]: Poll UPS [ffwzie-vm01@localhost] failed - Data stale

In this case the server has been running for 55 days. So after 55 days of operation now these error messages come. What could be the reason for this?

> lsusb
> Bus 001 Device 050: ID 06da:ffff Phoenixtec Power Co., Ltd.

> cat ups.conf
```
[ffwzie-vm01]
    driver = usbhid-ups
    port = auto
    vendorid = 06da
    desc = "ffwzie-vm01"
    pollinterval = 15
```

jimklimov, 2021-04-21:
For the most part the messages indicate that the OS lost the USB device,
not much that NUT can do to fix that part. Might be some power-saving
disabling the chip, or something else frozen on the hub and a USB reset
restarting every connection (either intentionally, or because some vendors
cut corners and implement a port reset same as hub reset).

Guessing by your device name, containing "vm", I might add potentially some
delays or flaws in virtualization/passthrough layer as a possible problem
source.

On a busy machine (hypervisor with many guests, or other
compute/IO/context-switch heavy systems) or on an under-powered machine
(preempted VM just not on CPU) there might also be some exceeded hardcoded
timeouts in any layer of the OS, drivers and software down to libusb used
by NUT driver, perceiving a loss of device because last data exchange was a
bit too long ago.

In that log I see the lost device reported all over April, but the driver
only aborted once? I believe that means retries succeeded to find (maybe
reinitialize) the USB device except once.

For context: which OS and version of NUT (packaged or custom built) is that
happening with? If you do/can run a custom build, are you in position to
check if native libusb-1.0 support from some of the competing branches in
issue #300 would behave better?

HyP3r-, 2021-04-21:
> For the most part the messages indicate that the OS lost the USB device,
> not much that NUT can do to fix that part. Might be some power-saving
> disabling the chip, or something else frozen on the hub [...]
This is a Debian 10 which runs on a HP ProLiant DL360. I mention this because I can't imagine that such a server has energy saving mechanisms for the USB system. How can I find out if something like this happened at this moment?

> Guessing by your device name, containing "vm", I might add potentially some
> delays or flaws in virtualization/passthrough layer as a possible problem
> source.
It is not a virtual machine but the hypervisor. As platform I use Debian 10 with Proxmox VE.
 
> On a busy machine (hypervisor with many guests, or other
> compute/IO/context-switch heavy systems) or on an under-powered machine
> (preempted VM just not on CPU) there might also be some exceeded [...]
As already described, the system actually runs a hypervisor. And the system is also relatively busy. But somehow I can't imagine that this is enough to hang up so much.

> In that log I see the lost device reported all over April, but the driver
> only aborted once? I believe that means retries succeeded to find (maybe
> reinitialize) the USB device except once.
It seems so yes: the last few months the connection is repeatedly interrupted, only unfortunately today it did not work to restore the connection.

> For context: which OS and version of NUT (packaged or custom built) is that
> happening with? If you do/can run a custom build, are you in position to
> check if native libusb-1.0 support from some of the competing branches in
> issue #300 would behave better?
I just tried a few times to compile the branches "libusb-1.0+0.1", "libusb-1.0" and "libusb-compat-1.0" but with no full success: The branches "libusb-1.0+0.1" and "libusb-1.0" I could compile but were unable to use them. I always get the following error message:

> upsdrvctl[44892]: Failed to read pid from /var/state/ups/usbhid-ups1-myups.pid
> upsdrvctl[44892]: Network UPS Tools - Generic HID driver 0.42 (2.7.4-420-g2999c95f)
> upsdrvctl[44892]: USB communication driver (libusb 1.0) 0.02
> upsdrvctl[44892]: No matching HID UPS found
> upsdrvctl[44892]: Driver failed to start (exit status=1)
> upsmon[18049]: Poll UPS [myups@localhost] failed - Driver not connected

jimklimov, 2021-04-21:
> upsmon[18049]: Poll UPS ***@***.*** failed - Driver not connected

That failure to start probably means that it probed available USB devices
and did not find a supported vendor/product ID. Might be that as of those
branches the IDs for your UPS were not known yet, which is unlikely at
least if you were running last release before that, or that the device
remains not seen on the OS side.

Try starting the driver with `-DDDDD` argument for debug with higher
verbosity; it would mention IDs it saw in particular.

HyP3r-, 2021-04-23:
After I called `configure` Debian conform the library could be compiled and used. I will now observe the whole thing for a quarter. If it doesn't hang anymore, the ticket can be closed.

I will inform you in this issue.

Tonny2442, 2022-01-13:
I am having a similar problem, Debian 11 and Proxmox VE 7.1-8. The UPS appears working fine until randomly, it disconnects from the NUT, going data stale, then disconnecting from the host a while later.

```
Jan 13 12:50:25 proxmox upsmon[126914]: Poll UPS [apcups@localhost] failed - Data stale
Jan 13 12:50:30 proxmox upsmon[126914]: Poll UPS [apcups@localhost] failed - Data stale
Jan 13 12:50:31 proxmox usbhid-ups[126886]: libusb_get_report: error sending control message: No such device
Jan 13 12:50:31 proxmox kernel: [51736.933173] usb 4-1.4: USB disconnect, device number 5
Jan 13 12:50:31 proxmox usbhid-ups[126886]: Parse error on sock: Invalid ctx buffer
Jan 13 12:50:31 proxmox kernel: [51737.153935] usb 4-1.4: new full-speed USB device number 6 using ehci-pci
Jan 13 12:50:31 proxmox kernel: [51737.264672] usb 4-1.4: New USB device found, idVendor=051d, idProduct=0002, bcdDevice= 0.90
Jan 13 12:50:31 proxmox kernel: [51737.264677] usb 4-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 13 12:50:31 proxmox kernel: [51737.264679] usb 4-1.4: Product: Back-UPS ES 600M1 FW:928.a8 .D USB FW:a8 
Jan 13 12:50:31 proxmox kernel: [51737.264680] usb 4-1.4: Manufacturer: American Power Conversion
Jan 13 12:50:31 proxmox kernel: [51737.264682] usb 4-1.4: SerialNumber: 4B1949P17279  
Jan 13 12:50:31 proxmox kernel: [51737.267902] hid-generic 0003:051D:0002.0004: hiddev0,hidraw0: USB HID v1.00 Device [American Power Conversion Back-UPS ES 600M1 FW:928.a8 .D USB FW:a8 ] on usb-0000:00:1d.0-1.4/input0
Jan 13 12:50:31 proxmox usbhid-ups[126886]: Parse error on sock: Invalid ctx buffer
Jan 13 12:50:31 proxmox usbhid-ups[126886]: Parse error on sock: Invalid ctx buffer
Jan 13 12:50:31 proxmox usbhid-ups[126886]: Parse error on sock: Invalid ctx buffer
Jan 13 12:50:31 proxmox usbhid-ups[126886]: Parse error on sock: Invalid ctx buffer
Jan 13 12:50:31 proxmox upsd[126888]: UPS [apcups] data is no longer stale
Jan 13 12:50:31 proxmox upsd[126888]: Data for UPS [apcups] is stale - check driver
Jan 13 12:50:32 proxmox upsd[126888]: UPS [apcups] data is no longer stale
Jan 13 12:50:35 proxmox upsmon[126914]: Communications with UPS apcups@localhost established
```
I have my time until marked as stale (whatever it was called) set to 30 seconds. This solution actually solved the issue with my UPS, but a friend still has the issue.

graemev, 2024-02-29:
OK, I hit this issue, a reboot "fixed" it but I gathered some info before the reboot , might help somebody, esp if the real fix is to raise a kernel defect. (BTW this happened about 10 days after a real power cut...that would have been annoying if it had failed):

The messages:

```
Feb 25 14:06:57 ybox usbhid-ups[727]: libusb_get_interrupt: could not claim interface 0: Device or resource busy                                                                                                              
Feb 25 14:07:07 ybox usbhid-ups[727]: libusb_get_interrupt: error sending control message: Protocol error                                                                                                                     
Feb 25 14:07:17 ybox usbhid-ups[727]: libusb_get_interrupt: error sending control message: Protocol error                                                                                                                     
... 33K of similar messages  
```

Stop and restart nut-server (debian bullseye)
```

root@ybox:/var/log# systemctl start  nut-server                                                                                                                                                                               
root@ybox:/var/log# systemctl status nut-server                                                                                                                                                                               
● nut-server.service - Network UPS Tools - power devices information server                                                                                                                                                   
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)                                                                                                                                 
     Active: active (running) since Thu 2024-02-29 10:50:45 GMT; 38s ago                                                                                                                                                      
    Process: 1066411 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)                                                                                                                                                     
   Main PID: 1066412 (upsd)                                                                                                                                                                                                   
      Tasks: 1 (limit: 4476)                                                                                                                                                                                                  
     Memory: 772.0K                                                                                                                                                                                                           
        CPU: 33ms                                                                                                                                                                                                             
     CGroup: /system.slice/nut-server.service                                                                                                                                                                                 
             └─1066412 /lib/nut/upsd                                      

Feb 29 10:50:45 ybox upsd[1066411]: listening on 127.0.0.1 port 3493                                                                                                                                                          
Feb 29 10:50:45 ybox upsd[1066411]: listening on 10.117.0.152 port 3493                                                                                                                                                       
Feb 29 10:50:45 ybox upsd[1066411]: listening on 127.0.0.1 port 3493                                                                                                                                                          
Feb 29 10:50:45 ybox upsd[1066411]: Can't connect to UPS [qnapups] (usbhid-ups-qnapups): No such file or directory                                                                                                            
Feb 29 10:50:45 ybox upsd[1066411]: Can't connect to UPS [qnapups] (usbhid-ups-qnapups): No such file or directory                                                                                                            
Feb 29 10:50:45 ybox upsd[1066411]: /etc/nut/upsd.users is world readable                                                                                                                                                     
Feb 29 10:50:45 ybox upsd[1066411]: /etc/nut/upsd.users is world readable                                                                                                                                                     
Feb 29 10:50:45 ybox upsd[1066412]: Startup successful                                                                                                                                                                        
[... cut ...]
                                                                                                                                          
           **Device Status:     0x0000  **                                                                                                                                                                                       
             (Bus Powered)                                                                                                                                                                                  
   

```

john8329, 2025-01-27:
Same problem, it works initially and it stops reading the UPS after a while, I tried every possible parameter change, it still happens. Is there a way to tell NUT to restart after 15 minutes of activity as a workaround?

graemev, 2025-01-27:
OK, the problem appears fixed for me at least:

Following my post I added a cronjob for root:

`08                09      *      *       *             chronic /usr/local/bin/check_ups`

Where

```
$ cat /usr/local/bin/check_ups
#!/bin/bash

# Who   Date    What
# GPV   09Oct24 Inital version , greps lsusb(1) and nut-scanner(1) output

# See: https://unix.stackexchange.com/questions/340718/how-do-i-bring-heredoc-text-into-a-shell-script-variable
eat_here() {
    res=$(cat)
}

# Make sure these have blank lines at start and end

eat_here <<'HEREDOC'

root@zbox:/home/graeme# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 2148:7022  USB2.0 HUB
Bus 001 Device 004: ID 8087:07dc Intel Corp. Bluetooth wireless interface
Bus 001 Device 003: ID 0bda:0153 Realtek Semiconductor Corp. 3-in-1 (SD/SDHC/SDXC) Card Reader
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 047: ID 051d:0002 American Power Conversion Uninterruptible Power Supply   <----- here
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

HEREDOC

expected_lsusb="${res}"

eat_here <<'HEREDOC'

graeme@zbox:~$ sudo nut-scanner -U
[sudo] password for graeme: 
Scanning USB bus.
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "051D"
        productid = "0002"
        product = "Back-UPS XS 700U   FW:924.Z5 .I USB FW:Z5"
        serial = "3B2041X34361"
        vendor = "American Power Conversion"
        bus = "001"

HEREDOC

expected_nut_scanner="${res}"

actual_lsusb="$(lsusb)"

actual_nut_scanner="$(nut-scanner -U)"

if [[ -z "${actual_lsusb}" ]] ; then
    actual_lsusb="<NOTHING>"
fi

if [[ -z "${actual_nut_scanner}" ]] ; then
    actual_nut_scanner="<NOTHING>"
fi

declare -i OK
declare -i RC

OK=1

if [[ "${actual_lsusb}" =~ "051d:0002" ]] ; then
    :
else
    OK=0
fi

if [[ "${actual_nut_scanner}" =~ "3B2041X34361" ]] ; then
    :
else
    OK=0
fi

if [[ ${OK} -eq 1 ]]; then
    echo "check_ups: all is OK"
    RC=0
else
    msg="Error WRT UPS: lsusb reports\n ${actual_lsusb}\n we expect\n ${expected_lsusb}\n\n and nut-scanner reports \n ${actual_nut_scanner} \n and we expected \n ${expected_nut_scanner}"

    printf "$msg"

    printf "$msg" | mail -s "check_ups: Lost contact with UPS $(hostname)" graeme@home
    RC=1
fi

exit ${RC}

```

After a few weeks, the email triggered. The device had just disappeared. I was on a nearby computer at the time, nothing unusual had happened, no power blip, no noise from the UPS.

I did some investigation and it seemed ALL the devices on  the USB hub where the UPS was attached had gone (IIRC , so had the hub itself).  It was an external USB 2.0 hub. Unplugging and re-plugging  "fixed" the problem ...but of no long term use.

So I bought a uGreen USB 3.0 USB hub (generally pleased with their kit) and the problem has not reoccurred in several months.

john8329, 2025-03-03:
@jimklimov Using the "NUT" add-on for Home Assistant, v0.14.1, UPS driver controller 2.8.0. I'm reading that from logs. Building the new package in the raspberry it's running on is a little complicated (it communicates over USB, can't forward that to a dev machine), so for now it's not worth the time for me (I already spent at least 8 hours on trying to make it work, I'm done with this). If you need detailed logs or any debug info feel free to ask though.

jimklimov, 2025-03-04:
FWIW, you can check the wiki, there were community-contributed notes on custom-building the HA plugin...