networkupstools/nut #1028: Trouble with Liebert PSI5-1500RT120LI

darconeous, 2021-05-03:
I recently purchased a [Vertiv Liebert PSI5-1500RT120LI](https://www.vertiv.com/en-us/products-catalog/critical-power/uninterruptible-power-supplies-ups/psi5-1500rt120li-vertiv-liebert-psi5-lithium-ion-ups-1500va1350w-120v-line-interactive-avr-/) (Admittedly not listed as an officially supported UPS), and after reading [this thread](https://www.mail-archive.com/nut-upsuser@alioth-lists.debian.net/msg01090.html) I assumed that simply adding something like this to my `/etc/nut/ups.conf` would do the trick:

```
[ups]
driver = usbhid-ups
port = auto
desc = "Libert PSI5 (Rack)"
productid = 0002
```

And this did kinda work, except that it only reports that it is always "on battery". I haven't tried running it down to see if it ever gets to the "LB" condition.

I'm a software developer and I'm not afraid to try hacking around to get this to work, but I'm not sure where to start. Any advice or pointers would be welcome.

Here is the output from running `usbhid-ups -DDDDD -a ups` manually:

```
$ sudo -u nut /lib/nut/usbhid-ups -DDDDD -a ups
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	send_to_all: SETINFO driver.parameter.port "auto"
   0.000150	send_to_all: SETINFO driver.parameter.productid "0002"
   0.000256	debug level is '5'
   0.001567	upsdrv_initups...
   0.026776	Checking device (1D6B/0003) (002/001)
   0.055284	- VendorID: 1d6b
   0.055336	- ProductID: 0003
   0.055377	- Manufacturer: unknown
   0.055416	- Product: unknown
   0.055452	- Serial Number: unknown
   0.055487	- Bus: 002
   0.055522	- Device release number: 0510
   0.055556	Trying to match device
   0.055601	Device does not match - skipping
   0.055997	Checking device (10AF/0002) (001/003)
   0.062778	- VendorID: 10af
   0.062824	- ProductID: 0002
   0.062862	- Manufacturer: Vertiv Co.
   0.062901	- Product: Liebert PSI5
   0.062940	- Serial Number: 0000000000000000
   0.062986	- Bus: 001
   0.063023	- Device release number: 0002
   0.063062	Trying to match device
   0.063154	Device matches
   0.063248	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.065797	HID descriptor, method 1: (9 bytes) => 09 21 10 01 00 01 22 1b 00
   0.065840	HID descriptor length (method 1) 27
   0.065879	i=0, extra[i]=09, extra[i+1]=21
   0.065926	HID descriptor, method 2: (9 bytes) => 09 21 10 01 00 01 22 1b 00
   0.065965	HID descriptor length (method 2) 27
   0.066006	HID descriptor length 27
   0.069283	Report Descriptor size = 27
   0.069337	Report Descriptor: (27 bytes) => 06 00 ff 09 01 a1 01 09 02 15 00 26 ff 00
   0.069386	 75 08 95 08 81 82 09 02 95 08 91 82 c0
   0.069614	Using subdriver: Belkin/Liebert HID 0.17
   0.069656	2 HID objects found
   0.069696	Entering libusb_get_report
   0.072032	libusb_get_report: error sending control message: Operation not permitted
   0.072084	Can't retrieve Report 00: Operation not permitted
   0.072131	hid_lookup_path: ff000001 -> not found in lookup table
   0.072171	hid_lookup_path: ff000002 -> not found in lookup table
   0.072210	Path: ff000001.ff000002, Type: Input, ReportID: 0x00, Offset: 0, Size: 8
   0.072247	Entering libusb_get_report
   0.074781	libusb_get_report: error sending control message: Operation not permitted
   0.074829	Can't retrieve Report 00: Operation not permitted
   0.074870	hid_lookup_path: ff000001 -> not found in lookup table
   0.074970	hid_lookup_path: ff000002 -> not found in lookup table
   0.075009	Path: ff000001.ff000002, Type: Output, ReportID: 0x00, Offset: 0, Size: 8
   0.075057	send_to_all: SETINFO ups.mfr "Vertiv Co."
   0.075096	send_to_all: SETINFO ups.model "Liebert PSI5"
[... 396 lines cut ...]
  18.354412	upsdrv_updateinfo...
  18.605465	libusb_get_interrupt: Connection timed out
  18.606411	Got 0 HID objects...
  18.607258	Quick update...
  18.765045	send_to_one: PONG
  20.354970	upsdrv_updateinfo...
  20.605782	libusb_get_interrupt: Connection timed out
  20.606713	Got 0 HID objects...
  20.607553	Quick update...
  22.356801	upsdrv_updateinfo...
  22.609190	libusb_get_interrupt: Connection timed out
  22.610102	Got 0 HID objects...
  22.610944	Quick update...
  24.358602	upsdrv_updateinfo...
  24.609920	libusb_get_interrupt: Connection timed out
  24.610829	Got 0 HID objects...
  24.611671	Quick update...
  24.826562	send_to_one: PONG
  26.360188	upsdrv_updateinfo...
  26.611208	libusb_get_interrupt: Connection timed out
  26.611307	Got 0 HID objects...
  26.611373	Quick update...
 ```

I see a bunch of timeouts on the "get_interrupt", which I suspect is part of the problem. Any ideas?

darconeous, 2021-05-03:
A few additional notes:

* That `libusb_get_report: error sending control message: Operation not permitted` seems suspicious, but running it directly as root doesn't seem to make a difference.
* This is running on a Raspberry Pi 4 with the latest Raspbian. I understand there have been some problems recently with NUT and this platform. Could this be related?

darconeous, 2021-05-04:
As a separate datapoint, I tried hooking up a [cyberpower CP550HG](https://www.cyberpowersystems.com/product/ups/standby/cp550hg/) and it worked fine via USB, so I'm hoping this isn't an RPI-specific issue.

reubano, 2025-07-02:
Any idea if the same error occurs for the PSI5-1500RT120 model (not LI)?

jimklimov, 2025-07-02:
@reubano : one thing to note is that this report was with NUT v2.7.4, released about a decade ago now. So the question is if it misbehaves with current code base?

reubano, 2025-07-02:
@jimklimov yes, the current release. This is the only issue I was able to find regarding this PSI5. I want to purchase the non LI model but only if it is supported.

darconeous, 2025-07-06:
You'd think at this point all UPS manufacturers would aim for `nut` compatibility, given how popular it is. Alas.

I personally wouldn't risk buying a PSI5 if nut compatibility is important. The chances of it "just working" are pretty low.

I ended up biting the bullet and getting the network card, which was adequate for my needs but still kinda lame I had to shell out $100+ just to make it work with my setup.

reubano, 2025-07-06:
> I personally wouldn't risk buying a PSI5 if nut compatibility is important. The chances of it "just working" are pretty low.

Thanks for the info! 

@jrjparks any thoughts re #2369?

darconeous, 2025-07-06:
Oh I didn't see #2369. If it says it adds support for the PSI5, then I would assume that it works with at least the non-LI model. I don't have any reason to think that it wouldn't work if it has been added to the supported list. All that being said, I don't have the non-LI model, so all I can do is amend my previous comment.

jrjparks, 2025-07-06:
I'm currently running 2 PSI5-1500RM1201U in my home lab rack and a NUT server via USB to monitor them. There is a quirk that I haven't had time or energy to track down where they randomly drop off of USB and I need to reset the USB port they are connected to, but I just run that as a 1min cron and it's been working well enough for me for now. If you are looking to use this brand in a production environment I'd either go with a known supported brand as others have mentioned or be willing to do some troubleshooting if what I'm doing will not work for you.

---

`cat /home/jrparks/usbreset/usbreset.c`

`cc usbreset.c -o usbreset`

```
/* usbreset -- send a USB port reset to a USB device */

#include <stdio.h>
#include <unistd.h>
#include <fcntl.h>
#include <errno.h>
#include <sys/ioctl.h>

#include <linux/usbdevice_fs.h>

int main(int argc, char **argv)
{
    const char *filename;
    int fd;
    int rc;

    if (argc != 2) {
        fprintf(stderr, "Usage: usbreset device-filename\n");
        return 1;
    }
    filename = argv[1];

    fd = open(filename, O_WRONLY);
    if (fd < 0) {
        perror("Error opening output file");
        return 1;
    }

    printf("Resetting USB device %s\n", filename);
    rc = ioctl(fd, USBDEVFS_RESET, 0);
    if (rc < 0) {
        perror("Error in ioctl");
        return 1;
    }
    printf("Reset successful\n");

    close(fd);
    return 0;
}
```

`cat reset_ups_usb.sh`

```
#!/bin/env bash

for dev in $(lsusb -d 10af:0002 | cut -d: -f1 | cut -d' ' -f2,4 | sed 's/ /\//'); do
        sudo /usr/bin/usbreset /dev/bus/usb/$dev
done

echo "Reloading Systemctl Daemon"
sudo systemctl daemon-reload
echo "Restarting nut-drivers"
sudo systemctl restart 'nut-driver@UPS-*.service'
echo "Restarting nut-server"
sudo systemctl restart nut-server.service
```

jrjparks, 2025-07-12:
@reubano got them to stay responsive for another hour before getting the "Data stale" warning again. Seems like I just need to figure out either better values or what Vertiv does to keep the connection alive.