networkupstools/nut #1029: usbhid-ups stops working after 10-20 minutes

eyal0, 2021-05-04:
I have installed nut using the 2.7.4 distribution available in MX Linux.

Every 10-20 minutes, the usb monitoring programs stop getting information.  I can fix it by restarting the usbhid-ups program.  Why does it keep losing the connection?

In my testing, I type this:

`sudo /lib/nut/usbhid-ups -DD -a myups`

And all monitoring works just fine for about 10-20 minutes.  Eventually, the usbhid-ups program stops outputting anything to the screen and the UPS monitoring programs report stale data.  If I stop and restart the usbhid-ups, this fixes the problem for another 10-20 minutes.

There is no error output on the console, just the usually debug messages.  When the problem occurs, the debug output on the screen pauses.

eyal0, 2021-05-05:
I was able to solve my problem with the suggestions here: https://serverfault.com/a/995122

I added `pollinterval = 1` to /etc/nut/ups.conf and I also added `RUN+="/sbin/upsdrvctl stop; /sbin/upsdrvctl start"` to the line in my file in `/etc/udev/rules.d`.  I'm not sure which one fixed the problem or if perhaps both were needed but either way, it worked.

closed 2021-05-05 by eyal0

eyal0, 2021-05-13:
This only helped a little but didn't solve the problem.  I'm still getting a hang about once a day.

I was able to narrow it down to somewhere between the following two lines:

```
 737.021489	upsdrv_updateinfo...
 737.136305	Report[int]: (2 bytes) => 34 64
```

When it hangs, I see the first one but not the second.  I think that means that it is hanging in libhud.c HIDGetEvents.  I suspect that it might be for the get_interrupt.  HIDGetEvents is only called if `use_interrupt_pipe` is true, which I think is the same as the option to the driver `-x pollonly`.  So far, it seems to be working, even without the pollinterval.  And without pollinterval, I was usually getting a hang every hour.

So my solution is, instead of the pollinterval, to put `pollonly = "enabled"` into `/usr/nut/ups.conf`.  Hopefully this will work.

If I don't add another comment here in the next 2-3 days, it worked.

benzino77, 2022-01-21:
Setting `pollonly = "enabled"` for my Cyber Power USB connected UPS solved the problem. Thanks @eyal0 for the hint!

Dude4Linux, 2022-09-10:
I'm running Home Assistant on a Raspberry Pi4 using HAOS with the NUT Add-on and Integrations.  For months I've been struggling with this issue.  I happen to have a Tripplite SMART1500LCDT UPS.  I've tried about every combination of pollinterval, deadtime, and maxage that I found on other threads.  I even upgraded the Pi's firmware to the latest version. None of these suggestions seem to help for very long.

Now I tried setting pollonly = "enabled" and it appears to be working without disconnection (48 hours +).
Thanks @eyal0 for the hint!

jimklimov, 2022-11-01:
Referenced in https://github.com/networkupstools/nut/wiki/Troubleshooting-eventual-disconnections-(Data-stale)

JonLaliberte, 2024-01-03:
This was driving me crazy. I have a Cyberpower CP1500PFCLCDa and I had tried the other settings changing MAXAGE, maxretry, and some others IIRC... with no luck. Adding pollonly to ups.conf (not using the HA add-on) it's seemed to stay online for a week or two now.

eyal0, 2024-01-08:
@JonLaliberte glad that it helped you!

One way to help others would be to think about any other websites that you might have encountered during your search to solve this issue and go to those websites and leave a link to this page.