networkupstools/nut #1035: <Ippon> <BACK COMFO PRO II 650/850/1050> supported by <blazer_usb >

CindyZhan19891207, 2021-05-28:
Hello NUT developers:
     After my test, I confirm that  Ippon BCP(Back Comfo Pro) II models support NUT. Please help to list them in compatibility list. 
    upsc, upscmd, upsrw results are listed below. Shutdown sequences are tested and result is OK.

upsc result: 
    
[image]

upsrw result:  (as this UPS is entry level, no read/write commands are supported)
[image]

upscmd result: 
[image]

Let me know if any other inputs required. Thanks.

BR
CindyZ.

jimklimov, 2021-05-30:
Hello, and thanks for the report. Is it possible to paste these data again as text, so it can be forwarded to the Device Data Dump Library, please?

CindyZhan19891207, 2021-06-04:
Hello Jim,
Thanks for your reply. Here I update test result as text. 

// **************upsc test result*********** 
root@pc46:~# upsc ippon
Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.60
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 2.0
input.frequency: 50.1
input.frequency.nominal: 50
input.voltage: 227.4
input.voltage.fault: 227.4
input.voltage.nominal: 220
output.voltage: 227.4
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 6
ups.productid: 5161
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
ups.vendorid: 0665

// **************upsrw test result*********** 
no read write parameter supported. The return value is null by command “upsrw ippon”

//**************upscmd test result*********** 
root@pc46:/opt# upscmd -l ippon
Instant commands supported on UPS [ippon]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test

//**********shutdown sequence***********
Test result PASS

CindyZhan19891207, 2021-07-01:
Hello Nut developers,
  Do you need more information about this? Thanks

CindyZhan19891207, 2021-07-05:
@jimklimov  please help to check and update this ticket, thanks.

jimklimov, 2021-07-28:
Should be now seen at https://networkupstools.org/stable-hcl.html and https://networkupstools.org/ddl/Ippon/index.html

closed 2021-07-28 by jimklimov