networkupstools/nut #1042: No matching MIB found for sysOID '.1.3.6.1.4.1.318.1.3.2.7'

gurubert, 2021-06-10:
Starting the SNMP driver it says: 

Jun 10 11:50:40 pmhost3 upsdrvctl[15852]: No matching MIB found for sysOID '.1.3.6.1.4.1.318.1.3.2.7'!
Jun 10 11:50:40 pmhost3 upsdrvctl[15852]: Please report it to NUT developers, with an 'upsc' output for your device.
Jun 10 11:50:40 pmhost3 upsdrvctl[15852]: Going back to the classic MIB detection method.
Jun 10 11:50:40 pmhost3 upsdrvctl[15852]: Detected Smart-UPS 1500 RM on host 192.168.50.241 (mib: apcc 1.2)

Output of upsc is:

ambient.1.humidity.alarm.high: 90.00
ambient.1.humidity.alarm.low: 10.00
ambient.1.temperature.alarm.high: 60.00
ambient.1.temperature.alarm.low: 0.00
ambient.humidity: -1.00
ambient.temperature: -1.0
battery.charge: 100.00
battery.charge.restart: 60
battery.current: 0.00
battery.date: 05/10/07
battery.packs: 0.00
battery.packs.bad: 0.00
battery.runtime: 1260.00
battery.runtime.low: 120
battery.voltage: 27.00
battery.voltage.nominal: 0.00
device.mfr: APC
device.model: Smart-UPS 1500 RM
device.serial: AS0719232542
device.type: ups
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.50.241
driver.parameter.snmp_version: v1
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: apcc MIB 1.2
driver.version.internal: 0.97
input.frequency: 0.00
input.L1-L2.voltage: 0.00
input.L1-L2.voltage.maximum: 0.00
input.L1-L2.voltage.minimum: 0.00
input.L1.current: 0.00
input.L1.current.maximum: 0.00
input.L1.current.minimum: 0.00
input.phases: 0
input.sensitivity: high
input.transfer.high: 253
input.transfer.low: 208
input.transfer.reason: noTransfer
input.voltage: 230.00
input.voltage.maximum: 230.00
input.voltage.minimum: 228.00
output.current: 0.00
output.frequency: 50.00
output.L1-L2.voltage: 0.00
output.L1.current: 0.00
output.L1.current.maximum: 0.00
output.L1.current.minimum: 0.00
output.L1.power: 0.00
output.L1.power.maximum: 0.00
output.L1.power.maximum.percent: 0.00
output.L1.power.minimum: 0.00
output.L1.power.minimum.percent: 0.00
output.L1.power.percent: 0.00
output.phases: 0
output.voltage: 230.00
output.voltage.nominal: 230
ups.delay.shutdown: 180
ups.delay.start: 60
ups.firmware: 617.3.I
ups.id: UPS_IDEN
ups.load: 55.00
ups.mfr: APC
ups.mfr.date: 05/10/07
ups.model: Smart-UPS 1500 RM
ups.serial: AS0719232542
ups.status: OL RB
ups.temperature: 17.00
ups.test.date: 06/08/2021
ups.test.result: Failed

jimklimov, 2021-06-11:
Thank you for the report.

Cross-checked with NUT sources, and apc-mib.c does have a comment that
various APC models report different OIDs referenced from their IETF
standard sysObjectID node.

So the snmp-ups driver directly matches one of those values, and falls back
to other methods to guess the device and its MIB, as it reported in the log.

I wonder about the values returned - too many zeroes about input and output
currents (maybe just not measured?), although AC voltages seem sane. Less
sure about battery.voltage (27V) - not sure how that model is wired; two
charged 12V batteries (peaking around 14V) might be it.

Also if the battery.date is true and they are ~15 years old (failed test
notwithstanding), you might want them replaced so the particular UPS is
more than a power strip. Lead batteries are normally warrantied for 2-3,
rarely 5 years, some special techs had 10 though.

Jim

gurubert, 2021-06-11:
> Also if the battery.date is true and they are ~15 years old (failed test notwithstanding), you might want them replaced so the particular UPS is more than a power strip.

Yeah, the customer already tries to getting the batteries replaced. It was the first time this UPS was getting monitored…

aquette, 2021-06-11:
Confirming @jimklimov comment on the comment ;)
As for the zero decimals, I've fixed that in more recent version (still not yet released)
Cheers

kquinsland, 2024-05-04:
For what it's worth, I hit this exact issue today and the underlying cause was... unexpected.

I'll spare you the long rant and just say that the APC/SE website for looking up the appropriate firmware version(s) and file(s) for a particular device/network care are an absolute dumpster fire.

---

The UPS itsself has an underlying firmware and the network card has three different firmware files: boot, os, app.  Boot and os are likely to be the "same" but the application version must PRECISELY match.

Example, the `AP9630` card has _at least_ these firmware bundles available for it:

`APC_SY_712_EN`
`APC_SY3P_712_EN`
`APC_SUMX_712_EN`

You _can_ install any of those three but if you install the `SY` family (Symmetra) on a `SUMX` (Smart-UPS) then it'll flash/boot OK but in the web interface you'll get the "no connection to UPS" error. I was in that state when my snmp monitoring all of a sudden broke.

After installing the `SUMX` version on my smart-ups, I could log into the web UI and see that there _was_ a working serial connection. Quick check and snmp monitoring was working again.

**TL;DR:** Not the most intuitive failure mode in the world, but if there's a mis-match between the UPS firmware and the network card application (web ui shows "no connection to ups") then you can expect this `No matching MIB found for sysOID '.1.3.6.1.4.1.318.1.3.2.7'` error string.  Flashing the correct application to the net card restored my snmp.