networkupstools/nut #1049: 'hunnox' protocol of 'nutdrv_qx' don't work in XigmaNAS (FreeBSD)

Yellow-Horror, 2021-06-21:
I have a "Crown CMU-SP1200IEC LCD USB" UPS which interface is similar to [Hunnox HNX-850 mentioning here](https://github.com/networkupstools/nut/pull/638) and works with the 'hunnox' protocol in Linux. But when i try to setup the UPS in FreeBSD-based XigmaNAS, i have no success:
```
xigmanas: ~# /usr/local/libexec/nut/nutdrv_qx -DDDDD -a ups
Network UPS Tools - Generic Q* USB/Serial driver 0.30 (2.7.4.1)
USB communication driver 0.33
   0.000000     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000292     [D5] send_to_all: SETINFO driver.parameter.vendorid "0001"
   0.000402     [D5] send_to_all: SETINFO driver.parameter.productid "0000"
   0.000523     [D5] send_to_all: SETINFO driver.parameter.protocol "hunnox"
   0.000630     [D5] send_to_all: SETINFO driver.parameter.langid_fix "0x409"
   0.000710     [D5] send_to_all: SETINFO driver.flag.novendor "enabled"
   0.000834     [D5] send_to_all: SETINFO driver.flag.noscanlangid "enabled"
   0.001098     [D1] debug level is '5'
   0.008360     [D5] send_to_all: SETINFO device.type "ups"
   0.008460     [D1] upsdrv_initups...
   0.008897     [D2] Language ID workaround enabled (using '0x409')
   0.012319     [D3] usb_busses=0x800ea7928
   0.012409     [D2] Checking device (0001/0000) (/dev/usb//dev/ugen1.2)
   0.025152     [D2] - VendorID: 0001
   0.025230     [D2] - ProductID: 0000
   0.025269     [D2] - Manufacturer: unknown
   0.025302     [D2] - Product: MEC0003
   0.025335     [D2] - Serial Number: unknown
   0.025366     [D2] - Bus: /dev/usb
   0.025400     [D2] - Device: /dev/ugen1.2
   0.025443     [D2] - Device release number: 0100
   0.025477     [D2] Trying to match device
   0.025509     [D3] match_function_regex: matching a device...
   0.025602     [D2] Device matches
   0.025667     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.025739     [D5] send_to_all: SETINFO ups.vendorid "0001"
   0.025800     [D5] send_to_all: SETINFO ups.productid "0000"
   0.031109     [D2] Skipping protocol Voltronic 0.06
   0.031174     [D2] Skipping protocol Voltronic-QS 0.07
   0.031211     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.031251     [D2] Skipping protocol Mustek 0.07
   0.031287     [D2] Skipping protocol Megatec/old 0.07
   0.031323     [D2] Skipping protocol BestUPS 0.06
   0.031369     [D2] Skipping protocol Mecer 0.07
   0.031406     [D2] Skipping protocol Megatec 0.06
   0.031452     [D2] Skipping protocol Zinto 0.06
   0.031490     [D2] Skipping protocol Masterguard 0.01
   0.031617     [D3] send: Q1
   0.031670     [D4] command index: 0x03
   0.031723     [D3] asking for: 00
   0.077091     [D3] read: timeout (0)
   0.077244     [D2] qx_process_answer: short reply (input.voltage)
   0.077288     [D3] send: Q1
   0.077313     [D4] command index: 0x03
   0.077335     [D3] asking for: 0D
   0.105056     [D3] read: timeout (0)
   0.105107     [D2] qx_process_answer: short reply (input.voltage)
   0.105137     [D3] send: Q1
   0.105157     [D4] command index: 0x03
   0.122053     [D3] read: timeout (0)
   0.122092     [D2] qx_process_answer: short reply (input.voltage)
   0.122116     [D2] Skipping protocol Q1 0.07
   0.122137     Device not supported!
   0.122153     Device not supported!
```
Playing with the protocol settings i noticed that without 'langid_fix' it can read some info from the UPS, but the final result doesn't change:
```
xigmanas: ~# /usr/local/libexec/nut/nutdrv_qx -DDDDD -a ups
Network UPS Tools - Generic Q* USB/Serial driver 0.30 (2.7.4.1)
USB communication driver 0.33
   0.000000     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000509     [D5] send_to_all: SETINFO driver.parameter.vendorid "0001"
   0.000579     [D5] send_to_all: SETINFO driver.parameter.productid "0000"
   0.000641     [D5] send_to_all: SETINFO driver.parameter.protocol "hunnox"
   0.000691     [D5] send_to_all: SETINFO driver.flag.novendor "enabled"
   0.000749     [D5] send_to_all: SETINFO driver.flag.noscanlangid "enabled"
   0.000884     [D1] debug level is '5'
   0.005953     [D5] send_to_all: SETINFO device.type "ups"
   0.006006     [D1] upsdrv_initups...
   0.008737     [D3] usb_busses=0x800ea7928
   0.008792     [D2] Checking device (0001/0000) (/dev/usb//dev/ugen1.2)
   0.021362     [D2] - VendorID: 0001
   0.021416     [D2] - ProductID: 0000
   0.021438     [D2] - Manufacturer: unknown
   0.021462     [D2] - Product: MEC0003
   0.021482     [D2] - Serial Number: unknown
   0.021501     [D2] - Bus: /dev/usb
   0.021524     [D2] - Device: /dev/ugen1.2
   0.021545     [D2] - Device release number: 0100
   0.021565     [D2] Trying to match device
   0.021588     [D3] match_function_regex: matching a device...
   0.021640     [D2] Device matches
   0.021686     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.021735     [D5] send_to_all: SETINFO ups.vendorid "0001"
   0.021767     [D5] send_to_all: SETINFO ups.productid "0000"
   0.022059     [D2] Skipping protocol Voltronic 0.06
   0.022129     [D2] Skipping protocol Voltronic-QS 0.07
   0.022167     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.022206     [D2] Skipping protocol Mustek 0.07
   0.022245     [D2] Skipping protocol Megatec/old 0.07
   0.022282     [D2] Skipping protocol BestUPS 0.06
   0.022324     [D2] Skipping protocol Mecer 0.07
   0.022363     [D2] Skipping protocol Megatec 0.06
   0.022433     [D2] Skipping protocol Zinto 0.06
   0.022474     [D2] Skipping protocol Masterguard 0.01
   0.022572     [D3] send: Q1
   0.022618     [D4] command index: 0x03
   0.022654     [D3] asking for: 00
   0.074379     [D5] read: (46 bytes) => 28 30 30 30 2e 30 20 30 30 30 2e 30 20 30 30 30 2e 30
   0.074467     [D5]  20 30 30 30 20 30 30 2e 30 20 30 2e 30 30 20 30 30 2e 30 20 30 30 30 30 30
   0.074495     [D5]  30 30 30
   0.074518     [D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.074617     [D2] qx_process_answer: short reply (input.voltage)
   0.074652     [D3] send: Q1
   0.074677     [D4] command index: 0x03
   0.074698     [D3] asking for: 0D
   0.108387     [D5] read: (46 bytes) => 28 30 30 30 2e 30 20 30 30 30 2e 30 20 30 30 30 2e 30
   0.108498     [D5]  20 30 30 30 20 30 30 2e 30 20 30 2e 30 30 20 30 30 2e 30 20 30 30 30 30 30
   0.108535     [D5]  30 30 30
   0.108570     [D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.108614     [D2] qx_process_answer: short reply (input.voltage)
   0.108661     [D3] send: Q1
   0.108697     [D4] command index: 0x03
   0.131379     [D5] read: (46 bytes) => 28 30 30 30 2e 30 20 30 30 30 2e 30 20 30 30 30 2e 30
   0.131489     [D5]  20 30 30 30 20 30 30 2e 30 20 30 2e 30 30 20 30 30 2e 30 20 30 30 30 30 30
   0.131529     [D5]  30 30 30
   0.131561     [D3] read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.131602     [D2] qx_process_answer: short reply (input.voltage)
   0.131644     [D2] Skipping protocol Q1 0.07
   0.131676     Device not supported!
   0.131707     Device not supported!
```
Please help.

jimklimov, 2021-07-05:
Hello, one idea that comes to mind is that NUT version on the NAS can be different from that on your successful Linux experiment. NUT version (e.g. 2.7.4.1 above) and driver version (e.g. nutdrv_qx 0.30) and subdriver (usb 0.33) and subprotocol (those many 0.xx numbers) can be seen at startup of the debug log, and if stuff connects successfully - should be exposed in data points seen e.g. via `upsc` lookups.

My guess is that if it works for you in Linux, something was fixed in the codebase since 2.7.4 version was released (which alas was too long ago, working hard on next one). So it may be that your OS distributions used different NUT source revisions equal to or newer than the 2.7.4 tag to "backport" some fixes from the master branch.

If you have a possibility to build NUT from source on the NAS, please see if that "just works"?

Yellow-Horror, 2021-07-05:
> If you have a possibility to build NUT from source on the NAS, please see if that "just works"?

I can't. It is an embedded installation that lacks resources to run compilation. Based on XigmaNAS current release changelist, it used nut-devel source as of 2021.05.24. So, all that i can do is wait for a XigmaNAS update with renewed NUT and try it again then.

Yellow-Horror, 2022-05-21:
The UPS works with NUT in the new release of XigmaNAS 12.3.0.4 (revision 9047). It is still unusable for my purpose because it don't show remaining battery charge (always 0%), but the connection itself works properly, the UPS state, load and voltages are shown correctly.

closed 2022-05-21 by Yellow-Horror

jimklimov, 2022-05-21:
Do you know which NUT version got into that release?

Yellow-Horror, 2022-05-30:
v2022.04.27 as described in the XigmaNAS changelog. But I think the problem disappearance is not related to the update of the NUT version, but to other changes in the system, such as the FreeBSD kernel or the USB driver. As I wrote above, in Linux for a long time there were no problems with connecting "Crown CMU-SP1200IEC LCD USB" to NUT.

P.S. Is it possible that the "hunnox" protocol of NUT will learn to get information about battery charge percentage? Crown's proprietary program is able to receive this data from the UPS.