networkupstools/nut #1050: Windows build is obsolete

LinoBarreca, 2021-06-22:
...and there are no build instructions for windows.
The software works (with the openssl libraries manually copied in the right place) but it lacks the features/fixes of the newer source code.
Any plan to update them or give proper instructions on how to build under windows?

jimklimov, 2021-06-25:
The plan is quite latent and currently lower in priority than many other things to rectify first, for the available team members, sadly... (that said, help and PRs would be welcome!)

There are numerous issues to be found in github history and perhaps some more in mailing lists, but the gist of it is that latest Windows builds were done as a branch from 2.6.5 baseline, and later main-branch releases diverged significantly so that getting a Windows build ability on top of current code would be a new effort at best inspired by comparisons of those branches vs. main-line 2.6.5 release.

I suppose compilers, headers and third-party deps available in Windows have marched on significantly as well. Notably, if the new effort were to concentrate on recent Windows releases and require WSL2 or at least shell tools usable for configure/make logic, it might be easier than a fully-native build with older Windows versions.

opoplawski, 2021-12-01:
It would be really nice to have a windows port of NUT.

jimklimov, 2021-12-03:
I do not disagree, but it was not a priority so I did not even get a system
set up for that although I intended to, for a year. So... PRs welcome ;)

Branch based on NUT release 2.6.5 should be there, but I suppose it would
not be an easy merge with modern master, and might refer to compiler/tools
versions no longer relevant. But really, I didn't try yet so may be it is
not all that sad.

Jim

kramttocs, 2022-01-19:
Agreed this would also be great. I can't get it working on my machine and not many docs out there for troubleshooting.

jimklimov, 2022-06-09:
See also #5

mrfrenzy, 2022-08-09:
There are a few quirks due to changes done in Windows over the years.

Here is a short writeup how to get NUT-Installer-2.6.5-6 properly working on Windows 11 64-bit:
Make sure the service is stopped
Copy libgcc_s_dw2-1.dll from bin to sbin
Download libeay32.dll and ssleay32.dll to sbin (https://github.com/openssl-net/openssl-net/tree/master/native/windows/x86)
Download ShutdownWithupdates.exe to sbin (https://github.com/dennisbabkin/ShutdownWithUpdates) - this is needed because builtin shutdown.exe installs updates before shutdown which can be catastrophic.

**nut.conf**
MODE=standalone

**ups.conf**
set the proper driver here for "myups"
Test it in an admin command line:
cd "\Program Files (x86)\NUT\bin"
genericups.exe -DDD -a myups _(for example depending on what driver you choose)_
(leave this running)

**upsd.users**
[upsmon]
password  = pass _(please change to a good password)_
upsmon master

**upsd.conf**
LISTEN 127.0.0.1 3493 _(ipv6 runs out of memory)_
Open a new admin prompt and run:
cd "\Program Files (x86)\NUT\sbin"
upsd -DD (you should see connected to UPS and status OL/OB etc)
(leave this running)

**upsmon.conf**
MONITOR myups@127.0.0.1 1 upsmon pass master _(not localhost)_
"c:\Program Files (x86)\NUT\sbin\ShutdownWithUpdates.exe" /s /nu /t 30 _(regular shutdown.exe installs updates before shutdown)_
NOTIFYCMD "C:\\WINDOWS\\Sysnative\\msg.exe *" _(system32 does not work since msg.exe is actually 64-bit)_

_Change all NOTIFY to EXEC for example:_
NOTIFYFLAG ONLINE	SYSLOG+EXEC

Open a new admin prompt and run:
cd "\Program Files (x86)\NUT\sbin"
upsmon -D (you should see logged into UPS)

Now test to disconnect line power, you should see messages in all windows and a popup, and after low bat the computer should shut down)

Close all windows and start the service, check event viewer - applications
Test disconnecting power

jimklimov, 2022-08-12:
See also #5 that coordinates (sort of) the effort to modernize NUT for Windows support. Might try a build of that branch. Packages not pursued yet, caveats / incomplete code are known (many listed in Project for that)

jimklimov, 2022-08-19:
FYI: To experiment with recent NUT codebase built (not packaged or otherwise integrated so far) for Windows you can try archives from https://ci.appveyor.com/project/nut-travis/nut/build/artifacts

cableghost, 2023-11-26:
> There are a few quirks due to changes done in Windows over the years.

I could use some help in following this...
1) Starting w/ nut.conf.. I'm not understanding where to make the changes?
2) Is there a means of setting the shutdown process to 'Hibernate' instead?
3) What if I only want the NUT server to provide access for my Home Assistant to view my UPS as a client, not to actually handle the shutting down process?

jimklimov, 2023-11-27:
At the moment I can only elaborate on point (3) generally, as I haven't had practical experience with NUT 2.6.5 based MSI's and what the prograns there can do.

Normally NUT has 3 layers - one or more drivers running on a system, a data server (upsd) which represents them all via network protocol, and local or remote clients. One of such clients is  upsmon that can handle shutdowns, alerts, etc. Generally any client greater than the read-only `upsc` needs server-side preparations (see docs on `upsd.users` config file to set up monitoring users or admins to run commands and set firmware variables).

The setup you seem to describe involves a driver and data server running on Windows (and maybe an upsmon watching for shutdowns of that Windows box, if it is fed by the UPS it monitors), and a separate Home Assistant with which you want to just watch the UPS state but not shut down HA.

So on the Windows side, prepare the driver in `ups.conf`, and data server via `upsd.user` as well as `LISTEN *` (or specific IP) in `upsd.conf`, and probably allow access to 3493/tcp on your LAN.

On HA side, if it runs NUT upsmon - in its config set `MONITOR` line for that `ups@server` as a `secondary` which feeds 0 PSUs (of the HA box). If it has some other client (graphs etc) - depends on such client; probably it is readonly and needs no logins, like `upsc`...