networkupstools/nut #1056: Is there an API to add a custom UPS ?

donnib, 2021-07-04:
Hi,
I have a solar system consisting of an item called Victron Cerbo GX which i can talk to Modbus TCP / MQTT etc so i can in NodeRed for example do something with the data for example i'd like to tell the NUT server i have a UPS and hand over information about it like when battery is low etc. Is there a API or something that i can call to integrate my own UPS ?

/donnib

jimklimov, 2021-07-05:
Hello, please read up the developer guide texts in the docs/ of the sources.

In short, NUT has a three-layer architecture, with vendor/protocol-specific device drivers talking to hardware over some media (serial/USB ports or networked SNMP are most common but not only options); these NUT drivers talk to an `upsd` server on same system (using a local pipe protocol and I guess you can say API based on using same libs and headers) and that server represents them using common networked NUT protocol to clients like `upsmon` (and many others) on local or remote systems.

What you are after here is making a new NUT driver for your device. I believe there were already some efforts for solar device support as a concept (similar but different from UPS or ePDU for example), as well as a few drivers for Modbus TCP, including a recent PR #1052. I'm traveling this week so can't quickly find if any efforts on MQTT were completed and merged, but some were started at least. Maybe there are PRs or experimental branches on github to that effect.

dtsecon, 2021-07-05:
Hello all,
generic_modbus driver (PR #1052) intends to support generic UPS devices with contact signals through modbus TCP/RTU gateways (also known as RIO - remote I/Os). 

[UPSD] <--- IPC ---> [GENERIC_UPS] <--- modbus TCP/RTU ---> MODBUS-RIO <--- contacts ---> [UPS DEVICE]

On the other hand you can setup any kind of modbus server, and configure generic_modbus driver to connect and read or write specific registers. Your application / modbus server could then drive NUT statuses (e.g. OL, OB, HB etc)  by writing over those registers.

jimklimov, 2021-09-21:
FYI: generic_modbus was recently merged, so you can try it with a build of NUT master branch now. Thanks @dtsecon !

closed 2021-11-14 by jimklimov