networkupstools/nut #1064: Sorry is not an issue about nut, there is some APK for android to monitor the ups remotely?

Andr20, 2021-07-08:
Sorry i don't find somethins on the web about it, i ask gently if there is some app to monitor nut remotely maybe also with widget, sorry is not the correct place to write it

dogtopus, 2021-07-10:
NUT has a web interface. You can use that on Android.

Andr20, 2021-07-10:
I tried to configure nut-cgi on OMV4 following many manuals on internet, but without results, is not working at all.

jimklimov, 2021-07-27:
I am not aware of anyone doing that yet, but given that all interactions between an arbitrary client (e.g. APK) and server (upsd) follow a documented protocol, with security constraints ("login" equivalents, SSL...) I suppose someone could write one.

At some point there was a jNut effort, that could help start the Java bindings to talk the protocol; I guess Android sources would need that anyway.