networkupstools/nut #1070: NUT doesn't support "inhibitor interface"

Atronach, 2021-08-02:
I would like NUT to hibernate my PC instead of shutting it down when ONBATT event is registered. So I've put:
_SHUTDOWNCMD "systemctl hibernate"_ to **/etc/ups/upsmon.conf**.

But this works only once, unless there's a hack in **/usr/lib/systemd/system-sleep/** which stops **nut-server.service** during hibernation and starts it during wake up, there also needs to be starting of **nut-monitor.service** specified. Otherwise **nut-monitor.service** reports to be "inactive (dead)" after waking up so no ONBATT event is registered and thus no hibernation takes place. In openSUSE's NUT package, currently there's a hack **/usr/lib/systemd/system-sleep/nut.sh** for it and is broken so it needs to be fixed manually.

After checking "man systemd-sleep", it states that scripts in **/usr/lib/systemd/system-sleep/** are considered hacks for applications not using the so called "inhibitor interface".

If NUT would use inhibitor interface, there wouldn't be a need for **/usr/lib/systemd/system-sleep/** hacks implemented by individual distros and my PC could be hibernated by NUT without problem.

Tested in OS: openSUSE Leap 15.3. with NUT v.2.7.4.
[Related mailing list discussion](https://alioth-lists.debian.net/pipermail/nut-upsuser/2021-July/012476.html)
[And continuation](https://alioth-lists.debian.net/pipermail/nut-upsuser/2021-August/012481.html)

jimklimov, 2024-04-06:
Seems related to #1833

jimklimov, 2024-08-15:
Some docs on this:
* https://systemd.io/INHIBITOR_LOCKS/ - with its examples, the feature should actually not be hard to implement
* https://www.freedesktop.org/software/systemd/man/latest/systemd-inhibit.html

jimklimov, 2024-08-19:
@Callisos : Cheers, FWIW, a PR with implementation attempt is posted, practical testing would be welcome :)

PR #2597 (2024-08-23): Implement (currently systemd) OS sleeping support in `upsmon` client
Main goal: detect going to sleep and subsequently waking up, in which case we neuter collected UPS status bits and last-read timestamps to avoid shutdowns because of stale data in a last-known-critical situation, and so rely only on up-to-date (re-)connections and data reads in this case. This problem is more likely if `SHUTDOWNCMD="systemctl suspend/hibernate/..."` or equivalent is used in `upsmon.conf` as the reaction to depleting UPS batteries.

Design is currently centered around libsystemd concepts, but is hopefully exte [... cut]
closed 2024-08-23 by jimklimov