networkupstools/nut #1072: The runtime time of APC SRT3000RMXLI is not output.

jc-lab, 2021-08-08:
### In the apcupsd tool, the output is as follows:

```
# ./apcaccess
APC      : 001,027,0692
VERSION  : 3.14.14 (31 May 2016) debian
CABLE    : USB Cable
DRIVER   : USB UPS Driver
UPSMODE  : Stand Alone
MODEL    : Smart-UPS SRT
STATUS   : ONLINE
BCHARGE  : 100.0 Percent
TIMELEFT : 112.1 Minutes
MBATTCHG : 5 Percent
MINTIMEL : 3 Minutes
MAXTIME  : 0 Seconds
ALARMDEL : 30 Seconds
BATTV    : 109.2 Volts
NUMXFERS : 0
TONBATT  : 0 Seconds
CUMONBATT: 0 Seconds
XOFFBATT : N/A
STATFLAG : 0x05000008
MANDATE  : 2017-12-19
SERIALNO : ...
NOMBATTV : 96.0 Volts
FIRMWARE : UPS 05.3 / ID=1010
```

**TIMELEFT : 112.1 Minutes**

### nut output :

```
battery.charge	:	99
battery.charge.low	:	10
battery.charge.warning	:	50
battery.runtime	:	0
battery.runtime.low	:	150
...
driver.name	:	usbhid-ups
driver.version	:	2.7.4
driver.version.data	:	APC HID 0.96
driver.version.internal	:	0.41
...
ups.model : Smart-UPS SRT
```

### usbhid-dump

```
004:002:000:DESCRIPTOR         1628390999.231533
 05 84 09 04 A1 01 09 24 A1 02 75 08 95 01 15 00
 26 FF 00 85 01 09 FE 79 02 B1 23 85 02 09 FF 79
 03 B1 23 85 03 09 FD 79 01 B1 23 05 85 85 04 09
 8F 79 01 B1 23 85 05 09 89 79 04 B1 23 85 06 09
 8B B1 23 09 2C B1 23 85 0E 09 83 25 64 B1 23 09
 67 B1 23 85 0C 09 66 81 23 09 66 B1 A3 85 10 09
 8D B1 23 09 8E B1 23 85 0F 09 8C B1 A2 85 11 09
 29 B1 A2 85 09 09 85 75 10 27 FF FF 00 00 B1 A3
 85 0D 09 68 66 01 10 55 00 75 20 17 FF FF FF FF
 25 00 81 A3 09 68 B1 A3 05 84 85 12 09 57 15 00
 27 FF FF 00 00 81 22 09 57 B1 A2 85 13 09 55 75
 20 17 FF FF FF FF 25 00 81 22 09 55 B1 A2 05 85
 85 08 09 2A 75 10 15 00 27 FF FF 00 00 81 22 09
 2A B1 A2 05 84 85 0A 09 40 67 21 D1 F0 00 55 05
 B1 23 85 0B 09 30 B1 A3 85 14 09 5A 75 08 15 01
 25 03 65 00 55 00 81 22 09 5A B1 A2 09 02 A1 02
 85 07 05 85 75 01 15 00 25 01 09 44 81 A3 09 44
 B1 A3 09 45 81 A3 09 45 B1 A3 09 D0 81 A3 09 D0
 B1 A3 09 D1 81 A3 09 D1 B1 A3 09 42 81 A3 09 42
 B1 A3 05 84 09 68 81 A2 09 68 B1 A2 09 69 81 A3
 09 69 B1 A3 05 85 09 43 81 A2 09 43 B1 A2 05 84
 09 73 81 A3 09 73 B1 A3 05 85 09 4B 81 A3 09 4B
 B1 A3 05 84 09 65 81 A3 09 65 B1 A3 05 85 09 DB
 81 A3 09 DB B1 A3 95 04 81 01 B1 01 C0 C0 06 86
 FF 85 89 09 FD 15 00 26 FF 00 75 08 95 3F 81 02
 85 90 09 FC 95 3F 91 02 85 96 09 F1 95 3F B1 22
 85 8D 09 F7 95 02 B1 22 85 8E 09 F6 B1 23 85 92
 09 F4 B1 23 85 93 09 F3 B1 22 85 94 09 F2 B1 23
 C0
```

### Code

The apcupsd tool seems to read it as "0x00850068".
* https://github.com/joseph-jclab/apcupsd/blob/00624effc90b7a88a2a31d58b42eb96453e03569/src/drivers/usb/usb.c#L119

nut also seems to bring this value:
* https://github.com/networkupstools/nut/blob/03c3bbe8df9a2caf3c09c120ae7045d35af99b76/drivers/libhid.c#L917

But why is there no output?

karelkryda, 2022-01-30:
[image]
Hi,
I have exact the same issue. Did you solved it somehow?
Thanks

jclab-joseph, 2022-01-30:
No, I didn't find a way either.

jimklimov, 2022-02-01:
In the past months several fixes for USB support have landed, some comments
implicating APC (and others') wire protocol imperfections (LogMin/LogNax
debacle notably). Can you please check if a local build of current NUT
master branch happens to solve this issue too?

jclab-joseph, 2022-12-04:
Still the same issue.
As of today, I have built the master branch.

[image]

```
<html>
<body>
<!--StartFragment-->

battery.charge | : | 100
-- | -- | --
battery.charge.low | : | 10
battery.charge.warning | : | 50
battery.runtime | : | -1
battery.runtime.low | : | 150
battery.type | : | PbAc
battery.voltage | : | 109.7
battery.voltage.nominal | : | 96.0
device.mfr | : | American Power Conversion
device.model | : | Smart-UPS SRT
device.serial | : | AS1751291634
device.type | : | ups
driver.name | : | usbhid-ups
driver.parameter.pollfreq | : | 30
driver.parameter.pollinterval | : | 2
driver.parameter.port | : | auto
driver.parameter.product | : | Smart-UPS SRT FW:UPS 05.3 / ID=1010
driver.parameter.productid | : | 0003
driver.parameter.serial | : | AS1751291634
driver.parameter.synchronous | : | auto
driver.parameter.vendor | : | American Power Conversion
driver.parameter.vendorid | : | 051D
driver.version | : | 2.8.0.1
driver.version.data | : | APC HID 0.100
driver.version.internal | : | 0.49
driver.version.usb | : | libusb-1.0.23 (API: 0x1000107)
ups.beeper.status | : | enabled
ups.delay.shutdown | : | 20
ups.firmware | : | UPS 05.3 / ID=1010
ups.mfr | : | American Power Conversion
ups.mfr.date | : | 2017/12/19
ups.model | : | Smart-UPS SRT
ups.productid | : | 0003
ups.serial | : | AS1751291634
ups.status | : | OL
ups.timer.reboot | : | -1
ups.timer.shutdown | : | 65535
ups.vendorid | : | 051d

<!--EndFragment-->
</body>
</html>
```