networkupstools/nut #1074: NUT and "Compensating for high input voltage" msg

djacobson, 2021-08-19:
I trying to replicate this issue at the moment but, I was wondering if NUT would report an APC's "Compensating for high input voltage" and "No longer compensating for high input voltage" warning msgs as logged in APC PowerChute?

jimklimov, 2023-08-29:
Hi, sorry this was not noticed in time... Still, I believe these conditions should pop up as `TRIM` and `BOOST` device states, when they are active.

jimklimov, 2025-03-19:
Reporting is done by clients, most commonly `upsmon`. Indeed it does not have a message template (nor code and variables for memory) about beginning/ending the power compensations: `BOOST`, `TRIM` (aka `BUCK` in some docs but no drivers publish such a token in `ups.status`), `OVER`(load) -- so the issue remains valid.

In fact, feels related to recent work done to report entering/exiting `ECO` state, getting or clearing `ALARM`'s, etc.

jimklimov, 2025-04-03:
Posted a PR to this effect, should be one of the last items remaining in NUT v2.8.3 scope, to finally release it.

PR #2873 (2025-04-06): clients/upsmon.{c,h} et al: introduce {NOT,}{OVER,TRIM,BOOST} notifications
Closes: #1074

FWIW, CC @desertwitch  :)
commit d07234c5 (2025-04-06): clients/upsmon.{c,h} et al: introduce {NOT,}{OVER,TRIM,BOOST} notifications [#1074]

Closes: #1074
closed 2025-04-06 by jimklimov