networkupstools/nut #1086: nut-monitor service fails even though nut-monitor seems to work

Gooberpatrol66, 2021-09-02:
The systemd service fails even though the output seems to indicate
that nut-monitor is communicating successfully with the UPS.

Gentoo Linux

Nut v2.7.4-r7 (installed from distro package)

UPS: Tripplite SU1500RTXLCD2U (controlled by pfsense)

Here's the unit file:

```
[Unit]
Description=Network UPS Tools - power device monitor and shutdown controller
After=local-fs.target network.target nut-server.service
OnFailure=systemd-monitor@%n.service

[Service]
ExecStart=/usr/sbin/upsmon -DDDDDDD
PIDFile=/run/upsmon.pid
Type=forking

[Install]
WantedBy=multi-user.target
```

journalctl output:

```
-- Boot 93e0d5b24bac491994a2ea5cce08cdcc --
Aug 16 23:45:40 gentoodesktop systemd[1]: Starting Network UPS Tools -
power device monitor and shutdown controller...
░░ Subject: A start job for unit nut-monitor.service has begun execution
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░
░░ A start job for unit nut-monitor.service has begun execution.
░░
░░ The job identifier is 107.
Aug 16 23:45:40 gentoodesktop upsmon[1943]: kill: No such process
Aug 16 23:45:40 gentoodesktop upsmon[1943]:    0.000000        UPS:
tripplite1@pfsense (slave) (power value 1)
Aug 16 23:45:40 gentoodesktop upsmon[1943]:    0.000019        UPS:
heartbeat@localhost (monitoring only)
Aug 16 23:45:40 gentoodesktop upsmon[1943]:    0.000051        Using
power down flag file /etc/killpower
Aug 16 23:45:40 gentoodesktop upsmon[1943]:    0.000127        debug
level is '8'
Aug 16 23:45:40 gentoodesktop upsmon[1944]:    0.003931        Trying
to connect to UPS [tripplite1@pfsense]
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.452582        UPS
[tripplite1@pfsense]: connect failed: No such host
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.452600
do_notify: ntype 0x0005 (COMMBAD)
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.452725        Trying
to connect to UPS [heartbeat@localhost]
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.453032        Can not
connect to localhost in SSL, continue uncrypted
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.457065        Logged
into UPS heartbeat@localhost
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.457104
pollups: heartbeat@localhost
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.457110
get_var: heartbeat@localhost / status
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.457142
parse_status: [OL]
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.457148        parsing: [OL]
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.457151
ups_on_line: heartbeat@localhost (first time)
Aug 16 23:45:52 gentoodesktop upsmon[1944]:   12.457154        Current
[... 372 lines cut ...]
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░
░░ The unit nut-monitor.service has entered the 'failed' state with
result 'timeout'.
Aug 16 23:47:12 gentoodesktop systemd[1]: Failed to start Network UPS
Tools - power device monitor and shutdown controller.
░░ Subject: A start job for unit nut-monitor.service has failed
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░
░░ A start job for unit nut-monitor.service has finished with a failure.
░░
░░ The job identifier is 107 and the job result is failed.
Aug 16 23:47:12 gentoodesktop systemd[1]: nut-monitor.service:
Triggering OnFailure= dependencies.
Aug 16 23:47:12 gentoodesktop systemd[1]: nut-monitor.service:
Consumed 2.054s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░
░░ The unit nut-monitor.service completed and consumed the indicated resources.
```

jimklimov, 2021-11-14:
I think there may be several issues in play here:

1) Early in the log, it says that "pfsense" hostname is not resolved - but it is later; probably a race between naming service startup and NUT components (maybe you should order the services predictably?) or just "pfsense" registers by DHCP or similar as it boots?

2) `upsmon` starts with debug enabled, this keeps it foregrounded (not sure quickly if it also forgoes the split of unprivileged monitoring child process vs `root` parent able to shutdown) - so a "Forking" service type is probably not applicable, and "Simple" or some such might be better (assuming systemd kills it by timeout since the child process never appears).

There are several discussions and PRs to separate foreground/background behavior from debug enablement, such as #349 and #683, so this may change in later NUT `master` branch iterations and eventually releases.

Gooberpatrol66, 2021-11-14:
I figured out this happened because I broke a symlink from /var/run -> /run, so systemd would look for a PID file in a different place than the daemon wrote it. So, this is invalid. My bad.

closed 2021-11-14 by Gooberpatrol66