networkupstools/nut #1087: No matching MIB found for sysOID '.1.3.6.1.4.1.318.1.3.4.5'

davidew98, 2021-09-04:
Log says:
[00:33:48] INFO: Starting the UPS drivers...
No matching MIB found for sysOID '.1.3.6.1.4.1.318.1.3.4.5'!
Please report it to NUT developers, with an 'upsc' output for your device.
Going back to the classic MIB detection method.
No supported device detected
Driver failed to start (exit status=1)
Network UPS Tools - Generic SNMP UPS driver 0.97 (2.7.4)
Network UPS Tools - UPS driver controller 2.7.4

My Device is a APC AP7932 Rack PDU:
Model Number:AP7932
Hardware Revision:B2
Manufacture Date:07/27/2013
Application Module
Name: rpdu
Version: v3.9.2
Date: 07/07/2016
Time: 10:40:30
APC OS (AOS)
Name: aos
Version: v3.9.2
Date: 07/06/2016
Time: 15:12:05

My config is:
`driver: snmp-ups
port: 10.0.0.228
snmp_version = v3
secLevel = authPriv
secName = MYsecNAME
authPassword = MYauthPAS
authProtocol = MD5
privPassword = MYprivPAS
privProtocol = DES
desc = APC PDU`

APC's Webpage has this MIB:
[https://download.schneider-electric.com/files?p_enDocType=Firmware+-+Released&p_Doc_Ref=APC_POWERNETMIB_432&p_File_Name=powernet432.mib](url)

RichardMore, 2021-11-02:
I have the same issue with an older AP7831 PDU [https://www.apc.com/shop/us/en/products/Rack-PDU-Metered-Zero-U-15A-100-120V-16-5-15/P-AP7831](https://www.apc.com/shop/us/en/products/Rack-PDU-Metered-Zero-U-15A-100-120V-16-5-15/P-AP7831) tried snmp v1 and v3 as well, with and without auth/priv options. The website only has MIB reference but no MIB file

jimklimov, 2021-11-08:
Regarding "Unknown mibs value" - a bug for that report was fixed recently (#1161) and in practice the message meant that the device did not respond on the tested OID.

Note that in the original post you seem to be running with the latest *release* of NUT that was alas a few years ago and we are still polishing code for the next official release.

Is it possible for you to test if a build with recent NUT `master` branch from this github repository would fare better?

Also/Alternately, can you check if an `snmpwalk` on these OIDs from the same system would work?

OTOH, check (or sniff) also for firewall settings. SNMP being based on session-less UDP, it may be that restrictive firewalls do not recognize replies from the device as a desired part of a dialog, so they never reach your `snmp-ups` program.

Likewise, do double-check the community names etc: with mismatching credentials (at least in SNMPv1) the servers on the devices tend to not recognize a request and do not respond at all.

RichardMore, 2021-11-09:
Snmpwalk did not have any issues, both with v1 and v3.
A fresh build of the master branch seems to work correctly with snmp-ups. Thank you.

closed 2021-11-10 by jimklimov