networkupstools/nut #1088: Is possible have monitoring power consumption ?

RobertusIT, 2021-09-04:
Hi,

I have a Back-UPS XS 1400U and I config it into Home Assistant.

Is possible have monitoring power consumption, to avoid use another device like Shelly 1 pm that show in Watt, the request of energy ?

jimklimov, 2021-11-13:
I am not sure what Home Assistant and their integration of NUT do; however if the UPS reports the Amps and Volts, you can estimate the Watts of power, and track the wattage values over time to estimate the energy used. Generally you would have to also know the phase shift between the two as a factor (not always 0.7).

I can't immediately say if NUT reports an "estimate" power use like above, I think there were experiments in this area, but might be confined to some drivers like snmp-ups and not spread over the generic codebase. Just not sure, need to read code to know better. It probably does not summarize wattage over time to count the energy use in current up-time; NUT's protocol consumers are better positioned to get and store and visualize such historical data.

Some devices are intended for "metering" and have built-in logic for tracking how much input energy and output per socket they have transmitted; this is generally intended for billing (like in hosting as a service) and reliable. NUT can report those data items for vendors who make them available, e.g. many SNMP ePDUs.

jimklimov, 2022-09-12:
Maybe related to #1428

franz82, 2022-10-31:
Hi all,
the same here, apparently no power comsumption from APC Back-UPS ES 550G. Available data from NUT are:

```
upsc es550g@192.168.x.y
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.date: not set
battery.mfr.date: 2019/05/22
battery.runtime: 2092
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 13.6
battery.voltage.nominal: 12.0
device.mfr: APC
device.model: Back-UPS ES 550G
device.serial: 5B1921T40700  
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
input.sensitivity: medium
input.transfer.high: 266
input.transfer.low: 180
input.transfer.reason: input voltage out of range
input.voltage: 224.0
input.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 870.O4 .I
ups.firmware.aux: O4 
ups.load: 16
ups.mfr: APC
ups.mfr.date: 2019/05/22
ups.model: Back-UPS ES 550G
ups.productid: 0002
ups.serial: 5B1921T40700  
ups.status: OL
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

Thanks,
-f

jimklimov, 2022-11-01:
In the data dump above I don't see any `output.*` values that the consumption could be at least calculated from. The nearest best option is `ups.load` as 16% of nominal ability (probably 550 units... W? VA?), which is hopefully reported by the device.

Out of curiosity, does any other software (vendor's, apcupsd) report anything beside this?

Is it possible to try newer NUT in case mappings to recognize more data points were added over time, and/or to follow the docs about generating an USB HID subdriver (essentially a walk and decode of whatever it reports at standard endpoints) to compare the list presented with your device to one in `apc-hid.c` (and if there are more lines served by yours - PR the additions)?

franz82, 2022-11-01:
> In the data dump above I don't see any `output.*` values that the consumption could be at least calculated from. The nearest best option is `ups.load` as 16% of nominal ability (probably 550 units... W? VA?), which is hopefully reported by the device.

Yeah, `output.*` values are missing as well as `nominal real power`, often used for load to watt calc.
Using this formula:
UPS Load * Nominal Power (W) / 100
16 * 330 / 100
I got more then 50W,
but measuring UPS Load with an external smart plus I got just 38W.

APC Back-UPS ES 550G has:
- 550VA nominal power in VA and
- 330W nominal power in W

> Out of curiosity, does any other software (vendor's, apcupsd) report anything beside this?

I cannot try it soon.