networkupstools/nut #1098: HP T1500 Intl No matching HID UPS found

sundodger, 2021-09-21:
Hi all, trying to get nut up and running on my raspberry pi to monitor the above model UPS. Apparently it should work.
As this UPS has a serial port i am using a usb to serial cable. 
I was following this guide [https://melgrubb.com/2016/12/11/rphs-v2-ups/](url)
However i seem to have a problem with the setup and get the above error.
I suspect it is something to do with the converter cable but have no clue how to check this.
Has anyone setup this model of UPS and if so how did you proceed.
All advice welcome.

sundodger, 2021-09-21:
Not sure if this is important i just plugged the converter cable into a laptop and it shows HP UPS in the device manager and wants drivers installed.
Does this eliminate the cable as the problem?

jimklimov, 2021-09-21:
Since it is an UPS with serial port, it probably does not implement an USB HID protocol -- at least not on a serial port. You should try one of the serial-port drivers using `port = /dev/ttyUSB0` or similar (which your USB-to-serial converter presents to the OS, probably can be seen in `dmesg` among other places).

sundodger, 2021-09-21:
I did try the bcmxcp driver as the usb thing did have me wondering even though usbhid-ups is the one they say works.
The bcmxcp did give the same error but maybe that is just not the right driver.
Will try all the ones listed for HP when i get a second and see does it connect.

sundodger, 2021-09-22:
tried a few drivers and no luck, If i start the drivers in debug mode, i can see it trying to match with the vendor id it sees connected to the usb port. Unfortunately it sees the usb to serial cable id and not the UPS and hence the error.
should the ups show as a device with lsusb as it currently does not.

jimklimov, 2021-11-14:
I think the USB-to-Serial converter would show up in `lsusb`, as providing a software device for the serial port. Normally it does not know or care what is on the other side - UPS, mouse, server console, modem, etc.

jimklimov, 2022-01-10:
Actually, upon revisiting this thread: this is not a USB driver issue since as far as NUT is concerned, this is a Serial-port UPS so should be monitored by a serial-port driver (at a specified /dev/... port device node name).

If you don't have other serial-port devices that could get "corrupted" by noise traffic, you can try `nut-scanner -E '0 9'` to scan the 10 supported port numbers to see if any of Eaton compatible serial protocols would match the device.

Otherwise, you would have to experiment manually with different drivers like `nutdrv_qx` or `blazer_ser` for serial port support...