networkupstools/nut #1166: Optimize nut-scanner multi-threaded loops

jimklimov, 2021-11-08:
After recent PRs for nut-scanner, including #1157, #1158 and #1164, there is quite a bit of repetitive code.

Find ways to:
[ ] Code the check for whether we have enough threads and spawn new ones (or wait until older ones complete) once
[ ] Combine code to portably have the benefits of both solutions mixed there now: of semaphores vs. "naive" counters, and of piecemeal `pthread_tryjoin_np()` where available vs. waiting for all threads to complete
[ ] Cover other network scanning protocols that iterate separate IP addresses (e.g. IPMI but not AVAHI, per cursory investigation)
[ ] Rename variables like start/end IP addresses to the same pattern

user01010111, 2026-09-14:
While checking balanced thread admission and cleanup, I found an unresolved case from #2539: if creation of the global named semaphore fails, the scanner logs the error and retains a NULL pointer, which the scan loops can pass to sem_wait(). Should scans stop in this case, or should they continue using a fallback limit mechanism? The existing per-protocol initialisation failure disables that protocol limit; I would like to confirm the intended global behaviour before changing it.

jimklimov, 2026-09-23:
I think the question was since investigated in the PR, but generally the fallback is better - to ensure we did scan everything we were asked for. If that is for some reason not feasible (e.g. scanning all IPv6 addresses aborted by some stack overflow), ideally we would report that we did not complete the task. Or the OS crash handler would...