networkupstools/nut #1178: Ubuntu on WSL - Can't start /lib/nut/snmp-ups: No such file or directory

scyto, 2021-11-12:
I am trying to test something with NUT in WSL using ubunti.

I installed NUT, SNMP configured my conf files but on start i get

```
Network UPS Tools - UPS driver controller 2.7.4
Can't start /lib/nut/snmp-ups: No such file or directory
```
that directory doesn't include and snmp-ups directory...  not sure what I am missing as i installed using `apt install nut`

scyto, 2021-11-12:
ok found i had to install nut-snmp

the docs are impenetrable

closed 2021-11-12 by scyto

jimklimov, 2021-11-12:
I suppose that would mean Ubuntu docs: like many other FOSS projects, NUT
leaves most of the packaging decisions (what goes where) to OS vendors and
mostly provides init/systemd/smf/... method scripts that those vendors
might use... or roll their own.

scyto, 2021-11-12:
Maybe. I don't know enough about linux (i prefer debian as windows user and just happened to have ubutu loaded up).
This is real ubuntu provided by them.  I dabble in linux when i am forced to, not out of desire.

Do some distro when one installs nut also install the XML, usb, snmp drivers?

If not all it really needs is a quickstart guide that assumes one wants to start with packages, rather than starting with building from source (i get historically why the docs the other way around)  if i get inspired i will write something up and do a docs PR.

Nice project, nice to see it so active, i finally built a private off master thanks to guidance from somoene else to test out the upcoming cyberpower snmp-ups driver changes.  They work well. 6 hours well spent, i learnt something new :-)

jimklimov, 2021-11-12:
Thanks, a PR would be welcome to help other readers coming from a
perspective and background like yours.

For good or bad, NUT tries to be systems agnostic, so every year from
questions like yours we find that some OS or product we never heard of
before ships it, and it's great!

Well, of course big names like Debian, Ubuntu or RHEL/CentOS are more
prominent than others, but there are also quite a few open BSD and Solaris
based systems, veterans like HPUX or AIX are possible, MacOS, etc. Nobody
alone even has them all under hand to look at, let alone packaging nuances
for different generations and forks of the distros :)