networkupstools/nut #1189: Wrong input voltage reported for APC Back-UPS XS 1400U

smainz, 2021-11-17:
I am trying to set up nut fpr my APC  Back-UPS XS 1400U formerly used with apcupsd.

Got almost all working, but nut us reporting wrong input voltage:
```
$ upsc usv01
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.date: 2001/09/25
battery.mfr.date: 2015/07/02
battery.runtime: 1464
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.1
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Back-UPS XS 1400U
device.serial: 3B1527X15613
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0002
driver.parameter.synchronous: no
driver.parameter.vendorid: 051d
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
input.sensitivity: medium
input.transfer.high: 280
input.transfer.low: 155
input.voltage: 98.0
input.voltage.nominal: 102
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 926.T1 .I
ups.firmware.aux: T1
ups.load: 17
ups.mfr: American Power Conversion
ups.mfr.date: 2015/07/02
ups.model: Back-UPS XS 1400U
ups.productid: 0002
ups.realpower.nominal: 700
ups.serial: 3B1527X15613
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

These values are defiitly wrong:

- input.voltage: 98.0
- input.voltage.nominal: 102

I am in Germany and we have 230V. apcupsd reports the correct values.

OS: Raspbian GNU/Linux 10 (buster)
nut version 2.7.4 aken from the official raspbian repositories

Maybe this helps to analyze:

```
pi@pi:~ $ sudo /lib/nut/usbhid-ups -a usv01 -DDD 2>&1
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
[... 249 lines cut ...]
   0.530791     Path: UPS.PresentStatus.BatteryPresent, Type: Feature, ReportID: 0x7a, Offset: 3, Size: 1, Value: 1
   0.530922     Report[buf]: (5 bytes) => 7a 0c 00 00 00
   0.531059     Path: UPS.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x7a, Offset: 4, Size: 1, Value: 0
   0.531191     Report[buf]: (5 bytes) => 7a 0c 00 00 00
   0.531327     Path: UPS.PresentStatus.ShutdownImminent, Type: Feature, ReportID: 0x7a, Offset: 5, Size: 1, Value: 0
   0.531458     Report[buf]: (5 bytes) => 7a 0c 00 00 00
   0.531592     Path: UPS.PresentStatus.RemainingTimeLimitExpired, Type: Feature, ReportID: 0x7a, Offset: 6, Size: 1, Value: 0
   0.531723     Report[buf]: (5 bytes) => 7a 0c 00 00 00
   0.531859     Path: UPS.PresentStatus.CommunicationLost, Type: Feature, ReportID: 0x7a, Offset: 7, Size: 1, Value: 0
   0.531990     Report[buf]: (5 bytes) => 7a 0c 00 00 00
   0.532129     Path: UPS.PresentStatus.NeedReplacement, Type: Feature, ReportID: 0x7a, Offset: 8, Size: 1, Value: 0
   0.532260     Report[buf]: (5 bytes) => 7a 0c 00 00 00
   0.532391     Path: UPS.PresentStatus.Overload, Type: Feature, ReportID: 0x7a, Offset: 9, Size: 1, Value: 0
   0.532517     Report[buf]: (5 bytes) => 7a 0c 00 00 00
   0.532655     Path: UPS.PresentStatus.VoltageNotRegulated, Type: Feature, ReportID: 0x7a, Offset: 10, Size: 1, Value: 0
   0.535121     Report[get]: (2 bytes) => 79 00
   0.535610     Path: UPS.APCPanelTest, Type: Feature, ReportID: 0x79, Offset: 0, Size: 8, Value: 0
   0.539347     Report[get]: (2 bytes) => 78 02
   0.539699     Path: UPS.AudibleAlarmControl, Type: Feature, ReportID: 0x78, Offset: 0, Size: 8, Value: 2
   0.542973     Report[get]: (3 bytes) => 75 62 00
   0.543236     Path: UPS.ff860029, Type: Feature, ReportID: 0x75, Offset: 0, Size: 16, Value: 98
   0.546584     Report[get]: (5 bytes) => 74 a9 1a 64 64
   0.546859     Path: UPS.ff86002a, Type: Feature, ReportID: 0x74, Offset: 0, Size: 32, Value: 1.68428e+09
   0.547205     Report descriptor retrieved (Reportlen = 1028)
```

jimklimov, 2021-11-22:
Can you please check if it performs better with a build of recent NUT codebase, checked out from master, https://github.com/networkupstools/nut/tree/libusb-1.0 or https://github.com/networkupstools/nut/tree/libusb-1.0+0.1 branch?

smainz, 2021-11-22:
I am using a Raspberry Pi 1 for my UPS, compiling takes some time.

Using the branch *libusb-1.0* and building with
```
apt-get remove nut-client nut-server  # remove nut packages
sudo apt-get install autoconf build-essentials automake libtool pkg-config m4
sudo apt-get install libusb-1.0 libusb-1.0.0-dev
autoreconf --install
./autogen.sh
autoconf
./configure --with-user=nut --with-group=nut --with-usb
make
sudo make install
```

and testing with
```
cd /usr/local/ups
cp -rp /etc/nut/* etc   # use my old config files
chown -R nut etc/*
sudo sbin/upsdrvctl start
sudo sbin/upsd
bin/upsc usv01
```

input.voltage is still wrong (but different):

```
pi@pi:/usr/local/ups $ bin/upsc usv01
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.date: 2001/09/25
battery.mfr.date: 2015/07/02
battery.runtime: 1248
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.1
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Back-UPS XS 1400U
device.serial: 3B1527X15613
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0002
driver.parameter.synchronous: no
driver.parameter.vendorid: 051d
driver.version: 2.7.4-3392-ga2a81dd1
driver.version.data: APC HID 0.96
driver.version.internal: 0.44
driver.version.usb: libusb-1.0 (API: 0x1000106)
input.sensitivity: medium
input.transfer.high: 280
input.transfer.low: 155
input.voltage: 140.0
input.voltage.nominal: 140
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 926.T1 .I
ups.firmware.aux: T1
ups.load: 19
ups.mfr: American Power Conversion
ups.mfr.date: 2015/07/02
ups.model: Back-UPS XS 1400U
ups.productid: 0002
ups.realpower.nominal: 700
ups.serial: 3B1527X15613
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

Will try with the other branches tomorrow

smainz, 2021-11-23:
Same result with branch *libusb-1.0+0.1*

branch master is compiling now, will report this evening.

smainz, 2021-11-23:
OK tried *master* as well, but still the wrong input.voltage and input.voltage.nominal.

Is there anything else I can do to help fix this?

andbez, 2022-02-03:
Same for my device Back-UPS XS 950U. I am living in Germany too with 230V. Any news on this issue?

smainz, 2022-02-04:
Unfortunately not. I had to switch back to apcupsd to make it work.

My setup is now, that I have an old Raspberry Pi with apcupsd monitoring the UPS on the USB-port and provide the UPS info to the local networt.

On other systems I use nut with the apcupsd-ups driver having the port pointing to the Pi.

jimklimov, 2022-02-04:
One more idea popped up like in other tickets: can you try following the
developer docs to try generating a usbhid-ups subdriver - there's a script
that effectively walks your device and generates a C file with mapping
table for whatever it finds.

Possibly your device serves the needed data in nodes not mapped (maybe also
scaled) into NUT names, standards are ambiguously implemented by many
market actors; then it would be a matter of adding the lines to your
originally used subdriver source (note: first hit in a walk wins), rebuild,
test - and if that helps - post a PR to update the subdriver.

smainz, 2022-02-04:
@jimklimov I will try this weekend and scream back on you if I need more advice.

Thanks for careing

smainz, 2022-02-06:
OK, no luck so far. I created the subdriver using the documented commands:

```
$ drivers/usbhid-ups -DD -u root -x explore -x vendorid=051D \
        -x port=auto -s ups 2>&1 | tee apcxs1400.txt
$ cd drivers
$ ../scripts/subdriver/gen-usbhid-subdriver.sh < ../apcxs1400.txt
```

Followed the instructions

```
Please enter a name for this driver. Use only letters and numbers. Use
natural (upper- and lowercase) capitalization, e.g., 'Belkin', 'APC'.
Name of subdriver: APCXS1400
Creating apcxs1400-hid.h
Creating apcxs1400-hid.c
Done.

Do not forget to:
* add #include "apcxs1400-hid.h" to drivers/usbhid-ups.c,
* add &apcxs1400_subdriver to drivers/usbhid-ups.c:subdriver_list,
* add apcxs1400-hid.c to USBHID_UPS_SUBDRIVERS in drivers/Makefile.am
* add apcxs1400-hid.h to dist_noinst_HEADERS in drivers/Makefile.am
* "autoreconf" from the top level directory
```

Placed my driver before the apc-hid already in usbhid-ups.c

and rebuilt it. 

Unfortunally it segfaults when executing 

```
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4536-gd46d5153)
USB communication driver (libusb 1.0) 0.42
   0.000000     [D1] debug level is '2'
   0.009538     [D2] Initializing an USB-connected UPS with library libusb-1.0.24 (API: 0x1000108) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.42')
   0.013648     [D1] upsdrv_initups (non-SHUT)...
   0.057151     [D2] Checking device 1 of 4 (051D/0002)
   0.099389     [D2] - VendorID: 051d
   0.103395     [D2] - ProductID: 0002
   0.105010     [D2] - Manufacturer: American Power Conversion
   0.106493     [D2] - Product: Back-UPS XS 1400U  FW:926.T1 .I USB FW:T1
   0.107006     [D2] - Serial Number: 3B1527X15613
   0.107381     [D2] - Bus: 001
   0.107748     [D2] - Device: unknown
   0.109048     [D2] - Device release number: 0106
   0.109443     [D2] Trying to match device
   0.110196     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.111621     [D2] Device matches
   0.112020     [D2] Reading first configuration descriptor
   0.113502     [D2] Claimed interface 0 successfully
   0.116668     [D2] HID descriptor length 1028
   0.298714     [D2] Report Descriptor size = 1028

   0.299895     Using subdriver: APCXS1400 HID 0.1
Speicherzugriffsfehler
```

What else can I do?

Files attached.
[apcxs1400.txt](https://github.com/networkupstools/nut/files/8010613/apcxs1400.txt)
[apcxs1400-hid.c.txt](https://github.com/networkupstools/nut/files/8010615/apcxs1400-hid.c.txt)

jimklimov, 2022-02-06:
Can you see if you can get a meaningful reason for that segfault? maybe
some null pointer or a buffer overflow (e.g. your HID custom name looks
long :))

Something like

gdb --args drivers/usbhid-ups -x ...

and when GDB starts, type "run", and when that fails - "bt" for backtrace.

You may need to embed this into the libtool shell script if
"drivers/usbhid-ups" is one (with binary in ".libs" subdir).

smainz, 2022-02-06:
Need more help:

```
Thread 1 "usbhid-ups" received signal SIGSEGV, Segmentation fault.
0x00000000 in ?? ()
(gdb) bt
#0  0x00000000 in ?? ()
#1  0x00012b94 in callback (argudev=<optimized out>, arghd=0x4a2f0 <curDevice>, rdbuf=<optimized out>, rdlen=<optimized out>)
    at usbhid-ups.c:1239
#2  0x00016d08 in nut_libusb_open (udevp=udevp@entry=0x4a2bc <udev>, curDevice=curDevice@entry=0x4a2f0 <curDevice>, matcher=0x0,
    callback=0x404, callback@entry=0x12ad0 <callback>) at libusb1.c:548
#3  0x00014730 in upsdrv_initups () at usbhid-ups.c:1037
#4  0x00012450 in main (argc=<optimized out>, argv=<optimized out>) at main.c:690
```

renamed apcxs1400 to xs, but no change

Running on a Raspberry Pi (early model), on buster

smainz, 2022-02-06:
OK, found the reason for the segfault. The script creates incorect c code.

instead of:

```
subdriver_t xs_subdriver = {
        XS_HID_VERSION,
        xs_claim,
        xs_utab,
        xs_hid2nut,
        xs_format_model,
        xs_format_mfr,
        xs_format_serial,
        fix_report_desc,
};
```

it creates

```
subdriver_t xs_subdriver = {
        XS_HID_VERSION,
        xs_claim,
        xs_utab,
        xs_hid2nut,
        xs_format_model,
        xs_format_mfr,
        xs_format_serial,
};
```
Note the missing last line.

smainz, 2022-02-06:
But it does not seem to be different from what the apc-hid subdriver does:

```
  64.798816     [D2] Path: UPS.Input.ConfigVoltage, Type: Feature, ReportID: 0x30, Offset: 0, Size: 8, Value: 140
  64.808107     [D2] Path: UPS.Input.HighVoltageTransfer, Type: Feature, ReportID: 0x33, Offset: 0, Size: 16, Value: 280
  64.814238     [D2] Path: UPS.Input.LowVoltageTransfer, Type: Feature, ReportID: 0x32, Offset: 0, Size: 16, Value: 155
  64.819999     [D2] Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 140
```
apcupsd reports:

```
LOTRANS  : 155.0 Volts
HITRANS  : 280.0 Volts
LINEV    : 228.0 Volts
```

Maybe LINEV is not equal to Input.Voltage?

smainz, 2022-02-07:
This is all the EXPLORE gave me, no additional voltage reading, nor a sign to indicate EU/US/... (I am in the EU).

```
0.322581	Using subdriver: EXPLORE HID 0.2
   0.322696	[D1] 104 HID objects found
   0.326426	[D1] Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.333161	[D1] Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2
   0.337649	[D1] Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 4
   0.342204	[D1] Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x04, Offset: 0, Size: 8, Value: 3
   0.350154	[D1] Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 1
   0.353826	[D1] Path: UPS.PowerSummary.Charging, Type: Input, ReportID: 0x06, Offset: 0, Size: 8, Value: 0
   0.354865	[D1] Path: UPS.PowerSummary.Charging, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 0
   0.356885	[D1] Path: UPS.PowerSummary.Discharging, Type: Input, ReportID: 0x06, Offset: 8, Size: 8, Value: 0
   0.358825	[D1] Path: UPS.PowerSummary.Discharging, Type: Feature, ReportID: 0x06, Offset: 8, Size: 8, Value: 0
   0.361031	[D1] Path: UPS.PowerSummary.ff860060, Type: Input, ReportID: 0x06, Offset: 16, Size: 8, Value: 8
   0.362884	[D1] Path: UPS.PowerSummary.ff860060, Type: Feature, ReportID: 0x06, Offset: 16, Size: 8, Value: 8
   0.366948	[D1] Path: UPS.PowerSummary.ManufacturerDate, Type: Feature, ReportID: 0x07, Offset: 0, Size: 16, Value: 18146
   0.371530	[D1] Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x08, Offset: 0, Size: 16, Value: 24
   0.374599	[D1] Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x09, Offset: 0, Size: 16, Value: 27.25
   0.383155	[D1] Path: UPS.PowerSummary.iManufacturer, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 8, Value: 3
   0.387652	[D1] Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 8, Value: 2
   0.391916	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x0c, Offset: 0, Size: 8, Value: 100
   0.392934	[D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 8, Value: 100
   0.394852	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x0c, Offset: 8, Size: 16, Value: 524
   0.396789	[D1] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x0c, Offset: 8, Size: 16, Value: 524
   0.399362	[D1] Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x0d, Offset: 0, Size: 8, Value: 100
   0.407140	[D1] Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 8, Value: 100
   0.412675	[D1] Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 8, Value: 50
   0.419637	[D1] Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x10, Offset: 0, Size: 8, Value: 1
   0.423883	[D1] Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x11, Offset: 0, Size: 8, Value: 10
   0.432659	[D1] Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x12, Offset: 0, Size: 8, Value: 1
   0.437273	[D1] Path: UPS.PowerSummary.ACPresent, Type: Input, ReportID: 0x13, Offset: 0, Size: 8, Value: 1
   0.439321	[D1] Path: UPS.PowerSummary.ACPresent, Type: Feature, ReportID: 0x13, Offset: 0, Size: 8, Value: 1
   0.444659	[D1] Path: UPS.PowerSummary.BelowRemainingCapacityLimit, Type: Input, ReportID: 0x14, Offset: 0, Size: 8, Value: 0
   0.445822	[D1] Path: UPS.PowerSummary.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x14, Offset: 0, Size: 8, Value: 0
   0.447452	[D1] Path: UPS.PowerSummary.ShutdownImminent, Type: Input, ReportID: 0x14, Offset: 8, Size: 8, Value: 0
   0.447576	[D1] Path: UPS.PowerSummary.ShutdownImminent, Type: Feature, ReportID: 0x14, Offset: 8, Size: 8, Value: 0
   0.453654	[D1] Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x15, Offset: 0, Size: 16, Value: -1
   0.457017	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x16, Offset: 0, Size: 1, Value: 0
   0.458068	[D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x16, Offset: 0, Size: 1, Value: 0
   0.459586	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Input, ReportID: 0x16, Offset: 1, Size: 1, Value: 0
   0.459710	[D1] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x16, Offset: 1, Size: 1, Value: 0
   0.459820	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x16, Offset: 2, Size: 1, Value: 1
   0.459926	[D1] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x16, Offset: 2, Size: 1, Value: 1
   0.460030	[D1] Path: UPS.PowerSummary.PresentStatus.BatteryPresent, Type: Input, ReportID: 0x16, Offset: 3, Size: 1, Value: 1
   0.460133	[D1] Path: UPS.PowerSummary.PresentStatus.BatteryPresent, Type: Feature, ReportID: 0x16, Offset: 3, Size: 1, Value: 1
   0.460597	[D1] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Input, ReportID: 0x16, Offset: 4, Size: 1, Value: 0
   0.460740	[D1] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x16, Offset: 4, Size: 1, Value: 0
   0.460848	[D1] Path: UPS.PowerSummary.PresentStatus.ShutdownImminent, Type: Input, ReportID: 0x16, Offset: 5, Size: 1, Value: 0
[... cut ...]
static hid_info_t xs_hid2nut[] = {

	{ "unmapped.ups.apcxs140010", 0, 0, "UPS.APCXS140010", NULL, "%.0f", 0, NULL },
	{ "unmapped.ups.apcxs140011", 0, 0, "UPS.APCXS140011", NULL, "%.0f", 0, NULL },
	{ "unmapped.ups.apcxs140015", 0, 0, "UPS.APCXS140015", NULL, "%.0f", 0, NULL },
	{ "unmapped.ups.apcxs14001.apcxs14005", 0, 0, "UPS.APCXS14001.APCXS14005", NULL, "%.0f", 0, NULL },
	{ "unmapped.ups.apcxs14001.apcxs14007", 0, 0, "UPS.APCXS14001.APCXS14007", NULL, "%.0f", 0, NULL },
```
Is this something I should have done?

jimklimov, 2022-02-07:
Just in case: the fix-up of faulty HID info was explored in PR #1245 adding a generic hook method that subdrivers may implement for known deficiencies as those surface...

jimklimov, 2022-02-07:
CC @nbriggs : Hi, maybe you can help make sense out of this?

I also wonder what the `explore-hid(.c?)` covers - maybe *it* may need some improvements from some source of modern standard tables? And likewise, what are those hex name components - something reported by the device that we don't know how to interpret?..

Alas, I'm "just" a maintainer to help things roll along, but not a deep expert in all the nuances...

smainz, 2022-02-07:
Tell me, what I can do to help, but it es gtting close to my skill level. I do not know the details of how USB works, not am I into the internals of nut.

Compiling and running stuff is ok and some C skills have resist the ravages of time.

nbriggs, 2022-02-07:
Hi @smainz -- nothing specific to do yet, but I expect that seeing the results of running some NUT programs in debug mode will be the most useful once I've had a chance to look at the code.  I have a "American Power Conversion Back-UPS RS 1500G", but I'm in the US and that unit is 120VAC *only* (you'd think that auto voltage sensing would make sense, but I guess the market is large enough they can save cost by selling models specific to the input voltage)

nbriggs, 2022-02-08:
The "EXPLORE" driver doesn't know about any of the vendor-specific usages (as defined for the APC units in `apc-hid.c`) so it will put out the hex value for those.  For example, you see
```
   0.361031	[D1] Path: UPS.PowerSummary.ff860060, Type: Input, ReportID: 0x06, Offset: 16, Size: 8, Value: 8
   0.362884	[D1] Path: UPS.PowerSummary.ff860060, Type: Feature, ReportID: 0x06, Offset: 16, Size: 8, Value: 8
```
in the explore output dump, where ff860060 actually corresponds to "APCStatusFlag" in the original dump that was submitted:
```
   0.307857     Report[buf]: (4 bytes) => 06 00 00 08
   0.308263     Path: UPS.PowerSummary.APCStatusFlag, Type: Input, ReportID: 0x06, Offset: 16, Size: 8, Value: 8
   0.309966     Report[buf]: (4 bytes) => 06 00 00 08
   0.310403     Path: UPS.PowerSummary.APCStatusFlag, Type: Feature, ReportID: 0x06, Offset: 16, Size: 8, Value: 8
```

nbriggs, 2022-02-08:
... so the newly generated sub-driver will only make things worse here, and any info that you learn about undecoded fields can be merged back into the apc-specific tables in apc-hid.c -- however, I don't see a path to finding any authoritative documentation unless Schneider Electric/APC is willing to give it up, and I am not hopeful about that.

smainz, 2022-02-08:
This is so crazy, apcupsd from 2016 can read the data and nut can not. I am not aware where the guys from apcupsd got their specs from and am too dumb to find it within a few minutes in their code/docs.

Is this something which rings a bell on you 

from src/drivers/usb/usb.c:

```
/*
 * This table is used when walking through the USB reports to see
 * what information found in the UPS that we want. If the usage_code 
 * and the physical code match, then we make an entry in the command
 * index table containing the usage information provided by the UPS
 * as well as the data type from this table. Entries in the table 
 * with ci == CI_NONE are not used, for the moment, they are
 * retained just so they are not forgotten.
 */
const UsbUpsDriver::s_known_info UsbUpsDriver::_known_info[] = {
   /*  Page 0x84 is the Power Device Page */
   /* CI                        USAGE       PHYSICAL   LOGICAL  TYPE        VOLATILE? */
   {CI_NONE,                    0x00840001, P_ANY,     P_ANY,   T_INDEX,    false},  /* iName */
   {CI_VLINE,                   0x00840030, P_INPUT,   P_ANY,   T_UNITS,    true },  /* Line Voltage */
   {CI_VOUT,                    0x00840030, P_OUTPUT,  P_ANY,   T_UNITS,    true },  /* Output Voltage */
   {CI_VBATT,                   0x00840030, P_BATTERY, P_ANY,   T_UNITS,    true },  /* Battery Voltage */
   {CI_VBATT,                   0x00840030, P_ANY,     P_PWSUM, T_UNITS,    true },  /* Battery Voltage (alternative) */
   {CI_NONE,                    0x00840031, P_ANY,     P_ANY,   T_UNITS,    false},  /* Current */
   {CI_FREQ,                    0x00840032, P_OUTPUT,  P_ANY,   T_UNITS,    true },  /* Frequency */
```

from example/hid-ups_info.c:
```
struct s_ups_info {
    unsigned usage;
    int type;
    const char *label;
} ups_info[] = {

    /* MGE & APC */
    { 0x000000, T_NONE,    "None" },
    /* Page 0x84 is the Power Device Page */
    { 0x840000, T_NONE,    "UPS-Power" },
    { 0x840001, T_INDEX,   "iName" },
    { 0x840002, T_NONE,    "PresentStatus" },
    { 0x840004, T_NONE,    "UPS" },
    { 0x840012, T_NONE,    "Battery" },
    { 0x840016, T_NONE,    "PowerConverter" },
    { 0x840018, T_NONE,    "OutletSystem" },
    { 0x840017, T_NONE,    "PowerConverterID" },
    { 0x840019, T_NONE,    "OutletSystemID" },
    { 0x84001a, T_NONE,    "Input" },
    { 0x84001c, T_NONE,    "Output" },
    { 0x84001e, T_NONE,    "Flow" }, 
    { 0x84001d, T_NONE,    "OutputID" },
    { 0x84001f, T_NONE,    "FlowID" },
    { 0x840020, T_NONE,    "Outlet" },
    { 0x840021, T_NONE,    "OutletID" },
    { 0x840024, T_NONE,    "PowerSummary" },
    { 0x840025, T_NONE,    "PowerSummaryID" },
    { 0x840030, T_UNITS,   "Voltage" },
    { 0x840031, T_UNITS,   "Current" },
```

nbriggs, 2022-02-08:
@smainz -- If your UPS is still under warranty (or you have any kind of support contract) you might consider opening a case with APC/Schneider Electric to get them to fix their firmware so that it reports the "correct" logical range for the input and configured input voltage in the HID report descriptor.  I suspect they just forgot to update those limits when they produced the unit for the European market.  If people complain, there's a small chance they might fix it.

nbriggs, 2022-02-08:
@smainz Could you try the branch [1189_APC_UPS_wrong_minmax_input_voltage](https://github.com/nbriggs/nut/tree/1189_APC_UPS_wrong_minmax_input_voltage) -- if you don't want to recompile the whole thing, copy `drivers/apc-hid.c` from there to the corresponding place in your previously compiled nut master branch and redo the make there.  Re-run your test with debug level 4 (-DDDD) and see if you see the right stuff.

smainz, 2022-02-09:
> If your UPS is still under warranty (or you have any kind of support contract) you might consider opening a case with APC/Schneider Electric to get them to fix their firmware so that it reports the [...]

My UPS is out of warranty and I do not have a support contract. In the net I have found a few complaints about the same problem on the same model. Maybe APC has changed the firmware in the mean time.

smainz, 2022-02-09:
> Could you try the branch [1189_APC_UPS_wrong_minmax_input_voltage](https://github.com/nbriggs/nut/tree/1189_APC_UPS_wrong_minmax_input_voltage) -- if you don't want to recompile the whole thing, [...]

Checked out your branch, my pi will be compiling a while and when I am back this evening I will report progress.
Thanks for taking a look.

smainz, 2022-02-09:
How much of the output do you need? I have the following:

```
  0.343381     Using subdriver: APC HID 0.97
   0.347043     [D3] Attempting Report Descriptor fix for UPS: Vendor: 051d, Product: 0002
   0.348353     [D4] Report Descriptor: hvt input LogMin: 280 LogMax: 280
   0.349634     [D4] Report Descriptor: voltage LogMin: 75 LogMax: 140
   0.350030     [D3] Fixing Report Descriptor. Set voltage LogMin = 0, LogMax = 560
   0.350407     [D4] Report Descriptor: configVoltage LogMin: 75 LogMax: 140
   0.351448     [D3] Fixing Report Descriptor. Set configVoltage LogMin = 0, LogMax = 255
   0.351942     [D2] Report Descriptor Fixed
   0.353638     [D1] 104 HID objects found
   0.354071     [D4] Entering libusb_get_report
   0.357943     [D3] Report[get]: (2 bytes) => 01 01
   0.358474     [D1] Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
```

```
  0.594872     [D1] Path: UPS.Battery.ff860018, Type: Feature, ReportID: 0x28, Offset: 0, Size: 32, Value: 8.90083e+08
   0.595259     [D4] Entering libusb_get_report
   0.602577     [D3] Report[get]: (2 bytes) => 30 e6
   0.603128     [D1] Path: UPS.Input.ConfigVoltage, Type: Feature, ReportID: 0x30, Offset: 0, Size: 8, Value: 230
   0.604434     [D4] Entering libusb_get_report
   0.606983     [D3] Report[get]: (3 bytes) => 31 e4 00
   0.607525     [D1] Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 228
   0.609183     [D4] Entering libusb_get_report
   0.613316     [D3] Report[get]: (3 bytes) => 32 9b 00
   0.614231     [D1] Path: UPS.Input.LowVoltageTransfer, Type: Feature, ReportID: 0x32, Offset: 0, Size: 16, Value: 155
   0.615553     [D4] Entering libusb_get_report
   0.618396     [D3] Report[get]: (3 bytes) => 33 18 01
   0.618939     [D1] Path: UPS.Input.HighVoltageTransfer, Type: Feature, ReportID: 0x33, Offset: 0, Size: 16, Value: 280
   0.619328     [D4] Entering libusb_get_report
   0.622674     [D3] Report[get]: (2 bytes) => 34 84
```

This looks right to me 228V Input Voltage. So your idea was correct, the min and max values are wrong and nut limits the values to these values.

Now what can I do to get this into a release?

PS: it was too early this morning and I made the little Pi compile the master branch, so I had do compile it again. So I was lucky about the message Fixing Report Descriptor. ...

nbriggs, 2022-02-09:
That output is sufficient, thanks.  Sorry to hear you ended up compiling twice.
I'll ask a couple of people (probably Jim K and the guy who added the scaffolding for the fix_report_descriptor feature) to review the changes I made and if they approve then Jim K will merge my changes into master.

I don't know what Jim's schedule is for creating NUT releases -- I see v2.7.4 from Sep 26, 2021.  I also don't know what constraints, if any, you have on what version of the code you run -- one of the sites I installed NUT for is happy to run off the master branch rather than waiting for a tagged release, others, not so much.

smainz, 2022-02-09:
@andbez Does the version of @nbriggs work for your UPS as well? 
If you can not comple it, I can provide you with binaries.

smainz, 2022-02-09:
> I don't know what Jim's schedule is for creating NUT releases -- I see v2.7.4 from Sep 26, 2021. I also don't know what constraints, if any, you have on what version of the code you run -- one of [...]

If I do it the normal way, I'd to wait until it ends up in a distribution, but if I see it will end up in nut some time, I am going to convert from apcupsd to nut in the near future. I will take the master branch and be happy.

Thank you so much.

andbez, 2022-02-09:
@nbriggs First let me thank all of you for your support. Please find attached the Output of usbhid-ups (Network UPS Tools - Generic HID driver 0.41 (2.7.4) on Debian Bullseye). 
[ups_device.log](https://github.com/networkupstools/nut/files/8036602/ups_device.log)
.

nbriggs, 2022-02-09:
@andbez Thanks.  Your log info shows exactly the same issue (wrong logical minimum/maximum in the report descriptor) as smainz's.  Also, the vendor ID and product ID are the same, and the report numbers of the problematic items (voltage and configVoltage) are also the same.  So... I believe if you were to compile my branch   [1189_APC_UPS_wrong_minmax_input_voltage](https://github.com/nbriggs/nut/tree/1189_APC_UPS_wrong_minmax_input_voltage) you'd find that running the same test would report the correct values for configVoltage and voltage -- are you in a position to try that?

andbez, 2022-02-09:
@nbriggs Not really. I am SQL and Python-Developer, but if @smainz can provide an Debian Intel/AMD-Binary, I can test it.

nbriggs, 2022-02-10:
I've built a Debian/Intel VM and built a usbhid-ups from my new code -- what's the best way to get it to you?  I don't think attaching a 700 KB file here is socially responsible ;-)

andbez, 2022-02-10:
I can provide a Nextcloud-Share. Can I send you a PM somehow?

nbriggs, 2022-02-10:
@andbez -- I remembered I have a generic web service from my ISP, so I put up http://nbriggs.users.sonic.net/usbhid-ups (note http, 'cause it's a "free" service) and you can curl or wget that to test with.

% openssl sha256 usbhid-ups
SHA256(usbhid-ups)= b33b5c77e5df1273025c98513b9a492d780757c833881f1a5da9096bdd212809

nbriggs, 2022-02-10:
@andbez -- sorry, had to correct the URL there.  It's http://nhbriggs.users.sonic.net/usbhid-ups (rather than nbriggs).

smainz, 2022-02-10:
@andbez I can't see these lines

```
  0.343381     Using subdriver: APC HID 0.97
   0.347043     [D3] Attempting Report Descriptor fix for UPS: Vendor: 051d, Product: 0002
   0.348353     [D4] Report Descriptor: hvt input LogMin: 280 LogMax: 280
```

in the output. Is this log taken with the correct verison?

Looks like the log from the old subdriver and as you see:
```
   0.325815	[D1] Path: UPS.Input.ConfigVoltage, Type: Feature, ReportID: 0x30, Offset: 0, Size: 8, Value: 140
   0.330057	[D3] Report[get]: (3 bytes) => 31 e6 00
   0.330107	[D1] Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 140
   0.333114	[D3] Report[get]: (3 bytes) => 32 9b 00
```
Input.Voltage is still 140

You APC is the same model as mine, except for the smaller battery.

andbez, 2022-02-10:
@nbriggs @smainz Tried with the new version and now the Input.Voltage-Value looks better:
  0.330174	[D1] Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 234
Whole log attached
[ups_log_new.log](https://github.com/networkupstools/nut/files/8043372/ups_log_new.log)
.

jimklimov, 2022-02-10:
Looks promising, thanks @nbriggs for the proposed fix and @andbez & @smainz for testing on short notice!

>  NUT releases -- I see v2.7.4 from Sep 26, 2021
It's actually "worse" than that: this release date is from "drafting a release" on github, which is independent of the release tag... would be 6 years since that point, soon :(

smainz, 2022-02-10:
> It's actually "worse" than that: this release date is from "drafting a release" on github, which is independent of the release tag... would be 6 years since that point, soon :( So tying up loose threads in order for 2.7.5 to not be an epic to talk about for eons ;)

I see what you mean, but the longer it is since the last release, the sooner we will see the next one :-)

Thank you for your effort.

nbriggs, 2022-02-11:
@flashydave -- quick question if you don't mind:  I'm using your fix_report_desc framework to fix the problem smainz reported here for an APC unit, and I noticed that in the CPS code you chose to implement `FindReport()` to find the entry by ReportID and node.  I'm wondering (a) if that should be in hidparser.c, along with `FindObject_with_path()` and `FindObject_with_ID()`, and (b) what was the reasoning behind your locating the entry to be fixed based on the ID+node rather than either of the existing methods?

flashydave, 2022-02-11:
I agree it is similar to those functions.
I see no reason not to relocate it if has wider use with other UPS
models.  I was simply trying to keep the fix local to the driver it
related to and the function itself fell out of the need to call it multiple times.  I would have preferred one of the existing funxtions but there were complications over access to parameters at the point where it was called and there were speed/performance issues to consider as mentioned in the HIDDumpTree() comments.

nbriggs, 2022-02-11:
@flashydave -- OK.

Re. speed & performance: The comments in `HIDDumpTree()` don't apply to our situation -- `HIDDumpTree()` is going to fetch the value from the UPS for each item in the tree, which as they note might take a non-trivial amount of time (especially if the UPS doesn't respond), while we are only inspecting the descriptor array itself and never fetch anything from the UPS, so both `FindReport()` and `FindObject_with_ID()` are O(n) (for n = number of descriptors) and run at memory access speed, once, at startup.

For the APC case I also need to look at the high voltage transfer threshold, but initially chose to use
```
pData=FindObject_with_ID(pDesc_arg, 0x33, 0, ITEM_FEATURE)
```
vs.
```
pData=FindReport(pDesc_arg, 16, (PAGE_POWER_DEVICE<<16)+USAGE_HIGHVOLTAGETRANSFER)
```
-- I checked the report descriptor to see that for the APC it sends report ID 0x33 (vs. 16 for your CPS) at an offset of 0 and as a FEATURE.  I think I prefer your approach to locating the specific item, but I was a little surprised to find that the headers didn't already have a standard set of definitions for the usage tables based on the USB-IF spec (pdcv11.pdf -- "Universal Serial Bus Usage Tables for HID Power Devices").  There's the array of string to code (`hid_usage_lkp`) defined in libhid.c, but it just uses the magic numbers rather than constructing them from individual defines (or enum elements -- style choice!) -- looking at what includes what, perhaps hidparser.h (where you put the USAGE_HIGHVOLTAGETRANSFER) would be the right place to put a complete set?

flashydave, 2022-02-12:
Agree with your point in general re performance and accessing hardware
but felt also the computational (and programming) effort constructing path objects wasnt worth it.  In my case numerics were sufficient to detect the buggy situation reasonably reliably and apply the fix.

I also was surprised a full set of magic numbers had not been defined
but I guess it starts to get complicated if you have to deal with vendor
unique numbers too should those be needed as they might not be
interpreted the same
across manufacturers. I just added the ones I needed.  I cant see any
reason not to add more if you need them..

nbriggs, 2022-02-12:
@flashydave, @jimklimov  -- I have a NUT style question: I've taken the `rd.conf` file from github.com:abend0c1/hidrdd.git (actually, a very slightly updated version of that for which I've submitted a PR), which incorporates the complete USB-IF usage definitions for power device and battery system pages and processed it with
```
grep -e ^0084 -e ^0085 rd.conf | sed 's/,.*$//;s/ *$//' | sed 's/ /_/g;s/_/ /' | tr '[:lower:]' '[:upper:]' | sed 's/\(0085.... \)/\1USAGE_BAT_/;s/\(0084.... \)/\1USAGE_POW_/;s/\([A-Z_]*\)_PAGE/PAGE_\1/' | awk '{print "#define "$2" 0x"$1}'
```
to produce defines for the usages that look like
```
#define PAGE_POWER_DEVICE 0x0084
#define USAGE_POW_UNDEFINED 0x00840000
#define USAGE_POW_I_NAME 0x00840001
#define USAGE_POW_PRESENT_STATUS 0x00840002
[...]
#define PAGE_BATTERY_SYSTEM 0x0085
#define USAGE_BAT_UNDEFINED 0x00850000
#define USAGE_BAT_SMB_BATTERY_MODE 0x00850001
#define USAGE_BAT_SMB_BATTERY_STATUS 0x00850002
#define USAGE_BAT_SMB_ALARM_WARNING 0x00850003
[...]
```
... since the power device and battery system pages have conflicting field names we need to include the `POW_` or `BAT_` to differentiate, but the real style question is: underscores in the variable part of the field names or not? -- that is `USAGE_BAT_SMB_BATTERY_MODE` or `USAGE_BAT_SMBBATTERYMODE`?

I'm happy either way but it's not my codebase so I'm asking...

flashydave, 2022-02-12:
I prefer the latter style so that the _ acts more like a heirarchical
separator but I'm sure someone else could suggest an alternative rule.
Its not my code either so I can't say definitively!

jimklimov, 2022-02-13:
I like underscores as more readable, especially where we can't have CamelCase etc. Maybe follow precedent of nut-ddl structured names with two underscores between tokens? like USAGE__BAT__SMB_BATTERY_STATUS ?

flashydave, 2022-02-13:
Thats starting to get messy. Not only does it become irksome to type
(especially for those using command line type editors) and reduces the
amount of code that can seen on any guven size of screen.   For example
my recent contributions were all developed using ssh from my phone!

jimklimov, 2022-02-13:
Well then, given a small dictionary of prefixes to juggle, single
underscores can be good and not ambiguous... at least, it's sure better to
get fixes delivered than to bikeshed over this :-)

flashydave, 2022-02-13:
I agree!

nbriggs, 2022-02-13:
OK... I've (force) pushed a revised version of [1189_APC_UPS_wrong_minmax_input_voltage](https://github.com/nbriggs/nut/tree/1189_APC_UPS_wrong_minmax_input_voltage) with all the changes and stored a new version of usbhid-ups for x86_64 (SHA256(usbhid-ups)= 17d3420fc892f1cd8fb3d5e4c3b766708890725b76dbfe90ea8a6ddaaa27d495) on http://nhbriggs.users.sonic.net/usbhid-ups if @smainz and/or @andbez want to re-test on their APC units.

smainz, 2022-02-14:
@nbriggs Is it enough for you when I run this commit on my Pi or do you need x86_64 too? The later means, i have to move cabinets.

Pi is compiling now, you will have feedback later this evening.

nbriggs, 2022-02-14:
@smainz -- Pi is enough, I think @andbez has the x86_64 but not the setup for compiling it.

smainz, 2022-02-14:
Here is:

```
$ sudo drivers/usbhid-ups -DDD -s usv01 -x port=auto  -u root
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4615-gbaae14d9)
USB communication driver (libusb 1.0) 0.42
```

```
   0.390418     Using subdriver: APC HID 0.97
   0.392100     [D3] Attempting Report Descriptor fix for UPS: Vendor: 051d, Product: 0002
   0.392516     [D3] Fixing Report Descriptor. Set voltage LogMin = 0, LogMax = 560
   0.392899     [D3] Fixing Report Descriptor. Set configVoltage LogMin = 75, LogMax = 255
   0.393278     [D2] Report Descriptor Fixed
```

```
   0.614673     [D1] Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 228
   0.618179     [D3] Report[get]: (3 bytes) => 32 9b 00
   0.618711     [D1] Path: UPS.Input.LowVoltageTransfer, Type: Feature, ReportID: 0x32, Offset: 0, Size: 16, Value: 155
   0.621399     [D3] Report[get]: (3 bytes) => 33 18 01
```

Looks good. Thank you

andbez, 2022-02-14:
Here is: ;-)

root@omv:~# ./usbhid-ups -a ups -u root -DDD 2>&1
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4540-gaefcdc0a)
USB communication driver (libusb 1.0) 0.42

   0.185032	Using subdriver: APC HID 0.97
   0.185036	[D3] Attempting Report Descriptor fix for UPS: Vendor: 051d, Product: 0002
   0.185041	[D3] Fixing Report Descriptor. Set voltage LogMin = 0, LogMax = 560
   0.185045	[D3] Fixing Report Descriptor. Set configVoltage LogMin = 75, LogMax = 255
   0.185048	[D2] Report Descriptor Fixed

   0.329770	[D1] Path: UPS.Input.Voltage, Type: Fea ture, ReportID: 0x31, Offset: 0, Size: 16, Value: 236
   0.334925	[D3] Report[get]: (3 bytes) => 32 9b 00
   0.334941	[D1] Path: UPS.Input.LowVoltageTransfer, Type: Feature, ReportID: 0x32, Offset: 0, Size: 16, Value: 155
   0.339027	[D3] Report[get]: (3 bytes) => 33 18 01

@nbriggs Thank you for providing the binary file.

nbriggs, 2022-02-14:
Thanks @smainz and @andbez for testing.   I re-read my code and realized there's one more minor change to make having to do with when it reports ("Report Descriptor Fixed") that it did (or did not) find the problem.  That probably only requires someone visually inspect the change, though. Famous last words and all that...

nbriggs, 2022-02-15:
Updated again with bumped driver version number and check as mentioned above.
% openssl sha256 usbhid-ups
SHA256(usbhid-ups)= 828a31c62d819f04c4ad305163adb9108f0acfd9d526e0285d5ff8fa77725d49
and updated at the same link as before (x86_64 version).
Submitted PR for it -- I think this one's done for now.

jimklimov, 2022-02-15:
Thanks all for reporting, fixing and testing! Will follow up on the posted
PR :)

smainz, 2022-02-15:
Do you need another test or do you trust in you famous last words?

nbriggs, 2022-02-15:
If you'd be happy to just eyeball the last change in source rather than build/test it that would be enough for me.  Can't speak to anyone else's wishes.

[2 more replies]

PR #1290 (2022-02-17): Fix incorrect limits for input and config voltages in APC report descriptor
Closes #1189 in which some APC devices operating in regions with higher (> 120V) nominal input voltages still report North American range limits for the input and configuration voltages and thus read out incorrect values.

Diagnoses the problem by checking if the logical maximum values for UPS.Input.Voltage and UPS.Input.ConfigVoltage in the report descriptor are consistent with the UPS.Input.HighVoltageTransfer logical maximum (which is observed to be correct) and if not, resets the logical maximum values to be consis [... cut]
closed 2022-02-17 by jimklimov