networkupstools/nut #1191: [HCL] Cyberpower PR1500RT2U supported by usbhid-ups

Brandoskey, 2021-11-21:
Cyberpower PR1500RT2U

upsc output
```
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 35
battery.charge.warning: 35
battery.mfr.date: CPS
battery.runtime: 6036
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 2.3
battery.voltage.nominal: 22
device.mfr: CPS
device.model: PR1500RT2U
device.serial: PWLLO2000072
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 121.0
input.voltage.nominal: 120
output.voltage: 265.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 12
ups.mfr: CPS
ups.model: PR1500RT2U
ups.productid: 0601
ups.realpower.nominal: 1500
ups.serial: PWLLO2000072
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: 0
ups.timer.start: 0
ups.vendorid: 0764
```
upsrw output

```
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 35

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 300

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: STRING
Maximum length: 10
Value: 30
```

upscmd output

```
Instant commands supported on UPS [ups]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

Tested shutdown and the master and all slaves shutdown at LB as expected (35%)

some commands don't appear to work went sent through upscmd and upsrw. battery.charge.low value doesn't seem to be user changeable.

Some of the reported values appear off as well. output.voltage I would expect to be ~120v and the battery voltages also appear incorrect. The UPS LCD reports 4 12v cells in series for 48v nominal ~54v actual

Finally a link to the UPS on the manufacturers website

https://www.cyberpowersystems.com/product/ups/new-smart-app-sinewave/pr1500rt2u/

jimklimov, 2021-12-28:
Detailed data also needs to be posted to nut-ddl, as commented in the PR for HCL.

hunter86bg, 2021-12-28:
Then we keep this one open till merged.

Brandoskey, 2021-12-28:
Is this something I needed to do?

jimklimov, 2021-12-28:
"no" - you were not required to, thanks for making the data available

...and "yes" - anyone weightlifting a PR posting is very welcome to do so, as there's only so much I can do in a day :)

closed 2022-04-25 by jimklimov

jimklimov, 2024-12-18:
PR #2718 is presumed to have fixed the broken CPS input/output voltage reports mentioned in this issue (although it did not focus on cases of "voltage too high" like here).

Testing would be welcome.