networkupstools/nut #1201: Standardze Status Flags and Instant Commands for ECO mode.

biergaizi, 2021-11-25:
Many online UPS systems support ECO mode. In ECO mode, the mains output bypasses the UPS during normal operation, and is only handed over to the UPS automatically during a power failure. This effectively converts an online UPS to an offline-like UPS with zero transfer time, the UPS no longer reconditions the power, but significant improves power efficiency.

So far NUT doesn't have a standard on ECO mode. The basic standardization needed are.

1. A `ups.status` flag for ECO mode, currently there's no standard. I looked around the codebase and found many drivers already use the flag `OL ECO`, so I think we could settle on that. Bun with a caveat - the ECO status flag only appears when the UPS is online, and can disappear when it's on battery. But this is how many UPS units report the current power status anyway. 
2. Instant commands for entering and exiting ECO mode. Currently NUT doesn't have standardized instant commands for that.

jimklimov, 2021-12-01:
Mailing-list discussion initiated with https://alioth-lists.debian.net/pipermail/nut-upsdev/2021-December/007616.html

biergaizi, 2026-08-31:
Related: https://github.com/networkupstools/nut/issues/2495.

NUT has too many issues, the same issue is often repeatedly submitted...