networkupstools/nut #1208: Numerous ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID is incorrect on device side from APC Back-UPS BX1600MI

asdavidov, 2021-11-28:
Hello,
I'm getting a lot of 
` ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID is incorrect on device side; skipping evaluation of these constraints`
from usbhid-ups driver polling the APC Back-UPS BX1600MI device.

I'm using usbhid-ups driver version:
```
Network UPS Tools - Generic HID driver 0.43 (2.7.4.1)
USB communication driver 0.33
```
The output from the 
`usbhid-ups -DDDDD -a APCBackUPS -d 2`
command is:
```
Network UPS Tools - Generic HID driver 0.43 (2.7.4.1)
USB communication driver 0.33
   0.000000     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000144     [D1] debug level is '5'
   0.003274     [D5] send_to_all: SETINFO device.type "ups"
   0.003327     [D2] Initializing an USB-connected UPS with library (null) (NUT subdriver name='USB communication driver' ver='0.33')
   0.003346     [D1] upsdrv_initups (non-SHUT)...
   0.005768     [D3] usb_busses=0x800287098
   0.005812     [D2] Checking device (051D/0002) (/dev/usb//dev/ugen0.2)
   0.029523     [D2] - VendorID: 051d
   0.029583     [D2] - ProductID: 0002
   0.029602     [D2] - Manufacturer: American Power Conversion
   0.029620     [D2] - Product: Back-UPS BX1600MI FW:294201G -302201G
   0.029638     [D2] - Serial Number: XXXXXXXXXX
   0.029656     [D2] - Bus: /dev/usb
   0.029673     [D2] - Device: /dev/ugen0.2
   0.029690     [D2] - Device release number: 0106
   0.029707     [D2] Trying to match device
   0.029725     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.029780     [D3] match_function_regex: matching a device...
   0.029802     [D2] Device matches
   0.029833     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.035544     [D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 e9 02
   0.035611     [D3] HID descriptor length (method 1) 745
   0.035631     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.035660     [D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 e9 02
   0.035681     [D3] HID descriptor length (method 2) 745
   0.035698     [D2] HID descriptor length 745
   0.043538     [D2] Report Descriptor size = 745
   0.043626     [D3] Report Descriptor: (745 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   0.043667     [D3]  79 02 75 08 95 01 15 00 26 ff 00 b1 22 85 02 09 ff 79 03 b1 22 85 03 05 85
   0.043706     [D3]  09 89 79 04 b1 22 85 04 09 8f 79 01 b1 22 85 05 09 8b b1 22 85 0a 05 84 09
   0.043755     [D3]  fd 75 08 26 ff 00 65 00 55 00 79 01 b1 22 85 0b 05 85 09 2c b1 03 75 08 85
   0.043793     [D3]  0c 09 66 25 ff b1 a2 09 66 81 a2 85 0d 09 83 75 08 25 64 65 00 b1 22 85 0e
   0.043830     [D3]  09 67 b1 22 85 0f 09 68 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 11 75 08 05
   0.043868     [D3]  85 09 29 15 01 b1 a2 85 13 09 d0 15 00 25 01 81 a2 09 d0 b1 a2 85 14 09 42
   0.043905     [D3]  81 a2 09 42 b1 a2 85 15 05 84 09 57 75 10 16 ff ff 26 ff 7f 66 01 10 b1 a2
   0.043942     [D3]  85 17 05 85 09 2a 95 01 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 18 05 84 09
   0.043978     [D3]  5a 75 08 15 01 25 03 65 00 b1 a2 09 02 a1 02 85 16 05 85 65 00 75 01 15 00
   0.044015     [D3]  25 01 09 44 81 a2 09 44 b1 a2 09 45 81 a2 09 45 b1 a2 09 d0 81 a2 09 d0 b1
   0.044051     [D3]  a2 09 d1 81 a2 09 d1 b1 a2 05 84 09 69 81 a2 09 69 b1 a2 05 85 09 43 81 a2
   0.044088     [D3]  09 43 b1 a2 05 85 09 4b 81 a2 09 4b b1 a2 05 84 09 65 81 a2 09 65 b1 a2 05
   0.044125     [D3]  85 09 db 81 a2 09 db b1 a2 95 17 81 01 b1 01 c0 c0 05 84 09 12 a1 00 95 01
   0.044161     [D3]  85 20 05 85 09 85 75 10 15 00 27 ff ff 00 00 81 22 09 85 b1 a2 85 21 05 84
   0.044206     [D3]  09 58 75 08 25 06 81 a2 09 58 b1 a2 85 22 05 85 09 66 25 ff b1 a2 85 23 09
   0.044243     [D3]  68 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 24 09 2a b1 a2 85 25 05 84 09 40
   0.044280     [D3]  67 21 d1 f0 00 55 05 b1 22 85 26 09 30 b1 a2 c0 05 84 09 1a a1 00 85 30 05
   0.044316     [D3]  84 09 40 75 08 15 00 26 ff 00 67 21 d1 f0 00 55 07 b1 22 85 31 09 30 75 10
   0.044352     [D3]  b1 a2 85 32 09 53 16 8c 00 26 96 00 b1 a2 85 33 09 54 16 22 01 26 2c 01 b1
   0.044388     [D3]  a2 85 35 75 08 06 86 ff 09 61 15 00 25 02 b1 a2 85 36 09 52 25 0d b1 a2 c0
   0.044425     [D3]  09 05 a1 00 85 40 09 7c 25 ff b1 a2 c0 05 84 09 16 a1 00 85 50 09 35 75 08
   0.044461     [D3]  15 00 25 64 65 00 55 00 b1 a2 85 52 05 84 09 44 75 10 15 00 27 ff ff 00 00
   0.044497     [D3]  66 21 d1 55 07 b1 a2 c0 85 7f 05 84 09 fe 79 05 75 08 95 01 15 00 26 ff 00
   0.044533     [D3]  65 00 55 00 b1 22 85 7e 06 86 ff 09 42 79 07 75 08 95 01 15 00 26 ff 00 b1
   0.044570     [D3]  22 85 7d 05 84 09 ff 79 03 b1 22 85 7c 09 fd 79 01 b1 22 85 7b 05 85 09 85
   0.044606     [D3]  75 10 27 ff ff 00 00 b1 a2 05 84 09 02 a1 02 85 7a 05 85 09 44 75 01 25 01
[... 1792 lines cut ...]
driver.version: 2.7.4.1
driver.version.data: APC HID 0.96
driver.version.internal: 0.43
input.sensitivity: medium
input.transfer.high: 295
input.transfer.low: 145
input.voltage: 234.0
input.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 294201G -302201G
ups.load: 9
ups.mfr: American Power Conversion
ups.mfr.date: 2021/03/11
ups.model: Back-UPS BX1600MI
ups.productid: 0002
ups.realpower.nominal: 900
ups.serial: XXXXXXXXXX
ups.status: OL
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
   0.797048     [D1] upsdrv_cleanup..
```

asdavidov, 2021-11-28:
According to the debug log that error appears when driver receives 
`LogMax = -1, LogMin = 0` from the UPS for some reason.

jimklimov, 2021-11-29:
OTOH the context for this is #1023, #1040, #1055, #1138 and @nbriggs was looking for more data from real hardware, bugs and all.

asdavidov, 2021-11-29:
Thank you for the update on this. As far as I can see there were a lot of work have been done to work around this issue.
Anyway even the version that I'm currently using is able to successfully poll this UPS and get data from it.
The main inconvenience is that usbhid-ups is constantly spamming the logs with that error message (up to several times per second), so it will be great if that will be changed in the future. 
If @nbriggs will need some additional data from this UPS I'll be glad to provide it.

jimklimov, 2021-11-30:
My understanding from earlier work linked above is that USB HID allows devices to send messages of different types and bit length of the number and (I think) imply whether the number is signed or not. So the problem is that the devices send an incorrect bit stream and an honest interpretation of those bits says that 65535 is -1. Ideally this is something to confirm, and then nag vendors to fix their firmware.

One thing I am not sure about from your report, and which might be a NUT bug compared to the goals, is whether the "error spam" happens all the time if the driver is running for a long time (and not just initializing the different items in data state map, which is about all it does in dump mode like `-d 1` looping over different data-points). The idea for these messages was to appear only during driver initialization, not for every chat with the device over many days.

nbriggs, 2021-11-30:
@jimklimov -- to ease integration, do you want me to start my changes from current main branch or somewhere else?

asdavidov, 2021-12-01:
> Ideally this is something to confirm, and then nag vendors to fix their firmware.

I agree that the best thing will be to force vendor to fix the firmware, but for such "consumer-grade" UPS's there is no official way to field-upgrade the device firmware. I've searched thru the APC site and didn't find any publicly available firmware for the Back-UPS family, but most of the Smart-UPS series do support firmware upgrade by consumer. So though the ideally such bugs should be fixed in firmware by vendors, I guess that the chances for that to happen are so low that we'll have to work-around this firmware bugs in the NUT software itself. =(

> One thing I am not sure about from your report, and which might be a NUT bug compared to the goals, is whether the "error spam" happens all the time if the driver is running for a long time

I can confirm that this "error spam" happens all the time the driver is running. And it was the main reason why I've opened this issue/bug report.
I actually thought that running usbhid-ups with `-d` option is a way to perform a distinct number of regular polls - and I didn't thought that this option means something else. Sorry for any confusion that I've made, but I thought that that dump that I've posted above is self-explanatory and shows that those error messages appears during each poll.

> The idea for these messages was to appear only during driver initialization, not for every chat with the device over many days.

I saw in one of the linked issues that you've posted that the supposed behavior was to produce these error messages only during the first UPS communication, but it is not the case with this UPS model. Those errors are constantly appears during the NUT runtime and I guess that they are appear for every chat with the UPS.
Here is the exempt from the logs just after the NUT software was started:

<html><body>

Dec  1 05:28:15 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
-- | -- | --
Dec  1 05:28:15 | snuffbox | root: nut service started
Dec  1 05:28:16 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:16 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:16 | snuffbox | root: nut_upslog service started
Dec  1 05:28:16 | snuffbox | root: nut_upsmon service started
Dec  1 05:28:17 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:19 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:19 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:22 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:23 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:23 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:26 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:28 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:28 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:30 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:32 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:32 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:34 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:36 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:36 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:38 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:40 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints
Dec  1 05:28:40 | snuffbox | usbhid-ups:  ERROR in GetValue: LogMin is greater than LogMax, possibly vendor HID  is incorrect on device side; skipping evaluation of these constraints

</body>
</html>

nbriggs, 2021-12-01:
I have a "*American Power Conversion Back-UPS RS 1500G FW:865.L6 .D USB FW:L6*" to test against.  It's usually monitored by a FreeBSD system with nut-2.7.4_13 installed, but I can move the USB link to a box I have more control over.

asdavidov, 2021-12-01:
FYI the system that express this behavior is also a FreeBSD box but with the nut-devel 2021.11.08 package installed.

jimklimov, 2021-12-01:
A lot of questions above, hope to address them in one go here :)
* Regarding "-d NUM" option, it is intended for data-dumping without daemonizing the driver, for easier testing of custom builds or updating the DDL. In this mode, the driver initializes (collects all data to make the dstate map for the particular device, including rarely- or never- refreshing points like device model name), runs NUM loops of usual data collection (voltages etc. that change frequently), and exits. Often NUM is 1 :)
* Regarding the branch to base fixes from, it really is a good question. Until recently, I'd say this should be based from master branch, and maybe that still is so to merge fixes quickly :)
** But a couple of week(end)s ago I had finally updated the long-pending libusb-1.0 and libusb-1.0+0.1 branches, merging into them the same version of master current at that time. I did not yet get to go over the differences of those resulting branches (more details in #300) to pick the codebase to put into master properly, but at least keep those in mind and maybe check with custom builds (rebasing your development over them locally) whether there are differences due to linking against libusb-0.1 vs libusb-1.0.

asdavidov, 2021-12-01:
> runs NUM loops of usual data collection (voltages etc. that change frequently), and exits. Often NUM is 1 :)

Ok, I see. So the "-d NUM" option works just as I supposed it to. Then I guess those several errors that appears during the normal poll should also be visible in that dump that I've provided earlier.
But it is not very clear to me from that dump (or log) what particular HID triggers those error messages during the usual data collection.

nbriggs, 2021-12-02:
Looking at the code now.  It's spamming the log because the changes in https://github.com/networkupstools/nut/pull/1146 are to the code that gets executed on every value read (vs. when the logical minimum and maximum are set up as the report descriptor(s) are read) and didn't get a once-only flag to suppress the warning messages after the first time.

More later...

nbriggs, 2021-12-02:
Hey @asdavidov -- are you able to try out this patch on your system and tell me whether it (a) gives you the correct values, and (b) reduces the log spam to a single message warning of the UPS's encoding issue
[hidparser-patch.txt](https://github.com/networkupstools/nut/files/7639130/hidparser-patch.txt)
?

asdavidov, 2021-12-02:
@nbriggs Should I apply this patch to the current master or to some other branch?

nbriggs, 2021-12-02:
@asdavidov it's based on the current master.  Thanks!

asdavidov, 2021-12-04:
I was able to successfully build the patched version on a FreeBSD 13.0-RELEASE-p5 system using modified version of the current nut-devel (2021.11.28) port. Unfortunately I'll be away from that UPS hardware in a next several days, so I'll be able to test that patched version only in a second half of the next week. 
I'm sorry for the delay.

nbriggs, 2021-12-04:
Excellent.  No worries about the delay.  I've got plenty to keep me busy.

asdavidov, 2021-12-14:
@nbriggs @jimklimov I've finally manage to test the patched version against the UPS hardware. The preparation steps took a bit longer than I expected, but now I've got the results.
The patch fixed the problem with the log spam and now the usbhid-ups runs quietly without any errors in the log files.
On the other hand the warnings at start-up are also not being logged. They appear just once on the system console during the driver start-up.
The dstate_dump seems to be similar between both versions.
Here is the debug output from the patched driver:
```
Network UPS Tools - Generic HID driver 0.43 (2.7.4.1)
USB communication driver 0.33
   0.000000     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000080     [D1] debug level is '5'
   0.000647     [D5] send_to_all: SETINFO device.type "ups"
   0.000669     [D2] Initializing an USB-connected UPS with library (null) (NUT subdriver name='USB communication driver' ver='0.33')
   0.000677     [D1] upsdrv_initups (non-SHUT)...
   0.001000     [D3] usb_busses=0x8002801b0
   0.001016     [D2] Checking device (051D/0002) (/dev/usb//dev/ugen0.2)
   0.007753     [D2] - VendorID: 051d
   0.007773     [D2] - ProductID: 0002
   0.007781     [D2] - Manufacturer: American Power Conversion
   0.007790     [D2] - Product: Back-UPS BX1600MI FW:294201G -302201G
   0.007801     [D2] - Serial Number: XXXXXXXXXXX
   0.007813     [D2] - Bus: /dev/usb
   0.007824     [D2] - Device: /dev/ugen0.2
   0.007839     [D2] - Device release number: 0106
   0.007851     [D2] Trying to match device
   0.007867     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.007900     [D3] match_function_regex: matching a device...
   0.007918     [D2] Device matches
   0.007941     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.008559     [D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 e9 02
   0.008571     [D3] HID descriptor length (method 1) 745
   0.008581     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.008611     [D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 e9 02
   0.008625     [D3] HID descriptor length (method 2) 745
   0.008637     [D2] HID descriptor length 745
   0.011760     [D2] Report Descriptor size = 745
   0.011783     [D3] Report Descriptor: (745 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   0.011822     [D3]  79 02 75 08 95 01 15 00 26 ff 00 b1 22 85 02 09 ff 79 03 b1 22 85 03 05 85
   0.011861     [D3]  09 89 79 04 b1 22 85 04 09 8f 79 01 b1 22 85 05 09 8b b1 22 85 0a 05 84 09
   0.011899     [D3]  fd 75 08 26 ff 00 65 00 55 00 79 01 b1 22 85 0b 05 85 09 2c b1 03 75 08 85
   0.011936     [D3]  0c 09 66 25 ff b1 a2 09 66 81 a2 85 0d 09 83 75 08 25 64 65 00 b1 22 85 0e
   0.011974     [D3]  09 67 b1 22 85 0f 09 68 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 11 75 08 05
   0.012012     [D3]  85 09 29 15 01 b1 a2 85 13 09 d0 15 00 25 01 81 a2 09 d0 b1 a2 85 14 09 42
   0.012050     [D3]  81 a2 09 42 b1 a2 85 15 05 84 09 57 75 10 16 ff ff 26 ff 7f 66 01 10 b1 a2
   0.012088     [D3]  85 17 05 85 09 2a 95 01 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 18 05 84 09
   0.012126     [D3]  5a 75 08 15 01 25 03 65 00 b1 a2 09 02 a1 02 85 16 05 85 65 00 75 01 15 00
   0.012163     [D3]  25 01 09 44 81 a2 09 44 b1 a2 09 45 81 a2 09 45 b1 a2 09 d0 81 a2 09 d0 b1
   0.012201     [D3]  a2 09 d1 81 a2 09 d1 b1 a2 05 84 09 69 81 a2 09 69 b1 a2 05 85 09 43 81 a2
   0.012238     [D3]  09 43 b1 a2 05 85 09 4b 81 a2 09 4b b1 a2 05 84 09 65 81 a2 09 65 b1 a2 05
   0.012276     [D3]  85 09 db 81 a2 09 db b1 a2 95 17 81 01 b1 01 c0 c0 05 84 09 12 a1 00 95 01
   0.012313     [D3]  85 20 05 85 09 85 75 10 15 00 27 ff ff 00 00 81 22 09 85 b1 a2 85 21 05 84
   0.012351     [D3]  09 58 75 08 25 06 81 a2 09 58 b1 a2 85 22 05 85 09 66 25 ff b1 a2 85 23 09
   0.012389     [D3]  68 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 24 09 2a b1 a2 85 25 05 84 09 40
   0.012426     [D3]  67 21 d1 f0 00 55 05 b1 22 85 26 09 30 b1 a2 c0 05 84 09 1a a1 00 85 30 05
   0.012464     [D3]  84 09 40 75 08 15 00 26 ff 00 67 21 d1 f0 00 55 07 b1 22 85 31 09 30 75 10
   0.012501     [D3]  b1 a2 85 32 09 53 16 8c 00 26 96 00 b1 a2 85 33 09 54 16 22 01 26 2c 01 b1
   0.012539     [D3]  a2 85 35 75 08 06 86 ff 09 61 15 00 25 02 b1 a2 85 36 09 52 25 0d b1 a2 c0
   0.012576     [D3]  09 05 a1 00 85 40 09 7c 25 ff b1 a2 c0 05 84 09 16 a1 00 85 50 09 35 75 08
   0.012620     [D3]  15 00 25 64 65 00 55 00 b1 a2 85 52 05 84 09 44 75 10 15 00 27 ff ff 00 00
   0.012657     [D3]  66 21 d1 55 07 b1 a2 c0 85 7f 05 84 09 fe 79 05 75 08 95 01 15 00 26 ff 00
   0.012695     [D3]  65 00 55 00 b1 22 85 7e 06 86 ff 09 42 79 07 75 08 95 01 15 00 26 ff 00 b1
   0.012733     [D3]  22 85 7d 05 84 09 ff 79 03 b1 22 85 7c 09 fd 79 01 b1 22 85 7b 05 85 09 85
   0.012771     [D3]  75 10 27 ff ff 00 00 b1 a2 05 84 09 02 a1 02 85 7a 05 85 09 44 75 01 25 01
   0.012809     [D3]  b1 a2 09 45 b1 a2 09 d0 b1 a2 09 d1 b1 a2 05 85 09 4b b1 a2 05 85 09 db b1
   0.012847     [D3]  a2 05 84 09 65 b1 a2 95 19 b1 01 c0 85 78 05 84 09 5a 75 08 95 01 15 01 25
   0.012885     [D3]  03 b1 a2 06 ff ff 09 ff a1 02 85 e3 09 ef 75 08 95 07 26 ff 00 15 00 81 82
   0.012905     [D3]  09 ef b1 82 c0 c0
   0.012982     HIDParse: LogMax is less than LogMin, possibly vendor HID report descriptor is incorrect; interpreting LogMax -1 as 255
   0.013037     HIDParse: LogMax is less than LogMin, possibly vendor HID report descriptor is incorrect; interpreting LogMax -1 as 255
[... cut ...]
input.transfer.high: 295
input.transfer.low: 145
input.voltage: 237.0
input.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 294201G -302201G
ups.load: 10
ups.mfr: American Power Conversion
ups.mfr.date: 2021/03/11
ups.model: Back-UPS BX1600MI
ups.productid: 0002
ups.realpower.nominal: 900
ups.serial: XXXXXXXXXXX
ups.status: OL
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
   0.557547     [D1] upsdrv_cleanup...
```

nbriggs, 2021-12-14:
@asdavidov Thanks!  This looks & behaves as I expected, the problematic entries it found in the Report Descriptor are:
```
   0.012982     HIDParse: LogMax is less than LogMin, possibly vendor HID report descriptor is incorrect; interpreting LogMax -1 as 255
   0.013037     HIDParse: LogMax is less than LogMin, possibly vendor HID report descriptor is incorrect; interpreting LogMax -1 as 255
   0.013063     HIDParse: LogMax is less than LogMin, possibly vendor HID report descriptor is incorrect; interpreting LogMax -1 as 255
```
They're logged at WARNING level, so if you've got warnings logged to the console that would explain that behavior.  I think the log level for this condition could be either NOTICE or WARNING, and while I have a slight preference for WARNING I'm happy to go with an argument either way.

I think I should include the ReportID of the descriptor it is parsing when it encounters this problem, otherwise it's harder than it needs to be to find the offending item.

asdavidov, 2021-12-14:
@nbriggs I'm glad to hear that the code now does the things it is expected to do. Thank you for the great work on fixing this issue.

I do agree that the WARNING log level for such messages is more appropriate than the NOTICE one.

It will also be great if you could include the ReportID of the problematic descriptor, since currently it is almost impossible for the novice user to figure out what is the actual name of the "HID report descriptor" it is complaining about.

Since I've got my test rig up and running, I could test some new patches with this UPS hardware if it's needed.

nbriggs, 2021-12-14:
OK, great -- here's a version with the ReportID in the log message.  It's the patch against the master as it was when I sent the last one, not a difference from the patched version, so just reset the patched hidparser.c and then apply this patch instead:
[hidparser-patch-v2.txt](https://github.com/networkupstools/nut/files/7714906/hidparser-patch-v2.txt)

nbriggs, 2021-12-14:
@asdavidov -- by the way, I recommend https://eleccelerator.com/usbdescreqparser/ if you ever have to debug USB report descriptors.  It has a couple of pretty minor issues, but it's really handy to be able paste in a whole report descriptor (with trivial edits, from the NUT driver log, or anywhere else) and get a sense for what it's trying to describe.  USB.org should be providing something like this (with extra diagnostics!) for validating report descriptors.

asdavidov, 2021-12-14:
> @asdavidov -- by the way, I recommend https://eleccelerator.com/usbdescreqparser/ if you ever have to debug USB report descriptors. It has a couple of pretty minor issues, but it's really handy to [...]

@nbriggs Thanks for the link to a such handy tool! 
Just tested it against the HID Report descriptor from one of my previous dumps and it was able to successfully decode it. =)

nbriggs, 2021-12-14:
@asdavidov you're welcome.  The two minor issues were that it interprets ReportIDs as signed 8-bit instead of unsigned 8-bit, and in certain circumstances it will report 0xFFFF as 65534 instead of 65535.   See the comments linked from the decoder page.

asdavidov, 2021-12-14:
> OK, great -- here's a version with the ReportID in the log message. It's the patch against the master as it was when I sent the last one, not a difference from the patched version, so just reset the [...]

@nbriggs Applied the v2 patch over the same source tree as the first one. 
ReportID information is now available during driver startup in the warning messages, thank you.
Here is a fresh dump from the usbhid-ups driver with v2 patch applied:
```
Network UPS Tools - Generic HID driver 0.43 (2.7.4.1)
USB communication driver 0.33
   0.000000     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000074     [D1] debug level is '5'
   0.000627     [D5] send_to_all: SETINFO device.type "ups"
   0.000649     [D2] Initializing an USB-connected UPS with library (null) (NUT subdriver name='USB communication driver' ver='0.33')
   0.000657     [D1] upsdrv_initups (non-SHUT)...
   0.000957     [D3] usb_busses=0x8002801b0
   0.000971     [D2] Checking device (051D/0002) (/dev/usb//dev/ugen0.2)
   0.007701     [D2] - VendorID: 051d
   0.007714     [D2] - ProductID: 0002
   0.007721     [D2] - Manufacturer: American Power Conversion
   0.007729     [D2] - Product: Back-UPS BX1600MI FW:294201G -302201G
   0.007735     [D2] - Serial Number: XXXXXXXXXXX
   0.007744     [D2] - Bus: /dev/usb
   0.007750     [D2] - Device: /dev/ugen0.2
   0.007757     [D2] - Device release number: 0106
   0.007764     [D2] Trying to match device
   0.007773     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.007791     [D3] match_function_regex: matching a device...
   0.007800     [D2] Device matches
   0.007812     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.008218     [D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 e9 02
   0.008229     [D3] HID descriptor length (method 1) 745
   0.008237     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.008251     [D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 e9 02
   0.008258     [D3] HID descriptor length (method 2) 745
   0.008264     [D2] HID descriptor length 745
   0.011475     [D2] Report Descriptor size = 745
   0.011496     [D3] Report Descriptor: (745 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   0.011517     [D3]  79 02 75 08 95 01 15 00 26 ff 00 b1 22 85 02 09 ff 79 03 b1 22 85 03 05 85
   0.011536     [D3]  09 89 79 04 b1 22 85 04 09 8f 79 01 b1 22 85 05 09 8b b1 22 85 0a 05 84 09
   0.011555     [D3]  fd 75 08 26 ff 00 65 00 55 00 79 01 b1 22 85 0b 05 85 09 2c b1 03 75 08 85
   0.011574     [D3]  0c 09 66 25 ff b1 a2 09 66 81 a2 85 0d 09 83 75 08 25 64 65 00 b1 22 85 0e
   0.011592     [D3]  09 67 b1 22 85 0f 09 68 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 11 75 08 05
   0.011611     [D3]  85 09 29 15 01 b1 a2 85 13 09 d0 15 00 25 01 81 a2 09 d0 b1 a2 85 14 09 42
   0.011630     [D3]  81 a2 09 42 b1 a2 85 15 05 84 09 57 75 10 16 ff ff 26 ff 7f 66 01 10 b1 a2
   0.011648     [D3]  85 17 05 85 09 2a 95 01 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 18 05 84 09
   0.011666     [D3]  5a 75 08 15 01 25 03 65 00 b1 a2 09 02 a1 02 85 16 05 85 65 00 75 01 15 00
   0.011685     [D3]  25 01 09 44 81 a2 09 44 b1 a2 09 45 81 a2 09 45 b1 a2 09 d0 81 a2 09 d0 b1
   0.011703     [D3]  a2 09 d1 81 a2 09 d1 b1 a2 05 84 09 69 81 a2 09 69 b1 a2 05 85 09 43 81 a2
   0.011721     [D3]  09 43 b1 a2 05 85 09 4b 81 a2 09 4b b1 a2 05 84 09 65 81 a2 09 65 b1 a2 05
   0.011740     [D3]  85 09 db 81 a2 09 db b1 a2 95 17 81 01 b1 01 c0 c0 05 84 09 12 a1 00 95 01
   0.011758     [D3]  85 20 05 85 09 85 75 10 15 00 27 ff ff 00 00 81 22 09 85 b1 a2 85 21 05 84
   0.011776     [D3]  09 58 75 08 25 06 81 a2 09 58 b1 a2 85 22 05 85 09 66 25 ff b1 a2 85 23 09
   0.011794     [D3]  68 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 24 09 2a b1 a2 85 25 05 84 09 40
   0.011813     [D3]  67 21 d1 f0 00 55 05 b1 22 85 26 09 30 b1 a2 c0 05 84 09 1a a1 00 85 30 05
   0.011831     [D3]  84 09 40 75 08 15 00 26 ff 00 67 21 d1 f0 00 55 07 b1 22 85 31 09 30 75 10
   0.011850     [D3]  b1 a2 85 32 09 53 16 8c 00 26 96 00 b1 a2 85 33 09 54 16 22 01 26 2c 01 b1
   0.011869     [D3]  a2 85 35 75 08 06 86 ff 09 61 15 00 25 02 b1 a2 85 36 09 52 25 0d b1 a2 c0
   0.011888     [D3]  09 05 a1 00 85 40 09 7c 25 ff b1 a2 c0 05 84 09 16 a1 00 85 50 09 35 75 08
   0.011907     [D3]  15 00 25 64 65 00 55 00 b1 a2 85 52 05 84 09 44 75 10 15 00 27 ff ff 00 00
   0.011925     [D3]  66 21 d1 55 07 b1 a2 c0 85 7f 05 84 09 fe 79 05 75 08 95 01 15 00 26 ff 00
   0.011947     [D3]  65 00 55 00 b1 22 85 7e 06 86 ff 09 42 79 07 75 08 95 01 15 00 26 ff 00 b1
   0.011966     [D3]  22 85 7d 05 84 09 ff 79 03 b1 22 85 7c 09 fd 79 01 b1 22 85 7b 05 85 09 85
   0.011984     [D3]  75 10 27 ff ff 00 00 b1 a2 05 84 09 02 a1 02 85 7a 05 85 09 44 75 01 25 01
   0.012003     [D3]  b1 a2 09 45 b1 a2 09 d0 b1 a2 09 d1 b1 a2 05 85 09 4b b1 a2 05 85 09 db b1
   0.012021     [D3]  a2 05 84 09 65 b1 a2 95 19 b1 01 c0 85 78 05 84 09 5a 75 08 95 01 15 01 25
   0.012040     [D3]  03 b1 a2 06 ff ff 09 ff a1 02 85 e3 09 ef 75 08 95 07 26 ff 00 15 00 81 82
   0.012050     [D3]  09 ef b1 82 c0 c0
   0.012092     HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID 0xc
   0.012129     HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID 0x22
[... cut ...]
input.transfer.high: 295
input.transfer.low: 145
input.voltage: 237.0
input.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 294201G -302201G
ups.load: 11
ups.mfr: American Power Conversion
ups.mfr.date: 2021/03/11
ups.model: Back-UPS BX1600MI
ups.productid: 0002
ups.realpower.nominal: 900
ups.serial: XXXXXXXXXXX
ups.status: OL
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
   0.488784     [D1] upsdrv_cleanup...
```

asdavidov, 2021-12-14:
@nbriggs Just two almost cosmetic notices:
1. If ReportID would use the same format as it being used in the later driver output (`ReportID:` instead of `ReportID`) it'll be easier to search thru the logs for the affected HID report id.
2. It looks like the leading zero is being stripped from the output in the warning message (`in ReportID 0xc` instead of actual '0x0c' )

nbriggs, 2021-12-15:
Yeah, ;-)  -- I realized just after I sent the patch that it would be better with the 2-digit hex id, and I hadn't thought about making it `ReportID:` but that makes a whole lot of sense.   Third time lucky?
[hidparser-patch-v3.txt](https://github.com/networkupstools/nut/files/7715468/hidparser-patch-v3.txt)

nbriggs, 2021-12-15:
I found another report descriptor decoder ( https://sourceforge.net/projects/hidrdd, but had to find/build a REXX interpreter https://sourceforge.net/projects/regina-rexx/ to run it, though) which decoded your APC report descriptor above per the attached file.  We can see a bunch of errors and warnings and suggestions in there!  This seems to be pretty much the USB HID report descriptor *validator* that I wanted.

[reportdescriptor-decoded.txt](https://github.com/networkupstools/nut/files/7715263/reportdescriptor-decoded.txt)

asdavidov, 2021-12-15:
> I found another report descriptor decoder ( https://sourceforge.net/projects/hidrdd, but had to find/build a REXX interpreter https://sourceforge.net/projects/regina-rexx/ to run it, though) which [...]

Hmm... Interesting....
This tool doesn't emit any warnings on the affected ReportID's that we've found:
```
85 22            (GLOBAL) REPORT_ID          0x22 (34)  
05 85            (GLOBAL) USAGE_PAGE         0x0085 Battery System Page 
09 66            (LOCAL)  USAGE              0x00850066 Remaining Capacity (Dynamic Value)  
25 FF            (GLOBAL) LOGICAL_MAXIMUM    0xFF (-1)  
B1 A2            (MAIN)   FEATURE            0x000000A2 (1 field x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 1=NoPrefState 0=NoNull 1=Volatile 0=Bitmap  <-- Error: LOGICAL_MAXIMUM (-1) is less than LOGICAL_MINIMUM (0)
85 23            (GLOBAL) REPORT_ID          0x23 (35)  
```
So it seems to think that  `25 FF            (GLOBAL) LOGICAL_MAXIMUM    0xFF (-1)` is perfectly fine in this case....

nbriggs, 2021-12-15:
It shows up when it hits the "FEATURE" and "INPUT" section for that report id --

```
85 0C            (GLOBAL) REPORT_ID          0x0C (12)
09 66            (LOCAL)  USAGE              0x00850066 Remaining Capacity (Dynamic Value)
25 FF            (GLOBAL) LOGICAL_MAXIMUM    0xFF (-1)
B1 A2            (MAIN)   FEATURE            0x000000A2 (1 field x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 1=NoPrefState 0=NoNull 1=Volatile 0=Bitmap  <-- Error: LOGICAL_MAXIMUM (-1) i\
s less than LOGICAL_MINIMUM (0)
09 66            (LOCAL)  USAGE              0x00850066 Remaining Capacity (Dynamic Value)
81 A2            (MAIN)   INPUT              0x000000A2 (1 field x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 1=NoPrefState 0=NoNull 1=Volatile 0=Bitmap  <-- Error: LOGICAL_MAXIMUM (-1) i\
s less than LOGICAL_MINIMUM (0)
```

asdavidov, 2021-12-15:
:man_facepalming: I didn't scroll all the line to the right before posting that. since I thought that it would mark it as Error at `25 FF` and not at `B1 A2` =) Sorry, my fault.
It really looks that it is a good candidate for a validator then. :+1:

asdavidov, 2021-12-15:
> Third time lucky? [hidparser-patch-v3.txt](https://github.com/networkupstools/nut/files/7715468/hidparser-patch-v3.txt)

It seems like it. =)
The build went fine, and the output from the driver patched with v3 patch is:
```
Network UPS Tools - Generic HID driver 0.43 (2.7.4.1)
USB communication driver 0.33
   0.000000     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000075     [D1] debug level is '5'
   0.000636     [D5] send_to_all: SETINFO device.type "ups"
   0.000657     [D2] Initializing an USB-connected UPS with library (null) (NUT subdriver name='USB communication driver' ver='0.33')
   0.000665     [D1] upsdrv_initups (non-SHUT)...
   0.000966     [D3] usb_busses=0x8002801b0
   0.000982     [D2] Checking device (051D/0002) (/dev/usb//dev/ugen0.2)
   0.007715     [D2] - VendorID: 051d
   0.007729     [D2] - ProductID: 0002
   0.007736     [D2] - Manufacturer: American Power Conversion
   0.007745     [D2] - Product: Back-UPS BX1600MI FW:294201G -302201G
   0.007752     [D2] - Serial Number: XXXXXXXXXXX
   0.007759     [D2] - Bus: /dev/usb
   0.007766     [D2] - Device: /dev/ugen0.2
   0.007773     [D2] - Device release number: 0106
   0.007779     [D2] Trying to match device
   0.007787     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.007805     [D3] match_function_regex: matching a device...
   0.007815     [D2] Device matches
   0.007826     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.008231     [D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 e9 02
   0.008242     [D3] HID descriptor length (method 1) 745
   0.008250     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.008263     [D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 e9 02
   0.008270     [D3] HID descriptor length (method 2) 745
   0.008277     [D2] HID descriptor length 745
   0.011383     [D2] Report Descriptor size = 745
   0.011403     [D3] Report Descriptor: (745 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   0.011428     [D3]  79 02 75 08 95 01 15 00 26 ff 00 b1 22 85 02 09 ff 79 03 b1 22 85 03 05 85
   0.011448     [D3]  09 89 79 04 b1 22 85 04 09 8f 79 01 b1 22 85 05 09 8b b1 22 85 0a 05 84 09
   0.011467     [D3]  fd 75 08 26 ff 00 65 00 55 00 79 01 b1 22 85 0b 05 85 09 2c b1 03 75 08 85
   0.011486     [D3]  0c 09 66 25 ff b1 a2 09 66 81 a2 85 0d 09 83 75 08 25 64 65 00 b1 22 85 0e
   0.011505     [D3]  09 67 b1 22 85 0f 09 68 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 11 75 08 05
   0.011524     [D3]  85 09 29 15 01 b1 a2 85 13 09 d0 15 00 25 01 81 a2 09 d0 b1 a2 85 14 09 42
   0.011542     [D3]  81 a2 09 42 b1 a2 85 15 05 84 09 57 75 10 16 ff ff 26 ff 7f 66 01 10 b1 a2
   0.011560     [D3]  85 17 05 85 09 2a 95 01 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 18 05 84 09
   0.011579     [D3]  5a 75 08 15 01 25 03 65 00 b1 a2 09 02 a1 02 85 16 05 85 65 00 75 01 15 00
   0.011597     [D3]  25 01 09 44 81 a2 09 44 b1 a2 09 45 81 a2 09 45 b1 a2 09 d0 81 a2 09 d0 b1
   0.011616     [D3]  a2 09 d1 81 a2 09 d1 b1 a2 05 84 09 69 81 a2 09 69 b1 a2 05 85 09 43 81 a2
   0.011634     [D3]  09 43 b1 a2 05 85 09 4b 81 a2 09 4b b1 a2 05 84 09 65 81 a2 09 65 b1 a2 05
   0.011653     [D3]  85 09 db 81 a2 09 db b1 a2 95 17 81 01 b1 01 c0 c0 05 84 09 12 a1 00 95 01
   0.011671     [D3]  85 20 05 85 09 85 75 10 15 00 27 ff ff 00 00 81 22 09 85 b1 a2 85 21 05 84
   0.011690     [D3]  09 58 75 08 25 06 81 a2 09 58 b1 a2 85 22 05 85 09 66 25 ff b1 a2 85 23 09
   0.011709     [D3]  68 75 10 27 ff ff 00 00 66 01 10 b1 a2 85 24 09 2a b1 a2 85 25 05 84 09 40
   0.011727     [D3]  67 21 d1 f0 00 55 05 b1 22 85 26 09 30 b1 a2 c0 05 84 09 1a a1 00 85 30 05
   0.011746     [D3]  84 09 40 75 08 15 00 26 ff 00 67 21 d1 f0 00 55 07 b1 22 85 31 09 30 75 10
   0.011764     [D3]  b1 a2 85 32 09 53 16 8c 00 26 96 00 b1 a2 85 33 09 54 16 22 01 26 2c 01 b1
   0.011782     [D3]  a2 85 35 75 08 06 86 ff 09 61 15 00 25 02 b1 a2 85 36 09 52 25 0d b1 a2 c0
   0.011801     [D3]  09 05 a1 00 85 40 09 7c 25 ff b1 a2 c0 05 84 09 16 a1 00 85 50 09 35 75 08
   0.011820     [D3]  15 00 25 64 65 00 55 00 b1 a2 85 52 05 84 09 44 75 10 15 00 27 ff ff 00 00
   0.011838     [D3]  66 21 d1 55 07 b1 a2 c0 85 7f 05 84 09 fe 79 05 75 08 95 01 15 00 26 ff 00
   0.011857     [D3]  65 00 55 00 b1 22 85 7e 06 86 ff 09 42 79 07 75 08 95 01 15 00 26 ff 00 b1
   0.011875     [D3]  22 85 7d 05 84 09 ff 79 03 b1 22 85 7c 09 fd 79 01 b1 22 85 7b 05 85 09 85
   0.011893     [D3]  75 10 27 ff ff 00 00 b1 a2 05 84 09 02 a1 02 85 7a 05 85 09 44 75 01 25 01
   0.011911     [D3]  b1 a2 09 45 b1 a2 09 d0 b1 a2 09 d1 b1 a2 05 85 09 4b b1 a2 05 85 09 db b1
   0.011929     [D3]  a2 05 84 09 65 b1 a2 95 19 b1 01 c0 85 78 05 84 09 5a 75 08 95 01 15 01 25
   0.011947     [D3]  03 b1 a2 06 ff ff 09 ff a1 02 85 e3 09 ef 75 08 95 07 26 ff 00 15 00 81 82
   0.011956     [D3]  09 ef b1 82 c0 c0
   0.011999     HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x0c
   0.012035     HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x22
   0.012051     HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x40
   0.012132     Using subdriver: APC HID 0.96
   0.012146     [D1] 71 HID objects found
[... 1625 lines cut ...]
driver.version: 2.7.4.1
driver.version.data: APC HID 0.96
[... cut ...]
input.transfer.high: 295
input.transfer.low: 145
input.voltage: 234.0
input.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 294201G -302201G
ups.load: 10
ups.mfr: American Power Conversion
ups.mfr.date: 2021/03/11
ups.model: Back-UPS BX1600MI
ups.productid: 0002
ups.realpower.nominal: 900
ups.serial: XXXXXXXXXXX
ups.status: OL
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
   0.307797     [D1] upsdrv_cleanup...
```

asdavidov, 2021-12-15:
Great work @nbriggs ! :+1: 
Now anyone can take the ReportID from the warning message (i.e. `ReportID: 0x0c` in our case) and see at what Vendor HID report descriptor the UPS firmware engineers have ~~screwed up the things~~ made a mistake ;-) by searching that ReportID thru the driver debug log - and in our case it is the:
```
   0.015843     [D4] Entering libusb_get_report
   0.016292     [D3] Report[get]: (2 bytes) => 0c 64
   0.016302     [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
   0.016309     [D5] Unit = 00000000, UnitExp = 0
   0.016315     [D5] Exponent = 0
   0.016322     [D5] hid_lookup_path: 00840004 -> UPS
   0.016329     [D5] hid_lookup_path: 00840024 -> PowerSummary
   0.016337     [D5] hid_lookup_path: 00850066 -> RemainingCapacity
   0.016347     [D1] Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 8, Value: 100
```
and several other places.

nbriggs, 2021-12-15:
@jimklimov Hi Jim -- I took a look at the libusb-1.0 and libusb-1.0+0.1 branches compared to the current master, and they both contain a much older version of `drivers/hidparser.c`.   I think I should start from master, and then conflicts between master and libusb-xxx branches should be resolved in favor of master for any hidparser.c related changes.

As you'll see in previous comments, I've been looking for a reasonable validator for the report descriptors, and found one that seems to do the job.   I'm also going to try to file a bug report with APC for the report descriptors that @asdavidov provided... we'll see if they just laugh.

I've still got this small concern that either the HID parser code is not fully complying with the USB-IF report descriptor model, or I'm just not completely understanding what it's doing.  However, I think the change I'll submit a PR for will be sufficient for now, though.

closed 2021-12-20 by jimklimov

asdavidov, 2021-12-20:
@nbriggs @jimklimov Thank you for fixing this issue!

nbriggs, 2021-12-21:
@asdavidov -- are you willing to share the serial number of this UPS unit with me so that I can continue the conversation with the APC/Schneider Electric tech support people?

asdavidov, 2021-12-22:
> @asdavidov -- are you willing to share the serial number of this UPS unit with me so that I can continue the conversation with the APC/Schneider Electric tech support people?

Yes, sure. 
How would you prefer to get it?

nbriggs, 2021-12-22:
@asdavidov how about e-mail to this throw-away address: south_03fielder at icloud.com [so that I don't get spammed from people scraping these comments :-(  ].   I'll also send you back the case number (from my real e-mail address) if they agree to continue.  It's a 1st level support person at this point who doesn't know the first thing about USB HID report descriptors and is more into whether you have the right version of the Powerchute Personal Edition software on the computer.   Not helpful!

asdavidov, 2021-12-23:
> @asdavidov how about e-mail to this throw-away address: south_03fielder at icloud.com [so that I don't get spammed from people scraping these comments :-( ]. 

@nbriggs I've send you an e-mail with the UPS serial number.
You can use the address that I've send that mail from to contact me directly.

>  It's a 1st level support person at this point who doesn't know the first thing about USB HID report descriptors and is more into whether you have the right version of the Powerchute Personal Edition software on the computer.

=) Lol. But usually it takes some time and patience to pass thru the 1st level of support. 
But this UPS that we are talking about is already registered at APC site and it is under an active manufacturer warranty, so they'll have to give you some answers.

javigmsuperka, 2023-10-09:
Hello, guys. Any news on this? We have the device duplicated with every Home Assistant update, and it seems to be related to this issue.

jimklimov, 2023-10-09:
@javigmsuperka : Posted an idea to that HA thread...

PR #1226 (2021-12-20): Move check for LogMax encoding error to parsing of report description
At least one UPS [APC Back-UPS BX1600MI FW:294201G -302201G] is known
to have encoding errors in its USB HID report description such that a
conforming interpretation of the encoded LogicalMaximum value
indicates that it is -1 and the resulting LogicalMin..LogicalMax range
is 0..-1.

The actual report values read are correctly encoded.  Rather than log
a range error each time a value is read we can detect the encoding
error when the report description is read (once, at startup) and
adjust for the encoding error at that time.
 [... cut]