networkupstools/nut #1212: /var/run/nut not created when missing with systemd

andydvsn, 2021-12-02:
Been fighting with this today and found that the runtime directory is never created, which is a particular problem when /var/run is held on a tmpfs. It can be easily solved by the inclusion of:

```
RuntimeDirectory=nut
RuntimeDirectoryMode=0775
```
In:
```
/lib/systemd/system/nut-driver-enumerator.service
/lib/systemd/system/nut-driver@.service
/lib/systemd/system/nut-monitor.service
/lib/systemd/system/nut-server.service
```
I also found it necessary to specify:

```
User=root
Group=nut
```
For more reliable startup it was also necessary to append `network.target` to the "After" section of `nut-driver@.service`:
```
After=local-fs.target network.target
```

jimklimov, 2021-12-03:
Just in case: was this seen with a recent master-branch build of NUT? There
should be support for systemd-tmpfiles mechanism to create and own dirs
before starting daemons.

Your approach sounds nice too, just gotta check the keywords are supported
with systemd versions in relevant distro versions.

jimklimov, 2022-02-10:
Seems there is indeed a problem with tmpfiles when building on systems with libraries not in `/usr/lib` or similar simple structure (more specifically, when the `*.pc` files are not in paths `pkg-config` looks for), e.g.:

````
[  287s] checking whether to install systemd unit files... using /lib/systemd/system
[  287s] checking whether to install systemd shutdown files... using /lib/systemd/system-shutdown
[  287s] checking whether to install systemd tmpfiles files... Package systemd was not found in the pkg-config search path.
[  287s] Perhaps you should add the directory containing `systemd.pc'
[  287s] to the PKG_CONFIG_PATH environment variable
[  287s] No package 'systemd' found
[  287s] no
````

PR #1283 (2022-02-11): configure.ac: refine delivery of systemd related files
Hopefully...
Closes: #1212
Closes: #1030

also:
* docs/man/Makefile.am: build and install nutdrv_siemens_sitop man pages
closed 2022-02-11 by jimklimov