networkupstools/nut #1222: Different event notification pattern for different monitored devices in upsmon.conf

SomebodySysop, 2021-12-09:
NUT running under Ubuntu OS on Raspberry Pi

Right now, my upsmon.conf is set up in the normal way.  I am monitoring 6 ups devices:

MONITOR cyberpower-d@localhost 1 admin secret master
MONITOR cyberpower-apc@localhost 1 admin secret master
MONITOR cyberpower-g@localhost 1 admin secret master
MONITOR cyberpower-h@localhost 1 admin secret master
MONITOR cyberpower-c@192.168.1.197 1 admin secret master
MONITOR cyberpower-f@192.168.1.197 1 admin secret master

Which all follow this event notification pattern

# Email script for NOTIFYCMD
NOTIFYCMD "/etc/nut/notifycmd.sh"
# Notification events
NOTIFYFLAG ONLINE     SYSLOG+WALL+EXEC
NOTIFYFLAG ONBATT     SYSLOG+WALL+EXEC
NOTIFYFLAG LOWBATT    SYSLOG+WALL+EXEC
NOTIFYFLAG FSD        SYSLOG+WALL+EXEC
# NOTIFYFLAG COMMOK     SYSLOG+WALL+EXEC
# NOTIFYFLAG COMMBAD    SYSLOG+WALL+EXEC
NOTIFYFLAG SHUTDOWN   SYSLOG+WALL+EXEC
NOTIFYFLAG REPLBATT   SYSLOG+WALL+EXEC
# NOTIFYFLAG NOCOMM     SYSLOG+WALL+EXEC
NOTIFYFLAG NOPARENT   SYSLOG+WALL+EXEC

However, suppose I only wanted 2 of the devices to follow the above pattern, and the other 4 to to follow this pattern:

# Email script for NOTIFYCMD
NOTIFYCMD "/etc/nut/notifycmd.sh"
# Notification events
NOTIFYFLAG ONLINE     SYSLOG+WALL+EXEC
NOTIFYFLAG ONBATT     SYSLOG+WALL+EXEC
NOTIFYFLAG LOWBATT    SYSLOG+WALL+EXEC
NOTIFYFLAG FSD        SYSLOG+WALL+EXEC
NOTIFYFLAG COMMOK     SYSLOG+WALL+EXEC
NOTIFYFLAG COMMBAD    SYSLOG+WALL+EXEC
NOTIFYFLAG SHUTDOWN   SYSLOG+WALL+EXEC
NOTIFYFLAG REPLBATT   SYSLOG+WALL+EXEC
NOTIFYFLAG NOCOMM     SYSLOG+WALL+EXEC
NOTIFYFLAG NOPARENT   SYSLOG+WALL+EXEC

Is this possible to do?

Why would I want to do that, you might ask?  Well, two of my devices fail with "stale data" errors quite often.  I suspect it is the USB ports of the computer they are connected to.  To fix this, I created a cronjob that restarts the nut driver every 15 minutes.  But, this causes the COMMOK and NOCOMM notifications to execute every 15 minutes.  That is why those events are commented out.  However, the other four devices run perfectly, and it would be nice to get the messages for them because they would more than likely mean real problems, and not just USB issues.

Of course,  I could resolve this by configuring upsmon.conf on the remote server (for the two devices that are giving me problems), but I still need the two in this upsmon.conf in order to be able to see ALL my devices in a single web gui (my ultimate goal for undertaking NUT in the first place):

[image]

Any suggestions?

jimklimov, 2021-12-10:
For one, I suppose the MONITOR lines should not all have the powervalue=1 column (that is how many power sources of this machine that UPS powers, so if that UPS outage is critical compared to MINSUPPLIES vs sum of powervalues - it would worry about shutting down ASAP... and I doubt your Raspberry is fed by all 6 at once)

For the main question, I'll bump the suggestions in `upsmon.conf.sample` but the docs/scheduling.txt details the solutions for notification (scripting or calling and configuring `upssched` for harder cases). Essentially, NUT calls the deployment-tailored helper (provided by you) and it is up to the helper to have the site-specific logic. Using the `$UPSNAME` in your script can help it differentiate the devices which had an event and whether messages should be sent.

Notably, the upssched is specifically written to delay posting of notifications so issues that resolve quickly are not spamming you.

Looking at your table here, it seems that your Raspberry is monitoring USB UPSes in many locations. How long is the cabling? I think USB is officially limited to 6 meters on a single piece of cable, might be less with cable-to-cable extensions; might be more with USB hubs along the way (assuming they don't go to sleep etc). Also check for the usual sources of noise near cabling (motors like the fridge, neon lights, etc.)

jimklimov, 2021-12-10:
Also, if your web-server table is generated with `upsstats.cgi`, that should be configurable by `hosts.conf` file and can refer to devices on remote `upsd` servers, if you eventually go that route with several NUT servers monitoring devices and one showning the dashboard.

SomebodySysop, 2021-12-10:
> Also, if your web-server table is generated with `upsstats.cgi`, that should be configurable by `hosts.conf` file and can refer to devices on remote `upsd` servers (not only `localhsot` by default), if you eventually go that route with several NUT servers monitoring devices and one showing the dashboard.
 ```
I didn't realize that remote devices aren't required to be included in upsmon.conf in a single server in order to be displayed in the upsstats.cgi of that same server.

Short term, this fixes my problem as I can now have a different email notification pattern for each NUT server by configuring the upsmon.conf of each accordingly.

Thank you!

SomebodySysop, 2021-12-10:
> For one, I suppose the MONITOR lines should not all have the powervalue=1 column (that is how many power sources of this machine that UPS powers, so if that UPS outage is critical compared to [...]
I followed a tutorial to get these servers running that recommended "1" as default.  I need to read up on what that value should actually be.  

> Looking at your table here, it seems that your Raspberry is monitoring USB UPSes in many locations. How long is the cabling? I think USB is officially limited to 6 meters on a single piece of cable, [...]

The USB cables on the server that is sending the COMBAD notifications are both less than 6ft.  No hubs.  But, there is a lot of other equipment (router, nas, HDD, computer, laptop) running between the UPS devices and the server which could be causing this noise.  Didn't think of that either.  

I will close this now as my primary issue has been addressed.  Many thanks for the assistance!

closed 2021-12-10 by SomebodySysop