networkupstools/nut #1225: Powercom driver setup with temperature metrics

dev-e, 2021-12-12:
Hi! I am trying to set up nut server for my Powercom UPS.
With `usbhid-ups` driver I get all needed metrics except the `ups.temperature`.
I have found [this issue](https://github.com/networkupstools/nut/issues/1031) and, as far as I understand, I should use `powercom` driver instead of `usbhid-ups` to get temperature. However, I cannot setup this driver correctly.
My /etc/nut/ups.conf:
```[powercom]
    driver = powercom
    port = auto
#    vendorid = 0d9f
#    productid = 0004
    desc = "UPS Powercom"
```
The nut-server service won't start with this error: `Can't connect to UPS [powercom] (powercom-powercom): No such file or directory`
nut-server package on Raspbian:
`nut-server                           2.7.4-8                                 armhf        network UPS tools - core system`
Could you please assist with the setup?

jimklimov, 2021-12-20:
The message means that the driver service for powercom was not (re-)started. Drivers talk a device protocol on one side, and NUT protocol for upsd (nut-server) on another.

dev-e, 2021-12-23:
Thanks, @jimklimov ! As far as I understand, the poercom driver is a part of the current repository and should work out of the box without any additional components. Here is what I get when trying to start the driver:
```
$ sudo upsdrvctl start      
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - PowerCom protocol UPS driver 0.17 (2.7.4)

Unable to open auto: No such file or directory

Things to try:

 - Check 'port=' in ups.conf

 - Check owner/permissions of all parts of path

Fatal error: unusable configuration
Driver failed to start (exit status=1)
```
It seems that the port should be specified explicitly, but I do not clearly understand what file should be used for it.
Here is my `dmesg` output:
```
[341819.923589] usb 1-1.3: USB disconnect, device number 69
[341820.223669] usb 1-1.3: new low-speed USB device number 70 using xhci_hcd
[341820.364725] usb 1-1.3: New USB device found, idVendor=0d9f, idProduct=0004, bcdDevice= 0.02
[341820.364747] usb 1-1.3: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[341820.364766] usb 1-1.3: Product: HID UPS Battery
[341820.364784] usb 1-1.3: Manufacturer: POWERCOM Co.,LTD
[341820.364801] usb 1-1.3: SerialNumber: 004-0D9F-000
[341820.416029] hid-generic 0003:0D9F:0004.00C1: hiddev96,hidraw0: USB HID v1.00 Device [POWERCOM Co.,LTD HID UPS Battery] on usb-0000:01:00.0-1.3/input0
```
Could you please assist with this? Or maybe there is the way to get temperature info when using `usbhid-ups` driver?

jimklimov, 2021-12-24:
Re-reading that linked story, it seems that the powercom-hid subdriver of usbhid-ups was meant; not the older (IIRC serial-port) powercom standalone driver. 

Looking at dates and comments, the fix may be in NUT source master by now, but certainly not in 2.7.4 release that you seem to be using (OS packaged?)

Is a build from source an option for you?

dev-e, 2022-01-07:
@jimklimov , thanks, building from source solved my problem!
Config `/usr/local/ups/etc/ups.conf`:
```
[powercom]
    driver = usbhid-ups
    port = auto
```
Start daemon:
`/usr/local/ups/sbin/upsdrvctl start && /usr/local/ups/sbin/upsd`
Now I am getting `battery.temperature` metric in the output:
```
$ upsc powercom@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 30
battery.date: 2014/01/15
battery.runtime: 1020
battery.temperature: 48.0
battery.type: PbAc
device.mfr: POWERCOM Co.,LTD
device.model: HID UPS Battery
device.serial: 004-0D9F-000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4-3723-g92ecace7
driver.version.data: PowerCOM HID 0.5
driver.version.internal: 0.43
input.frequency: 50.0
input.voltage: 240.0
input.voltage.nominal: 220
output.frequency: 50.0
output.voltage: 214.0
output.voltage.nominal: 220
ups.beeper.status: disabled
ups.date: 2014/01/15
ups.delay.shutdown: 20
ups.delay.start: 60
ups.load: 28
ups.mfr: POWERCOM Co.,LTD
ups.model: HID UPS Battery
ups.productid: 0004
ups.serial: 004-0D9F-000
ups.status: OL TRIM
ups.test.result: Done and passed
ups.timer.shutdown: 0
ups.timer.start: 817200
ups.vendorid: 0d9f
```

closed 2022-01-07 by dev-e