networkupstools/nut #1227: MinuteMan UPS/SNMP Support?

ShadowBullet, 2021-12-15:
Is it possible to include either USB support or basic SNMP support for minuteman UPS's? I would love to be able to monitor mine via NUT along with my other UPS's, I have the SNMP MIBs if necessary but this type of integration would be very beneficial to me, and based on my own research, to other individuals as well.

almulder, 2021-12-15:
I second this, I have 2 MinuteMan UPS both with SNMP cards installed. 

UPS Model: E1000RTXL2U
SNMP Card: Minuteman SNMP-NV6

jbeck1563, 2021-12-20:
I have a Minuteman EC1500RT2U available if there is a need to collect data for implementation.

jimklimov, 2021-12-20:
I believe NUT sources should include documentation and scripts for generation of a *-mib.c subdriver file with the mib2nut mappings table. I never did that myself though, so can't walk through that.

Also note that if the Minuteman supports IETF MIB, or standard USB HID, you might have luck with existing drivers.

jimklimov, 2021-12-21:
Update: that fix was small (adding VID/PID as known), so merged. Now you can test that with NUT master branch.