networkupstools/nut #1238: I'd like to add support for Armac UPS

blaa, 2021-12-23:
Hello,
  I have http://armac.pl/oferta/ups-rack-line-interactive-r2000ipsw/ and should be able to add support for it, but I wonder how to do it best and if there's some place to talk/consult the approach. I've sniffed and reveng a bit it's "native" 2005-old, x86, hideous driver. 

Bus 002 Device 008: ID 0925:1234 Lakeview Research UPS USB MON V1.4

It has USB port with "emulated serial" over control + interrupt transfers. It generally supports Q1-like protocol (I found Q1\r, I, F, Q for alarm, T01 for test commands).  Its driver seems to support the richcomm protocol but it doesn't seem to be used with this model, and richcomm_usb doesn't work with it either (connects, but decodes the status badly).

Commands are 4 byte long, whereas answers are *6* byte long. As far as I understand both Blazer and nutdrv_qx drivers use 8 byte long transfers which is(?) the main problem.

What should I do? Copy some matching subdriver of nutdrv_qx and try to code it there? Use blazer? Or maybe I'm missing something obvious?

Cheers,

jimklimov, 2021-12-23:
I hope a little that @zykh would chime in with more details, but haven't heard from our resident Qx protocol expert lately :\

In general, the best way forward would be to add a subdriver for nutdrv_qx, others may have been there first, but duplicate a lot of code and should be eventually obsoleted. That said, many of those older drivers for Qx devices did become parts of nutdrv_qx.c or specific subdrivers, so I suppose if you make it work by injecting new logic into an older driver, you would be able to import it into the new one as well.

blaa, 2021-12-23:
Linked driver from the producer site is http://armac.pl/wp-content/uploads/2018/08/PowerManagerII-EN.7z which doesn't seem to be really that Armac-specific. It's rather old (2005-2009) too. According google-found https://www.pythontr.com/makale/debian-wheezy-icin-ups-kurulumu-409 this might work with Digitech UPS. Vendor/product and partially protocol matches richcomm_usb (which is a limited experimental driver).

Grepping those names doesn't help finding easier solution than implementing the subdriver though, but maybe after implementation wider set of UPSes could be supported.

I'll try to do a PoC based on "fabula" if no better idea appears. ;)

blaa, 2021-12-24:
PoC:
```c
int armac_command(const char *cmd, char *buf, size_t buflen)
{
	int	ret = 0;
	size_t	i;
  int index = -1;

	struct {
      const char *str;
      char cmdbuf[4];
	} commands[] = {
      /* 1st byte of cmd, low-nibble indicates number of bytes in command. */
      { "Q1\r",	{0xa3, 'Q', '1', '\r'} },	/* Status */
      { "F\r",	{0xa2, 'F', '\r', 0x00 } },	/* Ratings */
      { "I\r",	{0xa2, 'I', '\r', 0x00 } },	/* Vendor infos */
      { "Q\r",	{0xa2, 'Q', '\r', 0x00 } },	/* Beeper toggle */
      // { "C\r",	0x0a, },	/* Cancel shutdown/Load on [0x(0..F)A]*/
      { NULL, "" }
	};
  /* Handle only supported commands */
	for (i = 0; commands[i].str; i++) {
      if (strcmp(cmd, commands[i].str) != 0)
          continue;
      index = i;
      break;
	}
  if (index == -1) {
      /* Echo the unknown command back */
      upsdebugx(3, "armac: unknown cmd '%s'   <\n", cmd);
      return snprintf(buf, buflen, "%s", cmd);
  }
  upsdebugx(3, "%s: Got a known command '%s' bl %lu   <\n",
            __func__, cmd, buflen);

  /* Send command to the UPS */
  ret = usb_control_msg(udev, USB_ENDPOINT_OUT + USB_TYPE_CLASS + USB_RECIP_INTERFACE,
                        0x09, 0x200, 0, commands[index].cmdbuf, 4, 5000);

  if (ret <= 0) {
			upsdebugx(3, "send control: %s (%d)", ret ? usb_strerror() : "timeout", ret);
			return ret;
  }

  memset(buf, 0, buflen);
  unsigned int bufpos = 0;
  char response[6];

  while (bufpos < buflen - 6) {
      /* Read data in 6-byte chunks */
      memset(response, 0, sizeof(*response)); /* TEMP for debug */

      ret = usb_interrupt_read(udev, 0x81, response, 6, 1000);
      /* Any errors here mean that we are unable to read a reply (which will happen after successfully writing a command to the UPS) */
      if (ret <= 0) {
          upsdebugx(3, "interrupt read error: %s (%d)", ret ? usb_strerror() : "timeout", ret);
          return ret;
      }

      upsdebugx(3, "read: ret %d buf %02hhx: %02hhx %02hhx %02hhx %02hhx %02hhx  >%c%c%c%c%c<", ret,
                response[0], response[1], response[2], response[3], response[4], response[5],
                response[1], response[2], response[3], response[4], response[5]);

      const char bytes_available = response[0] & 0x0f;
      if (bytes_available == 0) {
          upsdebugx(3, "message end found");
          break;
      }

      memcpy(buf + bufpos, response + 1, bytes_available);
      bufpos += bytes_available;
	}
  if (bufpos >= buflen - 6) {
      upsdebugx(3, "Protocol error, too much data read.");
      return -1;
  }
  upsdebugx(3, "final message '%s'", buf);
  return bufpos;
}
```

```  
                      [D3] armac_command: Got a known command 'Q1\r'                         
   2.121903     [D3] read: ret 6 buf c1: 28 30 30 30 30  >(0000<                  
   2.126943     [D3] read: ret 6 buf 81: 32 30 30 30 30  >20000<
   2.130978     [D3] read: ret 6 buf c1: 33 30 30 30 30  >30000<
   2.133998     [D3] read: ret 6 buf 81: 38 30 30 30 30  >80000<      
(...)
   2.188399     [D3] read: ret 6 buf c1: 2e 30 30 30 30  >.0000<
   2.192434     [D3] read: ret 6 buf 81: 30 30 30 30 30  >00000<
   2.196461     [D3] read: ret 6 buf c1: 20 30 30 30 30  > 0000<
   2.201500     [D3] read: ret 6 buf 81: 30 30 30 30 30  >00000<
   2.205522     [D3] read: ret 6 buf c1: 31 30 30 30 30  >10000<
   2.209558     [D3] read: ret 6 buf 81: 30 30 30 30 30  >00000<
   2.213593     [D3] read: ret 6 buf c1: 20 30 30 30 30  > 0000<
   2.217624     [D3] read: ret 6 buf 81: 35 30 30 30 30  >50000<
   2.221645     [D3] read: ret 6 buf c1: 30 30 30 30 30  >00000<
   2.226685     [D3] read: ret 6 buf 81: 2e 30 30 30 30  >.0000<
   2.230712     [D3] read: ret 6 buf c1: 30 30 30 30 30  >00000<
   2.234744     [D3] read: ret 6 buf 81: 20 30 30 30 30  > 0000<
   2.238767     [D3] read: ret 6 buf c1: 35 30 30 30 30  >50000<
   2.242798     [D3] read: ret 6 buf 81: 34 30 30 30 30  >40000<
   2.246830     [D3] read: ret 6 buf c1: 2e 30 30 30 30  >.0000<
   2.251873     [D3] read: ret 6 buf 81: 30 30 30 30 30  >00000<
   2.255897     [D3] read: ret 6 buf c1: 20 30 30 30 30  > 0000<
   2.259928     [D3] read: ret 6 buf 81: 33 30 30 30 30  >30000<
   2.263959     [D3] read: ret 6 buf c1: 33 30 30 30 30  >30000<
   2.266985     [D3] read: ret 6 buf 81: 2e 30 30 30 30  >.0000<
   2.271019     [D3] read: ret 6 buf c1: 30 30 30 30 30  >00000<
   2.276057     [D3] read: ret 6 buf 81: 20 30 30 30 30  > 0000<
   2.280086     [D3] read: ret 6 buf c1: 30 30 30 30 30  >00000<
   2.284108     [D3] read: ret 6 buf 81: 30 30 30 30 30  >00000<
   2.288141     [D3] read: ret 6 buf c1: 30 30 30 30 30  >00000<
   2.292175     [D3] read: ret 6 buf 81: 30 30 30 30 30  >00000<
   2.296203     [D3] read: ret 6 buf c1: 31 30 30 30 30  >10000<
   2.301240     [D3] read: ret 6 buf 81: 30 30 30 30 30  >00000<
   2.305261     [D3] read: ret 6 buf c1: 30 30 30 30 30  >00000<
   2.309290     [D3] read: ret 6 buf 81: 30 30 30 30 30  >00000<
0000<313330     [D3] read: ret 6 buf c1: 0d 30 30 30 30  >
0000<303671     [D3] read: ret 6 buf 80: 0d 30 30 30 30  >
   3.303688     [D3] message end found
'  3.303690     [D3] final message '(238.0 140.0 236.0 010 50.0 54.0 33.0 00001000
```

I'm unsure why it sends only single BYTE in each interrupt transfer. On my wireshark dumps it did send up to 5. Maybe the code is too fast and the UPS is not able to buffer enough bytes. Or there's bad usb configuration chosen before. but certainly reads something ok.

jimklimov, 2021-12-25:
I did not program for USB myself yet, but revised a lot of the codebase
recently. This way I know there are APIs for interrupt and bulk transfers -
both were seen in NUT sources. Try the other one, maybe it would behave
better? :)

blaa, 2021-12-25:
I made a PR and discussion can probably move there. I checked for higher API for USB control but have found nothing similar. Other communication functions in _qx driver seem to work in similar way. It would be probably grant to split this fie into multiple, but it uses some globals that make this non-trivial.

Sending/reading data in chunks could be extracted to separate functions, but the 1-byte header (0xa0 + bytes) makes this non-generic anyway

closed 2022-01-21 by jimklimov

Novocaine85, 2022-04-27:
Thank you @blaa .

Is it included in 2.8.0?

blaa, 2022-04-27:
As far as I know it's included. I'm still using a build from an old branch. I hope for debian packages to arrive.

Novocaine85, 2022-04-29:
can you confirm that the driver is "nutdrv_qx" instead of richcomm_usb?

i have a Vultech 2000VA.

if i put richcomm shows the error as mentioned in issue #640 

thanks

Novocaine85, 2022-04-29:
sooooooooooooooo

apparently after rebooting for some reason I can't see the ups from lsusb.

I have no clue why is doing this.
Sorry

blaa, 2022-04-29:
Well, you don't have to check it right *NOW*. Plugging/unplugging USB and of course restarting UPS (that's more complicated usually) can help. I have it within a VM and sometimes have to pass-through the USB

PR #1250 (2022-01-21): nutdrv_qx.c: add support for Armac UPS (or maybe other richcomm) from #1247
After my git-pushing accident (fat-fingered maintainer is sorry), this PR follows up from #1247 including some fixes over it.

> Based on Armac R/2000I/PSW, but should support other UPSes that work with "PowerManagerII" software from Richcomm Technologies from around 2004-2005.
> Closes #1238
> Tested to support querying, short and 10 minute tests. UPS is currently 300km from me so I can't pull the plug to test status change though. I'm unsure about battery measurement - seems to return different values than richcomm s [... cut]