networkupstools/nut #1251: [HCL] Digitus DN-170040/41 supported by blazer_usb

Evil2000, 2022-01-05:
The **Digitus** _DN-170040_ and _DN-170041_ is currently not listed as supported ups. I tested it with the `blazer_usb` driver and it works as expected. The same model was also sold as **Voltronic Power** _GalleonX9-RT LCD-1-3K_.
It is connected through USB and is recognized as: `Bus 001 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial`

Here's my ups.conf:
```
[digitus]
	driver = blazer_usb
	port = auto
	vendorid = 0665
	productid = 5161
	subdriver = cypress 
	desc = "Digitus DN-170041"
```

upsc reports:
```
battery.charge: 100
battery.voltage: 40.86
battery.voltage.high: 39.00
battery.voltage.low: 31.20
battery.voltage.nominal: 36.0
device.mfr: 
device.model: HV 1K5
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.subdriver: cypress
driver.parameter.synchronous: no
driver.parameter.vendorid: 0665
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 6.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 234.3
input.voltage.fault: 234.3
input.voltage.nominal: 230
output.voltage: 234.4
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: 00938.02
ups.load: 14
ups.mfr: 
ups.model: HV 1K5
ups.productid: 5161
ups.status: OL BYPASS
ups.temperature: 24.3
ups.type: online
ups.vendorid: 0665
```
The manufacturers driver is available from https://www.power-software-download.com/viewpower.html. But this is a huge Java program which has no ability to distribute the powerstate to remote systems nor is it lightweight, as nut is.

The datasheet and manual of the ups can be found here: [ftp://ftp.assmann.com/pub/DN-/DN-170041___4016032365907](http://ftp.assmann.com/pub/DN-/DN-170041___4016032365907)

Regards.

jimklimov, 2022-01-10:
Thank you for the report!

Would it be possible for you to check also, if this device works as well with the newer nutdrv-qx driver and similar configuration settings, please?

Evil2000, 2022-01-10:
This driver seems to give more information. It even detects the right `voltronic` protocol as well as all settings from the ups:
```
$ upsdrvctl start digitus
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
Using protocol: Voltronic 0.06
device.serial: UPS reported a non-significant serial [00000000000000]
battery_number: 3
battery_number, settable range: 1..9
bypass_alarm: enabled
battery_alarm: enabled
bypass_when_off: disabled
alarm_control: enabled
converter_mode: disabled
eco_mode: enabled
bypass_forbidding: disabled
site_fault_detection: disabled
advanced_eco_mode: enabled
max_bypass_volt: 264
max_bypass_volt, settable range: 230..264
min_bypass_volt: 170
min_bypass_volt, settable range: 170..220
max_bypass_freq: 53.0
max_bypass_freq, settable range: 51.0..60.0
min_bypass_freq: 47.0
min_bypass_freq, settable range: 40.0..49.0
```
The upsc output is as follows:
```
$ upsc digitus
battery.charge: 100
battery.energysave: no
battery.packs: 1
battery.protection: yes
battery.runtime: 19620
battery.voltage: 41.00
battery.voltage.high: 39
battery.voltage.low: 31.2
battery.voltage.nominal: 36
device.model: 1K5
device.type: ups
driver.name: nutdrv_qx
driver.parameter.chargetime: 14400
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.runtimecal: 210,100,660,50
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Voltronic 0.06
driver.version.internal: 0.28
input.current.nominal: 6.0
input.frequency: 50.0
input.frequency.nominal: 50.0
input.phases: 1
input.transfer.high: 242
input.transfer.high.max: 254
input.transfer.high.min: 237
input.transfer.low: 218
input.transfer.low.max: 223
input.transfer.low.min: 206
input.voltage: 234.0
input.voltage.nominal: 230.0
outlet.0.switchable: no
output.current: 1.0
output.current.nominal: 6
output.frequency: 50.0
output.frequency.nominal: 50.0
output.phases: 1
output.power.maximum.percent: 16
output.power.minimum.percent: 2
output.powerfactor: 0.9
output.voltage: 234.0
output.voltage.nominal: 230.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: 00938.02
ups.firmware.aux: P01
ups.load: 15
ups.power.nominal: 1500
ups.productid: 5161
ups.start.auto: yes
ups.start.battery: yes
ups.status: OL BYPASS
ups.temperature: 24.6
ups.type: online
ups.vendorid: 0665
```

My settings in the ups.conf are:
```
[digitus]
	driver = nutdrv_qx
	port = auto
	desc = "Digitus DN-170041"
	default.battery.voltage.high = 39
        default.battery.voltage.low = 31.2
        default.battery.voltage.nominal = 36
	runtimecal = 210,100,660,50
        chargetime = 14400
```
The values are set according the datasheet.

jimklimov, 2022-01-11:
Great, thanks for confirmation and details!

closed 2022-04-25 by jimklimov