networkupstools/nut #1252: Incorrect Status from Liebert PSI5-1500RM120

airforceixi, 2022-01-06:
Running NUT 2.7.4 on Fedora 35 connecting to a Liebert PSI5-1500RM120. Does not provide any valid status information. Here are some outputs:

# upsc ups@localhost
```
device.mfr: Vertiv Co.
device.model: Liebert PSI5
device.serial: ###
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0002
driver.parameter.synchronous: no
driver.parameter.vendorid: 10AF
driver.version: 2.7.4
driver.version.data: Belkin/Liebert HID 0.17
driver.version.internal: 0.41
ups.mfr: Vertiv Co.
ups.model: Liebert PSI5
ups.productid: 0002
ups.serial: ###
ups.status: OB
ups.vendorid: 10af
```

/etc/ups/ups.conf
```
[ups]
driver = usbhid-ups
port = auto
productid = 0002
vendorid = 10AF
```

usbhid-ups -DDDDD -a ups
```
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     send_to_all: SETINFO driver.parameter.port "auto"
   0.000031     send_to_all: SETINFO driver.parameter.productid "0002"
   0.000040     send_to_all: SETINFO driver.parameter.vendorid "10AF"
   0.000052     debug level is '5'
   0.001548     upsdrv_initups...
   0.006497     Checking device (8087/0024) (002/002)
   0.006534     Failed to open device, skipping. (Permission denied)
   0.006546     Checking device (1D6B/0002) (002/001)
   0.006560     Failed to open device, skipping. (Permission denied)
   0.006566     Checking device (5986/02D2) (001/006)
   0.006575     Failed to open device, skipping. (Permission denied)
   0.006583     Checking device (0A5C/21E6) (001/005)
   0.006599     Failed to open device, skipping. (Permission denied)
   0.006605     Checking device (147E/2020) (001/004)
   0.006617     Failed to open device, skipping. (Permission denied)
   0.006623     Checking device (10AF/0002) (001/014)
   0.009188     - VendorID: 10af
   0.009203     - ProductID: 0002
   0.009207     - Manufacturer: Vertiv Co.
   0.009212     - Product: Liebert PSI5
   0.009215     - Serial Number: ###
   0.009219     - Bus: 001
   0.009222     - Device release number: 0002
   0.009229     Trying to match device
   0.009267     Device matches
   0.009291     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.009823     HID descriptor, method 1: (9 bytes) => 09 21 10 01 00 01 22 1b 00
   0.009842     HID descriptor length (method 1) 27
   0.009851     i=0, extra[i]=09, extra[i+1]=21
   0.009867     HID descriptor, method 2: (9 bytes) => 09 21 10 01 00 01 22 1b 00
   0.009878     HID descriptor length (method 2) 27
[... 386 lines cut ...]
   6.272505     upsdrv_updateinfo...
   6.524942     libusb_get_interrupt: Connection timed out
   6.525038     Got 0 HID objects...
   6.525069     Quick update...
   8.274313     upsdrv_updateinfo...
   8.526732     libusb_get_interrupt: Connection timed out
   8.526780     Got 0 HID objects...
   8.526793     Quick update...
   8.806313     send_to_one: PONG
  10.275878     upsdrv_updateinfo...
  10.528313     libusb_get_interrupt: Connection timed out
  10.528368     Got 0 HID objects...
  10.528398     Quick update...
  12.277721     upsdrv_updateinfo...
  12.530170     libusb_get_interrupt: Connection timed out
  12.530219     Got 0 HID objects...
  12.530232     Quick update...
  14.279560     upsdrv_updateinfo...
  14.532043     libusb_get_interrupt: Connection timed out
  14.532097     Got 0 HID objects...
  14.532130     Quick update...
  14.812836     send_to_one: PONG
```

I can provide remote access for dev/testing if any devs are willing/available.

jimklimov, 2022-01-07:
Hello, before we get to in-depth testing, can you please check if a local
build of the current NUT master branch works for you?

Sadly, the last release was a few years ago and next one is still brewing,
so there were a lot of fixes (including at least 2 merged PRs tagged as
liebert) that are not yet packaged by distros.

> Message ID: ***@***.***>
>

airforceixi, 2022-01-07:
@jimklimov I switched from Fedora to Ubuntu and followed the instructions in the Wiki on how to build & install the drivers from master. Appears to be the same. See below:

upsc ups@localhost
```
Init SSL without certificate database
device.mfr: Vertiv Co.
device.model: Liebert PSI5
device.serial: ###
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0002
driver.parameter.synchronous: no
driver.parameter.vendorid: 10AF
driver.version: 2.7.4-3723-g92ecace7
driver.version.data: Belkin/Liebert HID 0.17
driver.version.internal: 0.43
ups.mfr: Vertiv Co.
ups.model: Liebert PSI5
ups.productid: 0002
ups.serial: ###
ups.status: OB
ups.vendorid: 10af
```

usbhid-ups -DDDDD -a ups
```
Network UPS Tools - Generic HID driver 0.43 (2.7.4-3723-g92ecace7)
USB communication driver 0.33
   0.000000     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000061     [D5] send_to_all: SETINFO driver.parameter.productid "0002"
   0.000086     [D5] send_to_all: SETINFO driver.parameter.vendorid "10AF"
   0.000124     [D1] debug level is '5'
   0.001586     [D5] send_to_all: SETINFO device.type "ups"
   0.001615     [D2] Initializing an USB-connected UPS with library (null) (NUT subdriver name='USB communication driver' ver='0.33')
   0.001630     [D1] upsdrv_initups (non-SHUT)...
   0.434003     [D3] usb_busses=0x563fa085af10
   0.434071     [D2] Checking device (1D6B/0003) (004/001)
   0.434302     [D2] - VendorID: 1d6b
   0.434339     [D2] - ProductID: 0003
   0.434406     [D2] - Manufacturer: unknown
   0.434465     [D2] - Product: unknown
   0.434512     [D2] - Serial Number: unknown
   0.434565     [D2] - Bus: 004
   0.434594     [D2] - Device: 001
   0.434646     [D2] - Device release number: 0504
   0.434689     [D2] Trying to match device
   0.434706     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.434724     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.434741     [D2] Device does not match - skipping
   0.434861     [D2] Checking device (1D6B/0002) (003/001)
   0.434988     [D2] - VendorID: 1d6b
   0.435011     [D2] - ProductID: 0002
   0.435031     [D2] - Manufacturer: unknown
   0.435050     [D2] - Product: unknown
   0.435062     [D2] - Serial Number: unknown
   0.435074     [D2] - Bus: 003
   0.435085     [D2] - Device: 001
   0.435104     [D2] - Device release number: 0504
   0.435123     [D2] Trying to match device
   0.435143     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.435168     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.435189     [D2] Device does not match - skipping
   0.435236     [D2] Checking device (8087/0024) (002/002)
   0.502519     [D2] - VendorID: 8087
   0.502582     [D2] - ProductID: 0024
   0.502649     [D2] - Manufacturer: unknown
   0.502675     [D2] - Product: unknown
[... 398 lines cut ...]
   4.832685     [D1] upsdrv_updateinfo...
   5.085258     [D2] libusb_get_interrupt: Connection timed out
   5.085305     [D1] Got 0 HID objects...
   5.085319     [D1] Quick update...
   6.834523     [D1] upsdrv_updateinfo...
   7.087756     [D2] libusb_get_interrupt: Connection timed out
   7.087805     [D1] Got 0 HID objects...
   7.087819     [D1] Quick update...
   8.836359     [D1] upsdrv_updateinfo...
   9.089257     [D2] libusb_get_interrupt: Connection timed out
   9.089306     [D1] Got 0 HID objects...
   9.089320     [D1] Quick update...
  10.838198     [D1] upsdrv_updateinfo...
  11.091463     [D2] libusb_get_interrupt: Connection timed out
  11.091510     [D1] Got 0 HID objects...
  11.091525     [D1] Quick update...
  12.840039     [D1] upsdrv_updateinfo...
  13.093250     [D2] libusb_get_interrupt: Connection timed out
  13.093298     [D1] Got 0 HID objects...
  13.093313     [D1] Quick update...
  14.841877     [D1] upsdrv_updateinfo...
  15.095180     [D2] libusb_get_interrupt: Connection timed out
  15.095230     [D1] Got 0 HID objects...
  15.095246     [D1] Quick update...
```

jimklimov, 2022-01-07:
Those "connection timed out" might be a HW problem, or might be a libusb(-0.1) issue. At least, IIRC there were a few reports where switching to libusb-1.0 branches helped. That is currently on trajectory to become part of NUT master branch, but if you're in a hurry - check the fightwarn-libusb-1.0+0.1-sysheaders branch as the currebt pinnacle.

airforceixi, 2022-01-07:
@jimklimov I built the fightwarn-libusb-1.0+0.1-sysheaders branch using the same instructions but usbhid-ups appears to still use libusb-0.1:
```
Network UPS Tools - Generic HID driver 0.43 (2.7.4-4033-g1b39f976)
USB communication driver (libusb 0.1) 0.35
```

jimklimov, 2022-01-08:
Do you have libusb-1.0-dev installed? Distros vary as to how they might call it though...

The new docs/config-prereq.txt lists the prerequisites on a number of platforms used by CI farm, you may want to check suggestions there.

airforceixi, 2022-01-08:
@jimklimov I was able to install libusb-1.0-0-dev and build it again. I confirmed it's using libusb 1.0 now but the same issues persist.

jimklimov, 2022-01-08:
Wondering if someone else (udev?) might have grabbed the USB device descriptor and not released it to the NUT driver...

Or if there is something about permissions?

Or a systems thing (controller/chipset/... preferring "bulk" I/O vs interrupts)?..

airforceixi, 2022-01-11:
I had to modify the udev rules that came with nut to add the PID:VID of this UPS so that the commands would work to the degree they do now. Outside of that, I don't see anything else that could cause this but I'm not a developer so I'm not 100% what to look for. I have this issue with 3 of these UPSes, 2 of them connected to a Raspberry Pi and 1 connected to a laptop (for testing) and both have the same problem, so I don't think it's related to the host.

As mentioned earlier, I can provide access to my testing laptop for someone to inspect/test if there's availability.

airforceixi, 2022-01-11:
I was able to install the Liebert Vertiv Power Assist software and that connects and reads the UPS values properly. Maybe that connectivity can be monitored for reverse engineering?

jimklimov, 2022-01-12:
After a bit of reading up to refresh my memory, it seems that the `pollonly` flag might make the difference between interrupt-driven and driver-initiated transfers.

jimklimov, 2022-01-12:
Reverse-engineering (by way of snooping on USB wire traffic) should be possible, is occasionally mentioned in issues or PR discussions here, but I haven't done that myself so can't really walk you (or other devs) through that.

What I know is that it's possible, and should be documented somewhere :\