networkupstools/nut #1261: APC Smart-UPS X 750 - some data is missing

kitor, 2022-01-12:
Hello,

```
battery.charge: 66
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 65535
battery.runtime.low: 150
battery.type: PbAc
battery.voltage: 52.0
battery.voltage.nominal: 48.0
device.mfr: American Power Conversion 
device.model: Smart-UPS X 750 
device.serial: <serial>
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: COM 03.8 / UPS 03.8
ups.mfr: American Power Conversion 
ups.mfr.date: 2011/11/28
ups.model: Smart-UPS X 750 
ups.productid: 0003
ups.serial:  <serial>
ups.status: OL CHRG
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d
```

Device connected via USB. It is lacking some data like ups load, input/output voltages and so on (visible on older Smart-UPS 1000 RM that I'm replacing).

I found some topics suggesting that it may use modbus, not sure if that's the case here (at least manual don't mention it, nor it is mentioned in menus). Any suggestions?

Below is the descriptor data:

```
Bus 002 Device 006: ID 051d:0003 American Power Conversion UPS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x051d American Power Conversion
  idProduct          0x0003 UPS
  bcdDevice            1.06
  iManufacturer           1 American Power Conversion 
  iProduct                2 Smart-UPS X 750  FW:COM 03.8 / UPS 03.8
  iSerial                 3 <serial>
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x60
      (Missing must-be-set bit!)
      Self Powered
[... 413 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              10
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

jimklimov, 2022-01-12:
Can you please check with a local build of current NUT master branch?
Packaged release 2.7.4 (which your "screenshots" imply) was a few years
ago, and some fixes might have landed for APC over time, and more
importantly for libusb-1.0 support which should be more stable WRT
'Resource temporarily unavailable' messages).

My quick guess is that this model exposes data at different HID points so
mapping is missing (at least in your build).

Note APC was also seen to not follow standards for USB data encoding, see
e.g. #1208

kitor, 2022-01-12:
> Packaged release 2.7.4 (which your "screenshots" imply) was a few years

Interesting, as this is Debian Stable (or to be exact, Proxmox VE 7.0.x).
I'll build latest release and report back.

jimklimov, 2022-01-12:
Note, not latest "release" (still is 2.7.4), but the master branch from
GitHub.

kitor, 2022-02-12:
Finally found some time to check that out. Likely a rookie mistake, but after building with
```
./configure -': ./configure --with-user=nut --with-group=nut --with-usb
make
make install
```

I went to `/usr/local/ups/bin` and tried to execute `./usbhid-ups -s ups -x port=auto`
```
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4613-ge1218569)
USB communication driver (libusb 0.1) 0.42
interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
No matching HID UPS found
```

BTW:  apcupsd can read all the values so it might be a good source to check how those are fetched:
```
root@pve02:/opt# apcaccess status
APC      : 001,027,0656
DATE     : 2022-02-12 19:44:45 +0100  
HOSTNAME : pve02
VERSION  : 3.14.14 (31 May 2016) debian
UPSNAME  : ups
CABLE    : USB Cable
DRIVER   : USB UPS Driver
UPSMODE  : Stand Alone
STARTTIME: 2022-02-12 19:44:43 +0100  
MODEL    : Smart-UPS X 750  
STATUS   : ONLINE 
BCHARGE  : 100.0 Percent
TIMELEFT : 1092.2 Minutes
MBATTCHG : 5 Percent
MINTIMEL : 3 Minutes
MAXTIME  : 0 Seconds
ALARMDEL : No alarm
BATTV    : 54.4 Volts
NUMXFERS : 0
TONBATT  : 0 Seconds
CUMONBATT: 0 Seconds
XOFFBATT : N/A
STATFLAG : 0x05000008
MANDATE  : 2011-11-28
SERIALNO : AS1149110011  
NOMBATTV : 48.0 Volts
FIRMWARE : COM 03.8 / UPS 03.8
```

jimklimov, 2022-02-13:
Thanks, can you please try to add some `-DDDDDD` flags to the driver
launch, to report some debugs that might shine a light on this? I'd expect
maybe unsupported VID/PID combo (new model for us, etc.), or device node
busy due to another driver instance, udev, upower etc., or the test running
as a user without permissions to the device node, etc.

kitor, 2022-02-13:
Logs:

```
# ./usbhid-ups -s ups -x port=auto -DDDDDDDD
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4613-ge1218569)
USB communication driver (libusb 0.1) 0.42
   0.000000	[D1] debug level is '8'
   0.005450	[D5] send_to_all: SETINFO device.type "ups"
   0.006061	[D2] Initializing an USB-connected UPS with library libusb-0.1 (or compat) (NUT subdriver name='USB communication driver (libusb 0.1)' ver='0.42')
   0.006561	[D1] upsdrv_initups (non-SHUT)...
   0.024062	[D3] usb_busses=0x5622c0ad99c0
//skipped other devices
   0.050910	[D2] Checking device (051D/0003) (001/003)
   0.053250	[D2] - VendorID: 051d
   0.053853	[D2] - ProductID: 0003
   0.053947	[D2] - Manufacturer: American Power Conversion 
   0.054146	[D2] - Product: Smart-UPS X 750  FW:COM 03.8 / UPS 03.8
   0.054231	[D2] - Serial Number: AS1149110011  
   0.054428	[D2] - Bus: 001
   0.054516	[D2] - Device: 003
   0.054689	[D2] - Device release number: 0106
   0.054948	[D2] Trying to match device
   0.055032	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.055239	interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
   0.055329	[D3] match_function_regex: matching a device...
   0.055527	[D2] Device matches
   0.055639	[D2] failed to claim USB device: could not claim interface 0: Device or resource busy
   0.058693	[D2] detached kernel driver from USB device...
   0.058782	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.060013	[D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 21 01 22 a9 01
   0.060055	[D2] Warning: HID descriptor, method 1 failed
   0.060082	[D3] HID descriptor length (method 1) -87
   0.060108	[D4] i=0, extra[i]=09, extra[i+1]=21
   0.060139	[D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 21 01 22 a9 01
   0.060163	[D2] Warning: HID descriptor, method 2 failed
   0.060186	[D3] HID descriptor length (method 2) -87
   0.060209	[D2] Unable to retrieve any HID descriptor
//skipped other devices
   0.061053	[D2] libusb0: No appropriate HID device found
   0.061085	No matching HID UPS found
```

```
# lsusb | grep UPS
Bus 001 Device 003: ID 051d:0003 American Power Conversion UPS
# ls -lah /dev/bus/usb/001/003 
crw-rw-r-- 1 root nut 189, 2 Feb 13 23:05 /dev/bus/usb/001/003
# usbhid-dump -a 1:3
001:003:000:DESCRIPTOR         1644789921.728014
 05 84 09 04 A1 01 09 24 A1 02 85 01 09 FE 79 02
 75 08 95 01 15 00 26 FF 00 B1 22 85 02 09 FF 79
 03 B1 22 85 03 09 FD 79 01 B1 22 05 85 85 04 09
 8F 79 01 B1 22 85 05 09 89 79 04 B1 22 85 06 09
 8B B1 22 09 2C B1 22 85 0E 09 83 25 64 B1 22 09
 67 B1 22 85 0C 09 66 81 22 09 66 B1 A2 85 10 09
 8D B1 22 09 8E B1 22 85 0F 09 8C B1 22 85 11 09
 29 B1 A2 85 09 09 85 75 10 27 FF FF 00 00 B1 A2
 85 0D 09 68 27 FF FF 00 00 66 01 10 55 00 81 A2
 09 68 B1 A2 05 84 85 12 09 57 16 FF FF 26 FF 7F
 B1 A2 85 13 09 55 75 18 17 FF FF FF FF 27 18 C5
 13 00 B1 A2 05 85 85 08 09 2A 27 64 05 00 00 16
 78 00 81 22 09 2A B1 A2 05 84 85 0A 09 40 75 10
 15 00 27 FF FF 00 00 67 21 D1 F0 00 55 05 B1 22
 85 0B 09 30 B1 A2 85 14 09 5A 75 08 15 01 25 03
 65 00 55 00 81 22 09 5A B1 A2 09 02 A1 02 85 07
 05 85 09 44 75 01 15 00 25 01 81 A2 09 44 B1 A2
 09 45 81 A2 09 45 B1 A2 09 D0 81 A2 09 D0 B1 A2
 09 D1 81 A2 09 D1 B1 A2 09 42 81 A2 09 42 B1 A2
 05 84 09 68 81 A2 09 68 B1 A2 09 69 81 A2 09 69
 B1 A2 05 85 09 43 81 A2 09 43 B1 A2 05 84 09 73
 81 A2 09 73 B1 A2 05 85 09 4B 81 A2 09 4B B1 A2
 05 84 09 65 81 A2 09 65 B1 A2 05 85 09 DB 81 A2
 09 DB B1 A2 06 86 FF 09 80 81 A2 09 80 B1 A2 95
 03 81 01 B1 01 C0 C0 06 86 FF 85 89 09 FD 15 00
 26 FF 00 75 08 95 3F 81 02 85 90 09 FC 15 00 26
 FF 00 75 08 95 3F 91 02 C0
```

If I launch distro provided packages, it is detected (as in first posts). Tests done on two machines, one had distro nut services disabled (and machine rebooted for a good measure), other had distro nut packages removed.

jimklimov, 2022-02-14:
Negative descriptor length looks odd. @nbriggs @clepple : any ideas?

Did some recent improvement to be stricter with the wire protocol make
another practical regression with APC? (2.7.4 apparently tolerated this and
talked to the device)

kitor, 2022-02-14:
Just a note: There seems to be multiple hardware revisions of this model. There are some with firmware 1.x that are not upgradable, mine is upgradable to 3.8 (as listed here), some can go up to 7.x series. So even though they all identify as Smart UPS 750-X, they have different internal model number.

I believe the same apply to 1000 and 1500W models, as they share the same chassis.

jimklimov, 2022-02-14:
Thanks for that heads-up. And just to clarify: was it the same device that
worked (at least somewhat) with packaged NUT 2.7.4 in earlier posts, and
did not get recognized with the later ones and master-branch builds?

kitor, 2022-02-14:
Yes, exactly the same device.
I tried updating the firmware, but as explained above - it turned out that my hw revision can accept up to 3.8, and it was already the revision I had.

nbriggs, 2022-02-14:
Yeah, it's the descriptor length that's wrong and preventing anything else from working correctly.

HID descriptor, method 1: (9 bytes) => 09 21 00 01 21 01 22 a9 01

so it should be doing 0xa9 | (0x01 << 8) and getting 425 (they are to be interpreted as unsigned bytes), but it's coming up with -87 because you've (newly?) declared the usb_ctrl_charbuf as signed characters for libusb 0.1 --

silversword% git blame drivers/usb-common.h 
[...]
3f8d588d (Jim Klimov     2021-12-19 16:38:04 +0100 411)  /* Here MIN/MAX should not matter much, type mostly used for casting */
c2029ec6 (Jim Klimov     2021-12-13 14:16:44 +0100 412)  typedef char* usb_ctrl_charbuf;
3f8d588d (Jim Klimov     2021-12-19 16:38:04 +0100 413)  typedef char usb_ctrl_char;
3f8d588d (Jim Klimov     2021-12-19 16:38:04 +0100 414)  #define USB_CTRL_CHAR_MIN	CHAR_MIN
3f8d588d (Jim Klimov     2021-12-19 16:38:04 +0100 415)  #define USB_CTRL_CHAR_MAX	CHAR_MAX

-- Nick

nbriggs, 2022-02-15:
While there may be fewer warnings... I suspect there will be other new bugs from having changed the buffers from unsigned to signed bytes in the libusb 0.1 case.

jimklimov, 2022-02-15:
Thanks a lot for the analysis and catch; I think the buffer type changes trickled down from libusb-1.0(+0.1) reworks under issue #300 and I believe they were supposed to match API headers of the two library versions, to call their methods with types they expected (and it is a huge mismatched mess there even inside one version).

So, the correction would be to make sure `unsigned char (*)` is used in sources of NUT drivers and libs, and cast to `usb_ctrl_char(buf)` anything that squeaks just in libusb API calls?

nbriggs, 2022-02-15:
I haven't looked that hard at the code (beyond just the negative length), but someone could look at anything in NUT code that does arithmetic on the buffer elements for issues.  I'm not sure which way round to cast is the right way -- declare the buffers that will get passed (or received from?) the libusb 0.1 code with what the libusb declarations expect and just cast for the arithmetic (probably a small number of cases...?) or declare the buffers unsigned in the NUT code and cast to what the libusb 0.1 expects when they're passed.  I suspect the former would be more obvious.  I'd cast to `uint8_t` for the arithmetic if we're assuming C99 or better with stdint.h available.

jimklimov, 2022-02-15:
Yes, by now I think C99+ is a reasonable assumption; there are too many
things to fix and de-optimise to build in ANSI mode, so I left the door
open and CI ready, but won't commit to the cause myself (documented as such
along the way to NUT 2.7.5).

kitor, 2022-03-02:
Hello, why this ticket is closed? It was about missing data, not about the bug that was apparently fixed in master.

nbriggs, 2022-03-02:
Sorry, I tagged this as closed by the bug fix that was preventing the code from reading *any* data when using libusb-0.1.

Presumably the fix for your missing data is to check if there are missing mappings from the descriptors to the NUT variables.  Perhaps @jimklimov could reopen this.

jimklimov, 2022-03-02:
Sorry about the automated mess-up :)

kitor, 2022-03-02:
Thank you. I built head (like above), run `usbhid-ups -s ups -x port=auto`, then `upsd` and `upsc` output is:
```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 13574
battery.runtime.low: 150
battery.type: PbAc
battery.voltage: 54.5
battery.voltage.nominal: 48.0
device.mfr: American Power Conversion 
device.model: Smart-UPS X 750 
device.serial: <serial>
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.7.4-4757-g9e5e34d3
driver.version.data: APC HID 0.98
driver.version.internal: 0.45
driver.version.usb: libusb-0.1 (or compat)
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.firmware: COM 03.8 / UPS 03.8
ups.mfr: American Power Conversion 
ups.mfr.date: 2011/11/28
ups.model: Smart-UPS X 750 
ups.productid: 0003
ups.serial: <serial>
ups.status: OL
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d
```

Thus nothing new, except in driver fields. 

Is there any document/guide on hid descriptors? I'm fine even running "hacked up" version as lack of load and input/output voltages screwed my automation a bit :)

nbriggs, 2022-03-02:
The HID descriptor that the UPS sent is really minimal, I ran an HID decoder over it that knows most of the APC specific usages, plus all the standard usages, and below are all the values that it will report.  Percent load, and input/output voltage are not values it seems to be prepared to report.   Power device page "ConfigVoltage" and "Voltage" are the battery nominal and actual voltages that you're seeing.

```
05 84        (GLOBAL) USAGE_PAGE         0x0084 Power Device Page 
09 04        (LOCAL)  USAGE              0x00840004 UPS (Application Collection)  
A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0x00840004: Page=Power Device Page, Usage=UPS, Type=Application Collection)
09 24          (LOCAL)  USAGE              0x00840024 Power Summary (Logical or Physical Collection)  
A1 02          (MAIN)   COLLECTION         0x02 Logical (Usage=0x00840024: Page=Power Device Page, Usage=Power Summary, Type=Logical or Physical Collection)
09 FE            (LOCAL)  USAGE              0x008400FE i Product (Static Value)  
09 FF            (LOCAL)  USAGE              0x008400FF i Serial Number (Static Value)  
09 FD            (LOCAL)  USAGE              0x008400FD i Manufacturer (Static Value)  
05 85            (GLOBAL) USAGE_PAGE         0x0085 Battery System Page 
09 8F            (LOCAL)  USAGE              0x0085008F i OEMInformation (Static Value)  
09 89            (LOCAL)  USAGE              0x00850089 i Device Chemistry (Static Value)  
09 8B            (LOCAL)  USAGE              0x0085008B Rechargeable (Static Value)  
09 2C            (LOCAL)  USAGE              0x0085002C Capacity Mode (Dynamic Value)  
09 83            (LOCAL)  USAGE              0x00850083 Design Capacity (Static Value)  
09 67            (LOCAL)  USAGE              0x00850067 Full Charge Capacity (Dynamic Value)  
09 66            (LOCAL)  USAGE              0x00850066 Remaining Capacity (Dynamic Value)  
09 66            (LOCAL)  USAGE              0x00850066 Remaining Capacity (Dynamic Value)  
09 8D            (LOCAL)  USAGE              0x0085008D Capacity Granularity 1 (Static Value)  
09 8E            (LOCAL)  USAGE              0x0085008E Capacity Granularity 2 (Static Value)  
09 8C            (LOCAL)  USAGE              0x0085008C Warning Capacity Limit (Static Value)  
09 29            (LOCAL)  USAGE              0x00850029 Remaining Capacity Limit (Dynamic Value)  
09 85            (LOCAL)  USAGE              0x00850085 Manufacturer Date (Static Value)  
09 68            (LOCAL)  USAGE              0x00850068 Run Time To Empty (Dynamic Value)  
09 68            (LOCAL)  USAGE              0x00850068 Run Time To Empty (Dynamic Value)  
05 84            (GLOBAL) USAGE_PAGE         0x0084 Power Device Page 
09 57            (LOCAL)  USAGE              0x00840057 Delay Before Shutdown (Dynamic Value)  
09 55            (LOCAL)  USAGE              0x00840055 Delay Before Reboot (Dynamic Value)  
05 85            (GLOBAL) USAGE_PAGE         0x0085 Battery System Page 
09 2A            (LOCAL)  USAGE              0x0085002A Remaining Time Limit (Dynamic Value)  
09 2A            (LOCAL)  USAGE              0x0085002A Remaining Time Limit (Dynamic Value)  
05 84            (GLOBAL) USAGE_PAGE         0x0084 Power Device Page 
09 40            (LOCAL)  USAGE              0x00840040 Config Voltage (Static Value or Dynamic Value)  
09 30            (LOCAL)  USAGE              0x00840030 Voltage (Dynamic Value)  
09 5A            (LOCAL)  USAGE              0x0084005A Audible Alarm Control (Dynamic Value)  
09 5A            (LOCAL)  USAGE              0x0084005A Audible Alarm Control (Dynamic Value)  
09 02            (LOCAL)  USAGE              0x00840002 Present Status (Logical Collection)  
A1 02            (MAIN)   COLLECTION         0x02 Logical (Usage=0x00840002: Page=Power Device Page, Usage=Present Status, Type=Logical Collection)
05 85              (GLOBAL) USAGE_PAGE         0x0085 Battery System Page 
09 44              (LOCAL)  USAGE              0x00850044 Charging (Dynamic Flag)  
09 44              (LOCAL)  USAGE              0x00850044 Charging (Dynamic Flag)  
09 45              (LOCAL)  USAGE              0x00850045 Discharging (Dynamic Value)  
09 45              (LOCAL)  USAGE              0x00850045 Discharging (Dynamic Value)  
09 D0              (LOCAL)  USAGE              0x008500D0 AC Present (Dynamic Flag)  
09 D0              (LOCAL)  USAGE              0x008500D0 AC Present (Dynamic Flag)  
09 D1              (LOCAL)  USAGE              0x008500D1 Battery Present (Dynamic Flag)  
09 D1              (LOCAL)  USAGE              0x008500D1 Battery Present (Dynamic Flag)  
09 42              (LOCAL)  USAGE              0x00850042 Below Remaining Capacity Limit (Dynamic Flag)  
09 42              (LOCAL)  USAGE              0x00850042 Below Remaining Capacity Limit (Dynamic Flag)  
05 84              (GLOBAL) USAGE_PAGE         0x0084 Power Device Page 
09 68              (LOCAL)  USAGE              0x00840068 Shutdown Requested (Dynamic Flag)  
09 68              (LOCAL)  USAGE              0x00840068 Shutdown Requested (Dynamic Flag)  
09 69              (LOCAL)  USAGE              0x00840069 Shutdown Imminent (Dynamic Flag)  
09 69              (LOCAL)  USAGE              0x00840069 Shutdown Imminent (Dynamic Flag)  
05 85              (GLOBAL) USAGE_PAGE         0x0085 Battery System Page 
[... cut ...]
09 FC          (LOCAL)  USAGE              0xFF8600FC <-- Warning: Undocumented usage
15 00          (GLOBAL) LOGICAL_MINIMUM    0x00 (0)
26 FF00        (GLOBAL) LOGICAL_MAXIMUM    0x00FF (255)
75 08          (GLOBAL) REPORT_SIZE        0x08 (8) Number of bits per field
95 3F          (GLOBAL) REPORT_COUNT       0x3F (63) Number of fields
91 02          (MAIN)   OUTPUT             0x00000002 (63 fields x 8 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap
```

kitor, 2022-03-03:
I will disagree, as PowerChute BE reports those:

[image]

Of course I understand it may talk in some additional protocol, but does that give any clues where to look?

nbriggs, 2022-03-03:
That by itself doesn't give any clue about what PowerChute may be doing.  Can you capture all the USB traffic starting before you start PowerChute? USBTrace, USBPcap or other USB sniffing tools might make it possible to reverse engineer what it's doing.  You'd need to get the USB trace *and* the values that it showed in all the PowerChute reports so that it's possible to correlate the displayed values with any raw data that's visible in the trace, and to see any commands that the PowerChute software sent to the UPS to elicit any private reports.

kitor, 2022-03-03:
Sure, I can, next time I'll be on location.

jimklimov, 2022-04-10:
Hello @kitor : any updates on practical testing?

kitor, 2022-04-10:
I'll have a chance during next weekend.

kitor, 2022-04-18:
Packet capture from Win7 in attachment. [apc.zip](https://github.com/networkupstools/nut/files/8505633/apc.zip)

Screen capture as "data reference": https://www.youtube.com/watch?v=_aYcPjheZMM
I started packet capture before starting APC service, so hopefully all the handshakes are in there.
I changed some in-UPS settings back and forward, but I didn't touch outlets switching as device is deployed on prod.

Hope that is good reference, if not I can setup Linux VM and capture traffic there.

nbriggs, 2022-04-18:
I’m offline for the next 3 weeks, so I hope someone else can look at this.

kitor, 2022-07-01:
By a chance, did anyone manage to take a peek at this data?

jimklimov, 2023-08-31:
@nbriggs : Hello again, did/would you have a chance to revise the data dump above? :)

nbriggs, 2023-08-31:
@jimklimov - I didn't look at it (I assumed that there'd be someone else who might take on the dump decoding in a more timely manner), but I have some time over the next week or so so I will see if I can figure out what was going on in the raw USB capture and report back whether there's anything of interest.

jimklimov, 2023-08-31:
Sounds great, thanks a lot!

kitor, 2023-09-01:
Hey, just a reminder that there's a screen capture of me gathering that dump: https://www.youtube.com/watch?v=_aYcPjheZMM
So you should have a pretty good reference point.

In the meantime I also got a stand alone Smart-UPS 750 (SMT750I). If I'm correct, SMT and SMX series are siblings (released about the same time, share almost identical local LCD interface). SMX was a bit more advanced with external battery packs support.
It reports exactly the same set of data as SMX750I:

```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 4320
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.1
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion 
device.model: Smart-UPS 750
device.serial: AS1316113420  
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: APC HID 0.98
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: UPS 09.8 / ID=18
ups.mfr: American Power Conversion 
ups.mfr.date: 2013/04/19
ups.model: Smart-UPS 750
ups.productid: 0003
ups.serial: AS1316113420  
ups.status: OL
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d
```

nbriggs, 2023-09-01:
Yes, noted.  I have extracted the USB data from the PCAP file, but now I have to look at decoding those with reference to the HID report descriptor that the UPS sent. I think a possible first step will be to remove duplicate request/response pairs and then determine if there are any reports that *don't* correspond to the known report types (or lengths?) in the report descriptor that therefore might contain the extra information that PowerChute sees.

nbriggs, 2023-09-09:
Note to let you know that my free tie got preempted - I'm probably not going to be able to work on this for about another month.

jimklimov, 2023-09-10:
Awww, schucks! Thanks for the update, and we do hope to see you soon! :)

FWIW, I plan to also populate an `scripts/usb_exploration` area in NUT sources to track bits of know-how about making heads or tails of USB HID info, sort of like the investigations you did for LogMin/LogMax debacle and the tools involved. However I found I can't really write too much about it based on issue comment history alone - so some notes, example scripts, etc. from de-facto experts like you would be very welcome :)

jimklimov, 2024-04-04:
@kitor : would you be able to check if this UPS would actually respond better to the new `apc_modbus` driver shipped since NUT v2.8.1? Note that at this time you would need a custom build of `libmodbus` with USB support for that, check https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol

kitor, 2024-04-04:
Thanks for notification. I'll try it in next few days and get back to you with result.

kitor, 2024-05-06:
NUT built as in linked docs (with libmodbus built to `/opt/prefix`)
`./configure --with-drivers=apc_modbus --with-usb --with-modbus    --with-modbus-includes=-I/opt/prefix/include/modbus    --with-modbus-libs="-L/opt/prefix/lib -lmodbus"`
then I did `make`, `make install`. That's on Debian testing btw.

I tried using modubs driver but it fails with:
```
root@odroid:/usr/local/ups/bin# ./apc_modbus -s ups -x port=auto
Network UPS Tools - NUT APC Modbus driver 0.10 (2.8.2-195-g050e101a1)
_apc_modbus_read_registers: Read of 516:636 failed: Connection timed out (auto)
Can't read inventory information from the UPS
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

Not sure if any extra params needs to be set?
`Bus 001 Device 002: ID 051d:0003 American Power Conversion UPS`

kitor, 2024-05-06:
Nevermind, forgot to enable modbus in device menus. Now it started, I need to prepare some config for test.

kitor, 2024-05-06:
Okey, that is a lot more output. That's from SMT750i, I can't check SMX750i (from title) now as it is deployed in remote location. I'll try when I'll visit there.

```
root@odroid:/usr/local/ups/bin# ./upsc ups
battery.charge: 100.00
battery.date: 2021-04-15
battery.date.maintenance: 2024-10-07
battery.runtime: 6420
battery.temperature: 37.30
battery.voltage: 26.84
device.mfr: American Power Conversion
device.model: Smart-UPS 750
device.serial: AS1316113420
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: apc_modbus
driver.parameter.bus: 001
driver.parameter.device: 002
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2-195-g050e101a1
driver.version.internal: 0.10
driver.version.usb: libusb-0.1 (or compat)
experimental.output.energy: 525890
input.transfer.high: 253
input.transfer.low: 208
input.transfer.reason: AcceptableInput
input.voltage: 240.39
outlet.group.0.delay.reboot: 8
outlet.group.0.delay.shutdown: 180
outlet.group.0.delay.start: 0
outlet.group.0.name: UPS Outlets
outlet.group.1.delay.reboot: 8
outlet.group.1.delay.shutdown: 90
outlet.group.1.delay.start: 0
outlet.group.1.name: Outlet Group 1
outlet.group.2.delay.reboot: -1
outlet.group.2.delay.shutdown: -1
outlet.group.2.delay.start: -1
outlet.group.2.name: 
outlet.group.3.delay.reboot: -1
outlet.group.3.delay.shutdown: -1
outlet.group.3.delay.start: -1
outlet.group.3.name: 
outlet.group.count: 1
output.current: 0.31
output.frequency: 49.97
output.voltage: 239.00
ups.delay.reboot: 8
ups.delay.shutdown: 180
ups.delay.start: 0
ups.efficiency: LoadTooLow
ups.firmware: UPS 09.8
ups.id: APCUPS
ups.load: 5.19
ups.mfr: American Power Conversion
ups.mfr.date: 2013-04-19
ups.model: Smart-UPS 750
ups.power: 77.93
ups.power.nominal: 750
ups.productid: 0003
ups.realpower: 25.94
ups.realpower.nominal: 500
ups.serial: AS1316113420
ups.status: OL HE
ups.test.result: Passed, Source: Internal
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 051d
```

jimklimov, 2024-05-07:
Super, thanks for the report!

> Nevermind, forgot to enable modbus in device menus. Now it started, I need to prepare some config for test.

That would be a menu in the UPS LCD+buttons panel, right?

kitor, 2024-05-07:
Yes. From UPS LCD in Configuration submenu there are modbus enable/disable and modbus address options. That's SMT750i 

On SMX750i, according to my original post there was no modbus option visible in menus.

kitor, 2024-05-07:
Ok, I built libmodbus/nut on that remote machine where SMX750i is connected, however all I get is this:

```
root@pve03:/opt/libusb/bin# lsusb | grep UPS
Bus 001 Device 005: ID 051d:0003 American Power Conversion UPS
root@pve03:/opt/libusb/bin# ./apc_modbus -s ups -x port=auto -DDDDDDD
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - NUT APC Modbus driver 0.10 (2.8.2-195-g050e101a1)
   0.000037	[D5] send_to_all: SETINFO driver.version.usb "libusb-0.1 (or compat)"
   0.000039	[D1] Using USB implementation: libusb-0.1 (or compat)
   0.000044	[D3] main_arg: var='port' val='auto'
   0.000047	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000049	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000052	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000054	[D1] Network UPS Tools version 2.8.2-195-g050e101a1 (release/snapshot of 2.8.2.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/opt/libusb/include/modbus --with-modbus-libs='-L/opt/libusb/lib -lmodbus' --prefix=/opt/libusb
   0.000057	[D1] debug level is '7'
   0.000059	[D5] send_to_all: SETINFO driver.debug "7"
   0.000061	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000391	[D1] Succeeded to become_user(nobody): now UID=65534 GID=65534
   0.000400	[D1] Signalling UPS [ups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.000405	Can't open /var/state/ups/apc_modbus-ups: No such file or directory
   0.000407	[D1] Request for other driver to exit returned code -1
   0.000409	[D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.000412	[D5] send_to_all: SETINFO device.type "ups"
   0.000414	[D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 6
Considering device #1 (1b4f:9206)
libusb_open for device #1 failed: Access denied (insufficient permissions)
Considering device #2 (0624:0201)
libusb_open for device #2 failed: Access denied (insufficient permissions)
Considering device #3 (051d:0003)
libusb_open for device #3 failed: Access denied (insufficient permissions)
Considering device #4 (05e3:0608)
libusb_open for device #4 failed: Access denied (insufficient permissions)
No matching device found
   0.002325	modbus_connect: unable to connect: No such device
   0.002342	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.002347	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

Insufficient permissions? Some weird state after shutting down distro NUT? (I stopped nut-server, nut-client, nut-driver@ups, verified usbhid-ups is not running)

jimklimov, 2024-05-07:
You would probably need to add `051d:0003` into the udev/upower rules file so it assigns permissions for that devfs node to `nut:nut` typically. Note how you did not `configure` a user into the build, so that may be another reason - `Succeeded to become_user(nobody): now UID=65534 GID=65534`.

A quick check if the device would work with the driver can be done by using `-x vendorid=051d -x productid=0003 -x user=root` options to the driver, I believe.

kitor, 2024-05-07:
Ok, then this one likely will not work with new driver :(

```
./apc_modbus -x port=auto -s ups -x user=nut 
Network UPS Tools - NUT APC Modbus driver 0.10 (2.8.2-195-g050e101a1)
_apc_modbus_read_registers: Read of 516:636 failed: Connection timed out (auto)
Can't read inventory information from the UPS
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

According to https://community.se.com/t5/APC-UPS-Data-Center-Enterprise/SMX1500RM2U-Modbus-Support-Firmware-3-8/td-p/292812

> Just to add to what Bill said - while Modbus is not natively supported on SMX's, Modbus TCP is supported on those models using the AP9630 and any firmware revision greater than v6.4.0. Modbus RTU is supported on these units using the AP9635, with the same firmware requirements. Here is a link to the register map: http://www.apc.com/salestools/LFLG-A32G3L/LFLG-A32G3L_R1_EN.pdf

That's a shame, it means `apc_modbus` won't be a full solution. With that in mind I hope someone might attempt to decode that USB packet capture I made earlier and maybe have some guesses from that.

Full debug log, just in case:
```
root@pve03:/opt/libusb/bin# ./apc_modbus -x port=auto -s ups -x user=nut -DDDDDD
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - NUT APC Modbus driver 0.10 (2.8.2-195-g050e101a1)
   0.000036	[D5] send_to_all: SETINFO driver.version.usb "libusb-0.1 (or compat)"
   0.000038	[D1] Using USB implementation: libusb-0.1 (or compat)
   0.000042	[D3] main_arg: var='port' val='auto'
   0.000045	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000046	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000049	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000051	[D3] main_arg: var='user' val='nut'
   0.000053	[D6] testval_reloadable: var=user, oldval=nobody, newval=nut, reloadable=0, reload_flag=0
   0.000054	[D1] testval_reloadable: setting 'user' exists and differs: new value 'nut' vs. 'nobody'
   0.000056	[D6] testval_reloadable: verdict for (re)loading var=user value: 1
   0.000057	[D1] Overriding previously specified user 'nobody' with 'nut' specified for driver section
   0.000060	[D1] Network UPS Tools version 2.8.2-195-g050e101a1 (release/snapshot of 2.8.2.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --with-drivers=apc_modbus --with-usb --with-modbus --with-modbus-includes=-I/opt/libusb/include/modbus --with-modbus-libs='-L/opt/libusb/lib -lmodbus' --prefix=/opt/libusb
   0.000063	[D1] debug level is '6'
   0.000065	[D5] send_to_all: SETINFO driver.debug "6"
   0.000067	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000628	[D1] Succeeded to become_user(nut): now UID=110 GID=117
   0.000637	[D1] Signalling UPS [ups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.000643	Can't open /var/state/ups/apc_modbus-ups: No such file or directory
   0.000645	[D1] Request for other driver to exit returned code -1
   0.000646	[D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.000649	[D5] send_to_all: SETINFO device.type "ups"
   0.000652	[D5] send_to_all: SETINFO driver.state "init.device"
Number of USB devices: 6
Considering device #1 (1b4f:9206)
libusb_open for device #1 failed: Access denied (insufficient permissions)
Considering device #2 (0624:0201)
libusb_open for device #2 failed: Access denied (insufficient permissions)
Considering device #3 (051d:0003)
   0.004929	[D2] _apc_modbus_usb_callback: Matched American Power Conversion  Smart-UPS X 750  FW:COM 03.8 / UPS 03.8 (USB VID/PID 051d:0003)
   0.004937	[D2] _apc_modbus_usb_callback: Checking American Power Conversion  Smart-UPS X 750  FW:COM 03.8 / UPS 03.8 (USB VID/PID 051d:0003) report descriptors
   0.004981	[D1] _apc_modbus_usb_callback: Found report ids RX=0x90 TX=0x89
   0.004986	[D5] send_to_all: SETINFO ups.vendorid "051d"
   0.004988	[D5] send_to_all: SETINFO ups.productid "0003"
Found Device 3 (Path 1-9.1):
  Vendor ID: 0x051d
  Product ID: 0x0003
Bytes flushed (0)
   0.017119	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.017127	[D5] send_to_all: SETINFO driver.version "2.8.2-195-g050e101a1"
   0.017129	[D5] send_to_all: SETINFO driver.version.internal "0.10"
   0.017130	[D5] send_to_all: SETINFO driver.name "apc_modbus"
   0.017132	[D5] send_to_all: SETINFO driver.state "init.info"
[01][03][02][04][00][78]
Waiting for a confirmation...
ERROR Connection timed out: select
   0.520341	_apc_modbus_read_registers: Read of 516:636 failed: Connection timed out (auto)
Bytes flushed (0)
   1.532811	Can't read inventory information from the UPS
   1.532827	[D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   1.532941	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
   1.532948	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

jimklimov, 2024-05-09:
For kicks, did you try this one with `usbhid-ups`? Maybe they support the standard at least to some extent?.. (allegedly APC used to have good support until early 2010's with the move to Modbus)

Another idea is if they have it all published on a non-zero USB interface number and if the driver only queries the zero'th one. For USB HID investigations, users looked at verbose `lsusb` listings where the device exposes one or more interfaces, and (forgot the name) width(?) of the base report descriptor list on that interface - the one useful for USB HID being about 600 bytes, and ones with other goals being 37-50 bytes or so.

Recent NUT versions, like your custom build, should have the CLI/ups.conf options to specify a custom number, although practical support in particular driver programs may vary so far - it is a recent'ish effort.

kitor, 2024-05-09:
> For kicks, did you try this one with `usbhid-ups`? Maybe they support the standard at least to some extent?.. (allegedly APC used to have good support until early 2010's with the move to Modbus)

Nope, just tried - nothing new available in upsc output.

> Another idea is if they have it all published on a non-zero USB interface number and if the driver only queries the zero'th one. For USB HID investigations, users looked at verbose `lsusb` listings [...]

All the descriptor dumps I think are attached in first few messages of this issue. It doesn't look anything else is available there.

I wouldn't be surprised if they indeed use modbus internally, just there's some magic done to start that communication. After all it would be surprising they implemented two new protocols at once.

JahMark420, 2025-05-05:
Hello - believe im running into some issues here with a similar model. Did you ever sort this one out?

PR #1320 (2022-03-02): Cast to unsigned type when interpreting HID descriptor length bytes (libusb 0.1)

The libusb 0.1 interface definition declares a (signed) char type for
control messages.  The HID descriptor length contained within a control
message is intended to be interpreted as a pair of unsigned bytes so
we must cast to uint8_t when doing the arithmetic rather than trip over
the sign bit.

Closes #1261, closes #1312.
commit f2fb70d2 (2022-03-02): Cast to unsigned type when interpreting HID descriptor length bytes (libusb 0.1)

The libusb 0.1 interface definition declares a (signed) char type for
control messages.  The HID descriptor length contained within a control
message is intended to be interpreted as a pair of unsigned bytes so
we must cast to uint8_t when doing the arithmetic rather than trip over
the sign bit.

Closes #1261, closes #1312.