networkupstools/nut #1263: v2.7.5 release?

j-broo, 2022-01-14:
Please excuse me for opening an issue with this question, but is there any indication on when we can expect v2.7.5 to be released? How close is the current build to a release candidate?

Maybe make this issue pinned as I'm sure others have this question too and I don't see any info.

teoberi, 2022-01-17:
A sign [here](https://github.com/networkupstools/nut/milestones)?

jimklimov, 2022-01-18:
Good point, thanks for reminding of that mechanism... gotta remember now how to assign issues/PRs to milestones.

In practical terms, I hope to get the NUT CI farm (recently migrated to another Fosshost's another slower sponsor) to churn through some of the remaining open PRs for device supports and HCL bumps, possibly fix some smaller issues (that may wait for another release name, but would be cool to fix earlier), and go through some HW testing to catch some low-hanging-fruit issues.

With libusb-1.0 merged, and literally thousands of build warnings eradicated, as well as many screenfulls of added and updated drivers, etc. coming from the community, I hope we are quite close to releasing a solidly featured version. Pragmatic caution tells me to expect a stream of issue reports (due to the sheer amount of changes, something gotta give in!) and a subsequent release afterwards, though. Hopefully shortly afterwards, if due to that.

jimklimov, 2022-01-18:
Also thanks for the suggestion to pin. Referencing earlier similar questions for cross-linking:
* #819 
* #699

blaa, 2022-02-22:
Since 2.7.4 was released in 2016, and master is currently 4698 commits away from this tag, I believe it would make more sense to tag it as 2.8. I'm unsure if the project uses semantic versioning or not, but that's certainly not just a "patch" release, and 2.7.5 would be widely understood like a small change over 2.7.4.

Would be cool to have a release that would make it into the next stable Debian release. Bullseye was released 2021-08, so there's some time for that, but downstream packaging might take a lot of time as well.

jimklimov, 2022-02-24:
Well, it is supposed to be semantic. Speaking of it, the goals were indeed for a 2.7.5, but as I found in recent sprints - over time incompatibilities arose, such as new key words for ups.conf etc. that make config files acceptable for current master using its newest features not usable with 2.7.4. In that sense it may be indeed a 2.8 material, thanks for the reminder.

jimklimov, 2022-02-24:
OTOH I'd say one known "blocker" for a release now is an issue with libusb-1.0 vs 0.1 API, with signed/unsigned char types in buffers, and some maths suffering from that (e.g. ending up with negative value ranges).

All or most of the other things on the plate are nice-to-have's any time, the sooner the better - but landing to master after a new release and becoming part of a later one is not critical.

jimklimov, 2022-04-26:
The deed is done!

closed 2022-04-26 by jimklimov

gertvdijk, 2022-04-26:
@jimklimov Congratulations on this milestone! 🎉 👍🏼  Your (and others' work) is very much appreciated!