networkupstools/nut #1269: UPS Ever Duo II Pro 800 - not supported?

quexpl, 2022-01-21:
Hi, I'm having trouble with UPS Ever Duo II Pro 800. Seems like it should be supported ([hcl](https://networkupstools.org/stable-hcl.html?manufacturer=EVER&model=DUO%20II%20Pro%20series), [blog post](https://rootprompt.apatsch.net/2012/01/25/ups-ever-duo-ii-pro-with-nut-linux/)) but I'm not able to connect with it. 

I will be grateful for any suggestions and advise - if there is anything I can do to make it work with NUT.

Os details: FreeBSD 13.0-RELEASE
NUT version: 2.7.4

`ups.conf`:

```
[ever-blazer]
  driver = blazer_usb
  port = auto

[ever-nutdrv]
  driver = nutdrv_qx
  port = auto

```

I tried `blazer_usb` driver with no luck...
```
~> # /usr/local/libexec/nut/blazer_usb -a ever-blazer -DDDDD
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
   0.000000	debug level is '5'
   0.000582	Checking device (0665/5161) (/dev/usb//dev/ugen2.2)
   0.031651	- VendorID: 0665
   0.031674	- ProductID: 5161
   0.031678	- Manufacturer: INNO TECH
   0.031682	- Product: USB to Serial
   0.031685	- Serial Number: 20100826
   0.031689	- Bus: /dev/usb
   0.031692	- Device release number: 0002
   0.031696	Trying to match device
   0.031702	Device matches
   0.031716	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.031731	send_to_all: SETINFO ups.vendorid "0665"
   0.031738	send_to_all: SETINFO ups.productid "5161"
   0.031748	send_to_all: SETINFO device.type "ups"
   0.031753	send_to_all: SETINFO driver.version "2.7.4"
   0.031760	send_to_all: SETINFO driver.version.internal "0.12"
   0.031766	send_to_all: SETINFO driver.name "blazer_usb"
   0.031771	Trying megatec protocol...
   0.033666	send: Q1
   1.014568	read: Unknown error
   1.014591	blazer_status: short reply
   1.014596	Status read 1 failed
   1.016632	send: Q1
   2.000222	read: Unknown error
   2.000257	blazer_status: short reply
   2.000263	Status read 2 failed
   2.003776	send: Q1
   2.984589	read: Unknown error
   2.984613	blazer_status: short reply
   2.984619	Status read 3 failed
   2.984623	Trying mustek protocol...
   2.986560	send: QS
   3.967490	read: Unknown error
   3.967514	blazer_status: short reply
   3.967519	Status read 1 failed
   3.969490	send: QS
   4.950555	read: Unknown error
   4.950580	blazer_status: short reply
   4.950586	Status read 2 failed
   4.952496	send: QS
   5.933445	read: Unknown error
   5.933465	blazer_status: short reply
   5.933470	Status read 3 failed
   5.933474	Trying megatec/old protocol...
   5.935459	send: D
[... 203 lines cut ...]
            Item(Global): Report Count, data= [ 0x08 ] 8
            Item(Main  ): Input, data= [ 0x82 ] 130
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Volatile Bitfield
            Item(Local ): Usage, data= [ 0x02 ] 2
                            (null)
            Item(Global): Report Count, data= [ 0x08 ] 8
            Item(Main  ): Output, data= [ 0x82 ] 130
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              32
Device Status:     0x0000
  (Bus Powered)

```

jimklimov, 2022-01-26:
Hello, just in case: would it be possible for you to re-test with a local
build of current NUT master branch from github? in particular, libusb-1.0
support landed recently, maybe that would stabilize some interactions with
the OS and hardware further. Also, there's a PR open to add support for
Ever devices, that stalled on something and I did not get back recently to
review and merge (many other contributions piled up before it), possibly
building a local version of that PR source branch (or its local merge with
master) may fare even better.

closed 2025-02-11 by quexpl

jimklimov, 2025-02-11:
I suppose it works with the more recent NUT codebase than as of initial posting? Thanks for testing  :)

quexpl, 2025-02-11:
Unfortunately I don't have the UPS anymore.