networkupstools/nut #1275: [Question] How do I fix missing information from the RMCARD205 in my CyberPower UPS?

UntouchedWagons, 2022-02-06:
I have a RMCARD205 in the CyberPower OR1500LCDRT2U and I just got a source-built copy of NUT set up in a test VM running Debian. NUT is able to pull nearly all the information from the card, but four lines are missing (if I'm understanding this right).

Here is the output of the nut-driver service:

```
# journalctl -u nut-driver@snmpv3
-- Journal begins at Wed 2022-01-19 15:32:06 EST, ends at Sun 2022-02-06 17:44:52 EST. --
Feb 06 17:40:36 nut-snmp-test systemd[1]: Starting Network UPS Tools - device driver for snmpv3...
Feb 06 17:40:36 nut-snmp-test nut-driver@snmpv3[43205]: Detected OR1500LCDRT2U on host rmcard205.untouchedwagons.site:161 (mib: cyberpower 0.4)
Feb 06 17:40:36 nut-snmp-test nut-driver@snmpv3[43205]: [snmpv3] unhandled ASN 0x5 received from .1.3.6.1.4.1.3808.1.1.1.7.2.7.0
Feb 06 17:40:36 nut-snmp-test snmp-ups[43205]: [snmpv3] unhandled ASN 0x5 received from .1.3.6.1.4.1.3808.1.1.1.7.2.7.0
Feb 06 17:40:36 nut-snmp-test nut-driver@snmpv3[43205]: [snmpv3] unhandled ASN 0x5 received from .1.3.6.1.4.1.3808.1.1.1.2.2.2.0
Feb 06 17:40:36 nut-snmp-test snmp-ups[43205]: [snmpv3] unhandled ASN 0x5 received from .1.3.6.1.4.1.3808.1.1.1.2.2.2.0
Feb 06 17:40:36 nut-snmp-test nut-driver@snmpv3[43205]: [snmpv3] unhandled ASN 0x5 received from .1.3.6.1.4.1.3808.1.1.1.5.2.9.0
Feb 06 17:40:36 nut-snmp-test snmp-ups[43205]: [snmpv3] unhandled ASN 0x5 received from .1.3.6.1.4.1.3808.1.1.1.5.2.9.0
Feb 06 17:40:36 nut-snmp-test nut-driver@snmpv3[43205]: [snmpv3] unhandled ASN 0x5 received from .1.3.6.1.4.1.3808.1.1.1.5.2.11.0
Feb 06 17:40:36 nut-snmp-test snmp-ups[43205]: [snmpv3] unhandled ASN 0x5 received from .1.3.6.1.4.1.3808.1.1.1.5.2.11.0
Feb 06 17:40:36 nut-snmp-test snmp-ups[43206]: Startup successful
Feb 06 17:40:36 nut-snmp-test nut-driver@snmpv3[43204]: Network UPS Tools - UPS driver controller 2.7.4-4536-gd46d5153
Feb 06 17:40:36 nut-snmp-test systemd[1]: Started Network UPS Tools - device driver for snmpv3.
```

From what I can tell, the missing info is `device.serial`, `output.current`, `ups.serial` and possibly `ups.status`. Of course none of these are critical to know, but I'd like to have this information available.

UntouchedWagons, 2022-02-07:
Hmm none of the systemd services are staying enabled across reboots.

jimklimov, 2022-02-07:
I think there may be issues with snmpv3 parsing (in libsnmp? maybe due to
encryption etc.?) that were reported a few times, and re-runs with snmp v1
or v2c showed those data. Finding and fixing that would be great :)

Can you quickly check if the older versions indeed show that, and if
snmpwalk has that issue? maybe comparing their sources with how snmp-ups.c
sets up the v3 dialog would be helpful...

As for enabled services - are they outright disabled after reboot, or
somehow otherwise not starting? did you `systemctl daemon-reload` after
installing them? Is there any variant of nut-driver-enumerator running that
aims to generate those instances and interdependencies, and maybe went
crazy and deleted/disabled some services (shouldn't, but just in case...
and should leave a trail in its journal)?

UntouchedWagons, 2022-02-07:
I'm not sure I understand what you want me to check in regards to snmp.

Running `systemctl daemon-reload` after enabling the services does seem to have done the trick although it's odd I never had to do that for systemd services I wrote myself.

Something else I discovered is that on Debian, `/var/run` is a symlink to `/run` which is a tmpfs. So the `/var/run/nut/` folder that NUT depends on disappears after every reboot. What would serve as a better alternate folder?

jimklimov, 2022-02-08:
Hello, bear with me :)

Regarding snmp - can you check if snmpv1 connection in the driver fares better, and if the snmpwalk program (from libsnmp) can request those OIDs that NUT failed to?

As for (/var)/run, nut build and installation should have added a configuration file for systemd-tmpfiles that re-creates and re-chowns the paths it needs during boot, so services may run less privileged.

UntouchedWagons, 2022-02-08:
I tried using snmp1 with NUT and the same four codes are missing. I looked in the output of snmpwalk and those four codes appear to be missing: [the numbers mason.txt](https://github.com/networkupstools/nut/files/8019845/the.numbers.mason.txt)

Where would I find the tmpfiles config file?

jimklimov, 2022-02-10:
Seems `snmpwalk` did not look into that data sub-tree. Can you ask it again with argument(s) for these 4 OIDs explicitly, or maybe their common ancestor like .1.3.6.1.4.1.3808.1.1.1 ?

UntouchedWagons, 2022-02-11:
snmpwalk reports NULL for all four missing OIDs:

```
$ snmpwalk -v3 -l authPriv -A LandlordDreamMatch -X OnAngerDoubleNothing -a MD5 -x AES -u snmp3 rmcard205.untouchedwagons.site .1.3.6.1.4.1.3808.1.1.1.7.2.7.0
iso.3.6.1.4.1.3808.1.1.1.7.2.7.0 = NULL
$ snmpwalk -v3 -l authPriv -A LandlordDreamMatch -X OnAngerDoubleNothing -a MD5 -x AES -u snmp3 rmcard205.untouchedwagons.site .1.3.6.1.4.1.3808.1.1.1.2.2.2.0
iso.3.6.1.4.1.3808.1.1.1.2.2.2.0 = NULL
$ snmpwalk -v3 -l authPriv -A LandlordDreamMatch -X OnAngerDoubleNothing -a MD5 -x AES -u snmp3 rmcard205.untouchedwagons.site .1.3.6.1.4.1.3808.1.1.1.5.2.9.0
iso.3.6.1.4.1.3808.1.1.1.5.2.9.0 = NULL
$ snmpwalk -v3 -l authPriv -A LandlordDreamMatch -X OnAngerDoubleNothing -a MD5 -x AES -u snmp3 rmcard205.untouchedwagons.site .1.3.6.1.4.1.3808.1.1.1.5.2.11.0
iso.3.6.1.4.1.3808.1.1.1.5.2.11.0 = NULL
```