networkupstools/nut #1281: Njoy

Shiroe93, 2022-02-10:


jimklimov, 2022-02-10:
Probably because nobody coded and contributed a driver for that? And/or did not report that an existing driver works for those models and brands (quite possible since there are relatively few OEMs whose devices and/or protocols are resold by many vendors).

jimklimov, 2022-04-10:
According to #867, at least one nJoy is known working.

closed 2022-04-10 by jimklimov

Shiroe93, 2022-04-10:
I've managed to get it working partially with an snmp card and the snmp-ups driver unfortunately though I was not able to see temps [image]
This is the data i was able to pull in my home assistant instance

jimklimov, 2023-10-28:
@Shiroe93 : Did you have a chance to check with some `snmpwalk` or similar tool, and an NJoy vendor extensions MIB if they publish one, whether it *can* report the temperatures etc.?

The default IETF MIB mapping only has a `battery.temperature` data point...

Shiroe93, 2023-10-28:
At the moment I bought the lan add on card and I'm using that to retrieve the data via snmtp protocol

jimklimov, 2023-10-28:
Thanks. I mean this part of your most-recent post:

> I've managed to get it working partially with an snmp card and the snmp-ups driver unfortunately though I was not able to see temps 

Are you monitoring with NUT now or directly with some other SNMP client?
If you tried other SNMP clients - did you find a way to display temps there? (It may be in a non-standard OID sub-tree as defined by vendor extension).

Shiroe93, 2023-10-28:
I'm using the nut add on in home assistant

commit e195389c (2022-04-25): driver.list.in: Add HCL info for nJoy Aten PRO 3000 (SNMP)

Closes: #1281