networkupstools/nut #1284: UPS not in HW compatibility list

jjantoine, 2022-02-11:
Hi,
my UPS Vultech ups 1000va pure line:

"https://www.vultech.it/it/gruppi-di-continuit%C3%A0/916--ups1000va-pure-line-interactive-lcd-gruppo-di-continuit%C3%A0-8052780300896.html"

is not in the HW compatibility list.
Is there a driver compatible with my UPS?
Thank you

jimklimov, 2022-02-11:
Never heard of this brand yet. But if they happen to use a standard
protocol like USB-HID or a variant of popular Megatec Qx family, it might
just work, or might be open to generation of a (HID) subdriver with scripts
provided in NUT sources.

whc2001, 2022-02-13:
Looks like they have provided a software, and the OEM company is Richcomm Co, ltd (广州凝智科技有限公司). The software bundle contains `PMService.exe` which acts as a combination of `upsd` and `upsmon`, that communicates with the USB UPS through `RCUpsUsb.dll`, and it in turn calls `RCUsbHidComm.dll` to do the actual USB transaction. It also frequently calls `Watch()` in `RCAPI.dll` which appears to be some sort of logging mechanism. Some logging action regarding USB can be discovered:

```
Watch(
    "UsbWriteData:%02X %02X %02X %02X %02X %02X %02X ret:%d\r\n",
    (unsigned __int8)*a2,
    (unsigned __int8)a2[1],
    (unsigned __int8)a2[2],
    (unsigned __int8)a2[3],
    (unsigned __int8)a2[4],
    (unsigned __int8)a2[5],
    (unsigned __int8)a2[6],
    v3);
```

```
Watch(
        "Head:%02X %02X %02X %02X %02X %02X BIT VAR:%08s\r\n",
        LOBYTE(v40[0]),
        BYTE1(v40[0]),
        BYTE2(v40[0]),
        HIBYTE(v40[0]),
        LOBYTE(v40[1]),
        BYTE1(v40[1]),
        v9);
```

However I am not sure where the log content is going to. And as I don't have that device on hand I am not able to run the service program. But judging from those limited result, it's very likely that the protocol is binary based rather than string like Q*.

jimklimov, 2022-02-13:
Thanks for the research. 

For my 2c: The name richcomm is at least among existing drivers, and "hex"/"bin" vs "txt" suffix is part of some driver filenames and/or was in some older releases before some more drivers were deprecated and assimilated into nutdrv_qx and similar efforts.

jjantoine, 2022-02-16:
Thank you for your support. 

However, I have to add the UPS on TrueNAS and I didn't get if there is a solution, selecting something compatible, or i definitely can't configure my UPS on it. 

Thank you again for your answers.

jimklimov, 2022-02-24:
Sorry about the delay. 

The gist of an answer is "I don't know". With a device/vendor never heard of before, I can only suggest experimental trial-and-error approach, picking drivers and their options like specific subdrivers, vendor/id/bus/... pinning, etc. If you do get it working, even if partially (e.g. some values are bogus but online/onbatt status is real and useful), a report to update HCL and/or DDL would be great.

blaa, 2022-04-27:
Try the one I did recently: https://github.com/networkupstools/nut/issues/1238 it seems it uses the same oem software

mirh, 2023-10-10:
Supposedly this improved with #2005?