networkupstools/nut #1289: Cross-compile issues with temporary changes to fix

eklinedi, 2022-02-14:
Hello, these are the changes I had to do to successfully cross-compile NUT w/ LibUSB-1.0 (v1.0.24) and Net-SNMP (5.9.1) support on a PC build (x86_64-pc-linux-gnu) for a Raspberry Pi 3 Model B+ target/host (arm-linux-gnueabihf)

Packages required/needed on Ubuntu 20.04:
$ sudo apt-get install git autoconf automake libtool g++-arm-linux-gnueabihf gcc-arm-linux-gnueabihf pkg-config-arm-linux-gnueabihf

Compilation command lines:
LIBUSB-1.0 (checked out at v1.0.24)
$ ./autogen.sh --prefix=/opt/test/libusb --enable-static --build=x86_64-pc-linux-gnu --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf --enable-udev=no

NET-SNMP (5.9.1)
$ ./configure --prefix=/opt/test/netsnmp --enable-static --build=x86_64-pc-linux-gnu --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf --with-defaults --without-perl-modules --disable-embedded-perl --without-pcre

NUT - master branch as of 2/11/2022
$ ./autogen.sh && PKG_CONFIG_PATH='/opt/test/libusb/lib/pkgconfig:/opt/test/netsnmp/lib/pkgconfig' ./configure --prefix=/opt/test/nut --with-modbus=no --without-ssl --with-user=nut --with-group=nut --enable-static --enable-strip --build=x86_64-pc-linux-gnu --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf --with-usb=libusb-1.0 --with-snmp

=== files with issues encountered ===
configure.ac
 - Using AC_RUN_IFELSE when testing strcasecmp, strncasecmp, and strdup can't be used when cross-compiling
 - Temp fix was to change this to AC_COMPILE_IFELSE/AC_LINK_IFELSE - need alternative solution here when cross-compiling

m4/ax_c_pragmas.m4
 - AC_DEFUN([AX_C_PRINTF_STRING_NULL] test uses AC_RUN_IFELSE which can't be used when cross-compiling
 - Temp fix was to change this to AC_COMPILE_IFELSE/AC_LINK_IFELSE - need alternative solution here when cross-compiling

m4/nut_compiler_family.m4
 - AC_DEFUN([NUT_CHECK_COMPILE_FLAG] test uses AC_RUN_IFELSE which can't be used when cross-compiling
 - Temp fix was to change this to AC_COMPILE_IFELSE/AC_LINK_IFELSE - need alternative solution here when cross-compiling

m4/nut_check_libnetsnmp.m4
 - I had lots of problems with the test code checking for defined variables via the Net-SNMP headers but the variables were NOT actually being built into my build (ie; resolved).  I did various objdumps, code searches, etc. to track down the correct #ifdefs/#ifndefs.  I hope this will help someone to rework/modify this .m4 file.

1) Updated usmDESPrivProtocol usage to check for its actual #ifdef from the Net-SNMP code
oid * pProto = usmDESPrivProtocol;
#ifdef NETSNMP_DISABLE_DES
#error "NETSNMP_DISABLE_DES is defined"
#endif
---
2) Updated usmHMAC256SHA384AuthProtocol/usmHMAC384SHA512AuthProtocol usage to check for its actual #ifndef from the Net-SNMP code
oid * pProto = usmHMAC256SHA384AuthProtocol/usmHMAC384SHA512AuthProtocol;
#ifndef HAVE_EVP_SHA384
#error "HAVE_EVP_SHA384 is NOT defined"
#endif
---
3) Updated usmHMAC192SHA256AuthProtocol usage to check for its actual #ifndef from the Net-SNMP code
oid * pProto = usmHMAC192SHA256AuthProtocol;
#ifndef HAVE_EVP_SHA224
#error "HAVE_EVP_SHA224 is NOT defined"
#endif
---
4) Updated usmAES192PrivProtocol/usmAES256PrivProtocol usage to check for its actual #ifndef from the Net-SNMP code
oid * pProto = usmAES192PrivProtocol/usmAES256PrivProtocol;
#ifndef NETSNMP_DRAFT_BLUMENTHAL_AES_04
#error "NETSNMP_DRAFT_BLUMENTHAL_AES_04 is NOT defined"
#endif
---
5) Updated usmHMACMD5AuthProtocol usage to check for its actual #ifdef from the Net-SNMP code
 oid * pProto = usmHMACMD5AuthProtocol;
#ifdef NETSNMP_DISABLE_MD5
#error "NETSNMP_DISABLE_MD5 is defined"
#endif

===

I've talked to @jimklimov about a few of these already from 'nut_compiler_family.m4'

jimklimov, 2022-02-15:
Great thanks for the great details, hope to see this "productized" eventually (both the fixes to NUT in a way that existing build farm scenarios concur, and a new cross-build scenario)... Not sure I'd have time to follow up on this myself soon, though.

jimklimov, 2022-02-15:
(Updated the original post for formatting, so it's more readable in GitHub webUI)

jimklimov, 2022-03-12:
For posterity: trying to retrace the situation, with an Ubuntu impish (21.10) container I had around, I found that https://packages.ubuntu.com/search?keywords=pkg-config-arm-linux-gnueabihf is quite a rare package - available from 18.04 to 21.04. Fetched the deb file for 20.04 and `dpkg -i` did not mind, though.

jimklimov, 2022-03-12:
Also current net-snmp sources (as of today) have some different header definitions compared to what we use (tagged release `v5.9` seems ok in this regard), causing warnings and outright build failures:
````
snmp-ups.c: In function 'nut_snmp_init':
snmp-ups.c:819:46: error: 'ONE_SEC' undeclared (first use in this function)
  819 |         g_snmp_sess.timeout = snmp_timeout * ONE_SEC;
      |                                              ^~~~~~~
snmp-ups.c:819:46: note: each undeclared identifier is reported only once for each function it appears in
snmp-ups.c:901:55: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  901 |                         g_snmp_sess.securityAuthProto = usmHMACMD5AuthProtocol;
      |                                                       ^
snmp-ups.c:908:55: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  908 |                         g_snmp_sess.securityAuthProto = usmHMACSHA1AuthProtocol;
      |                                                       ^
snmp-ups.c:915:55: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  915 |                         g_snmp_sess.securityAuthProto = usmHMAC192SHA256AuthProtocol;
      |                                                       ^
snmp-ups.c:922:55: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  922 |                         g_snmp_sess.securityAuthProto = usmHMAC256SHA384AuthProtocol;
      |                                                       ^
snmp-ups.c:929:55: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  929 |                         g_snmp_sess.securityAuthProto = usmHMAC384SHA512AuthProtocol;
      |                                                       ^
snmp-ups.c:978:55: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  978 |                         g_snmp_sess.securityPrivProto = usmDESPrivProtocol;
      |                                                       ^
snmp-ups.c:985:55: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  985 |                         g_snmp_sess.securityPrivProto = usmAESPrivProtocol;
      |                                                       ^
snmp-ups.c: In function 'nut_snmp_perror':
snmp-ups.c:1532:33: warning: passing argument 1 of 'snmp_sess_error' from incompatible pointer type [-Wincompatible-pointer-types]
 1532 |                 snmp_sess_error(sess, &cliberr, &snmperr, &snmperrstr);
      |                                 ^~~~
      |                                 |
      |                                 struct snmp_session *
In file included from /opt/test/netsnmp/include/net-snmp/library/snmp_client.h:32,
                 from /opt/test/netsnmp/include/net-snmp/varbind_api.h:102,
                 from /opt/test/netsnmp/include/net-snmp/library/snmp_api.h:33,
                 from /opt/test/netsnmp/include/net-snmp/net-snmp-includes.h:25,
                 from snmp-ups.h:81,
                 from snmp-ups.c:38:
/opt/test/netsnmp/include/net-snmp/output_api.h:24:50: note: expected 'struct session_list *' but argument is of type 'struct snmp_session *'
   24 |     void    snmp_sess_error(struct session_list *slp, int *clib_errorno,
      |                             ~~~~~~~~~~~~~~~~~~~~~^~~
````

jimklimov, 2022-03-12:
Explicit `git checkout v5.9.1`, local build and install of net-snmp behaves better (no warnings like above, neither for NUT builds, nor for net-snmp itself including the agent).

PR #1334 (2022-03-12): Fix cross builds
Closes: #1289
closed 2022-03-12 by jimklimov