networkupstools/nut #1302: riello IPG 800 serial. Serial-to-USB cable

angelbarrera92, 2022-02-17:
Hi!

I am having problems configuring my Riello IPG 800 that comes with a serial port. I bought a serial-to-usb cable to connect to my proxmox server.

I am the only one trying to connect a UPS with a serial-to-usb cable? I tried configuring nut with both drivers: riello_ser and riello_usb. None working :(

Any tip to debug the issue?

jimklimov, 2022-02-17:
On one hand, wondering if you are using the (pre-packaged) older NUT
release 2.7.4, which does not benefit from later fixes (and maybe bugs) in
the master branch - and if you can deploy a custom build of NUT. In
particular, there were some riello fixes are in open or recently closed PRs.

On another hand, you would need a serial driver for the serial port device,
probably using "port = /dev/ttyUSB0" or similar config option.

Beside riello, you can try nutdrv_qx as the new common driver for many "Qx"
protocol devices.

Driver binaries can be started with debug args (-DDDDDD) to see better what
fails (might be not protocol mismatch, but e.g. FS permissions to access
the device node - can try running as `-u root` then).

angelbarrera92, 2022-02-19:
Hi @jimklimov 

Ye, you are right, i were trying the pre-packaged version (2.7.4). 
Today i've tested building nut by following this doc: https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu
But i'm facing the same error with all drivers i'm testing:

Configuration file looks like:
`ups.conf` file:
```bash
maxretry = 3
[sai]
  driver = nutdrv_qx
  port = /dev/ttyUSB0
```

Im using the `/dev/ttyUSB0` after checking `dmesg` logs:

```
[    5.062684] usb 1-4: ch341-uart converter now attached to ttyUSB0
```

Testing the riello_ser binary/driver:

```bash
/lib/nut/riello_ser -DDDDDDDDD -a sai -u root
Network UPS Tools - Riello serial driver 0.07 (2.7.4.1)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000	[D3] do_global_args: var='maxretry' val='3'
   0.000011	[D3] main_arg: var='driver' val='riello_ser'
   0.000017	[D3] main_arg: var='port' val='/dev/ttyUSB0'
   0.000024	[D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyUSB0"
   0.000031	[D1] Built-in default or configured user for drivers 'nut' was ignored due to 'root' specified on command line
   0.000034	[D1] debug level is '9'
   0.000801	[D5] send_to_all: SETINFO device.type "ups"
   0.000809	[D2] entering upsdrv_initups()
   0.001161	[D2] set baudrate 9600
   0.001332	[D2] try to detect SENTR
   5.004991	[D2] try to detect GPSER
  10.010020	[D2] set baudrate 1200
  10.010399	[D2] try to detect SENTR
  15.014676	[D2] try to detect GPSER
  20.016940	Can't connect to the UPS on port /dev/ttyUSB0!
```

Trying riello_usb:

```bash
/lib/nut/riello_usb -DDDDDDDDD -a sai -u root
Network UPS Tools - Riello USB driver 0.07 (2.7.4.1)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000	[D3] do_global_args: var='maxretry' val='3'
   0.000013	[D3] main_arg: var='driver' val='riello_usb'
   0.000017	[D3] main_arg: var='port' val='/dev/ttyUSB0'
   0.000022	[D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyUSB0"
   0.000032	[D1] Built-in default or configured user for drivers 'nut' was ignored due to 'root' specified on command line
   0.000034	[D1] debug level is '9'
   0.001089	[D5] send_to_all: SETINFO device.type "ups"
   0.117922	[D3] usb_busses=0x55d8e1ed9e10
   0.117938	[D2] Checking device (152D/0561) (002/002)
   0.118565	[D2] - VendorID: 152d
   0.118572	[D2] - ProductID: 0561
   0.118593	[D2] - Manufacturer: unknown
   0.118597	[D2] - Product: External USB 3.0
   0.118600	[D2] - Serial Number: DD564198838B4
   0.118604	[D2] - Bus: 002
   0.118607	[D2] - Device: 002
[... 167 lines cut ...]
  30.128255	[D3] read: timeout (0)
  30.128274	[D2] qx_process_answer: short reply (input.voltage)
  30.128350	[D3] send: 'Q1'
  31.128929	[D3] read: timeout (0)
  31.128950	[D2] qx_process_answer: short reply (input.voltage)
  31.129034	[D3] send: 'Q1'
  32.129525	[D3] read: timeout (0)
  32.129545	[D2] qx_process_answer: short reply (input.voltage)
  32.129620	[D3] send: 'Q1'
  33.130645	[D3] read: timeout (0)
  33.130662	[D2] qx_process_answer: short reply (input.voltage)
  33.130683	[D3] send: 'Q1'
  34.131705	[D3] read: timeout (0)
  34.131724	[D2] qx_process_answer: short reply (input.voltage)
  34.131810	[D3] send: 'Q1'
  35.132797	[D3] read: timeout (0)
  35.132818	[D2] qx_process_answer: short reply (input.voltage)
  35.132882	[D3] send: 'Q1'
  36.133910	[D3] read: timeout (0)
  36.133925	[D2] qx_process_answer: short reply (input.voltage)
  36.133928	Device not supported!
  36.133948	Device not supported!
```

I don't know if this UPS is supported :(

NilByte, 2022-02-23:
https://www.mail-archive.com/nut-upsdev@alioth-lists.debian.net/msg00124.html

Links to:

https://view.officeapps.live.com/op/view.aspx?src=https%3A%2F%2Fwww.mail-archive.com%2Fnut-upsdev%40alioth-lists.debian.net%2Fmsg00124%2FRiello_UPS_protocol_list_02-2021.xlsx&wdOrigin=BROWSELINK

IPG800 is supposed to be supported.

angelbarrera92, 2022-02-24:
> Seems #498 also touched on (poor) support of Riello IDG devices, with https://alioth-lists.debian.net/pipermail/nut-upsdev/2017-November/007357.html for details - however there apparently they had _some_ data. Some other PRs and issues about it like #530 #957 #1280 imply that for some devices of this name we have support, even if it was in need of improvements (no idea if same HW/protocol behind the name... vendors do mix things sometimes).
> 
> Is the serial-to-usb converter part of the UPS or a separate device? Can you check with something (ideally some COM console of another PC/server/switch/... or even by shorting the RX and TX pins) that it actually works as a serial port? Also, is the serial cable the one supplied with the UPS? Some vendors have special pin-outs, or even additional components inside the cable/connectors, etc. so a standard DB9-DB9 null-modem cable would not work for them; no idea if this can be the case here though.

About the seria-to-usb converter... I bought it in another famous online store (china). I don't think I have nothing at home to test the serial port :( 
I've sent an email to riello support to know what driver to use and so on...

> https://www.mail-archive.com/nut-upsdev@alioth-lists.debian.net/msg00124.html
>
> Links to: [...]

Ye, I know, but seems that there are two different version, the current one ships with a USB port. Mine seems to be an older one with a serial/com port.