networkupstools/nut #1308: MAKELSAN Lion 850VA UPS - help?

jradxl, 2022-02-19:
Anyone any experience of this MAKELSAN Lion 850VA UPS?
http://www.makelsan-ups.com/

# apt install nut-ipmi
# nut-scanner -U
Scanning USB bus.
[nutdev1]
	driver = "nutdrv_atcl_usb"
	port = "auto"
	vendorid = "0001"
	productid = "0000"
	bus = "001"

Certainly the USB IDs are correct
Bus 001 Device 010: ID 0001:0000 Fry's Electronics 

The UPS works on Windows 10, with the supplied UPSSmart v1.5 software, but the Linux version opens it's GUI but does not correct to the UPS.

The detected driver isn't right....
```
/lib/nut/nutdrv_atcl_usb -a nutdev1
Network UPS Tools - 'ATCL FOR UPS' USB driver 1.1 (2.7.4)
Warning: This is an experimental driver.
Some features may not function correctly.

Communications with UPS lost: Can't open USB device, retrying ...
Communications with UPS lost: Can't open USB device, retrying ...
Communications with UPS lost: Can't open USB device, retrying ...
Unable to find ATCL FOR UPS

Things to try:
 - Connect UPS device to USB bus
 - Run this driver as another user (upsdrvctl -u or 'user=...' in ups.conf).
   See upsdrvctl(8) and ups.conf(5).

Fatal error: unusable configuration
```

```
 /lib/nut/nutdrv_qx -a nutdev1 -DDD
  18.375777	Checking device (0001/0000) (001/010)
  18.377071	- VendorID: 0001
  18.377104	- ProductID: 0000
  18.377115	- Manufacturer: unknown
  18.377126	- Product: unknown
  18.377136	- Serial Number: unknown
  18.377165	- Bus: 001
  18.377176	- Device release number: 0100
  18.377188	Trying to match device
  18.377203	Device matches
  18.377243	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
  18.377264	send: Q1
  18.378547	read: error sending control message: Operation not permitted (-5)
  18.378610	qx_process_answer: short reply (input.voltage)

Anyone any suggestions?

```

jimklimov, 2022-02-22:
Unfortunately this extremely random, unique and USB-forum purchased set of IDs (just kidding) matches a large number of unrelated devices. Effectively, it is a non-re-identified chip used for USB wire protocol, not an ID of the actual final product and its vendor because they did not bother to get and write one.

I think ATCL was documented as "one of NUT supported devices" with this ID pair in particular, and comes first alphabetically ;\
Maybe a (IIRC) nutdrv_qx would try the correct protocol for that.

jradxl, 2022-02-23:
Thanks for your suggestions
Using the Windows 10 UPSmart v1.5 software, it's Mega(USB) option that connects
Does that mean anything to you? Perhaps megatec protocol?

On Linux:
I do get a device match with nutdrv_qx, but no further.
Using this command-line
`./nutdrv_qx -a nutdev1 -DDD -x port=auto -x vendorid=0001 -x productid=0000`
It's clearly probing the USB, but nothing gets found
Does the snippet of driver output mean anything?

```
24.779225	Checking device (0001/0000) (001/061)
  24.780549	- VendorID: 0001
  24.780583	- ProductID: 0000
  24.780594	- Manufacturer: unknown
  24.780604	- Product: unknown
  24.780614	- Serial Number: unknown
  24.780629	- Bus: 001
  24.780642	- Device release number: 0100
  24.780655	Trying to match device
  24.780680	Device matches
  24.780722	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
  24.780736	send: Q1
  24.782035	read: error sending control message: Operation not permitted (-5)
  24.782101	qx_process_answer: short reply (input.voltage)
  24.782126	Device not supported!
  24.782137	Device not supported!

```

jimklimov, 2022-03-02:
Here it seems like the driver process had no rights to probe the port. Can
you re-run that test with `-u root` option (and starting as root, eg. via
sudo)?

Also, it seems you are using the older (last released?) NUT version, so if
you could retry with a local build of current master branch, that could be
helpful too!

D13410N3, 2023-09-03:
Hello. Did you make this thing work? I'm planning to buy this one

jradxl, 2023-09-03:
Alas, no. I gave up. I did try to contact Makelsan but received no reply (but I've no idea if my enquiry got though).
As far as I remember it worked fine on ViewPower software. I just use it now without any use of the interface.
Sorry

D13410N3, 2023-09-04:
@jradxl Thanks for your answer.
Anyway, I have no choice - there are no any other UPS in local stores, so...

But I will try anyway

jimklimov, 2023-09-04:
Recently a few improvements for `nutdrv_qx` have landed (into master branch, probably not yet distro packages). Some of those revolved around devices/firmwares that have got the Megatec Qx protocols wrong - assuming statistically that this one also talks Megatec... 

One problem re-discovered and addressed there for some devices (but IIRC not yet holistically for "everything") is about hardware which ends its protocol report lines with a `0x00` NULL byte instead of CR. Many subdrivers consider this a "short read" (after debug-dumping a line with reasonable numbers, which obscured discovery of the problem), because the expected end-of-line was not found in the entirety of the read response.

Wondering if this would be the case here too, or if it just goes supported with existing drivers... 

The permissions problem (`Operation not permitted`) posted originally still puzzles me. However it may be related to seeing effectively no identification from libusb (vendor/product unknown) and so is something about OS, udev, permissions, mismatch of run-time user vs. what udev config or devfs filesystem bits set... or even (not reported) virtualization and USB pass-through involved - which adds the same considerations for host system (that the hypervisor does not grab so much of the device that there's little left accessible for NUT or even whole guest OS to see). An `lsusb` might be helpful a bit to rule out the latter - there were cases where it reported the device details but libusb (used by NUT) did not. Running a driver as `root` might help for permissions mess-up - not encouraged if avoidable, but always an option...