networkupstools/nut #1311: CyberPower OLS1000E stopped working with usbhid-ups driver

CRCinAU, 2022-02-21:
UPS comes up as:

```
Bus 001 Device 002: ID 0764:0601 Cyber Power System, Inc. PR1500LCDRT2U UPS
```

`nut-scanner -U` finds the matching USB IDs:
```
$ nut-scanner -U
Cannot load SNMP library (/usr/lib64/libnetsnmp.so.40.1.0) : /usr/lib64/libnetsnmp.so.40.1.0: undefined symbol: usmDESPrivProtocol. SNMP search disabled.
Neon library not found. XML search disabled.
Scanning USB bus.
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0601"
        bus = "001"
```

Checking the output of `usbhid-ups` throws the following:
```
$ usbhid-ups -DD -a ols1000e -u root
......
   0.028798     Checking device (0764/0601) (001/002)
   0.030554     - VendorID: 0764
   0.030572     - ProductID: 0601
   0.030577     - Manufacturer: unknown
   0.030582     - Product: unknown
   0.030587     - Serial Number: unknown
   0.030592     - Bus: 001
   0.030597     - Device release number: 0200
   0.030602     Trying to match device
   0.030660     Device matches
   0.031930     Unable to get HID descriptor (No such device or address)
   0.031960     HID descriptor length 656
   0.059936     Unable to get Report descriptor: No such device or address
......
```

`dmesg` seems to not show any error when booting with the UPS connected via USB:
```
[    0.780314] usb 1-6: New USB device found, idVendor=0764, idProduct=0601, bcdDevice= 2.00
[    0.780330] usb 1-6: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[    0.780331] usb 1-6: Product: OLS1000E
[    0.780332] usb 1-6: Manufacturer: CyberPower Systems
[    0.810985] hid-generic 0003:0764:0601.0001: hiddev96,hidraw0: USB HID v1.10 Device [CyberPower Systems OLS1000E] on usb-0000:00:14.0-6/input0
```

I plugged the UPS into a different system, then passed that USB port through to a Windows 10 LTSC VM with the CyberPower Business Suite software installed - and it communicated with the UPS fine - including retrieving stats / configuration changes.

This has been working previously - but something has just caused communications to fail with nut, and I can't quite figure out what.

System info:
```
$ cat /etc/fedora-release 
Fedora release 35 (Thirty Five)

$ rpm -qa | grep kernel | sort
kernel-5.16.10-200.fc35.x86_64
kernel-5.16.8-200.fc35.x86_64
kernel-5.16.9-200.fc35.x86_64
kernel-core-5.16.10-200.fc35.x86_64
kernel-core-5.16.8-200.fc35.x86_64
kernel-core-5.16.9-200.fc35.x86_64
kernel-modules-5.16.10-200.fc35.x86_64
kernel-modules-5.16.8-200.fc35.x86_64
kernel-modules-5.16.9-200.fc35.x86_64

$ cat /proc/version 
Linux version 5.16.10-200.fc35.x86_64 (mockbuild@bkernel01.iad2.fedoraproject.org) (gcc (GCC) 11.2.1 20220127 (Red Hat 11.2.1-9), GNU ld version 2.37-10.fc35) #1 SMP PREEMPT Wed Feb 16 13:28:00 UTC 2022

$ rpm -qa | grep nut-
nut-client-2.7.4-41.fc35.x86_64
nut-2.7.4-41.fc35.x86_64
```

I'd be grateful for any suggestions, as I'm kinda out of ideas on what went wrong now :(

jimklimov, 2022-02-22:
Try to tune up `-DDDDDD`ebugging, if you are trying a recent master-branch build we added libusb 0.1/1.0 support, but might have missed a few points about signedness of bytes between two APIs.

jimklimov, 2022-02-22:
Also, it does look very strange that the driver log above did not detect any info about the device beside the vendor/product - which may match the inability to see the rest of protocol too... usually I'd suspect permissions, but here I believe you ran as `root` to no avail. Or not? Should the `$` prompt be trusted as that you ran unprivileged initially? Then the driver won't elevate on its own accord, it can only lose rights it had at startup...

But even then I'd expect more an "Access denied" than "No such device or address", although that may be some mixed-up diagnostics (either in libusb or on NUT side).

Also from the description, it is not quite clear if you are trying with packaged NUT (so 2.7.4 at least 6 years old) or with a custom build of recent master branch state? ("Stopped working" might mean a regression on master compared to old release, so sanity-checking the report for that aspect)

CRCinAU, 2022-02-23:
Hi, sorry - the $ is just me adding manually instead of including my whole PS1 prompt :) All commands were run as root.

libusb packages are at version:
```
$ rpm -qa | grep libusb | sort
libusb-0.1.7-6.fc35.x86_64
libusb1-1.0.25-3.fc35.x86_64
```

Adding more D's:
```
$ usbhid-ups -DDDDDD -a ols1000e -u root                                                                                                       
Network UPS Tools - Generic HID driver 0.41 (2.7.4)                                                                                                           
USB communication driver 0.33                                                                                                                                 
   0.000000     send_to_all: SETINFO driver.parameter.port "auto"                                                                                             
   0.000064     send_to_all: SETINFO driver.parameter.vendorid "0764"                                                                                         
   0.000132     send_to_all: SETINFO driver.parameter.productid "0601"                                                                                        
   0.000159     debug level is '6'                                                                                                                            
   0.001499     upsdrv_initups...
(skipped non-related USB devices)
   0.029402     Checking device (0764/0601) (001/002)
   0.031113     - VendorID: 0764
   0.031136     - ProductID: 0601
   0.031142     - Manufacturer: unknown
   0.031149     - Product: unknown
   0.031155     - Serial Number: unknown
   0.031161     - Bus: 001
   0.031168     - Device release number: 0200
   0.031174     Trying to match device
   0.031247     Device matches
   0.031275     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0) 
   0.032287     Unable to get HID descriptor (No such device or address)
   0.032311     HID descriptor length (method 1) -1
   0.032320     i=0, extra[i]=09, extra[i+1]=21
   0.032330     HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 90 02
   0.032337     HID descriptor length (method 2) 656
   0.032346     HID descriptor length 656
   0.060111     Unable to get Report descriptor: No such device or address
(snipped out more unrelated devices)
   0.157529     No appropriate HID device found
   0.157542     No matching HID UPS found
```

Looking at the udev rules, I see:
```
lib/udev/rules.d/62-nut-usbups.rules:ATTR{idVendor}=="0764", ATTR{idProduct}=="0005", MODE="664", GROUP="dialout"
lib/udev/rules.d/62-nut-usbups.rules:ATTR{idVendor}=="0764", ATTR{idProduct}=="0501", MODE="664", GROUP="dialout"
lib/udev/rules.d/62-nut-usbups.rules:ATTR{idVendor}=="0764", ATTR{idProduct}=="0601", MODE="664", GROUP="dialout"
```

The groups seem to match the nut user:
```
$ groups nut
nut : nut tty dialout
```

It looks like its only build against libusb:
```
$ ldd /usr/sbin/usbhid-ups 
        linux-vdso.so.1 (0x00007fff368e7000)
        libusb-0.1.so.4 => /lib64/libusb-0.1.so.4 (0x00007f6922193000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f6921f89000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f69221e3000)
```

However I'm not sure what package version that would be.

I'm not sure what else could be useful - so very much open to suggestions...

CRCinAU, 2022-02-23:
Further investigation...... I can see the device at: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6

```
$ hexdump -C descriptors 
00000000  12 01 10 01 00 00 00 08  64 07 01 06 00 02 03 01  |........d.......|
00000010  00 01 09 02 29 00 01 01  00 e0 19 09 04 00 00 02  |....)...........|
00000020  03 00 00 00 09 21 10 01  21 01 22 90 02 07 05 81  |.....!..!.".....|
00000030  03 08 00 0a 07 05 02 03  08 00 14                 |...........|
0000003b
```

In that directory:
```
total 0
drwxr-xr-x 5 root root     0 Feb 23 21:01 1-6:1.0
-rw-r--r-- 1 root root  4096 Feb 23 21:04 authorized
-rw-r--r-- 1 root root  4096 Feb 23 21:04 avoid_reset_quirk
-r--r--r-- 1 root root  4096 Feb 23 21:01 bcdDevice
-rw-r--r-- 1 root root  4096 Feb 23 21:04 bConfigurationValue
-r--r--r-- 1 root root  4096 Feb 23 21:01 bDeviceClass
-r--r--r-- 1 root root  4096 Feb 23 21:04 bDeviceProtocol
-r--r--r-- 1 root root  4096 Feb 23 21:04 bDeviceSubClass
-r--r--r-- 1 root root  4096 Feb 23 21:04 bmAttributes
-r--r--r-- 1 root root  4096 Feb 23 21:04 bMaxPacketSize0
-r--r--r-- 1 root root  4096 Feb 23 21:04 bMaxPower
-r--r--r-- 1 root root  4096 Feb 23 21:04 bNumConfigurations
-r--r--r-- 1 root root  4096 Feb 23 21:04 bNumInterfaces
-r--r--r-- 1 root root  4096 Feb 23 21:02 busnum
-r--r--r-- 1 root root  4096 Feb 23 21:04 configuration
-r--r--r-- 1 root root 65553 Feb 23 21:01 descriptors
-r--r--r-- 1 root root  4096 Feb 23 21:04 dev
-r--r--r-- 1 root root  4096 Feb 23 21:02 devnum
-r--r--r-- 1 root root  4096 Feb 23 21:04 devpath
lrwxrwxrwx 1 root root     0 Feb 23 21:04 driver -> ../../../../../bus/usb/drivers/usb
drwxr-xr-x 3 root root     0 Feb 23 21:03 ep_00
lrwxrwxrwx 1 root root     0 Feb 23 21:04 firmware_node -> ../../../../LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:4c/device:4d/device:53
-r--r--r-- 1 root root  4096 Feb 23 21:01 idProduct
-r--r--r-- 1 root root  4096 Feb 23 21:01 idVendor
-r--r--r-- 1 root root  4096 Feb 23 21:04 ltm_capable
-r--r--r-- 1 root root  4096 Feb 23 21:01 manufacturer
-r--r--r-- 1 root root  4096 Feb 23 21:04 maxchild
lrwxrwxrwx 1 root root     0 Feb 23 21:04 port -> ../1-0:1.0/usb1-port6
drwxr-xr-x 2 root root     0 Feb 23 21:03 power
-r--r--r-- 1 root root  4096 Feb 23 21:01 product
-r--r--r-- 1 root root  4096 Feb 23 21:04 quirks
-r--r--r-- 1 root root  4096 Feb 23 21:04 removable
--w------- 1 root root  4096 Feb 23 21:04 remove
-r--r--r-- 1 root root  4096 Feb 23 21:04 rx_lanes
-r--r--r-- 1 root root  4096 Feb 23 21:02 speed
lrwxrwxrwx 1 root root     0 Feb 23 21:01 subsystem -> ../../../../../bus/usb
-r--r--r-- 1 root root  4096 Feb 23 21:04 tx_lanes
-rw-r--r-- 1 root root  4096 Feb 23 21:01 uevent
-r--r--r-- 1 root root  4096 Feb 23 21:04 urbnum
-r--r--r-- 1 root root  4096 Feb 23 21:04 version
```

The values for those:
```
$ grep -i . *
grep: 1-6:1.0: Is a directory
authorized:1
avoid_reset_quirk:0
bcdDevice:0200
bConfigurationValue:1
bDeviceClass:00
bDeviceProtocol:00
bDeviceSubClass:00
bmAttributes:e0
bMaxPacketSize0:8
bMaxPower:50mA
bNumConfigurations:1
bNumInterfaces: 1
busnum:1
grep: descriptors: binary file matches
dev:189:5
devnum:6
devpath:6
grep: driver: Is a directory
grep: ep_00: Is a directory
grep: firmware_node: Is a directory
idProduct:0601
idVendor:0764
ltm_capable:no
manufacturer:CyberPower Systems
maxchild:0
grep: port: Is a directory
grep: power: Is a directory
product:OLS1000E
quirks:0x0
removable:unknown
grep: remove: Permission denied
rx_lanes:1
speed:1.5
grep: subsystem: Is a directory
tx_lanes:1
uevent:MAJOR=189
uevent:MINOR=5
uevent:DEVNAME=bus/usb/001/006
uevent:DEVTYPE=usb_device
uevent:DRIVER=usb
uevent:PRODUCT=764/601/200
uevent:TYPE=0/0/0
uevent:BUSNUM=001
uevent:DEVNUM=006
urbnum:15
version: 1.10
```

CRCinAU, 2022-02-23:
Well, you're not going to believe this....

I reformatted the system that this UPS is hooked up to, and redid it with a UEFI bios instead of legacy, a different SSD, and a fresh install of Fedora 35... It all works fine.

It looks like its now at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2

A couple of values have changed, but I'm not sure if that's just the way it is:
```
# grep -i . *
grep: 1-2:1.0: Is a directory
authorized:1
avoid_reset_quirk:0
bcdDevice:0200
bConfigurationValue:1
bDeviceClass:00
bDeviceProtocol:00
bDeviceSubClass:00
bmAttributes:e0
bMaxPacketSize0:8
bMaxPower:50mA
bNumConfigurations:1
bNumInterfaces: 1
busnum:1
grep: descriptors: binary file matches
dev:189:2
devnum:3
devpath:2
grep: driver: Is a directory
grep: ep_00: Is a directory
grep: firmware_node: Is a directory
idProduct:0601
idVendor:0764
ltm_capable:no
manufacturer:CyberPower Systems
maxchild:0
grep: port: Is a directory
grep: power: Is a directory
product:OLS1000E
quirks:0x0
removable:unknown
grep: remove: Permission denied
rx_lanes:1
speed:1.5
grep: subsystem: Is a directory
tx_lanes:1
uevent:MAJOR=189
uevent:MINOR=2
uevent:DEVNAME=bus/usb/001/003
uevent:DEVTYPE=usb_device
uevent:DRIVER=usb
uevent:PRODUCT=764/601/200
uevent:TYPE=0/0/0
uevent:BUSNUM=001
uevent:DEVNUM=003
urbnum:2143
version: 1.10
```

CRCinAU, 2022-02-23:
Now when I run `usbhid-ups`:

```
# usbhid-ups -D -a ols1000e -u root                                                                                                                                                                                           
Network UPS Tools - Generic HID driver 0.41 (2.7.4)                                                                                                                                                                                           
USB communication driver 0.33                                                                                                                                                                                                                 
   0.000000     debug level is '1'                                                                                                                                                                                                            
   0.001269     upsdrv_initups...                                                                                                                                                                                                             
   0.048984     Using subdriver: CyberPower HID 0.4                                                                                                                                                                                           
   0.049022     67 HID objects found                                                                                                                                                                                                          
   0.049857     Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1                                                                                                                                  
   0.050783     Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2                                                                                                                             
   0.051637     Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 3                                                                                                                           
   0.052658     Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x04, Offset: 0, Size: 8, Value: 4                                                                                                                          
   0.053704     Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 1                                                                                                                              
   0.054615     Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x06, Offset: 0, Size: 8, Value: 2                                                                                                                              
   0.055540     Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 100                                                                                                                          
   0.055576     Path: UPS.PowerSummary.CapacityGranularity1, Type: Feature, ReportID: 0x07, Offset: 8, Size: 8, Value: 5                                                                                                                      
   0.055594     Path: UPS.PowerSummary.CapacityGranularity2, Type: Feature, ReportID: 0x07, Offset: 16, Size: 8, Value: 10                                                                                                                    
   0.055617     Path: UPS.PowerSummary.WarningCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 24, Size: 8, Value: 20                                                                                                                    
   0.055636     Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x07, Offset: 32, Size: 8, Value: 100                                                                                                                     
   0.055655     Path: UPS.PowerSummary.RemainingCapacityLimit, Type: Feature, ReportID: 0x07, Offset: 40, Size: 8, Value: 10                                                                                                                  
   0.056478     Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x08, Offset: 0, Size: 8, Value: 100                                                                                                                         
   0.056503     Path: UPS.PowerSummary.RemainingCapacity, Type: Feature, ReportID: 0x08, Offset: 0, Size: 8, Value: 100                                                                                                                       
   0.056527     Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x08, Offset: 8, Size: 16, Value: 11100                                                                                                                         
   0.056546     Path: UPS.PowerSummary.RunTimeToEmpty, Type: Feature, ReportID: 0x08, Offset: 8, Size: 16, Value: 11100                                                                                                          [... cut ...]
   4.094449     Quick update...
   6.093064     upsdrv_updateinfo...
   6.094305     Got 6 HID objects...
   6.094366     Quick update...
   8.095353     upsdrv_updateinfo...
   8.102366     Got 3 HID objects...
   8.102456     Quick update...
  10.097610     upsdrv_updateinfo...
  10.102434     Got 6 HID objects...
  10.102502     Quick update...
  12.099876     upsdrv_updateinfo...
  12.102433     Got 6 HID objects...
  12.102503     Quick update...
```

I really don't understand it - but it works now :\