networkupstools/nut #1312: Salicru Twin Pro 2

drunkly, 2022-02-22:
Hi

Did someone get this to work, I have tried the deiban package with apt and also installing from binaries but no luck:

I have this UPS installed in a Proxmox machine so I don't know if this is related. Here are some info I gathered:

`root@PROXMOX:/usr/local/ups/bin# ./nut-scanner
Scanning USB bus.
No start IP, skipping NUT bus (old connect method)
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "2E66"
        productid = "0201"
        product = "SLC TWIN PRO2<=3KVA"
        serial = "232019D70264"
        vendor = "Salicru"
        bus = "002"
`

`root@PROXMOX:/usr/local/ups/bin# ./usbhid-ups -DD -a salicru -u root
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4698-g381a81fc)
USB communication driver (libusb 0.1) 0.42
   0.000000     [D1] Built-in default or configured user for drivers 'nut' was ignored due to 'root' specified on command line
   0.000017     [D1] debug level is '2'
   0.001716     [D2] Initializing an USB-connected UPS with library libusb-0.1 (or compat) (NUT subdriver name='USB communication driver (libusb 0.1)' ver='0.42')
   0.001732     [D1] upsdrv_initups (non-SHUT)...
   0.139629     [D2] Checking device (8087/8002) (003/002)
   0.207815     [D2] - VendorID: 8087
   0.207844     [D2] - ProductID: 8002
   0.207847     [D2] - Manufacturer: unknown
   0.207850     [D2] - Product: unknown
   0.207856     [D2] - Serial Number: unknown
   0.207887     [D2] - Bus: 003
   0.207894     [D2] - Device: 002
   0.207902     [D2] - Device release number: 0005
   0.207911     [D2] Trying to match device
   0.207919     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.207932     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.207940     [D2] Device does not match - skipping
   0.207963     [D2] Checking device (1D6B/0002) (003/001)
   0.208072     [D2] - VendorID: 1d6b
   0.208084     [D2] - ProductID: 0002
   0.208098     [D2] - Manufacturer: Linux 5.4.162-1-pve ehci_hcd
   0.208105     [D2] - Product: EHCI Host Controller
   0.208113     [D2] - Serial Number: 0000:00:1d.0
   0.208120     [D2] - Bus: 003
   0.208127     [D2] - Device: 001
   0.208135     [D2] - Device release number: 0504
   0.208142     [D2] Trying to match device
   0.208149     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.208159     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.208166     [D2] Device does not match - skipping
   0.208179     [D2] Checking device (1D6B/0003) (004/001)
   0.208372     [D2] - VendorID: 1d6b
   0.208383     [D2] - ProductID: 0003
   0.208388     [D2] - Manufacturer: Linux 5.4.162-1-pve xhci-hcd
   0.208395     [D2] - Product: xHCI Host Controller
   0.208402     [D2] - Serial Number: 0000:00:14.0
   0.208405     [D2] - Bus: 004
   0.208408     [D2] - Device: 001
   0.208413     [D2] - Device release number: 0504
   0.208418     [D2] Trying to match device
   0.208423     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.208432     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.208437     [D2] Device does not match - skipping
   0.208538     [D2] Checking device (2E66/0201) (002/010)
   0.210053     [D2] - VendorID: 2e66
   0.210064     [D2] - ProductID: 0201
   0.210071     [D2] - Manufacturer: Salicru
[... 220 lines cut ...]
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

.......
....`

`root@PROXMOX:/usr/local/ups/bin# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
`

With debian package it just shows the message: "Driver failed to start". I assume it is due the debian package verison is outdated (2.7.4) so I tried to install from binaries with the otpion "--with-usb".

I  guess my main problem is related to the fact that driver can't find HID descriptor but I don't know how to continue from this point.

Can someone provide some tip please?

jimklimov, 2022-02-22:
> Error: Connection failure: Connection refused ***@***.***:/usr/local/ups/bin#

Permissions issues could be about udev or upower setup, to make the devfs
node accessible to nut daemon account (driver).

There was a salicru-specific subdriver merged recebtly yo master branch,
but there was no NUT release for a while so you'd have to build a custom
replacement.

At least for a quick test to see if the new usbhid-ups dtiver recognizes
the device, it would suffice to just build the git checkout and run the
driver in-place (perhaps with debug and `-u root` options), without
intruding into packaged set-up. If the bew driver works, then fuss about
building with same configure script options as in OS packaging, so `sudo
make install` would replace that.

jimklimov, 2022-02-22:
Sort of "is like this" :) Default `configure` places files under /usr/local/ups/... tree to not conflict with packages; including an /usr/local/ups/etc/ for config. A non-default build would ask for sysconfigroot=/etc (to use same config as packages, conveniently for experiments) or even more options to replace the packaged files (distro-dependent, everyone places them differently).

For the driver start-up, can you add more `-DDDDDD`ebugs? :)

Possibly you would have to spell out `subdriver = salicru` option in ups.conf (or via `-x` on driver command line while testing).

drunkly, 2022-02-23:
Hi, first of all hanks for spend your time on me.

I have tried with the subdriver with the next settings:
```
[salicru]
driver = usbhid-ups
port = auto
vendorid = 2E66
productid = 0201
product = "SLC TWIN PRO2<=3KVA"
serial = 232019D70264
vendor = Salicru
bus = 002
subdriver = salicru
```
```
root@PROXMOX:~# /usr/local/ups/bin/usbhid-ups -a salicru -u root
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4698-g381a81fc)
USB communication driver (libusb 0.1) 0.42

Fatal error: 'subdriver' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.

```

Also with -x flag:

```
root@PROXMOX:~# /usr/local/ups/bin/usbhid-ups -a salicru -u root -x salicru
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4698-g381a81fc)
USB communication driver (libusb 0.1) 0.42

Fatal error: 'subdriver' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
```

Then I remvoed to subdriver setting and run it with more debug flags:
```
USB communication driver (libusb 0.1) 0.42
   0.000000     [D1] debug level is '6'
   0.001415     [D5] send_to_all: SETINFO device.type "ups"
   0.001434     [D2] Initializing an USB-connected UPS with library libusb-0.1 (or compat) (NUT subdriver name='USB communication driver (libusb 0.1)' ver='0.42')
   0.001443     [D1] upsdrv_initups (non-SHUT)...
   0.145248     [D3] usb_busses=0x55d67ff598b0
   0.145300     [D2] Checking device (8087/8002) (004/002)
   0.213410     [D2] - VendorID: 8087
   0.213442     [D2] - ProductID: 8002
   0.213463     [D2] - Manufacturer: unknown
   0.213478     [D2] - Product: unknown
   0.213487     [D2] - Serial Number: unknown
   0.213494     [D2] - Bus: 004
   0.213502     [D2] - Device: 002
   0.213510     [D2] - Device release number: 0005
   0.213517     [D2] Trying to match device
   0.213525     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.213539     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.213547     [D2] Device does not match - skipping
   0.213571     [D2] Checking device (1D6B/0002) (004/001)
   0.213716     [D2] - VendorID: 1d6b
   0.213732     [D2] - ProductID: 0002
   0.213739     [D2] - Manufacturer: Linux 5.4.166-1-pve ehci_hcd
   0.213745     [D2] - Product: EHCI Host Controller
   0.213752     [D2] - Serial Number: 0000:00:1d.0
   0.213759     [D2] - Bus: 004
   0.213766     [D2] - Device: 001
   0.213773     [D2] - Device release number: 0504
[... 87 lines cut ...]
   0.289211     [D2] Trying to match device
   0.289218     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.289231     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.289239     [D2] Device does not match - skipping
   0.289255     [D2] Checking device (1D6B/0002) (001/001)
   0.289383     [D2] - VendorID: 1d6b
   0.289390     [D2] - ProductID: 0002
   0.289401     [D2] - Manufacturer: Linux 5.4.166-1-pve ehci_hcd
   0.289408     [D2] - Product: EHCI Host Controller
   0.289421     [D2] - Serial Number: 0000:00:1a.0
   0.289430     [D2] - Bus: 001
   0.289438     [D2] - Device: 001
   0.289447     [D2] - Device release number: 0504
   0.289456     [D2] Trying to match device
   0.289463     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.289474     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.289483     [D2] Device does not match - skipping
   0.289492     [D2] libusb0: No appropriate HID device found
   0.289502     No matching HID UPS found

```

I guess the error about subdriver is related to a mistake on my side running the configure/make install so I will re-read all the documentation again to see if I'm missing some path/flag.

Please let me know if you have a more ideas I can try.

drunkly, 2022-02-23:
I have read the help of usbhidups and runned again with right sintax but I'm getting the same error.

```
root@PROXMOX:~# /usr/local/ups/bin/usbhid-ups -a salicru -u root -x subdriver=salicru
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4698-g381a81fc)
USB communication driver (libusb 0.1) 0.42

Fatal error: 'subdriver' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
```

It is interesting that sub driver is not on the help message:

```
root@PROXMOX:~# /usr/local/ups/bin/usbhid-ups -h
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4698-g381a81fc)
USB communication driver (libusb 0.1) 0.42

usage: usbhid-ups (-a <id>|-s <id>) [OPTIONS]
  -a <id>        - autoconfig using ups.conf section <id>
                 - note: -x after -a overrides ups.conf settings

  -s <id>        - configure directly from cmd line arguments
                 - note: must specify all driver parameters with successive -x
                 - note: at least 'port' variable should be set
                 - note: to explore the current values on a device from an
                   unprivileged user account (with sufficient media access in
                   the OS - e.g. to query networked devices), you can specify
                   '-d 1' argument and `export NUT_STATEPATH=/tmp` beforehand

  -V             - print version, then exit
  -L             - print parseable list of driver variables
  -D             - raise debugging level (and stay foreground by default)
  -d <count>     - dump data to stdout after 'count' updates loop and exit
  -F             - stay foregrounded even if no debugging is enabled
  -B             - stay backgrounded even if debugging is bumped
  -q             - raise log level threshold
  -h             - display this help
  -k             - force shutdown
  -i <int>       - poll interval
  -r <dir>       - chroot to <dir>
  -u <user>      - switch to <user> (if started as root)
  -g <group>     - set pipe access to <group> (if started as root)
  -x <var>=<val> - set driver variable <var> to <val>
                 - example: -x cable=940-0095B

Acceptable values for -x or ups.conf in this driver:

Set low battery level, in % (default=30) : -x lowbatt=<value>
Set shutdown delay, in seconds (default=20) : -x offdelay=<value>
Set startup delay, in seconds (default=30) : -x ondelay=<value>
Set polling frequency, in seconds, to reduce data flow (default=30) : -x pollfreq=<value>
Don't use interrupt pipe, only use polling : -x pollonly
Regular expression to match UPS Manufacturer string : -x vendor=<value>
Regular expression to match UPS Product string : -x product=<value>
Regular expression to match UPS Serial number : -x serial=<value>
Regular expression to match UPS Manufacturer numerical ID (4 digits hexadecimal) : -x vendorid=<value>
Regular expression to match UPS Product numerical ID (4 digits hexadecimal) : -x productid=<value>
Regular expression to match USB bus name : -x bus=<value>
Regular expression to match USB device name : -x device=<value>
Force redundant call to usb_set_altinterface() (value=bAlternateSetting; default=0) : -x usb_set_altinterface=<value>
  Diagnostic matching of unsupported UPS : -x explore
Activate tweak for buggy APC Back-UPS firmware : -x maxreport
Don't use polling, only use interrupt pipe : -x interruptonly
Number of bytes to read from interrupt pipe : -x interruptsize=<value>
```

That make me think I'm not running the last version of nut so I will try to remove everything and re-instlal it again.

drunkly, 2022-02-23:
Added some debug lines to check the problem was related to subdriver:

```
  0.216510     [D2] Checking device (2E66/0201) (002/004)
   0.218023     [D2] - VendorID: 2e66
   0.218032     [D2] - ProductID: 0201
   0.218038     [D2] - Manufacturer: Salicru
   0.218041     [D2] - Product: SLC TWIN PRO2<=3KVA
   0.218046     [D2] - Serial Number: 232019D70264
   0.218051     [D2] - Bus: 002
   0.218056     [D2] - Device: 004
   0.218061     [D2] - Device release number: 0129
   0.218067     [D2] Trying to match device
   0.218072     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.218078     [D2] (non-SHUT mode): trying EXPLORE HID 0.2 subdriver...
   0.218084     [D2] trying EXPLORE HID 0.2 subdriver failed
   0.218089     [D2] (non-SHUT mode): trying MGE HID 1.45 subdriver...
   0.218095     [D2]  463 compare vendor id with 2e66
   0.218100     [D2]  463 compare vendor id with 2e66
   0.218104     [D2]  47c compare vendor id with 2e66
   0.218108     [D2]  592 compare vendor id with 2e66
   0.218118     [D2]  3f0 compare vendor id with 2e66
   0.218138     [D2]  3f0 compare vendor id with 2e66
   0.218146     [D2]  3f0 compare vendor id with 2e66
   0.218152     [D2]  3f0 compare vendor id with 2e66
   0.218160     [D2] 2b2d compare vendor id with 2e66
   0.218168     [D2]  6da compare vendor id with 2e66
   0.218176     [D2]  4b3 compare vendor id with 2e66
   0.218183     [D2] trying MGE HID 1.45 subdriver failed
   0.218191     [D2] (non-SHUT mode): trying APC HID 0.98 subdriver...
   0.218199     [D2]  51d compare vendor id with 2e66
   0.218206     [D2]  51d compare vendor id with 2e66
   0.218214     [D2]  51d compare vendor id with 2e66
   0.218221     [D2] trying APC HID 0.98 subdriver failed
   0.218228     [D2] (non-SHUT mode): trying Arduino HID 0.2 subdriver...
   0.218235     [D2] 2341 compare vendor id with 2e66
   0.218240     [D2] 2341 compare vendor id with 2e66
   0.218246     [D2] 2a03 compare vendor id with 2e66
   0.218251     [D2] 2a03 compare vendor id with 2e66
   0.218256     [D2] 2a03 compare vendor id with 2e66
   0.218261     [D2] 2a03 compare vendor id with 2e66
   0.218266     [D2] trying Arduino HID 0.2 subdriver failed
   0.218271     [D2] (non-SHUT mode): trying Belkin/Liebert HID 0.18 subdriver...
   0.218277     [D2]  50d compare vendor id with 2e66
   0.218281     [D2]  50d compare vendor id with 2e66
   0.218286     [D2]  50d compare vendor id with 2e66
   0.218291     [D2]  50d compare vendor id with 2e66
   0.218295     [D2]  50d compare vendor id with 2e66
   0.218299     [D2]  50d compare vendor id with 2e66
   0.218304     [D2]  50d compare vendor id with 2e66
   0.218309     [D2]  50d compare vendor id with 2e66
   0.218314     [D2]  50d compare vendor id with 2e66
   0.218319     [D2]  50d compare vendor id with 2e66
   0.218324     [D2] 10af compare vendor id with 2e66
   0.218329     [D2] 10af compare vendor id with 2e66
   0.218334     [D2] 10af compare vendor id with 2e66
   0.218339     [D2] 10af compare vendor id with 2e66
   0.218343     [D2] trying Belkin/Liebert HID 0.18 subdriver failed
   0.218347     [D2] (non-SHUT mode): trying CyberPower HID 0.6 subdriver...
   0.218352     [D2]  764 compare vendor id with 2e66
   0.218356     [D2]  764 compare vendor id with 2e66
   0.218361     [D2]  764 compare vendor id with 2e66
   0.218366     [D2] trying CyberPower HID 0.6 subdriver failed
   0.218371     [D2] (non-SHUT mode): trying Delta UPS HID 0.6 subdriver...
   0.218375     [D2]  5dd compare vendor id with 2e66
   0.218380     [D2] trying Delta UPS HID 0.6 subdriver failed
   0.218384     [D2] (non-SHUT mode): trying Ever HID 0.1 subdriver...
   0.218390     [D2]  483 compare vendor id with 2e66
   0.218395     [D2] 2e51 compare vendor id with 2e66
   0.218400     [D2] trying Ever HID 0.1 subdriver failed
[... 18 lines cut ...]
   0.218517     [D2]  d9f compare vendor id with 2e66
   0.218524     [D2]  d9f compare vendor id with 2e66
   0.218530     [D2]  d9f compare vendor id with 2e66
   0.218536     [D2]  d9f compare vendor id with 2e66
   0.218542     [D2] trying PowerCOM HID 0.6 subdriver failed
   0.218548     [D2] (non-SHUT mode): trying Powervar HID 0.20 subdriver...
   0.218555     [D2] 4234 compare vendor id with 2e66
   0.218561     [D2] trying Powervar HID 0.20 subdriver failed
   0.218567     [D2] (non-SHUT mode): trying Salicru HID 0.3 subdriver...
   0.218574     [D2] 2e66 compare vendor id with 2e66
   0.218580     [D2] claim status: 2  for salicru subdriver failed
   0.218587     [D3] match_function_regex: matching a device...
   0.218702     [D2] Device matches
   0.218729     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.218888     [D3] HID descriptor, method 1: (9 bytes) => 09 21 11 01 00 01 22 9c 05
   0.218895     [D2] Warning: HID descriptor, method 1 failed
   0.218898     [D3] HID descriptor length (method 1) -100
   0.218904     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.218911     [D3] HID descriptor, method 2: (9 bytes) => 09 21 11 01 00 01 22 9c 05
   0.218917     [D2] Warning: HID descriptor, method 2 failed
   0.218922     [D3] HID descriptor length (method 2) -100
   0.218928     [D2] Unable to retrieve any HID descriptor
```

In my eyes loos like a problem with lisusb0.1/libusb1.0 but knoweledges are not arriving so far to be sure. I have tried to force the usb lib 1.0 but I had no luck so if can give me some advice about how to continue from this point would be great.

jimklimov, 2022-02-24:
Sorry, my bad, dealt too much with Qx drivers lately, and an explicit keyword to select a `subdriver` is their specialty. The current usbhid-ups driver should just pick what it deems a best fit.

drunkly, 2022-02-25:
Don't have to aplogizes. We are on the same boat :)

Looking deeply I think the subdriver is working but there is a problem related to HID descriptor

```
HID descriptor, method N: (9 bytes) => 09 21 11 01 00 01 22 9c 05
HID descriptor length (method N2) -100
````

On lines
https://github.com/networkupstools/nut/blob/master/drivers/libusb0.c#L383
https://github.com/networkupstools/nut/blob/master/drivers/libusb0.c#L415

I will read some book about HID descriptors to learn how they works and get why are not working and publish the results if get some results.

drunkly, 2022-02-27:
I could get somehow more info about HID descriptor and looks like is reperting well the data but can't get to root of the issue.

I saw there is some kind of validation here to prevent issues with usb. Could you please explain why is doing this or how is working or where can I get more information about this please?:

rdlen2 = p[7] | (p[8] << 8);
https://github.com/networkupstools/nut/blob/master/drivers/libusb0.c#L383#410

Here are the info I got until now in case is for someone useful:

```
root@PROXMOX:~/hidrd# usbhid-dump -d 2E66:0201 | grep -v : | xxd -r -p | hidrd-convert -o spec
Usage Page (Power Device),                                          ; Power device (84h, power page)
Usage (04h),
Collection (Application),
    Usage (10h),
    Collection (Physical),
        Usage (12h),
        Collection (Physical),
            Usage (13h),
            Report ID (32),
            Report Size (8),
            Report Count (1),
            Logical Minimum (0),
            Logical Maximum (255),
            Unit,
            Unit Exponent (0),
            Feature (Constant, Variable),
            Usage Page (FFFFh),                                     ; FFFFh, vendor-defined
            Usage (90h),
            Report ID (33),
            Feature (Variable, Volatile),
            Usage Page (Power Device),                              ; Power device (84h, power page)
            Usage (02h),
            Collection (Logical),
                Usage (60h),
                Report ID (39),
                Report Size (8),
                Logical Maximum (255),
                Unit,
                Feature (Constant, Variable, Volatile),
                Usage (60h),
                Input (Constant, Variable, Bit7),
                Usage Page (FFFFh),                                 ; FFFFh, vendor-defined
                Usage (74h),
                Feature (Constant, Variable, Volatile),
                Usage (74h),
                Input (Constant, Variable, Bit7),
            End Collection,
            Usage Page (Power Device),                              ; Power device (84h, power page)
            Usage (58h),
            Report ID (36),
            Feature (Variable, Volatile),
            Usage Page (FFFFh),                                     ; FFFFh, vendor-defined
            Usage (E3h),
            Report ID (227),
            Feature (Variable, Volatile),
            Usage (45h),
            Report ID (37),
            Report Size (32),
            Logical Maximum (2147483647),
            Unit (Seconds),
            Feature (Variable, Volatile),
        End Collection,
        Usage Page (Power Device),                                  ; Power device (84h, power page)
        Usage (11h),
        Report ID (32),
        Report Size (8),
        Logical Maximum (255),
        Unit,
        Feature (Constant, Variable),
[... 585 lines cut ...]
        Usage (F0h),
        Report ID (240),
        Report Size (8),
        Report Count (1),
        Logical Minimum (0),
        Logical Maximum (255),
        Unit,
        Unit Exponent (0),
        Feature (Variable, Volatile),
        Usage (F1h),
        Report ID (241),
        Report Count (4),
        Feature (Variable, Volatile),
        Usage (FFh),
        Report ID (255),
        Report Size (8),
        Report Count (1),
        Logical Minimum (0),
        Logical Maximum (255),
        Unit,
        Unit Exponent (0),
        Feature (Variable, Volatile),
    End Collection,
End Collection
```

jimklimov, 2022-02-28:
Wondering if this is (one of) the points @nbriggs recommended to check recently, where I tried to cater to the two generations of libusb APIs and got it handling maths with signed/unsigned char buffers, so rdlen might come out `-87` or some such.

I was busy with other fixes a d did not get around to looking at this yet...

nbriggs, 2022-02-28:
Yes, I think this is the same problem as #1261.
```
   0.213616     [D3] HID descriptor, method 2: (9 bytes) => 09 21 11 01 00 01 22 9c 05
```
My sense is that if the high bit is set in either of the last two bytes (above, 9c 05) then the computed descriptor length will be wrong.  I would expect this to affect all USB connected devices using the HID code with libusb 0.1, so this would be a good thing to fix ASAP.    I'll have a look at this again in a few hours and perhaps propose a fix.

nbriggs, 2022-02-28:
A minimal fix would be:
```
diff --git a/drivers/libusb0.c b/drivers/libusb0.c
index adf098ba..e7f16c90 100644
--- a/drivers/libusb0.c
+++ b/drivers/libusb0.c
@@ -377,7 +377,7 @@ static int libusb_open(usb_dev_handle **udevp,
 
                                upsdebug_hex(3, "HID descriptor, method 1", buf, 9);
 
-                               rdlen1 = buf[7] | (buf[8] << 8);
+                               rdlen1 = (uint8_t)buf[7] | ((uint8_t)buf[8] << 8);
                        }
 
                        if (rdlen1 < -1) {
```
so that the code is being explicit about the interpretation of the bytes it is getting back from the call.
@drunkly are you in a position to try this and let us know if it works better for you?

drunkly, 2022-02-28:
Hi. Yes. I will try it and provide a feedback tomorrow early when I arrive to the office. Thanks for your efforts.

drunkly, 2022-03-01:
I updated rdlen1 and test and after that did the same change for rdlen2. This are the resutls:

``` 
root@PROXMOX:~/nut# sudo ./drivers/usbhid-ups -DDDDDD -u root -x productid=0201 -a salicru
.....
   0.209795     [D2] Device does not match - skipping
   0.209905     [D2] Checking device (2E66/0201) (002/005)
   0.211532     [D2] - VendorID: 2e66
   0.211547     [D2] - ProductID: 0201
   0.211559     [D2] - Manufacturer: Salicru
   0.211566     [D2] - Product: SLC TWIN PRO2<=3KVA
   0.211574     [D2] - Serial Number: 232019D70264
   0.211582     [D2] - Bus: 002
   0.211589     [D2] - Device: 005
   0.211595     [D2] - Device release number: 0129
   0.211602     [D2] Trying to match device
   0.211610     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.211619     [D2] claim status: 2  for salicru subdriver failed
   0.211627     [D3] match_function_regex: matching a device...
   0.211751     [D2] Device matches
   0.211787     [D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.211950     [D3] HID descriptor, method 1: (9 bytes) => 09 21 11 01 00 01 22 9c 05
   0.211958     [D3] HID descriptor length (method 1) 1436
   0.211964     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.211971     [D3] HID descriptor, method 2: (9 bytes) => 09 21 11 01 00 01 22 9c 05
   0.211974     [D3] HID descriptor length (method 2) 1436
   0.211978     [D2] HID descriptor length 1436
   0.219426     [D2] Report Descriptor size = 1436
   0.219446     [D3] Report Descriptor: (1436 bytes) => 05 84 09 04 a1 01 09 10 a1 00 09 12 a1
   0.219469     [D3]  00 09 13 85 20 75 08 95 01 15 00 26 ff 00 65 00 55 00 b1 03 06 ff ff 09 90
   0.219478     [D3]  85 21 b1 82 05 84 09 02 a1 02 09 60 85 27 75 08 26 ff 00 65 00 b1 83 09 60
   0.219488     [D3]  81 83 06 ff ff 09 74 b1 83 09 74 81 83 c0 05 84 09 58 85 24 b1 82 06 ff ff
   0.219498     [D3]  09 e3 85 e3 b1 82 09 45 85 25 75 20 27 ff ff ff 7f 66 01 10 b1 82 c0 05 84
   0.219507     [D3]  09 11 85 20 75 08 26 ff 00 65 00 b1 03 09 14 a1 00 09 15 b1 03 05 84 09 02
   0.219523     [D3]  a1 02 06 ff ff 09 79 85 27 b1 83 09 79 81 83 05 84 09 62 b1 83 09 62 81 83
   0.219538     [D3]  09 6d b1 83 09 6d 81 83 c0 c0 c0 05 84 09 1e a1 81 09 42 85 14 66 01 f0 b1
   0.219555     [D3]  83 09 40 85 15 67 21 d1 f0 00 55 07 b1 83 09 1f 85 0b 65 00 55 00 b1 03 c0
   0.219568     [D3]  09 1e a1 84 09 44 85 74 75 10 27 ff ff 00 00 66 21 d1 55 07 b1 83 09 43 66
   0.219581     [D3]  21 d1 b1 83 09 42 85 36 75 08 26 ff 00 66 01 f0 55 00 b1 82 09 45 85 16 65
   0.219595     [D3]  00 b1 83 09 40 85 72 67 21 d1 f0 00 55 07 b1 82 09 1f 85 0b 65 00 55 00 b1
   0.219609     [D3]  03 c0 09 16 a1 00 06 ff ff 09 41 85 4a b1 03 09 42 85 4c b1 83 09 4f 85 4b
   0.219622     [D3]  b1 82 05 84 09 1a a1 81 06 ff ff 09 43 85 73 b1 82 09 94 85 c1 b1 82 05 84
   0.219636     [D3]  09 1f 85 30 b1 03 09 32 85 31 75 10 27 ff ff 00 00 66 01 f0 55 0f b1 83 09
   0.219649     [D3]  1b 85 30 75 08 26 ff 00 65 00 55 00 b1 03 09 02 a1 02 09 64 85 32 b1 83 09
   0.219663     [D3]  64 81 83 09 63 b1 83 09 63 81 83 06 ff ff 09 78 b1 83 09 78 81 83 c0 05 84
   0.219676     [D3]  09 30 85 31 75 10 27 ff ff 00 00 67 21 d1 f0 00 55 07 b1 83 c0 05 84 09 1a
   0.219693     [D3]  a1 82 06 ff ff 09 72 85 51 75 08 26 ff 00 65 00 b1 82 05 84 09 1f 85 2f 75
   0.219710     [D3]  08 26 ff 00 65 00 55 00 b1 03 05 84 09 32 85 37 75 10 27 ff ff 00 00 66 01
   0.219727     [D3]  f0 55 0f b1 83 09 1b 85 2f 75 08 26 ff 00 65 00 55 00 b1 03 06 ff ff 09 54
   0.219741     [D3]  85 34 b1 82 05 84 09 02 a1 02 09 64 85 38 b1 83 09 64 81 83 09 61 b1 83 09
   0.219754     [D3]  61 81 83 09 65 b1 83 09 65 81 83 05 84 09 6d b1 83 09 6d 81 83 09 63 b1 83
   0.219779     [D3]  09 63 81 83 c0 09 6c 85 2d b1 82 09 30 85 37 75 10 27 ff ff 00 00 67 21 d1
   0.219797     [D3]  f0 00 55 07 b1 83 c0 05 84 09 1a a1 83 06 ff ff 09 46 85 39 75 08 26 ff 00
   0.219812     [D3]  65 00 55 00 b1 82 05 84 09 1f 85 30 b1 03 09 1b b1 03 09 02 a1 02 09 6d 85
   0.219826     [D3]  3b b1 83 09 6d 81 83 c0 06 ff ff 09 47 85 3a b1 82 c0 05 84 09 1a a1 85 09
   0.219840     [D3]  1f 85 30 b1 03 09 1b b1 03 09 02 a1 02 09 6d 85 3c b1 83 09 6d 81 83 c0 09
   0.219854     [D3]  6c 85 3f b1 82 c0 09 14 a1 00 09 15 85 40 b1 03 05 84 09 02 a1 02 09 6d 85
   0.219866     [D3]  41 b1 83 09 6d 81 83 06 ff ff 09 74 85 3e b1 83 09 74 81 83 09 75 b1 83 09
   0.219878     [D3]  75 81 83 c0 c0 05 84 09 1c a1 00 09 34 85 42 75 10 27 ff ff 00 00 66 21 d1
   0.219891     [D3]  55 07 b1 83 09 33 66 21 d1 b1 83 09 1f 85 40 75 08 26 ff 00 65 00 55 00 b1
   0.219905     [D3]  03 05 84 09 32 85 42 75 10 27 ff ff 00 00 66 01 f0 55 0f b1 83 06 ff ff 09
   0.219917     [D3]  f8 85 f8 75 10 27 ff ff 00 00 66 01 f0 55 0f b1 82 09 f9 85 f9 b1 82 09 fb
   0.219930     [D3]  85 fb 75 08 26 ff 00 65 00 55 00 b1 82 09 fc 85 fc b1 82 06 ff ff 09 58 85
   0.219942     [D3]  46 75 10 27 ff ff 00 00 67 21 d1 f0 00 55 07 b1 82 09 57 85 45 b1 82 06 ff
   0.219955     [D3]  ff 09 5b 85 68 75 08 15 00 26 ff 00 65 00 55 00 b1 82 09 5a 85 69 b1 82 05
   0.219969     [D3]  84 09 54 85 48 75 10 27 ff ff 00 00 67 21 d1 f0 00 55 07 b1 82 05 84 09 53
   0.219978     [D3]  85 47 b1 82 09 1d 85 40 75 08 26 ff 00 65 00 55 00 b1 03 05 84 09 02 a1 02
   0.219986     [D3]  06 ff ff 09 4a 85 49 65 00 b1 83 09 4a 81 83 c0 05 84 09 30 85 42 75 10 27
   0.219994     [D3]  ff ff 00 00 67 21 d1 f0 00 55 07 b1 83 c0 09 17 85 40 75 08 26 ff 00 65 00
[... cut ...]
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4698-g381a81fc)
USB communication driver (libusb 0.1) 0.42
No matching HID UPS found
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4698-g381a81fc)
USB communication driver (libusb 0.1) 0.42
No matching HID UPS found
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.45 (2.7.4-4698-g381a81fc)
USB communication driver (libusb 0.1) 0.42
No matching HID UPS found
Driver failed to start (exit status=1)
```

drunkly, 2022-03-01:
I have installed nut and overwrite the files from my /tmp/package and now is working:

```
root@PROXMOX:~/nut/drivers# upsc salicru@192.168.1.2:3494
Init SSL without certificate database
battery.charge: 100
battery.runtime: 2829
battery.type: PbAc
battery.voltage: 27.40
battery.voltage.nominal: 24
device.mfr: Salicru
device.model: SLC TWIN PRO2<=3KVA
device.serial: 232019D70264
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 002
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: SLC TWIN PRO2<=3KVA
driver.parameter.productid: 0201
driver.parameter.serial: 232019D70264
driver.parameter.synchronous: auto
driver.parameter.vendor: Salicru
driver.parameter.vendorid: 2E66
driver.version: 2.7.4-4755-ga1454772
driver.version.data: Salicru HID 0.3
driver.version.internal: 0.45
driver.version.usb: libusb-0.1 (or compat)
input.transfer.high: 300
input.transfer.low: 100
ups.beeper.status: enabled
ups.mfr: Salicru
ups.model: SLC TWIN PRO2<=3KVA
ups.productid: 0201
ups.serial: 232019D70264
ups.status: OL CHRG
ups.vendorid: 2e66
root@PROXMOX:~/nut/drivers#
```

The only thing is I miss some values like load/input voltage/output voltage, etc. 

Do you think I can help somehow to extend this functionality to show more values/descriptors??

nbriggs, 2022-03-01:
@drunkly thanks for noticing that rdlen2 had the same problem as rdlen1.  I've updated my proposed fix.

Regarding the missing values -- I see quite a few unknown vendor-private fields are being reported, I wonder it is hiding some of the information you are looking for in those fields.

drunkly, 2022-03-02:
Thanks I will research a little bit more about this to see if I can show them.

Btw  @nbriggs I don't fully understand the architecture of the librery and as far as I know only libusb0.1 is supported by nut however in my eyes it looks like this change should be also implemented on libusb1 to prepare it for future releases:

https://github.com/networkupstools/nut/blob/master/drivers/libusb1.c#L428
https://github.com/networkupstools/nut/blob/master/drivers/libusb1.c#L453

drunkly, 2022-03-02:
Ok, I realize what is happening is related to the names (of the descriptors?). When I change some variables names then the results comes on the output. So I guess I will rename all the values I need.

```root@PROXMOX:~/nut# git diff drivers/salicru-hid.c
diff --git a/drivers/salicru-hid.c b/drivers/salicru-hid.c
index c93f52a8..87438737 100644
--- a/drivers/salicru-hid.c
+++ b/drivers/salicru-hid.c
@@ -155,7 +155,7 @@ static hid_info_t salicru_hid2nut[] = {
   { "battery.voltage", 0, 0, "UPS.PowerSummary.Voltage", NULL, "%.2f", 0, NULL },

   /* UPS page */
-  { "ups.load", 0, 0, "UPS.Output.PercentLoad", NULL, "%.0f", 0, NULL },
+  { "ups.load", 0, 0, "UPS.PowerSummary.PercentLoad", NULL, "%.0f", 0, NULL },
   { "ups.beeper.status", 0, 0, "UPS.PowerSummary.AudibleAlarmControl", NULL, "%s", 0, beeper_info },
   { "ups.test.result", 0, 0, "UPS.Output.Test", NULL, "%s", 0, test_read_info },
   { "ups.realpower.nominal", 0, 0, "UPS.Output.ConfigActivePower", NULL, "%.0f", 0, NULL },
root@PROXMOX:~/nut# ./clients/upsc salicru@192.168.1.2:3494 ups.load
21
```

jimklimov, 2022-03-02:
Thanks for the fix proposal and tests.

For mappings, it should work to specify several variants (e.g. add new
`ups.load` and not delete the old one), first handled hit wins.

> Message ID: ***@***.***>
>

jimklimov, 2022-03-02:
As for libusb 1.0 vs 0.1 - both should now be supported by codebase (one
selected at build time), and the `usb_ctrl_char(buf)` type definitions
allow NUT code to seem the same and simple while the backebd library APIs
do differ. What bites here is that one handles and passes char buffers, and
another - unsigned char buffers.

closed 2022-03-02 by jimklimov

jimklimov, 2023-08-31:
Historic note: additional mapping descriptors were added by #1331 before NUT v2.8.0 release.

PR #1320 (2022-03-02): Cast to unsigned type when interpreting HID descriptor length bytes (libusb 0.1)

The libusb 0.1 interface definition declares a (signed) char type for
control messages.  The HID descriptor length contained within a control
message is intended to be interpreted as a pair of unsigned bytes so
we must cast to uint8_t when doing the arithmetic rather than trip over
the sign bit.

Closes #1261, closes #1312.
commit f2fb70d2 (2022-03-02): Cast to unsigned type when interpreting HID descriptor length bytes (libusb 0.1)

The libusb 0.1 interface definition declares a (signed) char type for
control messages.  The HID descriptor length contained within a control
message is intended to be interpreted as a pair of unsigned bytes so
we must cast to uint8_t when doing the arithmetic rather than trip over
the sign bit.

Closes #1261, closes #1312.